libusb/libusb #1383: Failed to open the device: Access denied (insufficient permissions)

maxwang1104, 2023-12-18:
I want to use libusb_open() to open my usb device. But it shows Access denied.
I have used Zadig to install WINUSB to my device. 
Also I tried use libusb_open() to open other device -> Successfully opened

OS: Windows10
Zadig: 2.8.782
libusb: 1.0.27

This is the debug code in ./xusb
`$ ./xusb.exe -d 0601:1234
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000025] [00002788] libusb: debug [libusb_init] created default context
[ 0.000041] [00002788] libusb: debug [libusb_init] libusb v1.0.26.11724
[ 0.000053] [00002788] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000508 events 0
[ 0.000064] [00002788] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000069] [00002788] libusb: debug [usbi_add_event_source] add HANDLE 000000000000050C events 0
[ 0.000104] [00002788] libusb: debug [get_windows_version] Windows 8 (or later) 64-bit
[ 0.000107] [00002788] libusb: debug [htab_create] using 1021 entries hash table
[ 0.006823] [00002788] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.009096] [00002788] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.018658] [00002788] libusb: info [windows_init] UsbDk backend is not available
[ 0.018751] [00002788] libusb: debug [libusb_get_device_list]
[ 0.024420] [00000670] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.038164] [00002788] libusb: debug [winusb_get_device_list] allocating new device for session [89]
[ 0.038264] [00002788] libusb: debug [winusb_get_device_list] allocating new device for session [8A]
[ 0.038353] [00002788] libusb: debug [winusb_get_device_list] allocating new device for session [8B]
[ 0.038438] [00002788] libusb: debug [winusb_get_device_list] allocating new device for session [8C]
[ 0.038521] [00002788] libusb: debug [winusb_get_device_list] allocating new device for session [8D]
[ 0.039513] [00002788] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.039778] [00002788] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.039786] [00002788] libusb: debug [winusb_get_device_list] allocating new device for session [63]
[ 0.040364] [00002788] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.040477] [00002788] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.040480] [00002788] libusb: debug [winusb_get_device_list] allocating new device for session [64]
[ 0.041162] [00002788] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.041313] [00002788] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.041319] [00002788] libusb: debug [winusb_get_device_list] allocating new device for session [72]
[ 0.042120] [00002788] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.042333] [00002788] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.042337] [00002788] libusb: debug [winusb_get_device_list] allocating new device for session [73]
[ 0.042903] [00002788] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.043009] [00002788] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.043012] [00002788] libusb: debug [winusb_get_device_list] allocating new device for session [74]
[ 0.043553] [00002788] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.043717] [00002788] libusb: debug [get_api_type] matched driver name against HID API
[ 0.043720] [00002788] libusb: debug [winusb_get_device_list] allocating new device for session [75]
[ 0.044207] [00002788] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.044328] [00002788] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.044331] [00002788] libusb: debug [winusb_get_device_list] allocating new device for session [76]
[ 0.044593] [00002788] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_51ED&SUBSYS_8B78103C&REV_01\3&11583659&0&A0' bus number 1
[ 0.044635] [00002788] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A71E&SUBSYS_8B78103C&REV_01\3&11583659&0&68' bus number 2
[ 0.046568] [00002788] libusb: debug [winusb_get_device_list] found existing device for session [89]
[ 0.046791] [00002788] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.046807] [00002788] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 41 bytes)
[ 0.046868] [00002788] libusb: debug [init_device] (bus: 1, addr: 35, depth: 2, port: 1): 'USB\VID_4989&PID_0100\6&2E5F6FA0&0&1'
[ 0.047113] [00002788] libusb: debug [winusb_get_device_list] found existing device for session [8A]
[ 0.047317] [00002788] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.047329] [00002788] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 41 bytes)
[ 0.047340] [00002788] libusb: debug [init_device] (bus: 1, addr: 14, depth: 1, port: 3): 'USB\VID_2109&PID_2822\MSFT20000000001'
[ 0.047532] [00002788] libusb: debug [winusb_get_device_list] found existing device for session [72]
[ 0.047704] [00002788] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.047715] [00002788] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 101 bytes)
[ 0.047727] [00002788] libusb: debug [init_device] (bus: 1, addr: 24, depth: 2, port: 2): 'USB\VID_0403&PID_6011\6&2E5F6FA0&0&2'
[ 0.047902] [00002788] libusb: debug [winusb_get_device_list] found existing device for session [75]
[ 0.048035] [00002788] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.048048] [00002788] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.048059] [00002788] libusb: debug [init_device] (bus: 1, addr: 20, depth: 2, port: 4): 'USB\VID_093A&PID_2510\6&2E5F6FA0&0&4'
[ 0.048443] [00002788] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&2E5461A6&0&0' reports 16 ports
[ 0.048509] [00002788] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&2E5461A6&0&0'
[... 70 lines cut ...]
[ 0.058418] [00002788] libusb: debug [libusb_get_device_descriptor]
[ 0.058419] [00002788] libusb: debug [libusb_get_device_descriptor]
[ 0.058420] [00002788] libusb: debug [libusb_get_device_descriptor]
[ 0.058421] [00002788] libusb: debug [libusb_get_device_descriptor]
[ 0.058422] [00002788] libusb: debug [libusb_open] open 1.38
[ 0.060694] [00002788] libusb: error [winusbx_open] could not open device \\?\USB#VID_0601&PID_1234#76543210#{A5DCBF10-6530-11D2-901F-00C04FB951ED} (interface 0): [5] ▒s▒▒▒Q▒ڡC
[ 0.060707] [00002788] libusb: debug [libusb_open] open 1.38 returns -3
[ 0.060712] [00002788] libusb: debug [libusb_unref_device] destroy device 1.24
[ 0.060717] [00002788] libusb: debug [libusb_unref_device] destroy device 1.20
[ 0.060721] [00002788] libusb: debug [libusb_unref_device] destroy device 1.13
[ 0.060734] [00002788] libusb: debug [libusb_unref_device] destroy device 1.5
[ 0.060738] [00002788] libusb: debug [libusb_unref_device] destroy device 1.18
[ 0.060745] [00002788] libusb: debug [libusb_unref_device] destroy device 1.2
[ 0.060750] [00002788] libusb: debug [libusb_unref_device] destroy device 1.38
[ 0.060752] [00002788] libusb: debug [libusb_unref_device] destroy device 1.35
[ 0.060754] [00002788] libusb: debug [libusb_unref_device] destroy device 1.14
[ 0.060759] [00002788] libusb: debug [libusb_unref_device] destroy device 1.15
[ 0.060761] [00002788] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.060764] [00002788] libusb: debug [libusb_unref_device] destroy device 2.0
  Failed.
[ 0.060771] [00002788] libusb: debug [libusb_exit] destroying default context
[ 0.060866] [00000670] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.062798] [00002788] libusb: debug [usbi_remove_event_source] remove HANDLE 000000000000050C
[ 0.062820] [00002788] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000508
`

mcuee, 2023-12-18:
@maxwang1104 

You are using libusb-1.0.26 release version. Please help to test with libusb-1.0.27-rc1. Thanks.
https://github.com/libusb/libusb/releases/tag/v1.0.27-rc1

And please tell us more about the device and how you install WinUSB driver. There are still some issues dealing with certain type of USB Composite devices under Windows. You may want to use WinUSB as the driver of the whole device (use WinUSB driver to replace USB composite parent driver) to see if that helps.

maxwang1104, 2023-12-18:
@mcuee Thanks for reply.

After I install the  libusb-1.0 1.0.27-rc1 and command "make check" , it shows in libusb-1.0 1.0.27-rc1: tests/test-suite.log
```

TOTAL: 4
PASS:  3
SKIP:  0
XFAIL: 0
FAIL:  1
XPASS: 0
ERROR: 0
.. contents:: :depth: 2

FAIL: stress_mt

Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
All threads discovered 0 devices (0 not opened)
All done, 8 errors
Thread 0 failed (iteration 1): LIBUSB_ERROR_NOT_SUPPORTED
Thread 1 failed (iteration 1): LIBUSB_ERROR_NOT_SUPPORTED
Thread 2 failed (iteration 1): LIBUSB_ERROR_NOT_SUPPORTED
Thread 3 failed (iteration 1): LIBUSB_ERROR_NOT_SUPPORTED
Thread 4 failed (iteration 1): LIBUSB_ERROR_NOT_SUPPORTED
Thread 5 failed (iteration 1): LIBUSB_ERROR_NOT_SUPPORTED
Thread 6 failed (iteration 1): LIBUSB_ERROR_NOT_SUPPORTED
Thread 7 failed (iteration 1): LIBUSB_ERROR_NOT_SUPPORTED
FAIL stress_mt.exe (exit status: 1)

```
Will it effect the functionality of libusb?

I'm trying to implement a DFU (interface 1) with dfu-util under my HID device(interface 0).
When I attach my HID device, it shows composite device.

I test several scenario, Open Zadig and replace with the below driver

1. HID(HidUsb driver), DFU(None)     -> HID could be open
2. HID(HidUsb driver), DFU(WinUSB) -> HID could not be open
3. HID(WinUSB), DFU(WinUSB)          -> HID could not be open

After Sending the DETACH request and reconfig the descriptor to DFU descriptor set, the dfu device couldn't be opened

mcuee, 2023-12-18:
@maxwang1104 
It is a known issue for `make check` to fail under Windows when using libusb-1.0.27-rc1. PR #1360 should be able fix the issue.

As for your device, please help to post the USB Descriptor information. Thanks.
Under Windows, you can use USB Device Tree Viewer or Microsoft USBView.
https://www.uwe-sieber.de/usbtreeview_e.html
https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/usbview?redirectedfrom=MSDN

mcuee, 2023-12-18:
> I'm trying to implement a DFU (interface 1) with dfu-util under my HID device(interface 0). When I attach my HID device, it shows composite device.
>
> I test several scenario, Open Zadig and replace [...]

You can try the work-around to use WinUSB as the device driver for the whole device (Composite Parent).
HID interface + DFU interface --> WinUSB driver (to replace usbccgp driver).

Under Zadig -- Options --> tick "List All Devices" and untick "Ignore Hubs or Composite Parent". Then select the USB Composite Device (Composite Parent), replace the original system provided `usbccgp` driver with WinUSB. Then you should be able to access both interfaces using libusb.

Example screenshots:
[image]
[image]

maxwang1104, 2023-12-18:
I following the steps to replace the composite parent driver to WinUSB
But it still can't open the device...

I tried to use xusb to see the error code, this is below
Opening device 0601:5678...
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_045E&PID_0750&MI_00\6&9A1EDFA&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_4989&PID_0100\6&2E5F6FA0&0&1'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_2109&PID_2822\MSFT20000000001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0403&PID_6011\6&2E5F6FA0&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_093A&PID_2510\6&2E5F6FA0&0&4'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\4&2E5461A6&0&0'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_045E&PID_0750\5&601D3B1&0&6'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0403&PID_6011&MI_03\7&E67A473&0&0003'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0403&PID_6011&MI_02\7&E67A473&0&0002'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0403&PID_6011&MI_01\7&E67A473&0&0001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0403&PID_6011&MI_00\7&E67A473&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0403&PID_6010&MI_00\7&2970E7E3&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_8087&PID_0033\5&601D3B1&0&10'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0403&PID_6010\6&2E5F6FA0&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0403&PID_6010&MI_01\7&2970E7E3&0&0001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_05C8&PID_0B06\200901010001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_045E&PID_0750&MI_01\6&9A1EDFA&0&0001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_2109&PID_0822\MSFT30000000001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\4&106120CF&0&0'
libusb: error [winusbx_open] could not open device \\?\USB#VID_0601&PID_5678#76543210#{A5DCBF10-6530-11D2-901F-00C04FB951ED} (interface 0):

mcuee, 2023-12-19:
@maxwang1104 

Please post the full debug log. Thanks.

And as mentioned before, please post more info about your device.
Under Windows, you can use USB Device Tree Viewer or Microsoft USBView.
https://www.uwe-sieber.de/usbtreeview_e.html
https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/usbview?redirectedfrom=MSDN

If you use linux, you can use `lsusb -vvv`.

maxwang1104, 2023-12-19:
> @maxwang1104
>
> Please post the full debug log. Thanks.
>
> And as mentioned before, please post more info about your device. Under Windows, you can use USB Device Tree Viewer or Microsoft USBView. [...]

@mcuee Thank for your help.

**This is the full debug log**
[DFU+HID_Composite_DebugTool_info_1219.txt](https://github.com/libusb/libusb/files/13712216/DFU%2BHID_Composite_DebugTool_info_1219.txt)

**It is the info about my device, the descriptors of device are configurated by myself. It may have some mistake on it**
[USBView_HID+DFU_Composite.txt](https://github.com/libusb/libusb/files/13712210/USBView_HID%2BDFU_Composite.txt)

mcuee, 2023-12-19:
> **It is the info about my device, the descriptors of device are configurated by myself. It may have some mistake on it** [USBView_HID+DFU_Composite.txt](https://github.com/libusb/libusb/files/13712210/USBView_HID%2BDFU_Composite.txt)

You can see there is no HID interface at all from the above. 

There are two child devices.
IAD -- Interface 0 + Interface 1 --> Video device (HP HD Camera)
IAD -- Interface 2 --> custom device (DFU).

This is rather complex device and I suspect the USB descriptors are not totally correct. You may want to remove the IAD stuff for Interface 2 and try again, as it is not really using IAD. 

```
Power State              : D3 (supported: D0, D3, wake from D0)
 Child Device 1          : HP HD Camera (USB 視訊裝置)
  Device Path 1          : \\?\USB#VID_05C8&PID_0B06&MI_00#6&21b6f2bc&0&0000#{6994ad05-93ef-11d0-a3cc-00a0c9223196}\global (AM_KSCATEGORY_VIDEO)
  Device Path 2          : \\?\USB#VID_05C8&PID_0B06&MI_00#6&21b6f2bc&0&0000#{e5323777-f976-4f5b-9b55-b94699c46e44}\global (STATIC_KSCATEGORY_VIDEO_CAMERA)
  Device Path 3          : \\?\USB#VID_05C8&PID_0B06&MI_00#6&21b6f2bc&0&0000#{65e8773d-8f56-11d0-a3b9-00a0c9223196}\global (AM_KSCATEGORY_CAPTURE)
  Kernel Name            : \Device\0000018d
  Device ID              : USB\VID_05C8&PID_0B06&MI_00\6&21B6F2BC&0&0000
  Class                  : Camera
  Driver KeyName         : {ca3e7ab9-b4c3-4ae6-8251-579ef933890f}\0000 (GUID_DEVCLASS_CAMERA)
  Service                : usbvideo
  Location               : 0000.0014.0000.001.000.000.000.000.000
  LocationPaths          : PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(1)#USBMI(0)  PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(1)#USB(1)  ACPI(_SB_)#ACPI(PC00)#ACPI(XHCI)#ACPI(RHUB)#ACPI(HS01)#ACPI(WCAM)
 Child Device 2          : Camera DFU Device (WinUsb 裝置)
  Device Path 1          : \\?\USB#VID_05C8&PID_0B06&MI_02#6&21b6f2bc&0&0002#{dee824ef-729b-4a0e-9c14-b7117d33a817} (GUID_DEVINTERFACE_WINUSB)
  Device Path 2          : \\?\USB#VID_05C8&PID_0B06&MI_02#6&21b6f2bc&0&0002#{ecceff35-1463-4ff3-acd9-8f992d09acdd}
  Kernel Name            : \Device\0000018e
  Device ID              : USB\VID_05C8&PID_0B06&MI_02\6&21B6F2BC&0&0002
  Class                  : USBDevice
  Driver KeyName         : {88bae032-5a81-49f0-bc3d-a4ff138216d6}\0000 (GUID_DEVCLASS_USBDEVICE)
  Service                : WINUSB
  Location               : 0000.0014.0000.001.000.000.000.000.000
  LocationPaths          : PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(1)#USBMI(2)  ACPI(_SB_)#ACPI(PC00)#ACPI(XHCI)#ACPI(RHUB)#ACPI(HS01)#ACPI(IRCM)
```

maxwang1104, 2023-12-19:
> You can see there is no HID interface at all from the above.
>
> There are two child devices. IAD -- Interface 0 + Interface 1 --> Video device (HP HD Camera) IAD -- Interface 2 --> custom device [...]

Sorry to upload the wrong file
This is the correct one!
[USBView_HID+DFU_Composite.txt](https://github.com/libusb/libusb/files/13712481/USBView_HID%2BDFU_Composite.txt)

And I also try without IAD, then install composite parent driver following above mention but still access denied
[USBView_HID+DFU_Composite_wout_IAD.txt](https://github.com/libusb/libusb/files/13712497/USBView_HID%2BDFU_Composite_wout_IAD.txt)

mcuee, 2023-12-19:
> Sorry to upload the wrong file This is the correct one! [USBView_HID+DFU_Composite.txt](https://github.com/libusb/libusb/files/13712481/USBView_HID%2BDFU_Composite.txt)
>
> And I also try without IAD, [...]

The first interface is a mouse, libusb Windows can not be used to access a mouse.
https://github.com/libusb/libusb/wiki/Windows#user-content-Known_Restrictions
HID keyboards and mice cannot be accessed using the native HID driver as Windows reserves exclusive access to them.

```
        ---------------- Interface Descriptor -----------------
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x04 (Interface Descriptor)
bInterfaceNumber         : 0x00 (Interface 0)
bAlternateSetting        : 0x00
bNumEndpoints            : 0x01 (1 Endpoint)
bInterfaceClass          : 0x03 (HID - Human Interface Device)
bInterfaceSubClass       : 0x01 (Boot Interface)
bInterfaceProtocol       : 0x02 (Mouse)
iInterface               : 0x04 (String Descriptor 4)
 Language 0x0409         : "Mouse"
Data (HexDump)           : 09 04 00 00 01 03 01 02 04                        .........

```

mcuee, 2023-12-20:
Just tested with a USB Composite Device with WinUSB (interface 0)+HID (interface 1) and it is okay under Windows 10.

```
MINGW64 /c/work/libusb/libusb-1.0.27-rc1/mingw64
$ ./examples/xusb.exe 16c0:05dc
Using libusb v1.0.27.11848

Opening device 16C0:05DC...
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057\C&1F76A113&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0424&PID_5744\C&1936BD58&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_5413\B&2EAF716D&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_0413\B&1637EEB3&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_413C&PID_B06E\B&2EAF716D&0&5'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_1A86&PID_7523\D&127B8A6A&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_8087&PID_0AAA\5&E9F3E45&0&14'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_02\6&1BD5FF13&0&0002'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0424&PID_2744\C&1F76A113&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_5487\A&46F7C3F&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_58FD&MI_02\6&1E7CBC97&0&0002'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057&MI_01\D&58BCC3D&0&0001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057&MI_00\D&58BCC3D&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_0487\A&46F7C3F&0&4'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_03\6&1BD5FF13&0&0003'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_2717&PID_5013\5&E9F3E45&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_402E\200901010001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_047F&PID_C056&MI_03\6&360CBCF3&0&0003'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\4&36020D6F&0&0'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842\0123456789ABCD'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\9&AF4BBD4&0&0'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0B95&PID_772B\C1B13D'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\7&2452366F&0&0'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_402E&MI_00\D&E651DD6&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_047F&PID_C056\BB305534D79EBD418B6757528EAC0C19'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_00\6&1BD5FF13&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_8153\111000001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_413C&PID_B06F\C&1F76A113&0&5'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_047F&PID_C056&MI_00\6&360CBCF3&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_58FD&MI_00\6&1E7CBC97&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_05E3&PID_0608\5&E9F3E45&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_01\6&1BD5FF13&0&0001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_58FD\200901010001'

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 16C0:05DC
         bcdDevice: 0110
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: 2 caps
    Container ID:
      {b9d357ad-6611-f843-8790-ebe14ddc7594}
    Platform descriptor:
      bLength                : 28
      PlatformCapabilityUUID : {df60ddd8-8945-c74c-9cd2-659d9e648a9f}

  00000000  00 00 03 06 36 01 5d 00                          ....6.].

Reading first configuration descriptor:
              total length: 50
         descriptor length: 9
             nb interfaces: 2
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 0
   Class.SubClass.Protocol: FF.00.00
              interface[1]: id = 1
interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 81
           max packet size: 0008
          polling interval: 0A
       endpoint[1].address: 01
           max packet size: 0008
          polling interval: 0A

Kernel driver attached for interface 0: (not supported)

Claiming interface 0...

Kernel driver attached for interface 1: (not supported)

Claiming interface 1...

Reading string descriptors:
   String (0x01): "www.fischl.de"
   String (0x02): "USBasp"
   String (0x03): "0001"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
Releasing interface 1...
Closing device...
```

maxwang1104, 2023-12-20:
I solve this problem with using another Windows platform.

closed 2023-12-20 by maxwang1104

mcuee, 2023-12-20:
> I solve this problem with using another Windows platform.

Just wondering what you mean by another Windows platform. Is it Windows 11? Thanks.