libusb/libusb #1384: Broken Dependency Integration after a new vcpkg/libusb release

R-Abbasi, 2023-12-18:
When trying to compile our project (https://github.com/skunkforce/OmniView) on Ubuntu, we faced the `CMake` error: _Could not find a package configuration file provided by "libusb" with any of the following names:
libusbConfig.cmake
libusb-config.cmake_

This is due to CMake not finding libusb via find_package which could also be reproduced in CI (https://github.com/skunkforce/omniview/actions/runs/7232145540/job/19706039412#step:9:1244)
After investigation, it was concluded, that CMake couldn't find `libusb`, since it was missing a `.cmake` file as is required for `Config Mode`. 
The last successful run appeared on 2023-12-12, which was the exact day of a new vcpkg release.
In that release which can be found here: https://github.com/microsoft/vcpkg/releases/tag/2023.12.12 a minor change happend.
Looking at the "updated ports" `libusb` changed `1.0.26.11791#3` to `1.0.26.11791#5`.

Modifing `CMakeLists.txt` to clone the 2023.11.20 Release of vcpkg makes CMake happy:

```
FetchContent_Populate(
vcpkg
GIT_REPOSITORY https://github.com/microsoft/vcpkg.git
GIT_TAG        2023.11.20
SOURCE_DIR     "${CMAKE_SOURCE_DIR}/vcpkg"
)
```
Although this seems to be a possibility to mitigate the problem, it's, however, not clear what unintended consequences this might have for the future. 
Hopefully you fix this issue and we don't face such an error each time a new libusb/vcpkg releases.

mcuee, 2023-12-18:
libusb does not support CMake, you may want to report the issue to vcpkg.

closed 2023-12-18 by mcuee

Youw, 2023-12-18:
Also, if your goal is to use libusb as part of your CMake-project, you may consider using https://github.com/libusb/libusb-cmake