libusb/libusb #1386: Segmentation fault with the kernel detaching function under macOS 14.2 using sudo

mcuee, 2023-12-19:
Reference:
* https://github.com/libusb/libusb/issues/1372#issuecomment-1859071588

Run log under macOS 14.2, Mac Mini M1 2020.

```
mcuee@mcuees-Mac-mini arm64 % sudo ./examples/xusb 0925:7001
Password:
Using libusb v1.0.27.11848

Opening device 0925:7001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 0925:7001
         bcdDevice: 0001
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 41
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 01
       endpoint[1].address: 01
           max packet size: 0040
          polling interval: 01

Kernel driver attached for interface 0: yes

Claiming interface 0...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device

Reading string descriptors:
   String (0x01): "Microchip Technology Inc."
   String (0x02): "Generic HID"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
zsh: segmentation fault  sudo ./examples/xusb 0925:7001
```

mcuee, 2023-12-19:
Debug log using the suggestion from @RReverser to build with `CFLAGS="-g - fsanitize=address"`.

```
mcuee@mcuees-Mac-mini arm64_debug % sudo ./examples/xusb 0925:7001
Password:
Using libusb v1.0.27.11848

Opening device 0925:7001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 0925:7001
         bcdDevice: 0001
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 41
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 01
       endpoint[1].address: 01
           max packet size: 0040
          polling interval: 01

Kernel driver attached for interface 0: yes

Claiming interface 0...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device

Reading string descriptors:
   String (0x01): "Microchip Technology Inc."
   String (0x02): "Generic HID"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
=================================================================
==1066==ERROR: AddressSanitizer: heap-use-after-free on address 0x000106200360 at pc 0x0001033e4d40 bp 0x00016cf65e50 sp 0x00016cf65e48
READ of size 8 at 0x000106200360 thread T0
    #0 0x1033e4d3c in darwin_reenumerate_device darwin_usb.c:2118
    #1 0x1033d1a54 in darwin_attach_kernel_driver darwin_usb.c:2844
    #2 0x1033cefb4 in darwin_capture_release_interface darwin_usb.c:2869
    #3 0x1033ab0f4 in libusb_release_interface core.c:1832
    #4 0x102e9e99c in test_device xusb.c:1027
    #5 0x102e9af90 in main xusb.c:1185
    #6 0x1828e90dc  (<unknown module>)

0x000106200360 is located 32 bytes inside of 128-byte region [0x000106200340,0x0001062003c0)
freed by thread T1 here:
    #0 0x1038a5bcc in wrap__ZdlPv+0x74 (libclang_rt.asan_osx_dynamic.dylib:arm64e+0x61bcc)
    #1 0x102edb490 in IOUSBIUnknown::release()+0x48 (IOUSBLib:arm64e+0x3490)
    #2 0x1033d63c4 in darwin_devices_detached darwin_usb.c:656
    #3 0x1863a5210 in IODispatchCalloutFromCFMessage+0xc0 (IOKit:arm64e+0x6210)
    #4 0x182d77490 in __CFMachPortPerform+0x124 (CoreFoundation:arm64e+0xad490)
    #5 0x182d47fd8 in __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE1_PERFORM_FUNCTION__+0x38 (CoreFoundation:arm64e+0x7dfd8)
    #6 0x182d47ef8 in __CFRunLoopDoSource1+0x204 (CoreFoundation:arm64e+0x7def8)
    #7 0x182d468c4 in __CFRunLoopRun+0x8c0 (CoreFoundation:arm64e+0x7c8c4)
[... 40 lines cut ...]
  0x000106200480: fa fa fa fa fa fa fa fa 00 00 00 00 00 00 00 00
  0x000106200500: 00 00 00 00 00 00 00 00 fa fa fa fa fa fa fa fa
  0x000106200580: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==1066==ABORTING
zsh: abort      sudo ./examples/xusb 0925:7001
```

mcuee, 2023-12-19:
The crash is not limited to USB HID device, the following is with a USB Card Reader (USB Mass Storage) device.

```
mcuee@mcuees-Mac-mini arm64_debug % sudo ./examples/xusb 05e3:0749
Using libusb v1.0.27.11848

Opening device 05E3:0749...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 2
           VID:PID: 05E3:0749
         bcdDevice: 1535
   iMan:iProd:iSer: 3:4:2
          nb confs: 1

Reading BOS descriptor: 2 caps
    USB 2.0 extension:
      attributes             : F0E
    USB 3.0 capabilities:
      attributes             : 00
      supported speeds       : 000E
      supported functionality: 01

Reading first configuration descriptor:
              total length: 44
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 08.06.50
       endpoint[0].address: 81
           max packet size: 0400
          polling interval: 00
                 max burst: 04   (USB 3.0)
        bytes per interval: 0000 (USB 3.0)
       endpoint[1].address: 02
           max packet size: 0400
          polling interval: 00
                 max burst: 04   (USB 3.0)
        bytes per interval: 0000 (USB 3.0)

Kernel driver attached for interface 0: yes

Claiming interface 0...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device

Reading string descriptors:
   String (0x03): "Generic"
   String (0x04): "USB3.0 Card Reader"
   String (0x02): "000000001536"

Reading OS string descriptor:

Reading Extended Compat ID OS Feature Descriptor (wIndex = 0x0004):

  00000000  28 00 00 00 00 01 04 00 01 00 00 00 00 00 00 00  (...............
  00000010  00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000020  00 00 00 00 00 00 00 00                          ........

Reading Extended Properties OS Feature Descriptor (wIndex = 0x0005):

  00000000  4c 00 00 00 00 01 05 00 01 00 42 00 00 00 04 00  L.........B.....
  00000010  00 00 30 00 53 00 65 00 6c 00 65 00 63 00 74 00  ..0.S.e.l.e.c.t.
  00000020  49 00 76 00 65 00 53 00 75 00 73 00 70 00 65 00  I.v.e.S.u.s.p.e.
  00000030  6e 00 64 00 45 00 6e 00 61 00 62 00 6c 00 65 00  n.d.E.n.a.b.l.e.
  00000040  64 00 00 00 04 00 00 00 00 00 00 00              d...........

Reading interface association descriptors (IADs) for first configuration:
[... 78 lines cut ...]
  0x000106000300: fa fa fa fa fa fa fa fa 00 00 00 00 00 00 00 00
  0x000106000380: 00 00 00 00 00 00 00 00 fa fa fa fa fa fa fa fa
  0x000106000400: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==1307==ABORTING
zsh: abort      sudo ./examples/xusb 05e3:0749
```

mcuee, 2023-12-19:
Run log for 1.0.26 for reference.

```

mcuee@mcuees-Mac-mini arm64_debug % pwd                           
/Users/mcuee/build/libusb/libusb-1.0.26/arm64_debug

mcuee@mcuees-Mac-mini arm64_debug % sudo ./examples/xusb 05e3:0749
Password:
Using libusb v1.0.26.11724

Opening device 05E3:0749...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 2
           VID:PID: 05E3:0749
         bcdDevice: 1535
   iMan:iProd:iSer: 3:4:2
          nb confs: 1

Reading BOS descriptor: 2 caps
    USB 2.0 extension:
      attributes             : F0E
    USB 3.0 capabilities:
      attributes             : 00
      supported speeds       : 000E
      supported functionality: 01

Reading first configuration descriptor:
              total length: 44
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 08.06.50
       endpoint[0].address: 81
           max packet size: 0400
          polling interval: 00
                 max burst: 04   (USB 3.0)
        bytes per interval: 0000 (USB 3.0)
       endpoint[1].address: 02
           max packet size: 0400
          polling interval: 00
                 max burst: 04   (USB 3.0)
        bytes per interval: 0000 (USB 3.0)

Kernel driver attached for interface 0: 1

Claiming interface 0...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device

Reading string descriptors:
   String (0x03): "Generic"
   String (0x04): "USB3.0 Card Reader"
   String (0x02): "000000001536"

Reading OS string descriptor:

Reading Extended Compat ID OS Feature Descriptor (wIndex = 0x0004):

  00000000  28 00 00 00 00 01 04 00 01 00 00 00 00 00 00 00  (...............
  00000010  00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000020  00 00 00 00 00 00 00 00                          ........

Reading Extended Properties OS Feature Descriptor (wIndex = 0x0005):

  00000000  4c 00 00 00 00 01 05 00 01 00 42 00 00 00 04 00  L.........B.....
  00000010  00 00 30 00 53 00 65 00 6c 00 65 00 63 00 74 00  ..0.S.e.l.e.c.t.
  00000020  49 00 76 00 65 00 53 00 75 00 73 00 70 00 65 00  I.v.e.S.u.s.p.e.
  00000030  6e 00 64 00 45 00 6e 00 61 00 62 00 6c 00 65 00  n.d.E.n.a.b.l.e.
  00000040  64 00 00 00 04 00 00 00 00 00 00 00              d...........
Reading Max LUN:
   Max LUN = 0
Sending Inquiry:
   sent 6 CDB bytes
   received 36 bytes
   VID:PID:REV "Generic ":"MassStor":"1536"
   Mass Storage Status: 00 (Success)
Reading Capacity:
   sent 10 CDB bytes
   Pipe error
libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
libusb: info [darwin_capture_release_interface] on attempt to reattach the kernel driver got ret=-5
   Input/Output Error
```

mcuee, 2023-12-19:
@hjelmn and @tormodvolden 

At least until commit https://github.com/libusb/libusb/tree/65e4a70e0db5dc85cdbce01be276329580e0d986, there is no crash.

Looks like a recent regression.

```
mcuee@mcuees-Mac-mini arm64_debug_65e4a70 % sudo ./examples/xusb 05e3:0749
Using libusb v1.0.26.11756

Opening device 05E3:0749...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 2
           VID:PID: 05E3:0749
         bcdDevice: 1535
   iMan:iProd:iSer: 3:4:2
          nb confs: 1

Reading BOS descriptor: 2 caps
    USB 2.0 extension:
      attributes             : F0E
    USB 3.0 capabilities:
      attributes             : 00
      supported speeds       : 000E
      supported functionality: 01

Reading first configuration descriptor:
              total length: 44
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 08.06.50
       endpoint[0].address: 81
           max packet size: 0400
          polling interval: 00
                 max burst: 04   (USB 3.0)
        bytes per interval: 0000 (USB 3.0)
       endpoint[1].address: 02
           max packet size: 0400
          polling interval: 00
                 max burst: 04   (USB 3.0)
        bytes per interval: 0000 (USB 3.0)

Kernel driver attached for interface 0: 1

Claiming interface 0...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device

Reading string descriptors:
   String (0x03): "Generic"
   String (0x04): "USB3.0 Card Reader"
   String (0x02): "000000001536"

Reading OS string descriptor:

Reading Extended Compat ID OS Feature Descriptor (wIndex = 0x0004):

  00000000  28 00 00 00 00 01 04 00 01 00 00 00 00 00 00 00  (...............
  00000010  00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000020  00 00 00 00 00 00 00 00                          ........

Reading Extended Properties OS Feature Descriptor (wIndex = 0x0005):

  00000000  4c 00 00 00 00 01 05 00 01 00 42 00 00 00 04 00  L.........B.....
  00000010  00 00 30 00 53 00 65 00 6c 00 65 00 63 00 74 00  ..0.S.e.l.e.c.t.
  00000020  49 00 76 00 65 00 53 00 75 00 73 00 70 00 65 00  I.v.e.S.u.s.p.e.
  00000030  6e 00 64 00 45 00 6e 00 61 00 62 00 6c 00 65 00  n.d.E.n.a.b.l.e.
  00000040  64 00 00 00 04 00 00 00 00 00 00 00              d...........
Reading Max LUN:
   Max LUN = 0
Sending Inquiry:
   sent 6 CDB bytes
   received 36 bytes
   VID:PID:REV "Generic ":"MassStor":"1536"
   Mass Storage Status: 00 (Success)
Reading Capacity:
   sent 10 CDB bytes
   Pipe error
libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
libusb: info [darwin_capture_release_interface] on attempt to reattach the kernel driver got ret=-5
   Input/Output Error
mcuee@mcuees-Mac-mini arm64_debug_65e4a70 % pwd
/Users/mcuee/build/libusb/libusb/arm64_debug_65e4a70
```

mcuee, 2023-12-19:
Commit ID[ ](https://github.com/libusb/libusb/commit/d66ffcdd1227a03228efe8535cfa438f9c19c6b6) is also okay.

mcuee, 2023-12-19:
Commit ID https://github.com/libusb/libusb/commit/13a6953379fa8c94bb2476dc460439d4ba2b7c17 is bad. Probably it is the commit which causes the issue.

```
mcuee@mcuees-Mac-mini arm64_debug_13a6953 % sudo ./examples/xusb 05e3:0749
Using libusb v1.0.26.11841

Opening device 05E3:0749...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 2
           VID:PID: 05E3:0749
         bcdDevice: 1535
   iMan:iProd:iSer: 3:4:2
          nb confs: 1

Reading BOS descriptor: 2 caps
    USB 2.0 extension:
      attributes             : F0E
    USB 3.0 capabilities:
      attributes             : 00
      supported speeds       : 000E
      supported functionality: 01

Reading first configuration descriptor:
              total length: 44
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 08.06.50
       endpoint[0].address: 81
           max packet size: 0400
          polling interval: 00
                 max burst: 04   (USB 3.0)
        bytes per interval: 0000 (USB 3.0)
       endpoint[1].address: 02
           max packet size: 0400
          polling interval: 00
                 max burst: 04   (USB 3.0)
        bytes per interval: 0000 (USB 3.0)

Kernel driver attached for interface 0: yes

Claiming interface 0...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device

Reading string descriptors:
   String (0x03): "Generic"
   String (0x04): "USB3.0 Card Reader"
   String (0x02): "000000001536"

Reading OS string descriptor:

Reading Extended Compat ID OS Feature Descriptor (wIndex = 0x0004):

  00000000  28 00 00 00 00 01 04 00 01 00 00 00 00 00 00 00  (...............
  00000010  00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000020  00 00 00 00 00 00 00 00                          ........

Reading Extended Properties OS Feature Descriptor (wIndex = 0x0005):

  00000000  4c 00 00 00 00 01 05 00 01 00 42 00 00 00 04 00  L.........B.....
  00000010  00 00 30 00 53 00 65 00 6c 00 65 00 63 00 74 00  ..0.S.e.l.e.c.t.
  00000020  49 00 76 00 65 00 53 00 75 00 73 00 70 00 65 00  I.v.e.S.u.s.p.e.
  00000030  6e 00 64 00 45 00 6e 00 61 00 62 00 6c 00 65 00  n.d.E.n.a.b.l.e.
  00000040  64 00 00 00 04 00 00 00 00 00 00 00              d...........

Reading interface association descriptors (IADs) for first configuration:
[... 78 lines cut ...]
  0x000105500480: fa fa fa fa fa fa fa fa 00 00 00 00 00 00 00 00
  0x000105500500: 00 00 00 00 00 00 00 00 fa fa fa fa fa fa fa fa
  0x000105500580: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==59347==ABORTING
zsh: abort      sudo ./examples/xusb 05e3:0749
```

RReverser, 2023-12-19:
> SUMMARY: AddressSanitizer: heap-use-after-free darwin_usb.c:2090 in darwin_reenumerate_device

Should be the most useful bit of information.

seanm, 2023-12-19:
Thread sanitizer could be revealing here too, since it's one thread that freed the memory, and a different one reading it after free...

seanm, 2023-12-19:
Hmm, this doesn't look quite thread safe to me:

https://github.com/libusb/libusb/blob/37dee8f9dd429558e4d5bc4a804ab586abe7262a/libusb/os/darwin_usb.c#L2045-L2050

Isn't an atomic-test-and-set needed here?

mcuee, 2023-12-20:
> Thread sanitizer could be revealing here too, since it's one thread that freed the memory, and a different one reading it after free...

I will try this.
Ref:
https://developer.apple.com/documentation/xcode/diagnosing-memory-thread-and-crash-issues-early

mcuee, 2023-12-20:
> Hmm, this doesn't look quite thread safe to me:
> 
> https://github.com/libusb/libusb/blob/37dee8f9dd429558e4d5bc4a804ab586abe7262a/libusb/os/darwin_usb.c#L2045-L2050
> 
> Isn't an atomic-test-and-set needed here?

@seanm 

What will be your suggestion to fix this? I can test it.

tormodvolden, 2023-12-22:
@mcuee I made an attempt at following up Sean's suggestion in #1399, maybe you try it out.

mcuee, 2023-12-23:
@tormodvolden 

Unfortunately PR #1399 does not fix the issue.

```
mcuee@mcuees-Mac-mini arm64_debug % pwd
/Users/mcuee/build/libusb/libusb_pr1399/arm64_debug
mcuee@mcuees-Mac-mini arm64_debug % sudo ./examples/xusb 05e3:0749
Using libusb v1.0.27.11855

Opening device 05E3:0749...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 2
           VID:PID: 05E3:0749
         bcdDevice: 1535
   iMan:iProd:iSer: 3:4:2
          nb confs: 1

Reading BOS descriptor: 2 caps
    USB 2.0 extension:
      attributes             : F0E
    USB 3.0 capabilities:
      attributes             : 00
      supported speeds       : 000E
      supported functionality: 01

Reading first configuration descriptor:
              total length: 44
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 08.06.50
       endpoint[0].address: 81
           max packet size: 0400
          polling interval: 00
                 max burst: 04   (USB 3.0)
        bytes per interval: 0000 (USB 3.0)
       endpoint[1].address: 02
           max packet size: 0400
          polling interval: 00
                 max burst: 04   (USB 3.0)
        bytes per interval: 0000 (USB 3.0)

Kernel driver attached for interface 0: yes

Claiming interface 0...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device

Reading string descriptors:
   String (0x03): "Generic"
   String (0x04): "USB3.0 Card Reader"
   String (0x02): "000000001536"

Reading OS string descriptor:

Reading Extended Compat ID OS Feature Descriptor (wIndex = 0x0004):

  00000000  28 00 00 00 00 01 04 00 01 00 00 00 00 00 00 00  (...............
  00000010  00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000020  00 00 00 00 00 00 00 00                          ........

Reading Extended Properties OS Feature Descriptor (wIndex = 0x0005):

  00000000  4c 00 00 00 00 01 05 00 01 00 42 00 00 00 04 00  L.........B.....
  00000010  00 00 30 00 53 00 65 00 6c 00 65 00 63 00 74 00  ..0.S.e.l.e.c.t.
  00000020  49 00 76 00 65 00 53 00 75 00 73 00 70 00 65 00  I.v.e.S.u.s.p.e.
[... 82 lines cut ...]
  0x000105f00480: fa fa fa fa fa fa fa fa 00 00 00 00 00 00 00 00
  0x000105f00500: 00 00 00 00 00 00 00 00 fa fa fa fa fa fa fa fa
  0x000105f00580: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==14145==ABORTING
zsh: abort      sudo ./examples/xusb 05e3:0749
```

mcuee, 2023-12-23:
> > Thread sanitizer could be revealing here too, since it's one thread that freed the memory, and a different one reading it after free...
> 
> I will try this. Ref: https://developer.apple.com/documentation/xcode/diagnosing-memory-thread-and-crash-issues-early

The following is the output of ThreadSanitizer using PR #1399.

<details>
<summary> Click for the output of xusb with Thread Sanitizer </summary>

```
mcuee@mcuees-Mac-mini arm64debug_1 % pwd                                                                                    
/Users/mcuee/build/libusb/libusb_pr1399/arm64debug_1
mcuee@mcuees-Mac-mini arm64debug_1 % sudo ./examples/xusb 05e3:0749                                                          
Password:
xusb(6760,0x1e2f61000) malloc: nano zone abandoned due to inability to reserve vm space.
Using libusb v1.0.27.11855

Opening device 05E3:0749...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 2
           VID:PID: 05E3:0749
         bcdDevice: 1535
   iMan:iProd:iSer: 3:4:2
          nb confs: 1

==================
WARNING: ThreadSanitizer: data race (pid=6760)
  Read of size 8 at 0x000107d03f48 by thread T1:
    #0 usbi_get_transfer_priv libusbi.h:883 (libusb-1.0.0.dylib:arm64+0x25ef0)
    #1 darwin_async_io_callback darwin_usb.c:2558 (libusb-1.0.0.dylib:arm64+0x25f70)
    #2 IODispatchCalloutFromCFMessage <null>:58379908 (IOKit:arm64e+0x62b8)

  Previous write of size 8 at 0x000107d03f48 by main thread:
    #0 libusb_alloc_transfer io.c:1313 (libusb-1.0.0.dylib:arm64+0x118d0)
    #1 libusb_control_transfer sync.c:114 (libusb-1.0.0.dylib:arm64+0x162e0)
    #2 libusb_get_descriptor libusb.h:2034 (libusb-1.0.0.dylib:arm64+0xb8e8)
    #3 libusb_get_bos_descriptor descriptor.c:857 (libusb-1.0.0.dylib:arm64+0xb5d4)
    #4 test_device xusb.c:872 (xusb:arm64+0x100001bc0)
    #5 main xusb.c:1185 (xusb:arm64+0x1000016b0)

  Location is heap block of size 272 at 0x000107d03e80 allocated by main thread:
    #0 calloc <null>:58379908 (libclang_rt.tsan_osx_dynamic.dylib:arm64e+0x562cc)
    #1 libusb_alloc_transfer io.c:1307 (libusb-1.0.0.dylib:arm64+0x11868)
    #2 libusb_control_transfer sync.c:114 (libusb-1.0.0.dylib:arm64+0x162e0)
    #3 libusb_get_descriptor libusb.h:2034 (libusb-1.0.0.dylib:arm64+0xb8e8)
    #4 libusb_get_bos_descriptor descriptor.c:857 (libusb-1.0.0.dylib:arm64+0xb5d4)
    #5 test_device xusb.c:872 (xusb:arm64+0x100001bc0)
    #6 main xusb.c:1185 (xusb:arm64+0x1000016b0)

  Thread T1 (tid=28440, running) created by main thread at:
    #0 pthread_create <null>:58379908 (libclang_rt.tsan_osx_dynamic.dylib:arm64e+0x3062c)
    #1 darwin_first_time_init darwin_usb.c:865 (libusb-1.0.0.dylib:arm64+0x1be4c)
    #2 darwin_init_context darwin_usb.c:892 (libusb-1.0.0.dylib:arm64+0x1bbf4)
    #3 darwin_init darwin_usb.c:904 (libusb-1.0.0.dylib:arm64+0x18b54)
    #4 libusb_init_context core.c:2523 (libusb-1.0.0.dylib:arm64+0x90c0)
    #5 main xusb.c:1172 (xusb:arm64+0x1000015a4)

SUMMARY: ThreadSanitizer: data race libusbi.h:883 in usbi_get_transfer_priv
==================
==================
WARNING: ThreadSanitizer: data race (pid=6760)
  Read of size 8 at 0x000107d03f00 by thread T1:
    #0 darwin_async_io_callback darwin_usb.c:2560 (libusb-1.0.0.dylib:arm64+0x25f84)
    #1 IODispatchCalloutFromCFMessage <null>:58379908 (IOKit:arm64e+0x62b8)

  Previous write of size 8 at 0x000107d03f00 by main thread:
    #0 libusb_submit_transfer io.c:1504 (libusb-1.0.0.dylib:arm64+0x11c7c)
[... 1242 lines cut ...]
    #2 do_sync_bulk_transfer sync.c:185 (libusb-1.0.0.dylib:arm64+0x16e4c)
    #3 libusb_bulk_transfer sync.c:283 (libusb-1.0.0.dylib:arm64+0x16d34)
    #4 test_mass_storage xusb.c:514 (xusb:arm64+0x100004b9c)
    #5 test_device xusb.c:1019 (xusb:arm64+0x100002e78)
    #6 main xusb.c:1185 (xusb:arm64+0x1000016b0)

  Thread T1 (tid=28440, running) created by main thread at:
    #0 pthread_create <null>:58379908 (libclang_rt.tsan_osx_dynamic.dylib:arm64e+0x3062c)
    #1 darwin_first_time_init darwin_usb.c:865 (libusb-1.0.0.dylib:arm64+0x1be4c)
    #2 darwin_init_context darwin_usb.c:892 (libusb-1.0.0.dylib:arm64+0x1bbf4)
    #3 darwin_init darwin_usb.c:904 (libusb-1.0.0.dylib:arm64+0x18b54)
    #4 libusb_init_context core.c:2523 (libusb-1.0.0.dylib:arm64+0x90c0)
    #5 main xusb.c:1172 (xusb:arm64+0x1000015a4)

SUMMARY: ThreadSanitizer: data race darwin_usb.c:2560 in darwin_async_io_callback
==================
   Pipe error
libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
libusb: info [darwin_capture_release_interface] on attempt to reattach the kernel driver got ret=-5
   Input/Output Error
ThreadSanitizer: reported 37 warnings
zsh: abort      sudo ./examples/xusb 05e3:0749
```

</details>

seanm, 2023-12-23:
I repro those TSan issues. I'll look at them over xmas if no one beats me to it...

mcuee, 2023-12-29:
@seanm 

Just want to confirm that you can reproduce the segfault issue as well, right. Thanks.

seanm, 2023-12-29:
@mcuee I've found a few issues and have gone down a pretty deep rabbit hole... I've started creating some PRs for others to comment on, and I'll have a few more coming today...

mcuee, 2024-01-03:
@seanm 
So you think that Issue #1386 is related to Issue #1366, right?

* #1386
* #1366

seanm, 2024-01-05:
>@mcuee Can you please try benbryantzoro1's patch from https://github.com/libusb/libusb/issues/1366?

Note, the patch has a copy-pasteo:

```
+    usbi_mutex_lock(&darwin_cached_devices_mutex);
+    darwin_deref_cached_device(cached_device);
+    usbi_mutex_lock(&darwin_cached_devices_mutex);
```

Which was helpfully caught by clang's -Wthread-safety as part of https://github.com/libusb/libusb/pull/1419 with:

`warning: Acquiring mutex 'darwin_cached_devices_mutex' that is already held`

(I think -Wthread-safety is going to be very valuable!)

After changing it to `usbi_mutex_unlock`, it does fix the double free that ASan detects.

I'm not sure about its correctness though. Notably, the change of `(*old_device->device)->Release(old_device->device)` to `darwin_deref_cached_device(old_device)` seems to directly contradict the comment a few lines above "do not dereference the device at this time". But that comment may be bunk.

Also, even with this change, TSan still reports races.  So there's probably additional problems here too.

Also, @mcuee when you bisected, were you running ASan or TSan, or no sanitizers?  Since this bug looks threading-related, it may have sometimes unluckily *not* reproduced for you, leading you to false bisect conclusions.  Perhaps this bug actually is in 1.0.26?

mcuee, 2024-01-05:
> Also, @mcuee when you bisected, were you running ASan or TSan, or no sanitizers? Since this bug looks threading-related, it may have sometimes unluckily _not_ reproduced for you, leading you to false bisect conclusions. Perhaps this bug actually is in 1.0.26?

@seanm 

AScan is used.

I have asked @tormodvolden to create a version of `stress_mt` for libusb-1.0.26 and I will try it under macOS as well.
* https://github.com/libusb/libusb/issues/1372#issuecomment-1873255729

```
git checkout v1.0.26
git checkout v1.0.27-rc1 tests/stress_mt.c
git checkout ab612960 -- tests/Makefile.am
sed -i 's/libusb_init_context\([^,]*\)[^)]*/libusb_init\1/' tests/stress_mt.c
```

mcuee, 2024-01-05:
> > @mcuee Can you please try benbryantzoro1's patch from #1366?
> 
> Note, the patch has a copy-pasteo:
> 
> ```
> +    usbi_mutex_lock(&darwin_cached_devices_mutex);
> +    darwin_deref_cached_device(cached_device);
> +    usbi_mutex_lock(&darwin_cached_devices_mutex); --> this should be unlock
> ```

So the patch should be updated to the following. I will try it under macOS.

```
@@ -40,7 +40,7 @@

 /* Default timeout to 10s for reenumerate. This is needed because USBDeviceReEnumerate
  * does not return error status on macOS. */
-#define DARWIN_REENUMERATE_TIMEOUT_US (10 * USEC_PER_SEC)
+#define DARWIN_REENUMERATE_TIMEOUT_US (20 * USEC_PER_SEC)

 #include <AvailabilityMacros.h>
 #if MAC_OS_X_VERSION_MIN_REQUIRED >= 1060 && MAC_OS_X_VERSION_MIN_REQUIRED < 101200
@@ -402,11 +402,7 @@ static void darwin_devices_detached (void *ptr, io_iterator_t rem_devices) {
                     session, locationID);

           /* the device object is no longer usable so go ahead and release it */
-          if (old_device->device) {
-            (*(old_device->device))->Release(old_device->device);
-            old_device->device = NULL;
-          }
-
+          darwin_deref_cached_device(old_device);
           is_reenumerating = true;
         } else {
           darwin_deref_cached_device (old_device);
@@ -1093,7 +1089,9 @@ static enum libusb_error darwin_get_cached_device(struct libusb_context *ctx, io
                new_device->dev_descriptor.bDeviceClass, new_device->dev_descriptor.bDeviceSubClass);
     }
   } while (0);
-
+  if (ret == 0 && new_device) {
+    darwin_ref_cached_device(new_device);
+  }
   usbi_mutex_unlock(&darwin_cached_devices_mutex);

   return ret;
@@ -1213,6 +1211,9 @@ static enum libusb_error darwin_scan_devices(struct libusb_context *ctx) {
     (void) process_new_device (ctx, cached_device, old_session_id);

     IOObjectRelease(service);
+    usbi_mutex_lock(&darwin_cached_devices_mutex);
+    darwin_deref_cached_device(cached_device);
+    usbi_mutex_unlock(&darwin_cached_devices_mutex);
   }

   IOObjectRelease(deviceIterator);
~

```

mcuee, 2024-01-05:
@seanm and @tormodvolden 

Just wondering if you can reproduce this segfault issue under your Mac machine. You do not need any special HW, just any devices with build-in Apple kernel driver (like USB HID device or USB Mass Storage).

seanm, 2024-01-05:
@mcuee yes, I reproduce using `xusb` as root connecting to a mouse.

mcuee, 2024-01-05:
> Also, @mcuee when you bisected, were you running ASan or TSan, or no sanitizers? Since this bug looks threading-related, it may have sometimes unluckily _not_ reproduced for you, leading you to false bisect conclusions. Perhaps this bug actually is in 1.0.26?

@seanm 

AScan is used in all the bisected build.

libusb-1.0.26 is okay, please refer to the previous comment in this thread (again build with AScan). I will test with TScan as well later.
* https://github.com/libusb/libusb/issues/1386#issuecomment-1862111386

seanm, 2024-01-05:
Ah ha! I found what part of 13a6953379fa8c94bb2476dc460439d4ba2b7c17 caused this... I'll make a surgical patch to "fix" this...

seanm, 2024-01-05:
https://github.com/libusb/libusb/pull/1427

mcuee, 2024-01-05:
@seanm and @tormodvolden 

Unfortunately PR #1427 does not fix the issue yet.

```
git clone https://github.com/seanm/libusb.git -b issues1386 libusb_pr1427
cd libusb_pr1427
./bootstrap.sh
mkdir arm64_debug
cd arm64_debug
CFLAGS="-g -fsanitize=address" ../configure --enable-tests-build --enable-examples-build
make
mcuee@mcuees-Mac-mini arm64_debug % sudo ./examples/xusb 05e3:0749
Using libusb v1.0.27.11855

Opening device 05E3:0749...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 2
           VID:PID: 05E3:0749
         bcdDevice: 1535
   iMan:iProd:iSer: 3:4:2
          nb confs: 1

Reading BOS descriptor: 2 caps
    USB 2.0 extension:
      attributes             : F0E
    USB 3.0 capabilities:
      attributes             : 00
      supported speeds       : 000E
      supported functionality: 01

Reading first configuration descriptor:
              total length: 44
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 08.06.50
       endpoint[0].address: 81
           max packet size: 0400
          polling interval: 00
                 max burst: 04   (USB 3.0)
        bytes per interval: 0000 (USB 3.0)
       endpoint[1].address: 02
           max packet size: 0400
          polling interval: 00
                 max burst: 04   (USB 3.0)
        bytes per interval: 0000 (USB 3.0)

Kernel driver attached for interface 0: yes

Claiming interface 0...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device

Reading string descriptors:
   String (0x03): "Generic"
   String (0x04): "USB3.0 Card Reader"
   String (0x02): "000000001536"

Reading OS string descriptor:

Reading Extended Compat ID OS Feature Descriptor (wIndex = 0x0004):

  00000000  28 00 00 00 00 01 04 00 01 00 00 00 00 00 00 00  (...............
  00000010  00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000020  00 00 00 00 00 00 00 00                          ........

[... 85 lines cut ...]
  0x000103d00900: fa fa fa fa fa fa fa fa 00 00 00 00 00 00 00 00
  0x000103d00980: 00 00 00 00 00 00 00 00 fa fa fa fa fa fa fa fa
  0x000103d00a00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==22412==ABORTING
zsh: abort      sudo ./examples/xusb 05e3:0749
```

mcuee, 2024-01-05:
@seanm 
Just wondering if PR #1427 fixed your issue when testing with the mouse. If that is the case, maybe you want to try another devices.

mcuee, 2024-01-05:
@tormodvolden 

Apparently #1266 is causing the regression but then it fixed some other issues. So I am not suggesting to reverse the PR.

On the other hand, if I want to carry out tests by reversing that PR, what should I do? Just for testing purpose.

seanm, 2024-01-05:
@mcuee I've tried 4 different USB devices, and with all of them I repro consistently running master, and cannot repro with PR #1427 applied (tried 5x per device).

seanm, 2024-01-05:
@mcuee I've updated the patch with a few more similar reversions in other functions. can you try again?

tormodvolden, 2024-01-05:
> Apparently #1266 is causing the regression but then it fixed some other issues.

It fixes some cases of running a macOS binary on another macOS version than it was compiled on. Which is a long-standing issue, so we wouldn't cause a regression if we released without this commit. So in hindsight it might have been better to release 1.0.27 earlier, without waiting for that fix to get ready.

> On the other hand, if I want to carry out tests by reversing that PR, what should I do?

It would be some work to revert it for testing, but if you instead check out the commit just before, 43c6fe0b5, there has hardly been any commits in master afterwards that would affect the darwin build. Maybe cherry-pick commit 87d6686a for stress_mt.

mcuee, 2024-01-05:
> @mcuee I've updated the patch with a few more similar reversions in other functions. can you try again?

@seanm 

Great, now it works fine for my tests.

I run the following command 5 times and it works fine.

```
mcuee@mcuees-Mac-mini arm64_debug % sudo ./examples/xusb 05e3:0749
Using libusb v1.0.27.11855

Opening device 05E3:0749...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 2
           VID:PID: 05E3:0749
         bcdDevice: 1535
   iMan:iProd:iSer: 3:4:2
          nb confs: 1

Reading BOS descriptor: 2 caps
    USB 2.0 extension:
      attributes             : F0E
    USB 3.0 capabilities:
      attributes             : 00
      supported speeds       : 000E
      supported functionality: 01

Reading first configuration descriptor:
              total length: 44
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 08.06.50
       endpoint[0].address: 81
           max packet size: 0400
          polling interval: 00
                 max burst: 04   (USB 3.0)
        bytes per interval: 0000 (USB 3.0)
       endpoint[1].address: 02
           max packet size: 0400
          polling interval: 00
                 max burst: 04   (USB 3.0)
        bytes per interval: 0000 (USB 3.0)

Kernel driver attached for interface 0: yes

Claiming interface 0...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device

Reading string descriptors:
   String (0x03): "Generic"
   String (0x04): "USB3.0 Card Reader"
   String (0x02): "000000001536"

Reading OS string descriptor:

Reading Extended Compat ID OS Feature Descriptor (wIndex = 0x0004):

  00000000  28 00 00 00 00 01 04 00 01 00 00 00 00 00 00 00  (...............
  00000010  00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00000020  00 00 00 00 00 00 00 00                          ........

Reading Extended Properties OS Feature Descriptor (wIndex = 0x0005):

  00000000  4c 00 00 00 00 01 05 00 01 00 42 00 00 00 04 00  L.........B.....
  00000010  00 00 30 00 53 00 65 00 6c 00 65 00 63 00 74 00  ..0.S.e.l.e.c.t.
  00000020  49 00 76 00 65 00 53 00 75 00 73 00 70 00 65 00  I.v.e.S.u.s.p.e.
  00000030  6e 00 64 00 45 00 6e 00 61 00 62 00 6c 00 65 00  n.d.E.n.a.b.l.e.
  00000040  64 00 00 00 04 00 00 00 00 00 00 00              d...........

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Reading Max LUN:
   Max LUN = 0

Sending Inquiry:
   sent 6 CDB bytes
   received 36 bytes
   VID:PID:REV "Generic ":"MassStor":"1536"
   Mass Storage Status: 00 (Success)

Reading Capacity:
   sent 10 CDB bytes
   Pipe error
libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
libusb: info [darwin_capture_release_interface] on attempt to reattach the kernel driver got ret=-5
   Input/Output Error
```

tormodvolden, 2024-01-05:
Wonderful! Maybe we now have a possibility to release 1.0.27. There are certainly underlying issues (on darwin) that should be fixed, but these may have been there since a long time, and shouldn't hold up the release.

mcuee, 2024-01-06:
More tests: I ran each commands 5 times and it works fine without an issue.

You can see that my USB device topology is actually a bit complicated.

Mac Mini M1 -- USB 3.0 Hub -- USB 2.0 Hub

```
mcuee@mcuees-Mac-mini arm64_debug % ./examples/testlibusb 
Dev (bus 2, device 15): 05E3 - 0749 speed: 5G
  Manufacturer:              Generic
  Product:                   USB3.0 Card Reader
Dev (bus 2, device 8): 1915 - 1025 speed: 12M
  Manufacturer:              ZY.Ltd
  Product:                   ZY Control Mic
Dev (bus 2, device 7): 1A40 - 0201 speed: 480M
  Product:                   USB 2.0 Hub [MTT]
Dev (bus 2, device 6): 1A40 - 0201 speed: 480M
  Product:                   USB 2.0 Hub [MTT]
Dev (bus 2, device 5): 0BDA - 5411 speed: 480M
  Manufacturer:              Generic
  Product:                   4-Port USB 2.0 Hub
Dev (bus 2, device 3): 0BDA - 0411 speed: 5G
  Manufacturer:              Generic
  Product:                   4-Port USB 3.0 Hub
Dev (bus 2, device 2): 0BDA - 0411 speed: 5G
  Manufacturer:              Generic
  Product:                   4-Port USB 3.0 Hub
Dev (bus 2, device 1): 0BDA - 5411 speed: 480M
  Manufacturer:              Generic
  Product:                   4-Port USB 2.0 Hub
mcuee@mcuees-Mac-mini arm64_debug % ./examples/listdevs  
05e3:0749 (bus 2, device 15) path: 5.1.1
1915:1025 (bus 2, device 8) path: 1.4.5.6
1a40:0201 (bus 2, device 7) path: 1.4.5
1a40:0201 (bus 2, device 6) path: 1.4
0bda:5411 (bus 2, device 5) path: 1.1
0bda:0411 (bus 2, device 3) path: 5.1
0bda:0411 (bus 2, device 2) path: 5
0bda:5411 (bus 2, device 1) path: 1
mcuee@mcuees-Mac-mini arm64_debug % sudo ./examples/xusb 1a40:0201
Password:
Using libusb v1.0.27.11855

Opening device 1A40:0201...
libusb: warning [darwin_open] USBDeviceOpen: another process has device opened for exclusive access

Reading device descriptor:
            length: 18
      device class: 9
               S/N: 0
           VID:PID: 1A40:0201
         bcdDevice: 0100
   iMan:iProd:iSer: 0:1:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 41
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 09.00.01
       endpoint[0].address: 81
           max packet size: 0001
          polling interval: 0C
interface[0].altsetting[1]: num endpoints = 1
   Class.SubClass.Protocol: 09.00.02
       endpoint[0].address: 81
           max packet size: 0001
          polling interval: 0C

   Failed (error -5) Entity not found
Kernel driver attached for interface 0: 
Claiming interface 0...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device

Reading string descriptors:
   String (0x01): "USB 2.0 Hub [MTT]"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
libusb: warning [darwin_open] USBDeviceOpen: another process has device opened for exclusive access
libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
libusb: info [darwin_capture_release_interface] on attempt to reattach the kernel driver got ret=-5
Closing device...

mcuee@mcuees-Mac-mini arm64_debug % sudo ./examples/xusb 1915:1025
Using libusb v1.0.27.11855

Opening device 1915:1025...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 1915:1025
         bcdDevice: 0173
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 158
         descriptor length: 9
             nb interfaces: 4
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 0
   Class.SubClass.Protocol: 01.01.00
              interface[1]: id = 1
interface[1].altsetting[0]: num endpoints = 0
   Class.SubClass.Protocol: 01.02.00
interface[1].altsetting[1]: num endpoints = 1
   Class.SubClass.Protocol: 01.02.00
       endpoint[0].address: 87
           max packet size: 0010
          polling interval: 01
              interface[2]: id = 2
interface[2].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.01.01
       endpoint[0].address: 81
           max packet size: 0020
          polling interval: 08
              interface[3]: id = 3
interface[3].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 82
           max packet size: 0020
          polling interval: 08

Kernel driver attached for interface 0: yes

Claiming interface 0...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device

Kernel driver attached for interface 1: none

Claiming interface 1...

Kernel driver attached for interface 2: none

Claiming interface 2...

Kernel driver attached for interface 3: none

Claiming interface 3...

Reading string descriptors:
   String (0x01): "ZY.Ltd"
   String (0x02): "ZY Control Mic"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
libusb: info [darwin_capture_release_interface] on attempt to reattach the kernel driver got ret=-5
Releasing interface 1...
Releasing interface 2...
Releasing interface 3...
Closing device...
```

mcuee, 2024-01-06:
> Wonderful! Maybe we now have a possibility to release 1.0.27. There are certainly underlying issues (on darwin) that should be fixed, but these may have been there since a long time, and shouldn't hold up the release.

@tormodvolden 

I agree.

@seanm

Thanks a lot in fixing this issue.

commit 9401e6c7 (2024-01-19): darwin: Revert seemingly harmless introduction of temporary variable

Commit 13a69533 introduced a temporary variable for the device structure
(that originally was meant to be reverted, it turns out).

Here is a follow-up to revert this part, to avoid threading issues and
reported crashes.

The temporary variable would be harmless if there was no multithreading
happening, but there is:

On the hotplug background thread, darwin_devices_detached() is called,
which in turn calls Release() on the device, which frees the memory, and
then sets old_device->device to NULL. Shortly after,
darwin_device [... cut]
closed 2024-01-19 by tormodvolden