libusb/libusb #1387: Haiku OS: build fails

tormodvolden, 2023-12-20:
For Haiku OS, the build issue is still there, similar to 1.0.26 release. gcc version is 11.2.0.
https://www.haiku-os.org/get-haiku/r1beta4/
* https://github.com/libusb/libusb/issues/1099#issuecomment-1080067439

<details>
<summary> build log -- failed under Haiku OS R1 Beta4 </summary>

```
~/Desktop/build/libusb-1.0.27-rc1> make -i
make  all-recursive
make[1]: Entering directory '/boot/home/Desktop/build/libusb-1.0.27-rc1'
Making all in libusb
make[2]: Entering directory '/boot/home/Desktop/build/libusb-1.0.27-rc1/libusb'
  CXX      os/haiku_usb_backend.lo
cc1plus: warning: '-Werror=' argument '-Werror=implicit-function-declaration' is not valid for C++
cc1plus: warning: '-Werror=' argument '-Werror=implicit-int' is not valid for C++
cc1plus: warning: '-Werror=' argument '-Werror=missing-prototypes' is not valid for C++
cc1plus: warning: '-Werror=' argument '-Werror=strict-prototypes' is not valid for C++
In file included from ./libusbi.h:106,
                 from os/haiku_usb.h:25,
                 from os/haiku_usb_backend.cpp:27:
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:40:9: error: '_Atomic' does not name a type
   40 | typedef _Atomic _Bool atomic_bool;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:41:9: error: '_Atomic' does not name a type
   41 | typedef _Atomic char atomic_char;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:42:9: error: '_Atomic' does not name a type
   42 | typedef _Atomic signed char atomic_schar;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:43:9: error: '_Atomic' does not name a type
   43 | typedef _Atomic unsigned char atomic_uchar;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:44:9: error: '_Atomic' does not name a type
   44 | typedef _Atomic short atomic_short;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:45:9: error: '_Atomic' does not name a type
   45 | typedef _Atomic unsigned short atomic_ushort;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:46:9: error: '_Atomic' does not name a type
   46 | typedef _Atomic int atomic_int;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:47:9: error: '_Atomic' does not name a type
   47 | typedef _Atomic unsigned int atomic_uint;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:48:9: error: '_Atomic' does not name a type
   48 | typedef _Atomic long atomic_long;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:49:9: error: '_Atomic' does not name a type
   49 | typedef _Atomic unsigned long atomic_ulong;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:50:9: error: '_Atomic' does not name a type
   50 | typedef _Atomic long long atomic_llong;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:51:9: error: '_Atomic' does not name a type
   51 | typedef _Atomic unsigned long long atomic_ullong;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:52:9: error: '_Atomic' does not name a type
   52 | typedef _Atomic __CHAR16_TYPE__ atomic_char16_t;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:53:9: error: '_Atomic' does not name a type
   53 | typedef _Atomic __CHAR32_TYPE__ atomic_char32_t;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:54:9: error: '_Atomic' does not name a type
   54 | typedef _Atomic __WCHAR_TYPE__ atomic_wchar_t;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:55:9: error: '_Atomic' does not name a type
   55 | typedef _Atomic __INT_LEAST8_TYPE__ atomic_int_least8_t;
      |         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/11.2.0/include/stdatomic.h:56:9: error: '_Atomic' does not name a type
[... 426 lines cut ...]
      |         sig_atomic_t
Makefile:708: recipe for target 'os/haiku_usb_raw.lo' failed
make[2]: [os/haiku_usb_raw.lo] Error 1 (ignored)
  CXXLD    libusb_haiku.la
libtool:   error: 'os/haiku_usb_backend.lo' is not a valid libtool object
Makefile:625: recipe for target 'libusb_haiku.la' failed
make[2]: [libusb_haiku.la] Error 1 (ignored)
  CCLD     libusb-1.0.la
libtool:   error: cannot find the library 'libusb_haiku.la' or unhandled argument 'libusb_haiku.la'
Makefile:613: recipe for target 'libusb-1.0.la' failed
make[2]: [libusb-1.0.la] Error 1 (ignored)
make[2]: Leaving directory '/boot/home/Desktop/build/libusb-1.0.27-rc1/libusb'
Making all in examples
make[2]: Entering directory '/boot/home/Desktop/build/libusb-1.0.27-rc1/examples'
  CC       dpfp.o
make[2]: *** No rule to make target '../libusb/libusb-1.0.la', needed by 'dpfp'.  Stop.
make[2]: Leaving directory '/boot/home/Desktop/build/libusb-1.0.27-rc1/examples'
Makefile:491: recipe for target 'all-recursive' failed
make[1]: [all-recursive] Error 1 (ignored)
make[1]: Leaving directory '/boot/home/Desktop/build/libusb-1.0.27-rc1'
```

</details>

_Originally posted by @mcuee in https://github.com/libusb/libusb/issues/1372#issuecomment-1864393268_

tormodvolden, 2023-12-20:
@pulkomandy @i80and Would you be interested in helping out to get libusb to build on Haiku OS again?

pulkomandy, 2023-12-20:
Hello,

The gcc version used in Haiku is currently gcc 13. You can update to that from the beta4 image by running "pkgman full-sync" and rebooting (I don't know how your Haiku CI is done).

We are packaging libusb in haikuports with a patch reverting to nonstandard gcc atomics: https://github.com/haikuports/haikuports/blob/d259b85e59a130ae28877b4892376c52f1fdbcfc/dev-libs/libusb/patches/libusb-1.0.26.patchset

I don't know if that's still needed with gcc13. Would that be acceptable (with an appropriate platform check to do it only on Haiku?)

mcuee, 2023-12-20:
@pulkomandy 

Thanks for the tip. I will upgrade the beta4 image using `pkgman full-sync` and test later.

@tormodvolden 

In case the build still fails with gcc 13, I tend to think it is okay to treat Haiku OS specially using the patch.

mcuee, 2023-12-20:
The issue is still there with gcc 13.2.0.

```
~/Desktop/build/libusb-1.0.27-rc1> make -i
make  all-recursive
make[1]: Entering directory '/boot/home/Desktop/build/libusb-1.0.27-rc1'
Making all in libusb
make[2]: Entering directory '/boot/home/Desktop/build/libusb-1.0.27-rc1/libusb'
  CXX      os/haiku_usb_backend.lo
cc1plus: warning: '-Werror=' argument '-Werror=implicit-function-declaration' is not valid for C++
cc1plus: warning: '-Werror=' argument '-Werror=implicit-int' is not valid for C++
cc1plus: warning: '-Werror=' argument '-Werror=missing-prototypes' is not valid for C++
cc1plus: warning: '-Werror=' argument '-Werror=strict-prototypes' is not valid for C++
In file included from os/haiku_usb.h:25,
                 from os/haiku_usb_backend.cpp:27:
./libusbi.h:107:9: error: 'atomic_long' does not name a type
  107 | typedef atomic_long usbi_atomic_t;
      |         ^~~~~~~~~~~
./libusbi.h:374:9: error: 'usbi_atomic_t' does not name a type; did you mean 'sig_atomic_t'?
  374 |         usbi_atomic_t hotplug_ready;
      |         ^~~~~~~~~~~~~
      |         sig_atomic_t
./libusbi.h:499:9: error: 'usbi_atomic_t' does not name a type; did you mean 'sig_atomic_t'?
  499 |         usbi_atomic_t refcnt;
      |         ^~~~~~~~~~~~~
      |         sig_atomic_t
./libusbi.h:513:9: error: 'usbi_atomic_t' does not name a type; did you mean 'sig_atomic_t'?
  513 |         usbi_atomic_t attached;
      |         ^~~~~~~~~~~~~
      |         sig_atomic_t
Makefile:708: recipe for target 'os/haiku_usb_backend.lo' failed
make[2]: [os/haiku_usb_backend.lo] Error 1 (ignored)
  CXX      os/haiku_pollfs.lo
cc1plus: warning: '-Werror=' argument '-Werror=implicit-function-declaration' is not valid for C++
cc1plus: warning: '-Werror=' argument '-Werror=implicit-int' is not valid for C++
cc1plus: warning: '-Werror=' argument '-Werror=missing-prototypes' is not valid for C++
cc1plus: warning: '-Werror=' argument '-Werror=strict-prototypes' is not valid for C++
In file included from os/haiku_usb.h:25,
                 from os/haiku_pollfs.cpp:9:
./libusbi.h:107:9: error: 'atomic_long' does not name a type
  107 | typedef atomic_long usbi_atomic_t;
      |         ^~~~~~~~~~~
./libusbi.h:374:9: error: 'usbi_atomic_t' does not name a type; did you mean 'sig_atomic_t'?
  374 |         usbi_atomic_t hotplug_ready;
      |         ^~~~~~~~~~~~~
      |         sig_atomic_t
./libusbi.h:499:9: error: 'usbi_atomic_t' does not name a type; did you mean 'sig_atomic_t'?
  499 |         usbi_atomic_t refcnt;
      |         ^~~~~~~~~~~~~
      |         sig_atomic_t
./libusbi.h:513:9: error: 'usbi_atomic_t' does not name a type; did you mean 'sig_atomic_t'?
  513 |         usbi_atomic_t attached;
      |         ^~~~~~~~~~~~~
      |         sig_atomic_t
Makefile:708: recipe for target 'os/haiku_pollfs.lo' failed
make[2]: [os/haiku_pollfs.lo] Error 1 (ignored)
  CXX      os/haiku_usb_raw.lo
cc1plus: warning: '-Werror=' argument '-Werror=implicit-function-declaration' is not valid for C++
cc1plus: warning: '-Werror=' argument '-Werror=implicit-int' is not valid for C++
cc1plus: warning: '-Werror=' argument '-Werror=missing-prototypes' is not valid for C++
cc1plus: warning: '-Werror=' argument '-Werror=strict-prototypes' is not valid for C++
In file included from os/haiku_usb.h:25,
                 from os/haiku_usb_raw.cpp:27:
./libusbi.h:107:9: error: 'atomic_long' does not name a type
  107 | typedef atomic_long usbi_atomic_t;
      |         ^~~~~~~~~~~
./libusbi.h:374:9: error: 'usbi_atomic_t' does not name a type; did you mean 'sig_atomic_t'?
  374 |         usbi_atomic_t hotplug_ready;
      |         ^~~~~~~~~~~~~
      |         sig_atomic_t
[... 12 lines cut ...]
Makefile:625: recipe for target 'libusb_haiku.la' failed
make[2]: [libusb_haiku.la] Error 1 (ignored)
  CCLD     libusb-1.0.la
libtool:   error: cannot find the library 'libusb_haiku.la' or unhandled argument 'libusb_haiku.la'
Makefile:613: recipe for target 'libusb-1.0.la' failed
make[2]: [libusb-1.0.la] Error 1 (ignored)
make[2]: Leaving directory '/boot/home/Desktop/build/libusb-1.0.27-rc1/libusb'
Making all in examples
make[2]: Entering directory '/boot/home/Desktop/build/libusb-1.0.27-rc1/examples'
  CC       dpfp.o
make[2]: *** No rule to make target '../libusb/libusb-1.0.la', needed by 'dpfp'.  Stop.
make[2]: Leaving directory '/boot/home/Desktop/build/libusb-1.0.27-rc1/examples'
Makefile:491: recipe for target 'all-recursive' failed
make[1]: [all-recursive] Error 1 (ignored)
make[1]: Leaving directory '/boot/home/Desktop/build/libusb-1.0.27-rc1'
~/Desktop/build/libusb-1.0.27-rc1> gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/boot/system/develop/tools/bin/../lib/gcc/x86_64-unknown-haiku/13.2.0/lto-wrapper
Target: x86_64-unknown-haiku
Configured with: /sources/gcc-13.2.0/configure --build=x86_64-unknown-haiku --prefix=/packages/gcc-13.2.0_2023_08_10-3/.self/develop/tools --libexecdir=/packages/gcc-13.2.0_2023_08_10-3/.self/develop/tools/lib --mandir=/packages/gcc-13.2.0_2023_08_10-3/.self/documentation/man --docdir=/packages/gcc-13.2.0_2023_08_10-3/.self/documentation/packages/gcc --enable-threads=posix --disable-nls --enable-shared --with-gnu-ld --with-gnu-as --enable-version-specific-runtime-libs --enable-languages=c,c++,fortran,jit,objc --enable-lto --enable-frame-pointer --with-pkgversion=2023_08_10 --enable-__cxa-atexit --with-system-zlib --enable-checking=release --with-bug-url=http://dev.haiku-os.org/ --with-default-libstdcxx-abi=gcc4-compatible --enable-libssp --enable-host-shared --disable-multilib
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 13.2.0 (2023_08_10) 
```

tormodvolden, 2023-12-20:
It would be nice to be able to build this out of the box without patching, so we could consider adding a tweak for Haiku. OTOH it affects very few people, since most Haiku users would get libusb from haikuports.

Ironically the change that broke Haiku generally made libusb more portable, by using c11 stdatomic instead of GCC builtins.
Why isn't c11 stdatomic available on Haiku? Does it require OS features missing on Haiku?

Apparently the compiler finds a <stdatomic.h> header but it is not standard-compliant. Is this from the "lib:libatomic" mentioned in the haikuports recipe?

The haikuports gcc-13.2.0_2023_08_10.patchset even has [patches](https://github.com/haikuports/haikuports/blob/7e5dbb0115a6c0909a7d44f934ea13e81b9656b0/sys-devel/gcc/patches/gcc-13.2.0_2023_08_10.patchset#L6908) for building libatomic on Haiku. What is going on here?

pulkomandy, 2023-12-20:
The Haiku code has parts built as C++. This results in stdatomic.h from libstdc++ being used instead of the gcc standard one. However, stdatomic.h in libstcsdc++ has a guard to be enabled only in C++23. It has an include_next that would normally include the C version, but that is guarded by an if defined(__clang__). This seems to pe a bug in libstdc++ (I checked the version in upstream gcc, this is not a change coming from Haiku), I think most of the content of this file should be enabled for c++11.

Changing -std=gnu++11 to -std=c++23 to build this file fixes the problem. I'm not sure how the libusb buildsystem decided to use "gnu++11" for C++ files, since only a C language standard version is being set.

tormodvolden, 2023-12-20:
Thanks for the explanation. I see the same clang-only guard in /usr/include/c++/12/stdatomic.h on Debian 12.

Our configure.ac has:
`AM_CXXFLAGS="-std=${c_dialect}++11 ${EXTRA_CFLAGS} ${SHARED_CFLAGS} -Wmissing-declarations"`
(where c_dialect optionally is "gnu") and this was added for the sake of Haiku in commit 7c0cea7.

It seems it would be a solution to patch libstdc++ instead of libusb in Haiku, but I understand that brings more risk of breaking something.

commit 2f2e072c (2024-04-20): haiku: Use gcc atomic builtins to fix build

Fixes #1387
Closes #1482
closed 2024-04-20 by tormodvolden