libusb/libusb #1389: `make check` failed under OpenBSD 7.4

mcuee, 2023-12-20:
Reported here.
* https://github.com/libusb/libusb/issues/1372#issuecomment-1864460535

`make check` failed under OpenBSD due to similar issues like Windows. It is just the error code is now `LIBUSB_ERROR_OTHER`.
git HEAD will still fail `make check` for OpenBSD since we have not dealt with the error code.

```
openbsdvmn100$ sudo make check
Making check in libusb
Making check in examples
Making check in tests
make  check-TESTS
PASS: stress
FAIL: stress_mt
PASS: set_option
PASS: init_context
============================================================================
Testsuite summary for libusb-1.0 ..
============================================================================
# TOTAL: 4
# PASS:  3
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0
============================================================================
See tests/test-suite.log
Please report to libusb-devel@lists.sourceforge.net
============================================================================
*** Error 1 in tests (Makefile:900 'test-suite.log': @case $- in *e*) set +e;; esac; {  mgn= red= grn= lgn= blu= brg= std=;  am__color_tests...)
*** Error 2 in tests (Makefile:911 'check-TESTS': @set +e; bases='stress.log stress_mt.log set_option.log init_context.log';  bases=`for i i...)
*** Error 2 in tests (Makefile:1013 'check-am')
*** Error 1 in /home/mcuee/build/libusb (Makefile:514 'check-recursive': @fail=;  if (target_option=k; case ${target_option-} in  ?) ;;  *) ...)
openbsdvmn100$ sudo ./tests/stress_mt
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
Thread 0 failed (iteration 3): LIBUSB_ERROR_OTHER
Thread 1 failed (iteration 1): LIBUSB_ERROR_OTHER
Thread 2 failed (iteration 2): LIBUSB_ERROR_OTHER
Device count mismatch: Thread 3 discovered 3 devices instead of 0
Thread 4 failed (iteration 1): LIBUSB_ERROR_OTHER
Thread 5 failed (iteration 1): LIBUSB_ERROR_OTHER
Thread 6 failed (iteration 1): LIBUSB_ERROR_OTHER
Thread 7 failed (iteration 1): LIBUSB_ERROR_OTHER
All done, 8 errors
```

Full debug log for `stress_mt`.
[openbsd_stress_mt_failed_log.txt](https://github.com/libusb/libusb/files/13727799/openbsd_stress_mt_failed_log.txt)

mcuee, 2023-12-21:
@tormodvolden's comment for NetBSD may be applicable to OpenBSD as well.

* https://github.com/libusb/libusb/issues/1390#issuecomment-1865146031

> Different from the Windows case, where the devices simply cannot be opened, it looks like here on NetBSD it is a real bug. Thread 0 discovers no devices, but thread 1 discovers and opens one device. I don't know if the NetBSD backend even tries to be thread-safe.

mcuee, 2023-12-21:
@botovq and @kucharskim 

Just wondering if you can try with a real OpenBSD system (I am using a VM) to see if the issue is a real problem or not. Thanks.

botovq, 2023-12-21:
Looks good here on an Apple M1 mini and a Thinkpad T14g2

PASS: stress
PASS: stress_mt
PASS: set_option
PASS: init_context
============================================================================
Testsuite summary for libusb-1.0 ..
============================================================================
# TOTAL: 4
# PASS:  4
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
# ERROR: 0
============================================================================

botovq, 2023-12-21:
I should have said: That's OpenBSD 7.4-current, not release, but I don't
think all that much has changed in the USB subsystem since the release.

mcuee, 2023-12-21:
> I should have said: That's OpenBSD 7.4-current, not release, but I don't think all that much has changed in the USB subsystem since the release.

Glad to hear that you have no issues on real HW.

Just wondering how many USB device you attached to the system. For my VM, if I attach no USB devices then it will pass.

I will try with OpenBSD 7.4-current as well later.

botovq, 2023-12-21:
A few, but none really in use.

On the M1:

```
Bus 000 Device 001: ID 0000:0000
Bus 001 Device 001: ID 0000:0000
Bus 002 Device 001: ID 1b73:0000 Fresco Logic
Bus 002 Device 002: ID 0409:005a NEC Corp. HighSpeed Hub
Bus 002 Device 003: ID 0853:0100 Topre Corporation HHKB Professional
```

on the T14

```
Bus 000 Device 001: ID 1022:0000 Shinko Shoji Co., Ltd
Bus 000 Device 002: ID 1050:0407 Yubico.com
Bus 001 Device 001: ID 1022:0000 Shinko Shoji Co., Ltd
```

mcuee, 2023-12-21:
> A few, but none really in use.
>
> On the M1:
>
> ```
> Bus 000 Device 001: ID 0000:0000
> Bus 001 Device 001: ID 0000:0000
> Bus 002 Device 001: ID 1b73:0000 Fresco Logic
> Bus 002 Device 002: ID 0409:005a NEC [...]

It will be good to have some devices using the ugen driver to be attached for the testing. It seems to me the device you attached are not using ugen (either USB HUB or USB HID device).

botovq, 2023-12-21:
> It will be good to have some devices using the ugen driver to be attached for the testing. It seems to me the device you attached are not using ugen (either USB HUB or USB HID device).

Same result with

ugen0 at uhub1 port 2 "Lenovo Lenovo K6" rev 2.00/3.18 addr 2

mcuee, 2023-12-21:
Okay, then this may not be a real issue. I will check again.

mcuee, 2024-01-13:
> Okay, then this may not be a real issue. I will check again.

At least for my Proxmox PVE 8.0 environment, I still see the issue with OpenBSD latest snapshot. But then again it is not a regression as libusb-1.0.26 also has the issue. So this is not a regression and not a release blocker for 1.0.27 release.

git HEAD
```
openbsd74snapshotn100# pwd
/home/mcuee/build/libusb
openbsd74snapshotn100# ./tests/stress_mt
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
Thread 0 failed (iteration 1): LIBUSB_ERROR_OTHER
Thread 1 failed (iteration 1): LIBUSB_ERROR_OTHER
Device count mismatch: Thread 2 discovered 3 devices instead of 0
Thread 3 failed (iteration 1): LIBUSB_ERROR_OTHER
Thread 4 failed (iteration 3): LIBUSB_ERROR_OTHER
Thread 5 failed (iteration 2): LIBUSB_ERROR_OTHER
Thread 6 failed (iteration 1): LIBUSB_ERROR_OTHER
Thread 7 failed (iteration 3): LIBUSB_ERROR_OTHER
All done, 8 errors
```

libusb-1.0.26 with back ported stess_mt.c
```
openbsd74snapshotn100# pwd
/home/mcuee/build/libusb-1.0.26_stress_mt/libusb
openbsd74snapshotn100# ./tests/stress_mt
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
Thread 1 failed (iteration 1): LIBUSB_ERROR_OTHER
Thread 2 failed (iteration 3): LIBUSB_ERROR_OTHER
Thread 3 failed (iteration 3): LIBUSB_ERROR_OTHER
Thread 4 failed (iteration 1): LIBUSB_ERROR_OTHER
Thread 5 failed (iteration 1): LIBUSB_ERROR_OTHER
Thread 6 failed (iteration 9): LIBUSB_ERROR_OTHER
Thread 7 failed (iteration 1): LIBUSB_ERROR_OTHER
All threads discovered 3 devices (0 not opened)
All done, 7 errors
```

mcuee, 2024-01-13:
Tested on a physical machine and there are no issues.

Machine: Intel(R) Celeron(R) N4500 based mini PC, external USB 3.0 hub and USB 2.0 hub.

```
openbsd74$ uname -a
OpenBSD openbsd74.my.domain 7.4 GENERIC.MP#1397 amd64
openbsd74$ sudo ./examples/listdevs
Password:
8086:0000 (bus 0, device 1)
046d:c52b (bus 0, device 2) path: 1
0bda:5411 (bus 0, device 3) path: 4
0bda:5411 (bus 0, device 4) path: 1
1a40:0201 (bus 0, device 5) path: 4
1a40:0201 (bus 0, device 6) path: 5
1915:1025 (bus 0, device 7) path: 6
0bda:c821 (bus 0, device 8) path: 8
0bda:0411 (bus 0, device 9) path: 10
0bda:0411 (bus 0, device 10) path: 1
openbsd74$ sudo ./examples/testlibusb
Dev (bus 0, device 1): 8086 - 0000 speed: 5G
  Manufacturer:              Intel
  Product:                   xHCI root hub
Dev (bus 0, device 2): 046D - C52B speed: 12M
  Manufacturer:              Logitech
  Product:                   USB Receiver
Dev (bus 0, device 3): 0BDA - 5411 speed: 480M
  Manufacturer:              Generic
  Product:                   4-Port USB 2.0 Hub
Dev (bus 0, device 4): 0BDA - 5411 speed: 480M
  Manufacturer:              Generic
  Product:                   4-Port USB 2.0 Hub
Dev (bus 0, device 5): 1A40 - 0201 speed: 480M
  Product:                   USB 2.0 Hub [MTT]
Dev (bus 0, device 6): 1A40 - 0201 speed: 480M
  Product:                   USB 2.0 Hub [MTT]
Dev (bus 0, device 7): 1915 - 1025 speed: 12M
  Manufacturer:              ZY.Ltd
  Product:                   ZY Control Mic
Dev (bus 0, device 8): 0BDA - C821 speed: 12M
  Manufacturer:              Realtek
  Product:                   Bluetooth Radio
Dev (bus 0, device 9): 0BDA - 0411 speed: 5G
  Manufacturer:              Generic
  Product:                   4-Port USB 3.0 Hub
Dev (bus 0, device 10): 0BDA - 0411 speed: 5G
  Manufacturer:              Generic
  Product:                   4-Port USB 3.0 Hub
openbsd74$ sudo make check
Making check in libusb
Making check in examples
Making check in tests
make  check-TESTS
PASS: stress
PASS: stress_mt
PASS: set_option
PASS: init_context
============================================================================
Testsuite summary for libusb-1.0 ..
============================================================================
# TOTAL: 4
# PASS:  4
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
# ERROR: 0
============================================================================
openbsd74$ sudo ./tests/stress_mt
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
All threads discovered 10 devices (0 not opened)
All done, 0 errors
```

tormodvolden, 2024-01-19:
I wonder if it is related to the similar issue on darwin/macOS, where it fails on a VM sometimes but nobody saw it on real hardware.

mcuee, 2024-01-20:
> I wonder if it is related to the similar issue on darwin/macOS, where it fails on a VM sometimes but nobody saw it on real hardware.

I will close this issue for now. We can always reopen if someone has the issue with real HW.

closed 2024-01-20 by mcuee

mcuee, 2024-01-20:
Interesting there are no more issues after I update Proxmox, using libusb git. So this is indeed not a libusb issue.

```
openbsdvmn100$ uname -a
OpenBSD openbsdvmn100.my.domain 7.4 GENERIC.MP#1397 amd64
openbsdvmn100$ sudo make check
Making check in libusb
Making check in examples
Making check in tests
make  check-TESTS
PASS: stress
PASS: stress_mt
PASS: set_option
PASS: init_context
============================================================================
Testsuite summary for libusb-1.0 ..
============================================================================
# TOTAL: 4
# PASS:  4
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
# ERROR: 0
============================================================================
openbsdvmn100$ sudo ./tests/stress_mt
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
All threads discovered 2 devices (0 not opened)
All done, 0 errors
```

mcuee, 2024-01-20:
Just to make sure, I tested 1.0.27-rc1 and it is the same -- no more issues. Looks like it was a Proxmox issue which is now fixed.