libusb/libusb #1390: `make check` failed under NetBSD 9.4

mcuee, 2023-12-20:
Reported here:
* https://github.com/libusb/libusb/issues/1372#issuecomment-1864460535

`make check` failed under NetBSD due to similar issues like Windows. It is just the error code is now `LIBUSB_ERROR_IO`.
git HEAD will still fail `make check` under NetBSD since we have not dealt with the error code.

```
localhost$ sudo make check
Password:
Making check in libusb
Making check in examples
Making check in tests
make  check-TESTS
PASS: stress
FAIL: stress_mt
PASS: set_option
PASS: init_context
============================================================================
Testsuite summary for libusb-1.0 1.0.27-rc1
============================================================================
# TOTAL: 4
# PASS:  3
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0
============================================================================
See tests/test-suite.log
Please report to libusb-devel@lists.sourceforge.net
============================================================================
*** Error code 1

Stop.
make[3]: stopped in /home/mcuee/build/libusb/tests
*** Error code 1

Stop.
make[2]: stopped in /home/mcuee/build/libusb/tests
*** Error code 1

Stop.
make[1]: stopped in /home/mcuee/build/libusb/tests
*** Error code 1

Stop.

make: stopped in /home/mcuee/build/libusb
localhost$ sudo ./tests/stress_mt
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
Thread 0 failed (iteration 21): LIBUSB_ERROR_IO
Device count mismatch: Thread 1 discovered 1 devices instead of 0
All done, 2 errors

localhost$ sudo ./tests/stress_mt
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
Thread 3 failed (iteration 4): LIBUSB_ERROR_IO
Thread 7 failed (iteration 48): LIBUSB_ERROR_IO
All threads discovered 1 devices (0 not opened)
All done, 2 errors
```

It is a bit difficult to reproduce the error when using `LIBUSB_DEBUG=4`.

tormodvolden, 2023-12-20:
Different from the Windows case, where the devices simply cannot be opened, it looks like here on NetBSD it is a real bug. Thread 0 discovers no devices, but thread 1 discovers and opens one device. I don't know if the NetBSD backend even tries to be thread-safe.

mcuee, 2023-12-21:
@efuss 

Just wondering if you can try with a real NetBSD system (I am using a VM) to see if the issue is a real problem or not. Thanks.

mcuee, 2023-12-31:
Here is the debug log.

[error.zip](https://github.com/libusb/libusb/files/13800988/error.zip)

mcuee, 2023-12-31:
Debug log for examples (`listdev`, `testlibusb -v` and `xusb`)

```
localhost# ./examples/listdevs
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000304] [00000001] libusb: debug [libusb_init_context] created default context
[ 0.000396] [00000001] libusb: debug [libusb_init_context] libusb v1.0.27.11848-rc1
[ 0.000491] [00000001] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000632] [00000001] libusb: debug [libusb_get_device_list]
[ 0.000710] [00000001] libusb: debug [netbsd_get_device_list]
[ 0.000812] [00000001] libusb: debug [_cache_active_config_descriptor] fd 5
[ 0.000889] [00000001] libusb: debug [_cache_active_config_descriptor] active bLength 9
[ 0.000988] [00000001] libusb: debug [_cache_active_config_descriptor] index -1, len 32
[ 0.001377] [00000001] libusb: debug [libusb_get_device_descriptor]
03eb:2104 (bus 3, device 2)
[ 0.001560] [00000001] libusb: debug [libusb_unref_device] destroy device 3.2
[ 0.001633] [00000001] libusb: debug [netbsd_destroy_device]
[ 0.001711] [00000001] libusb: debug [libusb_exit] destroying default context
[ 0.001824] [00000001] libusb: debug [usbi_remove_event_source] remove fd 3
localhost# ./examples/testlibusb -v
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000343] [00000001] libusb: debug [libusb_init_context] created default context
[ 0.000448] [00000001] libusb: debug [libusb_init_context] libusb v1.0.27.11848-rc1
[ 0.000573] [00000001] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000673] [00000001] libusb: debug [libusb_get_device_list]
[ 0.000753] [00000001] libusb: debug [netbsd_get_device_list]
[ 0.000858] [00000001] libusb: debug [_cache_active_config_descriptor] fd 5
[ 0.000990] [00000001] libusb: debug [_cache_active_config_descriptor] active bLength 9
[ 0.001085] [00000001] libusb: debug [_cache_active_config_descriptor] index -1, len 32
[ 0.001490] [00000001] libusb: debug [libusb_get_device_descriptor]
Dev (bus 3, device 2): 03EB - 2104 speed: 12M
[ 0.001674] [00000001] libusb: debug [libusb_open] open 3.2
[ 0.001776] [00000001] libusb: debug [netbsd_open] open /dev/ugen0.00: fd 5
[ 0.001871] [00000001] libusb: debug [libusb_submit_transfer] transfer 0x7ebce2703088
[ 0.002060] [00000001] libusb: debug [netbsd_submit_transfer]
[ 0.002186] [00000001] libusb: debug [_sync_control_transfer] type 0x80 request 0x6 value 0x300 index 0 length 4 timeout 1000
[ 0.002703] [00000001] libusb: debug [_sync_control_transfer] transferred 4
[ 0.002822] [00000001] libusb: debug [libusb_get_next_timeout] next timeout in 0.999236s
[ 0.002929] [00000001] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.003012] [00000001] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.003095] [00000001] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 1000ms
[ 0.003181] [00000001] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.003257] [00000001] libusb: debug [handle_event_trigger] event triggered
[ 0.003366] [00000001] libusb: debug [usbi_handle_transfer_completion] transfer 0x7ebce2703088 has callback 0x7ebce240b080
[ 0.003441] [00000001] libusb: debug [sync_transfer_cb] actual_length=4
[ 0.003520] [00000001] libusb: debug [libusb_free_transfer] transfer 0x7ebce2703088
[ 0.003608] [00000001] libusb: debug [libusb_submit_transfer] transfer 0x7ebce2703088
[ 0.003707] [00000001] libusb: debug [netbsd_submit_transfer]
[ 0.003781] [00000001] libusb: debug [_sync_control_transfer] type 0x80 request 0x6 value 0x301 index 1033 length 255 timeout 1000
[ 0.004199] [00000001] libusb: debug [_sync_control_transfer] transferred 12
[ 0.004297] [00000001] libusb: debug [libusb_get_next_timeout] next timeout in 0.999410s
[ 0.004374] [00000001] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.004490] [00000001] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 1000ms
[ 0.004578] [00000001] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.004652] [00000001] libusb: debug [handle_event_trigger] event triggered
[ 0.004725] [00000001] libusb: debug [usbi_handle_transfer_completion] transfer 0x7ebce2703088 has callback 0x7ebce240b080
[ 0.004800] [00000001] libusb: debug [sync_transfer_cb] actual_length=12
[ 0.004901] [00000001] libusb: debug [libusb_free_transfer] transfer 0x7ebce2703088
  Manufacturer:              ATMEL
[ 0.005045] [00000001] libusb: debug [libusb_submit_transfer] transfer 0x7ebce2703088
[ 0.005123] [00000001] libusb: debug [netbsd_submit_transfer]
[ 0.005196] [00000001] libusb: debug [_sync_control_transfer] type 0x80 request 0x6 value 0x300 index 0 length 4 timeout 1000
[ 0.005586] [00000001] libusb: debug [_sync_control_transfer] transferred 4
[ 0.005707] [00000001] libusb: debug [libusb_get_next_timeout] next timeout in 0.999416s
[ 0.005784] [00000001] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.005860] [00000001] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 1000ms
[ 0.005935] [00000001] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.006028] [00000001] libusb: debug [handle_event_trigger] event triggered
[... 212 lines cut ...]
[... cut ...]
[ 0.009924] [00000001] libusb: debug [libusb_release_interface] interface 0
Closing device...
[ 0.009945] [00000001] libusb: debug [libusb_close]
[ 0.009970] [00000001] libusb: debug [netbsd_close] close: fd 5
[ 0.009985] [00000001] libusb: debug [libusb_unref_device] destroy device 3.2
[ 0.009996] [00000001] libusb: debug [netbsd_destroy_device]
[ 0.010009] [00000001] libusb: debug [libusb_exit] destroying default context
[ 0.010019] [00000001] libusb: debug [usbi_remove_event_source] remove fd 3
```

mcuee, 2024-01-05:
This is not a regression, as libusb-1.0.26 will have the same issue if we backport `stress_mt` to libusb-1.0.26.

```
localhost$ pwd
/home/mcuee/build/libusb-1.0.26_stress_mt/libusb/netbsd_debug
localhost$ sudo ./tests/stress_mt
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
All threads discovered 1 devices (0 not opened)
All done, 0 errors
```

Backport process by @tormodvolden 
* https://github.com/libusb/libusb/issues/1372#issuecomment-1873255729

mcuee, 2024-01-05:
Therefore this is not a release blocker for libusb-1.0.27.

mcuee, 2024-01-13:
Hopefully someone with a physical machine installation (not virtual machine) of NetBSD can help to check. There is a possibility that the issue does not exist on real physical machine.

mcuee, 2024-01-20:
No issues with NetBSD 10.0 rc3 Virtual machine (using the same PVE 8.0 host on Intel N100 mini-PC, and attach the same USB device).

```
NetBSD 10.0_RC3 (GENERIC) #0: Tue Jan 16 08:28:51 UTC 2024

Welcome to NetBSD!

This is a release candidate for NetBSD.

Bug reports: https://www.NetBSD.org/support/send-pr.html
Donations to the NetBSD Foundation: https://www.NetBSD.org/donations/
netbsd10vm$ cd build/libusb/
netbsd10vm$ uname -a
NetBSD netbsd10vm.home 10.0_RC3 NetBSD 10.0_RC3 (GENERIC) #0: Tue Jan 16 08:28:51 UTC 2024  mkrepro@mkrepro.NetBSD.org:/usr/src/sys/arch/amd64/compile/GENERIC amd64
netbsd10vm$ pwd
/home/mcuee/build/libusb
netbsd10vm$ sudo make check
Password:
Making check in libusb
Making check in examples
Making check in tests
make  check-TESTS
PASS: stress
PASS: stress_mt
PASS: set_option
PASS: init_context
============================================================================
Testsuite summary for libusb-1.0 1.0.27-rc1
============================================================================
# TOTAL: 4
# PASS:  4
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
# ERROR: 0
============================================================================
netbsd10vm$ sudo ./tests/stress_mt
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
All threads discovered 1 devices (0 not opened)
All done, 0 errors
```

mcuee, 2024-01-20:
I will close this issue for now.

We can always re-open if there are users reporting similar issues with a physical machine.

closed 2024-01-20 by mcuee

mcuee, 2024-01-20:
In fact, no issues using libusb git with the NetBSD 9.3 VM either.
```
-bash-5.2$ uname -a
NetBSD localhost 9.3 NetBSD 9.3 (GENERIC) #0: Thu Aug  4 15:30:37 UTC 2022  mkrepro@mkrepro.NetBSD.org:/usr/src/sys/arch/amd64/compile/GENERIC amd64
-bash-5.2$ sudo ./tests/stress_mt
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
All threads discovered 1 devices (0 not opened)
All done, 0 errors
-bash-5.2$ sudo make check
Making check in libusb
Making check in examples
Making check in tests
make  check-TESTS
PASS: stress
PASS: stress_mt
PASS: set_option
PASS: init_context
============================================================================
Testsuite summary for libusb-1.0 1.0.27-rc1
============================================================================
# TOTAL: 4
# PASS:  4
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
# ERROR: 0
============================================================================
```

mcuee, 2024-01-20:
Just to make sure, I tested libusb-1.0.27-rc1 again and no more issues. Looks like this is a Proxmox issue and not a libusb/NetBSD issue.

```
-bash-5.2$ pwd
/home/mcuee/build/libusb-1.0.27-rc1

-bash-5.2$ sudo ./tests/stress_mt
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
All threads discovered 1 devices (0 not opened)
All done, 0 errors

-bash-5.2$ sudo make check
Making check in libusb
Making check in examples
Making check in tests
make  check-TESTS
PASS: stress
PASS: stress_mt
PASS: set_option
PASS: init_context
============================================================================
Testsuite summary for libusb-1.0 1.0.27-rc1
============================================================================
# TOTAL: 4
# PASS:  4
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
# ERROR: 0
============================================================================
```