libusb/libusb #1394: To downgrade the warning messages related to `no DeviceInterfaceGUID registered' to info

mcuee, 2023-12-21:
From here by me.
* https://github.com/libusb/libusb/issues/1372#issuecomment-1863771546

Just wondering if we can downgrade the warning messages related to `no DeviceInterfaceGUID registered' to info.

```
MINGW64 /c/work/libusb/libusb-1.0.27-rc1/mingw64
$ ./examples/xusb.exe 0424:2740
Using libusb v1.0.27.11848

Opening device 0424:2740...
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057\C&1F76A113&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0424&PID_5744\C&1936BD58&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_5413\B&2EAF716D&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_0413\B&1637EEB3&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_413C&PID_B06E\B&2EAF716D&0&5'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_1A86&PID_7523\D&127B8A6A&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_8087&PID_0AAA\5&E9F3E45&0&14'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_02\6&1BD5FF13&0&0002'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0424&PID_2744\C&1F76A113&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_5487\A&46F7C3F&0&2'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_58FD&MI_02\6&1E7CBC97&0&0002'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057&MI_01\D&58BCC3D&0&0001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_24AE&PID_4057&MI_00\D&58BCC3D&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_0487\A&46F7C3F&0&4'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_03\6&1BD5FF13&0&0003'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_2717&PID_5013\5&E9F3E45&0&3'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_402E\200901010001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_047F&PID_C056&MI_03\6&360CBCF3&0&0003'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\4&36020D6F&0&0'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842\0123456789ABCD'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\9&AF4BBD4&0&0'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\7&2452366F&0&0'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_402E&MI_00\D&E651DD6&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_047F&PID_C056\BB305534D79EBD418B6757528EAC0C19'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_00\6&1BD5FF13&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_8153\111000001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_413C&PID_B06F\C&1F76A113&0&5'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_047F&PID_C056&MI_00\6&360CBCF3&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_58FD&MI_00\6&1E7CBC97&0&0000'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0A5C&PID_5842&MI_01\6&1BD5FF13&0&0001'
libusb: warning [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0BDA&PID_58FD\200901010001'

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 0424:2740
         bcdDevice: 0200
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: 1 caps
    USB 2.0 extension:
      attributes             : A21E

Reading first configuration descriptor:
              total length: 18
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 0
   Class.SubClass.Protocol: FF.FF.FF

Kernel driver attached for interface 0: (not supported)

Claiming interface 0...

Reading string descriptors:
   String (0x01): "Microchip Tech"
   String (0x02): "Hub Controller"

Reading OS string descriptor:

Reading Extended Compat ID OS Feature Descriptor (wIndex = 0x0004):

  00000000  28 00 00 00 00 01 04 00 01 00 00 00 00 00 00 00  (...............
  00000010  00 01 57 49 4e 55 53 42 00 00 00 00 00 00 00 00  ..WINUSB........
  00000020  00 00 00 00 00 00 00 00                          ........

Reading Extended Properties OS Feature Descriptor (wIndex = 0x0005):
   Failed: Pipe error
Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
Closing device...
```

mcuee, 2023-12-28:
The warning comes from PR #1322 

* #1322

tormodvolden, 2024-01-20:
It is now demoted to "info". However, for xusb, I think it sets debug level to "info" by default. So do these messages still show up there?

mcuee, 2024-01-21:
> It is now demoted to "info". However, for xusb, I think it sets debug level to "info" by default. So do these messages still show up there?

Yes it will show up. I think we can live with this for now.

```
 MINGW64 /c/work/libusb/libusb
$ ./examples/xusb.exe 1ea7:0064
Using libusb v1.0.27.11874

Opening device 1EA7:0064...
libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_05E3&PID_0612\5&1A00B1C&0&2'
libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_8087&PID_0026\5&586B51A&0&10'
libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_1EA7&PID_0064\5&586B51A&0&3'
libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\4&12C539F&0&0'
libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_04F2&PID_B6DD&MI_00\6&2F8D90E&0&0000'
libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_05E3&PID_0610\5&586B51A&0&2'
libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\4&1A63DC7C&0&0'
libusb: warning [hid_open] could not open HID device in R/W mode (keyboard or mouse?) - trying without

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 1EA7:0064
         bcdDevice: 0200
   iMan:iProd:iSer: 0:1:0
          nb confs: 1

Reading BOS descriptor: libusb: warning [_hid_get_descriptor] unsupported
libusb: error [libusb_get_bos_descriptor] failed to read BOS (-12)
no descriptor

Reading first configuration descriptor:
              total length: 34
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.01.02
       endpoint[0].address: 81
           max packet size: 0008
          polling interval: 02

Kernel driver attached for interface 0: (not supported)

Claiming interface 0...

Reading string descriptors:
   String (0x01): "2.4G Mouse"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
Closing device...
```

tormodvolden, 2024-01-21:
Maybe it should be an info for some devices and ignored for others, I dunno. Maybe @pazourek who added the warning or @sonatique would know.

Droid-MAX, 2024-04-03:
I have a vaguely bad feeling about this, could be the next potential supply chain poisoning similar to xz-utils?
And I found that programs built with the libusb-1.0.26 source code at some point in time always output some Libusb-related warning information, which had not been the case before. A warning message appears even after the upgrade to 1.0.27.

https://github.com/Droid-MAX/SNANDer/issues/15

mcuee, 2024-04-03:
Re-open this issue.

Droid-MAX, 2024-04-03:
Perhaps a compilation parameter can be defined to control whether debugging information is output?

mcuee, 2024-04-03:
> Perhaps a compilation parameter can be defined to control whether debugging information is output?

It is already there.
https://libusb.sourceforge.io/api-1.0/group__libusb__lib.html#ga07d4ec54cf575d672ba94c72b3c0de7c

mcuee, 2024-04-03:
> Maybe it should be an info for some devices and ignored for others, I dunno. Maybe @pazourek who added the warning or @sonatique would know.

@pazourek 
Please help here. Thanks.

@sonatique 
Please take a look as well. Thanks.

Droid-MAX, 2024-04-03:
so, the default behavior is enable all debug information output?

Droid-MAX, 2024-04-03:
If I want to turn off all debugging output by default, what do I do at compile time? Any reference examples?

mcuee, 2024-04-03:
> so, the default behavior is enable all debug information output?

The default should be `LIBUSB_LOG_LEVEL_NONE `.

```
LIBUSB_LOG_LEVEL_NONE 	
(0) : No messages ever emitted by the library (default)

LIBUSB_LOG_LEVEL_ERROR 	
(1) : Error messages are emitted

LIBUSB_LOG_LEVEL_WARNING 	
(2) : Warning and error messages are emitted

LIBUSB_LOG_LEVEL_INFO 	
(3) : Informational, warning and error messages are emitted

LIBUSB_LOG_LEVEL_DEBUG 	
(4) : All messages are emitted
```

But you change to use `LIBUSB_LOG_LEVEL_INFO`.
https://github.com/Droid-MAX/SNANDer/blob/f8742b3b5dc7579c720cdfd31fec6bd140623ab9/src/ch341a_spi.c#L413C1-L417C7
```
#if LIBUSB_API_VERSION >= 0x01000106
	libusb_set_option(NULL, LIBUSB_OPTION_LOG_LEVEL, 3);
#else
	libusb_set_debug(NULL, 3); // Enable information, warning and error messages (only).
#endif
```

Droid-MAX, 2024-04-03:
[image]

Droid-MAX, 2024-04-03:
> > so, the default behavior is enable all debug information output?
> > If I want to turn off all debugging output by default, what do I do at compile time? Any reference examples?
> 
> The default should be `LIBUSB_LOG_LEVEL_NONE `.
> 
> ```
> LIBUSB_LOG_LEVEL_NONE 	
> (0) : No messages ever emitted by the library (default)
> 
> LIBUSB_LOG_LEVEL_ERROR 	
> (1) : Error messages are emitted
> 
> LIBUSB_LOG_LEVEL_WARNING 	
> (2) : Warning and error messages are emitted
> 
> LIBUSB_LOG_LEVEL_INFO 	
> (3) : Informational, warning and error messages are emitted
> 
> LIBUSB_LOG_LEVEL_DEBUG 	
> (4) : All messages are emitted
> ```
> 
> But you change to use `LIBUSB_LOG_LEVEL_INFO`. https://github.com/Droid-MAX/SNANDer/blob/f8742b3b5dc7579c720cdfd31fec6bd140623ab9/src/ch341a_spi.c#L413C1-L417C7
> 
> ```
> #if LIBUSB_API_VERSION >= 0x01000106
> 	libusb_set_option(NULL, LIBUSB_OPTION_LOG_LEVEL, 3);
> #else
> 	libusb_set_debug(NULL, 3); // Enable information, warning and error messages (only).
> #endif
> ```

thank you for your reply, I will handle it right now.

Droid-MAX, 2024-04-03:
Well done, thank you again for your assistance!

pazourek, 2024-04-03:
The libusb 1.0.26 did not contain this warning. This warning is useless for many devices, but it gives us an important information related to composite devices on MS Win platform. Without the GUIDs, the libusb is not able to access  individual interfaces of the composite device.

It would probably make sense to report this warning only for composite devices. But I am not sure if it is possible to simply detect whether the current interface belongs to the composite device during listing the device interfaces.

pazourek, 2024-04-03:
I vote for keeping this warning message in the code. I would be fine with downgrading this warning message to debug level. Since the previous release did not show this warning at all, it was hard to find what is going on and why the composite device does not work for me.

tormodvolden, 2024-04-04:
Yes, if we cannot limit the message to composite devices, or until we can, we should demote it to debug level. It seems it can be too verbose as of the example linked above, and if something doesn't work it is natural to enable debug logging anyway.

mcuee, 2024-04-04:
> Yes, if we cannot limit the message to composite devices, or until we can, we should demote it to debug level. It seems it can be too verbose as of the example linked above, and if something doesn't work it is natural to enable debug logging anyway.

I agree.

tormodvolden, 2024-05-26:
I made it a debug message.

commit f9ae36b1 (2024-01-19): windows: Downgrade get_guid DeviceInterfaceGUID warning to info

Closes #1394
commit de1398db (2024-05-26): windows: Downgrade get_guid DeviceInterfaceGUID message to debug level

Ideally it should be a warning only for composite devices but it is
difficult to discern that case. The message is too invasive and
confusing for the non-composite case, so make it debug instead.

Follow-up of commit f9ae36b

Closes #1394
closed 2024-05-26 by tormodvolden