libusb/libusb #1396: `-Wswitch-enum` compiler warning for `core.c`

mcuee, 2023-12-21:
Reference: github action build log

https://github.com/libusb/libusb/actions/runs/7255523540/job/19766296506
macOS build
```
/Applications/Xcode_14.2.app/Contents/Developer/usr/bin/make  all-recursive
Making all in libusb
  CC       core.lo
  CC       descriptor.lo
  CC       hotplug.lo
  CC       io.lo
../../libusb/core.c:2490:10: warning: 4 enumeration values not explicitly handled in switch: 'LIBUSB_OPTION_LOG_LEVEL', 'LIBUSB_OPTION_USE_USBDK', 'LIBUSB_OPTION_NO_DEVICE_DISCOVERY'... [-Wswitch-enum]
                switch(options[i].option) {
                       ^~~~~~~~~~~~~~~~~
../../libusb/core.c:2490:10: note: add missing switch cases
                switch(options[i].option) {
  CC       strerror.lo
                       ^
1 warning generated.
```

Linux build
https://github.com/libusb/libusb/actions/runs/7255523524/job/19766296509
```
make[2]: Entering directory '/home/runner/work/libusb/libusb/build-netlink/libusb'
  CC       core.lo
  CC       hotplug.lo
  CC       descriptor.lo
  CC       io.lo
../../libusb/core.c: In function ‘libusb_init_context’:
../../libusb/core.c:2490:17: warning: enumeration value ‘LIBUSB_OPTION_LOG_LEVEL’ not handled in switch [-Wswitch-enum]
 2490 |                 switch(options[i].option) {
      |                 ^~~~~~
../../libusb/core.c:2490:17: warning: enumeration value ‘LIBUSB_OPTION_USE_USBDK’ not handled in switch [-Wswitch-enum]
../../libusb/core.c:2490:17: warning: enumeration value ‘LIBUSB_OPTION_NO_DEVICE_DISCOVERY’ not handled in switch [-Wswitch-enum]
../../libusb/core.c:2490:17: warning: enumeration value ‘LIBUSB_OPTION_MAX’ not handled in switch [-Wswitch-enum]
```

Same for other CI build.

seanm, 2024-01-04:
@mcuee fixed https://github.com/libusb/libusb/pull/1426

mcuee, 2024-01-04:
@seanm 

Yes, PR #1426 fixed this issue.

closed 2024-01-19 by tormodvolden