libusb/libusb #1398: usbutils lsusb devices still referenced

mcuee, 2023-12-21:
Tested under Linux Ubuntu 20.04 and libusb git head (libusb v1.0.27.11855-rc1)

```
mcuee@UbuntuSwift3 ~/build/libusb/libusb_git (master)$ ldd /usr/bin/lsusb 
	linux-vdso.so.1 (0x00007ffc491f9000)
	libusb-1.0.so.0 => /usr/local/lib/libusb-1.0.so.0 (0x00007f7de39f5000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f7de39c8000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f7de37d6000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f7de37b3000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f7de3a8b000)
mcuee@UbuntuSwift3 ~/build/libusb/libusb_git (master)$ export LIBUSB_DEBUG=4
mcuee@UbuntuSwift3 ~/build/libusb/libusb_git (master)$ lsusb
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000021] [00002eda] libusb: debug [libusb_init_context] libusb v1.0.27.11855-rc1
[ 0.000055] [00002eda] libusb: debug [usbi_add_event_source] add fd 4 events 1
[ 0.000066] [00002eda] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000069] [00002eda] libusb: debug [usbi_add_event_source] add fd 5 events 1
[ 0.000080] [00002eda] libusb: debug [get_kernel_version] reported kernel version is 5.15.0-91-generic
[ 0.000102] [00002eda] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000111] [00002eda] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000119] [00002eda] libusb: debug [op_init] sysfs is available
[ 0.000594] [00002edb] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.004566] [00002eda] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.004569] [00002eda] libusb: debug [linux_get_device_address] scan usb1
[ 0.004581] [00002eda] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.004582] [00002eda] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.004583] [00002eda] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.004626] [00002eda] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.004628] [00002eda] libusb: debug [linux_get_device_address] scan usb2
[ 0.004635] [00002eda] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.004636] [00002eda] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.004637] [00002eda] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.004677] [00002eda] libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
[ 0.004679] [00002eda] libusb: debug [linux_get_device_address] scan 2-1
[ 0.004686] [00002eda] libusb: debug [linux_get_device_address] bus=2 dev=2
[ 0.004688] [00002eda] libusb: debug [linux_enumerate_device] busnum 2 devaddr 2 session_id 514
[ 0.004689] [00002eda] libusb: debug [linux_enumerate_device] allocating new device for 2/2 (session 514)
[ 0.004699] [00002eda] libusb: debug [linux_get_parent_info] dev 0x55de8aae1ec0 (2-1) has parent 0x55de8aae22b0 (usb2) port 1
[ 0.004731] [00002eda] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.004733] [00002eda] libusb: debug [linux_get_device_address] scan usb3
[ 0.004740] [00002eda] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.004741] [00002eda] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.004742] [00002eda] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.004783] [00002eda] libusb: debug [linux_get_device_address] getting address for device: 3-10 detached: 0
[ 0.004784] [00002eda] libusb: debug [linux_get_device_address] scan 3-10
[ 0.004792] [00002eda] libusb: debug [linux_get_device_address] bus=3 dev=6
[ 0.004794] [00002eda] libusb: debug [linux_enumerate_device] busnum 3 devaddr 6 session_id 774
[ 0.004795] [00002eda] libusb: debug [linux_enumerate_device] allocating new device for 3/6 (session 774)
[ 0.004804] [00002eda] libusb: debug [linux_get_parent_info] dev 0x55de8aae25a0 (3-10) has parent 0x55de8aae26a0 (usb3) port 10
[ 0.004837] [00002eda] libusb: debug [linux_get_device_address] getting address for device: 3-2 detached: 0
[ 0.004839] [00002eda] libusb: debug [linux_get_device_address] scan 3-2
[ 0.004847] [00002eda] libusb: debug [linux_get_device_address] bus=3 dev=2
[ 0.004848] [00002eda] libusb: debug [linux_enumerate_device] busnum 3 devaddr 2 session_id 770
[ 0.004849] [00002eda] libusb: debug [linux_enumerate_device] allocating new device for 3/2 (session 770)
[ 0.004858] [00002eda] libusb: debug [linux_get_parent_info] dev 0x55de8aac9120 (3-2) has parent 0x55de8aae26a0 (usb3) port 2
[ 0.004892] [00002eda] libusb: debug [linux_get_device_address] getting address for device: 3-3 detached: 0
[ 0.004894] [00002eda] libusb: debug [linux_get_device_address] scan 3-3
[ 0.004902] [00002eda] libusb: debug [linux_get_device_address] bus=3 dev=3
[ 0.004904] [00002eda] libusb: debug [linux_enumerate_device] busnum 3 devaddr 3 session_id 771
[ 0.004905] [00002eda] libusb: debug [linux_enumerate_device] allocating new device for 3/3 (session 771)
[ 0.004917] [00002eda] libusb: debug [linux_get_parent_info] dev 0x55de8aac9520 (3-3) has parent 0x55de8aae26a0 (usb3) port 3
[ 0.004951] [00002eda] libusb: debug [linux_get_device_address] getting address for device: 3-5 detached: 0
[ 0.004953] [00002eda] libusb: debug [linux_get_device_address] scan 3-5
[ 0.004961] [00002eda] libusb: debug [linux_get_device_address] bus=3 dev=4
[ 0.004962] [00002eda] libusb: debug [linux_enumerate_device] busnum 3 devaddr 4 session_id 772
[ 0.004964] [00002eda] libusb: debug [linux_enumerate_device] allocating new device for 3/4 (session 772)
[ 0.004973] [00002eda] libusb: debug [linux_get_parent_info] dev 0x55de8aac8130 (3-5) has parent 0x55de8aae26a0 (usb3) port 5
[ 0.005006] [00002eda] libusb: debug [linux_get_device_address] getting address for device: 3-7 detached: 0
[ 0.005007] [00002eda] libusb: debug [linux_get_device_address] scan 3-7
[... 21 lines cut ...]
[ 0.005336] [00002eda] libusb: debug [libusb_get_device_descriptor]  
Bus 003 Device 006: ID 8087:0026 Intel Corp. 
[ 0.005372] [00002eda] libusb: debug [libusb_get_device_descriptor]  
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
[ 0.005401] [00002eda] libusb: debug [libusb_get_device_descriptor]  
Bus 002 Device 002: ID 05e3:0612 Genesys Logic, Inc. Hub
[ 0.005430] [00002eda] libusb: debug [libusb_get_device_descriptor]  
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
[ 0.005459] [00002eda] libusb: debug [libusb_get_device_descriptor]  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
[ 0.005510] [00002eda] libusb: debug [libusb_exit]  
[ 0.005607] [00002edb] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
[ 0.005694] [00002eda] libusb: debug [usbi_remove_event_source] remove fd 5
[ 0.005702] [00002eda] libusb: debug [usbi_remove_event_source] remove fd 4
[ 0.005705] [00002eda] libusb: warning [libusb_exit] device 4.1 still referenced
[ 0.005707] [00002eda] libusb: warning [libusb_exit] device 3.5 still referenced
[ 0.005708] [00002eda] libusb: warning [libusb_exit] device 3.4 still referenced
[ 0.005709] [00002eda] libusb: warning [libusb_exit] device 3.3 still referenced
[ 0.005710] [00002eda] libusb: warning [libusb_exit] device 3.2 still referenced
[ 0.005712] [00002eda] libusb: warning [libusb_exit] device 3.6 still referenced
[ 0.005713] [00002eda] libusb: warning [libusb_exit] device 3.1 still referenced
[ 0.005714] [00002eda] libusb: warning [libusb_exit] device 2.2 still referenced
[ 0.005715] [00002eda] libusb: warning [libusb_exit] device 2.1 still referenced
[ 0.005717] [00002eda] libusb: warning [libusb_exit] device 1.1 still referenced
```

mcuee, 2023-12-21:
On the other hand, no issues with libusb examples.
```
mcuee@UbuntuSwift3 ~/build/libusb/libusb_git (master)$ ./examples/testlibusb
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000101] [00004be0] libusb: debug [libusb_init_context] created default context
[ 0.000123] [00004be0] libusb: debug [libusb_init_context] libusb v1.0.27.11855-rc1
[ 0.000138] [00004be0] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000148] [00004be0] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000151] [00004be0] libusb: debug [usbi_add_event_source] add fd 4 events 1
[ 0.000161] [00004be0] libusb: debug [get_kernel_version] reported kernel version is 5.15.0-91-generic
[ 0.000181] [00004be0] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000184] [00004be0] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000191] [00004be0] libusb: debug [op_init] sysfs is available
[ 0.000523] [00004bee] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.004989] [00004be0] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.004993] [00004be0] libusb: debug [linux_get_device_address] scan usb1
[ 0.005004] [00004be0] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.005006] [00004be0] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.005007] [00004be0] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.005048] [00004be0] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.005050] [00004be0] libusb: debug [linux_get_device_address] scan usb2
[ 0.005057] [00004be0] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.005058] [00004be0] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.005059] [00004be0] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.005098] [00004be0] libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
[ 0.005101] [00004be0] libusb: debug [linux_get_device_address] scan 2-1
[ 0.005113] [00004be0] libusb: debug [linux_get_device_address] bus=2 dev=2
[ 0.005114] [00004be0] libusb: debug [linux_enumerate_device] busnum 2 devaddr 2 session_id 514
[ 0.005117] [00004be0] libusb: debug [linux_enumerate_device] allocating new device for 2/2 (session 514)
[ 0.005146] [00004be0] libusb: debug [linux_get_parent_info] dev 0x557b4f4f3430 (2-1) has parent 0x557b4f4f2de0 (usb2) port 1
[ 0.005186] [00004be0] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.005188] [00004be0] libusb: debug [linux_get_device_address] scan usb3
[ 0.005201] [00004be0] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.005203] [00004be0] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.005205] [00004be0] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.005250] [00004be0] libusb: debug [linux_get_device_address] getting address for device: 3-1 detached: 0
[ 0.005252] [00004be0] libusb: debug [linux_get_device_address] scan 3-1
[ 0.005265] [00004be0] libusb: debug [linux_get_device_address] bus=3 dev=7
[ 0.005266] [00004be0] libusb: debug [linux_enumerate_device] busnum 3 devaddr 7 session_id 775
[ 0.005268] [00004be0] libusb: debug [linux_enumerate_device] allocating new device for 3/7 (session 775)
[ 0.005281] [00004be0] libusb: debug [linux_get_parent_info] dev 0x557b4f4cda30 (3-1) has parent 0x557b4f4f3330 (usb3) port 1
[ 0.005323] [00004be0] libusb: debug [linux_get_device_address] getting address for device: 3-1.1 detached: 0
[ 0.005325] [00004be0] libusb: debug [linux_get_device_address] scan 3-1.1
[ 0.005340] [00004be0] libusb: debug [linux_get_device_address] bus=3 dev=8
[ 0.005341] [00004be0] libusb: debug [linux_enumerate_device] busnum 3 devaddr 8 session_id 776
[ 0.005344] [00004be0] libusb: debug [linux_enumerate_device] allocating new device for 3/8 (session 776)
[ 0.005357] [00004be0] libusb: debug [linux_get_parent_info] dev 0x557b4f4cdfa0 (3-1.1) has parent 0x557b4f4cda30 (3-1) port 1
[ 0.005396] [00004be0] libusb: debug [linux_get_device_address] getting address for device: 3-10 detached: 0
[ 0.005398] [00004be0] libusb: debug [linux_get_device_address] scan 3-10
[ 0.005411] [00004be0] libusb: debug [linux_get_device_address] bus=3 dev=6
[ 0.005412] [00004be0] libusb: debug [linux_enumerate_device] busnum 3 devaddr 6 session_id 774
[ 0.005414] [00004be0] libusb: debug [linux_enumerate_device] allocating new device for 3/6 (session 774)
[ 0.005427] [00004be0] libusb: debug [linux_get_parent_info] dev 0x557b4f4cdc50 (3-10) has parent 0x557b4f4f3330 (usb3) port 10
[ 0.005467] [00004be0] libusb: debug [linux_get_device_address] getting address for device: 3-2 detached: 0
[ 0.005468] [00004be0] libusb: debug [linux_get_device_address] scan 3-2
[ 0.005481] [00004be0] libusb: debug [linux_get_device_address] bus=3 dev=2
[ 0.005483] [00004be0] libusb: debug [linux_enumerate_device] busnum 3 devaddr 2 session_id 770
[ 0.005485] [00004be0] libusb: debug [linux_enumerate_device] allocating new device for 3/2 (session 770)
[ 0.005498] [00004be0] libusb: debug [linux_get_parent_info] dev 0x557b4f4ce5e0 (3-2) has parent 0x557b4f4f3330 (usb3) port 2
[... cut ...]
[ 0.007762] [00004bf2] libusb: debug [libusb_unref_device] destroy device 3.1
[ 0.007766] [00004bf2] libusb: debug [libusb_unref_device] destroy device 2.2
[ 0.007770] [00004bf2] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.007773] [00004bf2] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.007795] [00004c00] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
[ 0.007869] [00004bf2] libusb: debug [usbi_remove_event_source] remove fd 4
[ 0.007875] [00004bf2] libusb: debug [usbi_remove_event_source] remove fd 3
```

tormodvolden, 2023-12-21:
Isn't this just a bug in usbutils? It uses libusb_get_device_list() so all devices will be reference-counted. In the end it does libusb_free_device_list(list, 0) so without dereferencing the devices. There's no libusb_unref_device() in the code.

Without looking too close at the code, I guess it becomes correct If this is changed to libusb_free_device_list(list, 1).

mcuee, 2023-12-21:
@tormodvolden 

You are right and your proposed fix sorted out the issue.

```
mcuee@UbuntuSwift3 ~/build/libusb/usbutils (master)$ git diff
diff --git a/lsusb.c b/lsusb.c
index 5c64f42..df4aa87 100644
--- a/lsusb.c
+++ b/lsusb.c
@@ -3688,7 +3688,7 @@ static int list_devices(libusb_context *ctx, int busnum, int devnum, int vendori
                        dumpdev(dev);
        }
 
-       libusb_free_device_list(list, 0);
+       libusb_free_device_list(list, 1);
 error:
        return status;
 }

mcuee@UbuntuSwift3 ~/build/libusb/usbutils (master)$ export LIBUSB_DEBUG=4
mcuee@UbuntuSwift3 ~/build/libusb/usbutils (master)$ lsusb
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000021] [00001c02] libusb: debug [libusb_init_context] libusb v1.0.27.11855-rc1
[ 0.000050] [00001c02] libusb: debug [usbi_add_event_source] add fd 4 events 1
[ 0.000061] [00001c02] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000064] [00001c02] libusb: debug [usbi_add_event_source] add fd 5 events 1
[ 0.000079] [00001c02] libusb: debug [get_kernel_version] reported kernel version is 5.15.0-91-generic
[ 0.000101] [00001c02] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000104] [00001c02] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000115] [00001c02] libusb: debug [op_init] sysfs is available
[ 0.000595] [00001c03] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.004328] [00001c02] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.004331] [00001c02] libusb: debug [linux_get_device_address] scan usb1
[ 0.004341] [00001c02] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.004342] [00001c02] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.004343] [00001c02] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.004384] [00001c02] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.004385] [00001c02] libusb: debug [linux_get_device_address] scan usb2
[ 0.004392] [00001c02] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.004393] [00001c02] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.004394] [00001c02] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.004431] [00001c02] libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
[ 0.004432] [00001c02] libusb: debug [linux_get_device_address] scan 2-1
[ 0.004439] [00001c02] libusb: debug [linux_get_device_address] bus=2 dev=2
[ 0.004440] [00001c02] libusb: debug [linux_enumerate_device] busnum 2 devaddr 2 session_id 514
[ 0.004441] [00001c02] libusb: debug [linux_enumerate_device] allocating new device for 2/2 (session 514)
[ 0.004451] [00001c02] libusb: debug [linux_get_parent_info] dev 0x56117b65dec0 (2-1) has parent 0x56117b65e2b0 (usb2) port 1
[ 0.004482] [00001c02] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.004483] [00001c02] libusb: debug [linux_get_device_address] scan usb3
[ 0.004490] [00001c02] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.004491] [00001c02] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.004492] [00001c02] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.004529] [00001c02] libusb: debug [linux_get_device_address] getting address for device: 3-10 detached: 0
[ 0.004530] [00001c02] libusb: debug [linux_get_device_address] scan 3-10
[ 0.004537] [00001c02] libusb: debug [linux_get_device_address] bus=3 dev=6
[ 0.004538] [00001c02] libusb: debug [linux_enumerate_device] busnum 3 devaddr 6 session_id 774
[ 0.004539] [00001c02] libusb: debug [linux_enumerate_device] allocating new device for 3/6 (session 774)
[ 0.004547] [00001c02] libusb: debug [linux_get_parent_info] dev 0x56117b65e5a0 (3-10) has parent 0x56117b65e6a0 (usb3) port 10
[ 0.004578] [00001c02] libusb: debug [linux_get_device_address] getting address for device: 3-2 detached: 0
[ 0.004579] [00001c02] libusb: debug [linux_get_device_address] scan 3-2
[ 0.004586] [00001c02] libusb: debug [linux_get_device_address] bus=3 dev=2
[ 0.004587] [00001c02] libusb: debug [linux_enumerate_device] busnum 3 devaddr 2 session_id 770
[ 0.004588] [00001c02] libusb: debug [linux_enumerate_device] allocating new device for 3/2 (session 770)
[ 0.004596] [00001c02] libusb: debug [linux_get_parent_info] dev 0x56117b645120 (3-2) has parent 0x56117b65e6a0 (usb3) port 2
[ 0.004627] [00001c02] libusb: debug [linux_get_device_address] getting address for device: 3-3 detached: 0
[ 0.004628] [00001c02] libusb: debug [linux_get_device_address] scan 3-3
[ 0.004635] [00001c02] libusb: debug [linux_get_device_address] bus=3 dev=3
[ 0.004636] [00001c02] libusb: debug [linux_enumerate_device] busnum 3 devaddr 3 session_id 771
[... 41 lines cut ...]
Bus 003 Device 004: ID 04f2:b6dd Chicony Electronics Co., Ltd HD User Facing
[ 0.005180] [00001c02] libusb: debug [libusb_get_device_descriptor]  
[ 0.005181] [00001c02] libusb: debug [libusb_get_device_descriptor]  
Bus 003 Device 005: ID 1c7a:0575 LighTuning Technology Inc. EgisTec EH575
[ 0.005236] [00001c02] libusb: debug [libusb_get_device_descriptor]  
[ 0.005237] [00001c02] libusb: debug [libusb_get_device_descriptor]  
[... cut ...]
[ 0.005339] [00001c02] libusb: debug [libusb_unref_device] destroy device 3.1
[ 0.005341] [00001c02] libusb: debug [libusb_unref_device] destroy device 2.2
[ 0.005342] [00001c02] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.005344] [00001c02] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.005461] [00001c03] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
[ 0.005492] [00001c02] libusb: debug [usbi_remove_event_source] remove fd 5
[ 0.005496] [00001c02] libusb: debug [usbi_remove_event_source] remove fd 4
```

closed 2023-12-21 by mcuee