libusb/libusb #1400: call op_detach_kernel_driver(linux_usbfs.c) failed, return -1, and errno= 113

pengxianheng-nreal, 2023-12-24:
when I call op_detach_kernel_driver on  Android,  It‘s failed， and logs :

12-22 16:26:11.091 15259 15790 E libusb  : [10.626199] [00003dae] libusb: error [op_detach_kernel_driver] detach failed error -1 errno 113

what does 113 errno means?

pengxianheng-nreal, 2023-12-24:
The detailed logs is below:

12-22 16:26:11.089 15259 15790 D libusb  : [10.624803] [00003dae] libusb: debug [libusb_init] created default context
12-22 16:26:11.089 15259 15790 D libusb  : [10.624850] [00003dae] libusb: debug [libusb_init] libusb v1.0.23.11399
12-22 16:26:11.090 15259 15790 D libusb  : [10.625463] [00003dae] libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
12-22 16:26:11.090 15259 15790 D libusb  : [10.625518] [00003dae] libusb: debug [get_kernel_version] reported kernel version is 5.15.94-android13-8-1222390-abS9110ZCU2BWK7
12-22 16:26:11.090 15259 15790 D libusb  : [10.625529] [00003dae] libusb: debug [op_init] bulk continuation flag supported
12-22 16:26:11.090 15259 15790 D libusb  : [10.625537] [00003dae] libusb: debug [op_init] zero length packet flag supported
12-22 16:26:11.090 15259 15790 D libusb  : [10.625545] [00003dae] libusb: debug [op_init] max iso packet length is (likely) 49152 bytes
12-22 16:26:11.090 15259 15790 D libusb  : [10.625578] [00003dae] libusb: debug [op_init] sysfs can relate devices
12-22 16:26:11.090 15259 15790 D libusb  : [10.625586] [00003dae] libusb: debug [op_init] sysfs has complete descriptors
12-22 16:26:11.090 15259 15790 E libusb  : [10.625721] [00003dae] libusb: error [linux_netlink_start_event_monitor] failed to create netlink socket (13)
12-22 16:26:11.090 15259 15790 E libusb  : [10.625774] [00003dae] libusb: error [sysfs_get_device_list] opendir devices failed errno=13
12-22 16:26:11.090 15259 15790 D libusb  : [10.625810] [00003dae] libusb: debug [usbi_add_pollfd] add fd 155 events 1
12-22 16:26:11.090 15259 15790 D libusb  : [10.625921] [00003dae] libusb: debug [libusb_wrap_sys_device] wrap_sys_device 0x9a
12-22 16:26:11.090 15259 15790 D libusb  : [10.625931] [00003dae] libusb: debug [linux_get_device_address] getting address for device: (null) detached: 1
12-22 16:26:11.090 15259 15790 D libusb  : [10.625990] [00003dae] libusb: debug [op_wrap_sys_device] allocating new device for fd 154
12-22 16:26:11.090 15259 15790 W libusb  : [10.626056] [00003dae] libusb: warning [usbfs_get_active_config] get_configuration failed ret=-1 errno=113
12-22 16:26:11.090 15259 15790 D libusb  : [10.626069] [00003dae] libusb: debug [usbi_add_pollfd] add fd 154 events 4
12-22 16:26:11.090 15259 15790 D libusb  : [10.626078] [00003dae] libusb: debug [libusb_get_config_descriptor] index 0
12-22 16:26:11.090 15259 15790 D libusb  : [10.626086] [00003dae] libusb: debug [parse_configuration] skipping descriptor 0xb
12-22 16:26:11.091 15259 15790 D libusb  : [10.626165] [00003dae] libusb: debug [parse_endpoint] skipping descriptor 25
12-22 16:26:11.091 15259 15790 D libusb  : [10.626176] [00003dae] libusb: debug [parse_endpoint] skipping descriptor b
12-22 16:26:11.091 15259 15790 D libusb  : [10.626186] [00003dae] libusb: debug [libusb_detach_kernel_driver] interface 0
12-22 16:26:11.091 15259 15790 E libusb  : [10.626199] [00003dae] libusb: error [op_detach_kernel_driver] detach failed error -1 errno 113
12-22 16:26:11.091 15259 15790 D libusb  : [10.626209] [00003dae] libusb: debug [libusb_alloc_transfer] transfer 0xb400007a61b754d0
12-22 16:26:11.091 15259 15790 D libusb  : [10.626216] [00003dae] libusb: debug [libusb_submit_transfer] transfer 0xb400007a61b754d0
12-22 16:26:11.091 15259 15790 E libusb  : [10.626227] [00003dae] libusb: error [submit_control_transfer] submiturb failed error -1 errno=113
12-22 16:26:11.091 15259 15790 D libusb  : [10.626234] [00003dae] libusb: debug [libusb_free_transfer] transfer 0xb400007a61b754d0

mcuee, 2023-12-24:
I can not find errno = 113 for libusb. Maybe it is from the lower level.  And it seems to me you are not using an official version of libusb. Please use libusb-1.0.26 release or git and try again.

What is the Android device you are testing with? 

Or better you may want to give PR #1164 a try.

pengxianheng-nreal, 2023-12-26:
@mcuee The libusb library I used is forked with this github 4 years ago.
I use the Android Galaxy S23. It's not rooted , and we get fd from android java (UsbManager, getFileDescriptor), and has right permission for my usb device.  On other android phone, My usb devices and my app can work fine.  On Galaxy S23,It's failed.

mcuee, 2023-12-26:
> @mcuee The libusb library I used is forked with this github 4 years ago. I use the Android Galaxy S23. It's not rooted , and we get fd from android java (UsbManager, getFileDescriptor), and has [...]

I see. In that case, the error codes may come from Android. Please check.

Unfortunately I can not help you much here as I have zero experiences with Android. I feel it is a problematic platform to support especially since the libusb main line is not working well and people have to use a fork.