libusb/libusb #1401: Wrong info for HID Usage and HID Usage page

tks11111, 2023-12-26:
I was trying to get HID report descriptor by using libUSB v1.0.26, and I found that there is a weird thing in windows_winusb.c. 
When we call libusb_open, the function might use the function hid_open in windows_winusb.c, and it will initialize the hid_handle first. 
The hid_handle will be stored in handle_priv->interface_handle[i].api_handle.
However, after storing the handle info in api_handle, it will use the final interface handle to get the usage and usagepage directly. 
This might cause confusion if we try to get different interface HID report descriptor by using control transfer like the following code shows.

```
uint16_t wValue = LIBUSB_DT_REPORT<<8;
uint16_t wLength = 9; // get first 9 byte data
r = libusb_control_transfer(dev_handle, 
        LIBUSB_ENDPOINT_IN | 
        LIBUSB_REQUEST_TYPE_STANDARD | 
        LIBUSB_RECIPIENT_INTERFACE, 
	LIBUSB_REQUEST_GET_DESCRIPTOR, wValue, 
        interface->bInterfaceNumber, DataInBuffer, 
        wLength, 100);
if(r < 0)
{
	printf("Error code: %d, data Out fail\n", r);
	libusb_close(dev_handle);
}
else
{
	printf("Data IN:\n ");
	for(int j = 0; j < r; j++)
	{
		if(0 == (j & 0x0f))
			printf("\n%d: ", j);
		printf("0x%02x ", DataInBuffer[j]);
	}
	printf("\n");
}
```
I wonder if there are any concerns that we only receive the usage and usage page of the last interface instead of storing the usage and usage page information for each interface.
Therefore, I add a workaround in hid_submit_control_transfer to update the usage and usagepage by interface. 
I'm new to libUSB, not sure if this is the best place to add this. Or is there a function that we can use to update the usage/usagepage ourselves?

```
switch (LIBUSB_REQ_TYPE(setup->RequestType)) {
case LIBUSB_REQUEST_TYPE_STANDARD:
	switch (setup->Request) {
	case LIBUSB_REQUEST_GET_DESCRIPTOR:
		// workaround for HID report of different interface 
		if((setup->Value >> 8) & 0xFF == LIBUSB_DT_REPORT)
		{
			PHIDP_PREPARSED_DATA preparsed_data = NULL;
			HIDP_CAPS capabilities;
			// Get the maximum input and output report size
			if (!HidD_GetPreparsedData(hid_handle, &preparsed_data) || !preparsed_data) {
				usbi_err(HANDLE_CTX(dev_handle), "could not read HID preparsed data (HidD_GetPreparsedData)");
				break;
			}
			if (HidP_GetCaps(preparsed_data, &capabilities) != HIDP_STATUS_SUCCESS) {
				usbi_err(HANDLE_CTX(dev_handle), "could not parse HID capabilities (HidP_GetCaps)");
			}
			priv->hid->usagePage = capabilities.UsagePage;
			priv->hid->usage = capabilities.Usage;
		}
		r = _hid_get_descriptor(dev_handle->dev, hid_handle, LIBUSB_REQ_RECIPIENT(setup->RequestType),
			(setup->Value >> 8) & 0xFF, setup->Value & 0xFF, transfer->buffer + LIBUSB_CONTROL_SETUP_SIZE, &size);
		break;
```
Let me know if I need to provide any.
Thanks.

mcuee, 2023-12-26:
The recommendation is to use HIDAPI library and not libusb.
Please give HIDAPI a try first.
https://github.com/libusb/libusb/wiki/FAQ#user-content-Does_libusb_support_USB_HID_devices
https://github.com/libusb/hidapi

mcuee, 2024-01-21:
@tks11111 

Any objection if I close this issue? 

The Windows HID backen will not receive much attentions unless there are sever bugs.

mcuee, 2024-01-21:
Reference:
https://github.com/libusb/libusb/wiki/FAQ#does-libusb-support-usb-hid-devices
Warning: use of the HID backend is highly discouraged. libusb project recommends the user to switch to [hidapi](https://github.com/libusb/hidapi).

tormodvolden, 2024-01-21:
We would consider a patch or merge request though, if if is well explained and with demonstrations of before and after. Note there might not be much people around that are familiar with the HID code and can review it properly.

To convince me, someone without much HID or HID backend insight, it would need a clear explanation of what goes wrong in this bug report, more than "this might cause confusion". You are receiving a DT report corresponding to the last interface instead of the interface that you specify?

mcuee, 2024-01-21:
Okay, I will keep this issue open.