libusb/libusb #1403: usb_bulk_write returned -110 should be 9

1458861693, 2023-12-26:
There is a Ketek SDD（silicon drift detector）in my laboratory，which data is read by a USB Device（electronic instrument）.
I have used the USB device in EPICS（experimental physic and industry Control system）on CentOS7. 
when I start the system, it appears error as the following picture. 
I think this error may be caused by libusb, can you give me some help to solve it? 
Looking forward to your reply, Thanks！
[image]
developing environment：
Linux version：
Linux localhost,localdomain 3.10.0-1160.105.l.el7.x86 64 #1 SMP Thu Dec 7 15:39:45 UTC 2023 x86 64 x86 64 x86 64 GNU/Linux
libusb version：
libusb-devel-0.1.4-3el7.x8664

mcuee, 2023-12-28:
Looks like you are using very old version of libusb and libusb-compat-0.1.

Please upgrade to latest version of libusb and libusb-compat-0.1 and test again.

mcuee, 2023-12-28:
Basically you can try your luck with CentOS or Redhat if you want the support. I will close this issue.

closed 2023-12-28 by mcuee

mcuee, 2023-12-28:
Reference:
* https://github.com/epics-modules/dxp/issues/10

1458861693, 2023-12-28:
Thanks for your reply very much.
This reference is also committed by me, but this problem is not solved yet.
Thanks for your help very much，I will try the suggested version of libusb on Ubuntu 20.04.
Thanks again.

mcuee, 2023-12-28:
> Thanks for your reply very much. This reference is also committed by me, but this problem is not solved yet. Thanks for your help very much，I will try the suggested version of libusb on Ubuntu 20.04. Thanks again.

If you use newer version of CentOS or use Fedora, they will use libusb-compat-0.1 (on top of libusb-1.0).

1458861693, 2023-12-28:
I try the new version of libusb-1.0.26 on CentOS7，but when I compile this package，it appears this error：
[image]

Is this mean the version of my gcc compiler is old？ Should I upgrade it？ 
I have tried libusb-0.1.10a，libusb-1.0.26 with libusb-compat-0.1.8 on Ubuntu20.04 respectively，but it also doesn't work.
I really wish you can give me more help to solve it.

mcuee, 2023-12-28:
I will not be able to help you much on CentOS 7. 

But libusb 1.0.26 and libusb-compat-0.1 should work fine under Ubuntu 20.04. You should try them and post your log here.