libusb/libusb #1415: Device hotplug event notification

mehtadharmesh, 2023-12-31:
Hello,
I am using released version of libusb 1.0.26. I found for Windows 10 as well as for CentOS 7, device hotplug event notification is not working for me.
Does for anybody, it is working? For me, for both platform it says hotplug event notification is not supported.
It will be nice if we list supported platform/OS version.
-DM

mcuee, 2023-12-31:
Windows hotplug is not implemented yet. You can try PR #1406.
* https://github.com/libusb/libusb/pull/1406

As for Linux, it should work. Please post the debug log. Make sure you use libusb-1.0.26 or better 1.0.27-rc1 and not the bundled old version inside CentOS 7.

Please post the debug log for CentOS 7 using libusb-1.0.27-rc1. Thanks.

mcuee, 2023-12-31:
The documentation for reference:
https://libusb.sourceforge.io/api-1.0/group__libusb__hotplug.html#details

This page details how to use the libusb hotplug interface, where available.

Be mindful that not all platforms currently implement hotplug notification and that you should first call on [libusb_has_capability()](https://libusb.sourceforge.io/api-1.0/group__libusb__misc.html#ga9b8e324d28c624cd0b8e7ba21607b8db) with parameter [LIBUSB_CAP_HAS_HOTPLUG](https://libusb.sourceforge.io/api-1.0/group__libusb__misc.html#ggaab1b3fa0728c06fafbee897795889bd5a8b1bbccc648a6677cb9030a6ba827438) to confirm that hotplug support is available.

mcuee, 2024-01-21:
Close this for now.

Please re-open if you can provide more information.

closed 2024-01-21 by mcuee