libusb/libusb #1429: Add support for SuperSpeedPlus USB Device Capability

fabiensanglard, 2024-01-06:
See specs in USB 3.1 specs in section: 9.6.2.5 SuperSpeedPlus USB Device Capability

I have a WIP PR for it: https://github.com/libusb/libusb/pull/1428

fabiensanglard, 2024-01-24:
PR https://github.com/libusb/libusb/pull/1428 is no longer considered WIP. If you want to change the API/design, let's discuss here.

fabiensanglard, 2024-01-26:
@mcuee, @tormodvolden, @hjelmn Do you guys have a preference about which way this should go (expose raw data or parse it and expose it in custom struc)?

mcuee, 2024-01-27:
> @mcuee, @tormodvolden, @hjelmn Do you guys have a preference about which way this should go (expose raw data or parse it and expose it in custom struc)?

My prefere nce is to parse it and expose it in custom struct.

fabiensanglard, 2024-01-27:
The more I looked at it, the more  "custom struct" made sense. I have refactored it. Take a look when you get a chance.

tormodvolden, 2024-02-20:
> I have done it this way to be like the other descriptors we already have

Indeed that would have been more consistent with the other descriptors we have. But consistency is only one virtue. If a different structure makes it easier for the user, we should consider deviating from the others. Are there factors here that makes it more useful with digested data than was the case for the other, similar APIs?

"Descriptor" has a well-defined meaning in USB terminology. The existing, similar APIs return structures that exactly mirrors the "physical" descriptor on the bus, bar endian-conversion (well, there are some exception with extra fields added). If you are instead returning a completely different structure, I would consider using another name than *_descriptor. For instance, just *_capability (or -ies?).

tormodvolden, 2024-02-21:
All the descriptors that we expose have their bLength and bDescriptorType, for instance.
Try `git grep -A2 '_descriptor {$' libusb/libusb.h`

fabiensanglard, 2024-02-21:
I can change the name from `_descriptor` to `_capability` no problem.

But let me make one last attempt though at keeping the name :) :P !

It seems to be me that `bLength` and `bDescriptorType` have no value for users (at least I don't know of any). These fields are just here to allow parsing. I don't know why they are even exposed.

tormodvolden, 2024-02-21:
Yes, I don't see much reason for those fields either. It is probably based on the idea that the descriptors structs should match the real descriptor.

I just want to point out this change from existing APIs. I want it to be a intentional change that is agreed to and not something slipped in by oversight. @hjelmn What do you think about this?

fabiensanglard, 2024-02-28:
@hjelmn Ping on this?

fabiensanglard, 2024-03-08:
@hjelmn Ping :) ?

hjelmn, 2024-05-03:
Seems fine to me. It is inconsistent but we should probably do the right thing going forward and not include the useless fields in future descriptor structs.

We can always remove them and increment the ABI/API if needed. I doubt anyone actually uses them.

tormodvolden, 2024-05-07:
@hjelmn Thanks for confirming. And when it comes to ABI/API bumping I wouldn't rush for "esthetic" reasons.

@fabiensanglard I agree to using the *_descriptor name. All the information is coming from the USB descriptor and it would just make things more opaque and vague to conceal that in the name.

closed 2024-05-07 by tormodvolden

tormodvolden, 2024-07-28:
Doxygen warnings caused by this commit:

```
libusb/descriptor.c:1012: warning: Compound internal_ssplus_capability_descriptor is not documented.
```
Maybe it needs to be moved?

```
libusb/libusb.h:1077: warning: unable to resolve reference to 'libusb_ssplus_usb_device_capability_descriptor.numSublinkSpeedAttributes=' for \ref command
```
Caused by the trailing `=`. What was the intention with those `=`?

```
libusb/descriptor.c:1013: warning: Member bLength (variable) of struct internal_ssplus_capability_descriptor is not documented.
libusb/descriptor.c:1014: warning: Member bDescriptorType (variable) of struct internal_ssplus_capability_descriptor is not documented.
libusb/descriptor.c:1015: warning: Member bDevCapabilityType (variable) of struct internal_ssplus_capability_descriptor is not documented.
libusb/descriptor.c:1016: warning: Member bReserved (variable) of struct internal_ssplus_capability_descriptor is not documented.
libusb/descriptor.c:1017: warning: Member bmAttributes (variable) of struct internal_ssplus_capability_descriptor is not documented.
libusb/descriptor.c:1018: warning: Member wFunctionalitySupport (variable) of struct internal_ssplus_capability_descriptor is not documented.
libusb/descriptor.c:1019: warning: Member wReserved (variable) of struct internal_ssplus_capability_descriptor is not documented.
```
Should they be added?

```
libusb/libusb.h:1077: warning: unable to resolve reference to 'libusb_ssplus_usb_device_capability_descriptor.numSublinkSpeedAttributes=' for \ref command
```
Caused by the trailing `=`

fabiensanglard, 2024-08-01:
Sorry about this mistake. Here is a fix: PR 1545

tormodvolden, 2024-08-02:
What about the other points? And can you please include https://github.com/libusb/libusb/pull/1502#discussion_r1616315205 as well?

fabiensanglard, 2024-08-02:
Done

commit f00f06e9 (2024-05-07): Add support for SuperSpeed+ Capability Descriptors

See specs in USB 3.1 specs in section:

9.6.2.5 SuperSpeedPlus USB Device Capability

Closes #1428
Fixes #1429