libusb/libusb #1433: Usb printer not working on web Application

ujjain090, 2024-01-09:
I am below code for printer connectio
from escpos.printer import Usb
printer = Usb(0x04B8, 0x0202)

My problem is that when the client itself comes to the web application on the system the code deployed on the server system gives no command for print because the client connects the printer to the PC and shares the vendor ID and product ID but not the printer found from backed code. when we are updating vendor id and product id in our code

Code deployed on server system.
Please guide me how to print locally on client system with USB connection printer

mcuee, 2024-01-09:
This does not belong to libusb project as you are using a Python binding.

mcuee, 2024-01-09:
Reference:
* https://github.com/python-escpos/python-escpos/issues/617

closed 2024-01-09 by mcuee