libusb/libusb #1435: Valgrind Memory Leak

attackoncs, 2024-01-12:
I am using usbutils version 1:010-3 and libusb version 2:1.0.22-2. When I run 
"`valgrind --tool=memcheck --leak-check=full --show-reachable=yes --trace-children=yes lsusb`"
it shows a memory leak. Below is the Valgrind report. How can I resolve this issue?
[lsusb.txt](https://github.com/libusb/libusb/files/13916840/lsusb.txt)

attackoncs, 2024-01-12:
I guess it is because `usbi_alloc_device` allocated memory to create dev, and then added dev to `ctx->usb_devs` in `usbi_connect_device`, but didn't free dev in ctx->usb_devs when exiting?

attackoncs, 2024-01-12:
oh,sorry,I know what is really reason,I just change libusb_free_device_list(list, 0); to libusb_free_device_list(list, 1); it is work fine,same as https://github.com/libusb/libusb/issues/1398

mcuee, 2024-01-12:
BTW, next time when you want to report a libusb issue, please use the latest release or RC version, and then latest git, not using an older release like libusb-1.0.22.

closed 2024-01-12 by mcuee