libusb/libusb #1441: `hotplugtest` results are not predictable under Linux for external USB hubs

mcuee, 2024-01-22:
From here:
https://github.com/libusb/libusb/pull/1425#issuecomment-1879482085

Somehow I think the output of hotplugtest is not predictable under Linux for external hubs.

Adding a USB 3.0 USB Mass Storage to the USB Type-C hub.

The USB Type-C hub will show as two devices.
1) 05e3:0612 -- USB 3.0 hub, the USB 3.0 card reader (05e3:0749) is attached to it.
2) 05e3:0610 -- USB 2.0 hub, Microchip PICKit 4 (03eb:2177) is attached to it.

```
mcuee@UbuntuSwift3 ~/build/libusb/libusb_git (master)$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 007: ID 05e3:0612 Genesys Logic, Inc. Hub
Bus 002 Device 008: ID 05e3:0749 Genesys Logic, Inc. USB3.0 Card Reader
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 003: ID 1ea7:0064 SHARKOON Technologies GmbH 2.4G Mouse
Bus 003 Device 004: ID 04f2:b6dd Chicony Electronics Co., Ltd HD User Facing
Bus 003 Device 005: ID 1c7a:0575 LighTuning Technology Inc. EgisTec EH575
Bus 003 Device 006: ID 8087:0026 Intel Corp. 
Bus 003 Device 019: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 003 Device 024: ID 03eb:2177 Atmel Corp. MPLAB PICkit 4 CMSIS-DAP
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

mcuee@UbuntuSwift3 ~/build/libusb/libusb_git (master)$ lsusb -t
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/1p, 480M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/4p, 10000M
    |__ Port 001: Dev 007, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 003: Dev 008, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
/:  Bus 003.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 002: Dev 019, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 002: Dev 024, If 0, Class=Human Interface Device, Driver=usbhid, 480M
        |__ Port 002: Dev 024, If 1, Class=Communications, Driver=cdc_acm, 480M
        |__ Port 002: Dev 024, If 2, Class=CDC Data, Driver=cdc_acm, 480M
    |__ Port 003: Dev 003, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 005: Dev 004, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 005: Dev 004, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 007: Dev 005, If 0, Class=Vendor Specific Class, Driver=[none], 480M
    |__ Port 010: Dev 006, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 010: Dev 006, If 1, Class=Wireless, Driver=btusb, 12M
/:  Bus 004.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/4p, 10000M
```

Detaching the hub -- only shows PICKit 4 and the associated USB 2.0 hub device.
```
mcuee@UbuntuSwift3 ~/build/libusb/libusb_git (master)$ sudo ./examples/hotplugtest 
Device detached: 03eb:2177
Device detached: 05e3:0610
```

Attaching the hub: only shows the associated USB 2.0 hub and USB 3.0 Hub.
```
mcuee@UbuntuSwift3 ~/build/libusb/libusb_git (master)$ sudo ./examples/hotplugtest 
Device attached: 05e3:0610
Device attached: 05e3:0612
```

mcuee, 2024-01-22:
This is probably a side effect of PR #1350.
* https://github.com/libusb/libusb/pull/1350

We can either live with this or maybe mention in the `hotplugtest` example.

mcuee, 2024-01-22:
This may well happen under macOS as well but there is a current issue with macOS.
* https://github.com/libusb/libusb/issues/1366

tormodvolden, 2024-01-22:
Note that hotplugtest is a bit primitive. It will quit after two events, whether attach or detach. Try changing `while (done < 2)` and you'll might see all three devices.

Also, it keeps one handle, from the last device that was attached (whether it was able to open it or not), and then when a device is detached, it will close that handle, regardless of which device was detached.

It is strictly meant to be used with one device, specified on the command line (VID and optionally PID). And this device should be attached, then detached. So (un)plugging several devices at a time, without filtering, causes various issues.

mcuee, 2024-01-23:
> It is strictly meant to be used with one device, specified on the command line (VID and optionally PID). And this device should be attached, then detached. So (un)plugging several devices at a time, without filtering, causes various issues.

@tormodvolden 

That is why I think PR #1350 is wrong.

mcuee, 2024-01-24:
> Note that hotplugtest is a bit primitive. It will quit after two events, whether attach or detach. Try changing `while (done < 2)` and you'll might see all three devices.

This seems to work for my Ubuntu 20.04 system, with a nested external hubs setup.

Before attached the external hubs.
```

mcuee@UbuntuSwift3 ~/build/libusb/libusb (master)$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 003: ID 1ea7:0064 SHARKOON Technologies GmbH 2.4G Mouse
Bus 003 Device 004: ID 04f2:b6dd Chicony Electronics Co., Ltd HD User Facing
Bus 003 Device 005: ID 1c7a:0575 LighTuning Technology Inc. EgisTec EH575
Bus 003 Device 006: ID 8087:0026 Intel Corp. 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
mcuee@UbuntuSwift3 ~/build/libusb/libusb (master)$ lsusb | wc -l
8
mcuee@UbuntuSwift3 ~/build/libusb/libusb (master)$ lsusb -t
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/1p, 480M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 003.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 003: Dev 003, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 005: Dev 004, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 005: Dev 004, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 007: Dev 005, If 0, Class=Vendor Specific Class, Driver=[none], 480M
    |__ Port 010: Dev 006, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 010: Dev 006, If 1, Class=Wireless, Driver=btusb, 12M
/:  Bus 004.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/4p, 10000M
```

After attaching the USB hubs.
```
mcuee@UbuntuSwift3 ~/build/libusb/libusb (master)$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 003: ID 1ea7:0064 SHARKOON Technologies GmbH 2.4G Mouse
Bus 003 Device 004: ID 04f2:b6dd Chicony Electronics Co., Ltd HD User Facing
Bus 003 Device 005: ID 1c7a:0575 LighTuning Technology Inc. EgisTec EH575
Bus 003 Device 006: ID 8087:0026 Intel Corp. 
Bus 003 Device 027: ID 0bda:5411 Realtek Semiconductor Corp. 4-Port USB 2.0 Hub
Bus 003 Device 028: ID 0bda:5411 Realtek Semiconductor Corp. 4-Port USB 2.0 Hub
Bus 003 Device 029: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 003 Device 030: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub
Bus 003 Device 031: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub
Bus 003 Device 032: ID 1915:1025 Nordic Semiconductor ASA ZY Control Mic
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 017: ID 0bda:0411 Realtek Semiconductor Corp. 4-Port USB 3.0 Hub
Bus 004 Device 018: ID 0bda:0411 Realtek Semiconductor Corp. 4-Port USB 3.0 Hub
Bus 004 Device 019: ID 05e3:0616 Genesys Logic, Inc. hub
Bus 004 Device 020: ID 05e3:0749 Genesys Logic, Inc. USB3.0 Card Reader
Bus 004 Device 021: ID 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet
mcuee@UbuntuSwift3 ~/build/libusb/libusb (master)$ lsusb | wc -l
19
mcuee@UbuntuSwift3 ~/build/libusb/libusb (master)$ lsusb -t | uniq
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/1p, 480M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 003.Port 001: Dev 001, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 001: Dev 027, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 001: Dev 028, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 002: Dev 029, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 003: Dev 030, If 0, Class=Hub, Driver=hub/7p, 480M
            |__ Port 005: Dev 031, If 0, Class=Hub, Driver=hub/7p, 480M
                |__ Port 006: Dev 032, If 0, Class=Audio, Driver=snd-usb-audio, 12M
                |__ Port 006: Dev 032, If 1, Class=Audio, Driver=snd-usb-audio, 12M
                |__ Port 006: Dev 032, If 2, Class=Human Interface Device, Driver=usbhid, 12M
                |__ Port 006: Dev 032, If 3, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 003: Dev 003, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 005: Dev 004, If 0, Class=Video, Driver=uvcvideo, 480M
[... 32 lines cut ...]
mcuee@UbuntuSwift3 ~/build/libusb/libusb (master)$ sudo ./examples/hotplugtest 
Device detached: 0bda:5411
Device detached: 05e3:0610
Device detached: 1915:1025
Device detached: 1a40:0201
Device detached: 1a40:0201
Device detached: 0bda:5411
Device detached: 05e3:0749
Device detached: 0bda:0411
Device detached: 0b95:1790
Device detached: 05e3:0616
Device detached: 0bda:0411
mcuee@UbuntuSwift3 ~/build/libusb/libusb (master)$ sudo ./examples/hotplugtest 
Device attached: 0bda:5411
Device attached: 0bda:0411
Device attached: 05e3:0616
Device attached: 0bda:5411
Device attached: 1a40:0201
Device attached: 05e3:0610
Device attached: 0bda:0411
Device attached: 05e3:0749
Device attached: 1a40:0201
Device attached: 0b95:1790
Device attached: 1915:1025
```

mcuee, 2024-01-24:
@tormodvolden 

Maybe hotplugtest can be enhanced to be able to list all the devices attached/detched.

mcuee, 2024-01-24:
BTW, #1425 is also good to have.
* https://github.com/libusb/libusb/pull/1425