libusb/libusb #1444: segfault error in libusb when usb device disconnected

chiragjethava-cpi, 2024-01-23:
I am seeing an segfault in libusb when USB device disconnected .

This segfault occurs every-time when I disconnected USB device except first time 

sample code used :  hotplugtest.c

https://github.com/libusb/libusb/blob/master/examples/hotplugtest.c

```
user@user:~$ sudo ./a.out 

Device detached: 2109:8817
Device detached: 045e:07f8
Device detached: 046d:c52b
Device detached: 0bda:8153
Device detached: 2109:2817
Device detached: 2109:0817

Device attached: 2109:0817
Device attached: 2109:2817
Device attached: 0bda:8153
Device attached: 046d:c52b
Device attached: 045e:07f8
Device attached: 2109:8817

Device detached: 2109:8817
Device detached: 045e:07f8
Device detached: 046d:c52b
Device detached: 0bda:8153
Device detached: 2109:2817
Device detached: 2109:0817

-> USB Hub disconnected

[  166.037615] usb 3-3: USB disconnect, device number 2
[  166.037626] usb 3-3.2: USB disconnect, device number 5
[  166.291289] usb 3-3.3: USB disconnect, device number 8
[  166.670607] usb 3-3.5: USB disconnect, device number 10

-> USB Hub connected

[  166.945235] usb 3-3: new high-speed USB device number 13 using xhci_hcd
[  167.099245] usb 3-3: New USB device found, idVendor=2109, idProduct=2817, bcdDevice= 5.33
[  167.099261] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  167.099265] usb 3-3: Product: USB2.0 Hub             
[  167.099269] usb 3-3: Manufacturer: VIA Labs, Inc.         
[  167.101376] hub 3-3:1.0: USB hub found
[  167.101522] hub 3-3:1.0: 5 ports detected
[  167.681471] usb 3-3: USB disconnect, device number 13
[  167.721440] usb 4-1: USB disconnect, device number 2
[  167.721468] r8152-cfgselector 4-1.1: USB disconnect, device number 3
[  167.722858] r8152 4-1.1:1.0 enx5c857e3d381a: Stop submitting intr, status -108
[  198.681809] usb 4-1: new SuperSpeed USB device number 4 using xhci_hcd
[  198.704436] usb 4-1: New USB device found, idVendor=2109, idProduct=0817, bcdDevice= 5.33
[  198.704449] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  198.704453] usb 4-1: Product: USB3.0 Hub             
[  198.704456] usb 4-1: Manufacturer: VIA Labs, Inc.         
[  198.706517] hub 4-1:1.0: USB hub found
[  198.706691] hub 4-1:1.0: 4 ports detected
[  198.833555] usb 3-3: new high-speed USB device number 14 using xhci_hcd
[  198.987533] usb 3-3: New USB device found, idVendor=2109, idProduct=2817, bcdDevice= 5.33
[  198.987542] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  198.987546] usb 3-3: Product: USB2.0 Hub             
[  198.987549] usb 3-3: Manufacturer: VIA Labs, Inc.         
[  198.990969] hub 3-3:1.0: USB hub found
[  198.991151] hub 3-3:1.0: 5 ports detected
[  199.301732] usb 4-1.1: new SuperSpeed USB device number 5 using xhci_hcd
[  199.323139] usb 4-1.1: New USB device found, idVendor=0bda, idProduct=8153, bcdDevice=30.00
[  199.323147] usb 4-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=6
[  199.323148] usb 4-1.1: Product: USB 10/100/1000 LAN
[  199.323150] usb 4-1.1: Manufacturer: CMI
[  199.323150] usb 4-1.1: SerialNumber: 000001
[  199.409953] r8152-cfgselector 4-1.1: reset SuperSpeed USB device number 5 using xhci_hcd
[  199.464170] r8152 4-1.1:1.0: load rtl8153a-4 v2 02/07/20 successfully
[  199.494700] r8152 4-1.1:1.0 eth0: v1.12.13
[  199.725460] usb 3-3.2: new full-speed USB device number 15 using xhci_hcd
[  199.877753] usb 3-3.2: New USB device found, idVendor=046d, idProduct=c52b, bcdDevice=12.11
[  199.877768] usb 3-3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  199.877774] usb 3-3.2: Product: USB Receiver
[  199.877778] usb 3-3.2: Manufacturer: Logitech
[  199.894142] logitech-djreceiver 0003:046D:C52B.000E: hiddev0,hidraw1: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:00:14.0-3.2/input2
[  200.023356] input: Logitech M510 as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.2/3-3.2:1.2/0003:046D:C52B.000E/0003:046D:4051.000F/input/input35
[  200.024036] logitech-hidpp-device 0003:046D:4051.000F: input,hidraw2: USB HID v1.11 Mouse [Logitech M510] on usb-0000:00:14.0-3.2/input2:1
[  200.149531] usb 3-3.3: new low-speed USB device number 16 using xhci_hcd
[  200.304191] usb 3-3.3: New USB device found, idVendor=045e, idProduct=07f8, bcdDevice= 3.00
[  200.304204] usb 3-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  200.304207] usb 3-3.3: Product: Wired Keyboard 600
[  200.304210] usb 3-3.3: Manufacturer: Microsoft
[  200.314327] input: Microsoft Wired Keyboard 600 as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.3/3-3.3:1.0/0003:045E:07F8.0010/input/input36
[  200.378349] hid-generic 0003:045E:07F8.0010: input,hidraw4: USB HID v1.11 Keyboard [Microsoft Wired Keyboard 600] on usb-0000:00:14.0-3.3/input0
[  200.381393] input: Microsoft Wired Keyboard 600 Consumer Control as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.3/3-3.3:1.1/0003:045E:07F8.0011/input/input37
[  200.441923] input: Microsoft Wired Keyboard 600 System Control as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.3/3-3.3:1.1/0003:045E:07F8.0011/input/input38
[  200.442363] hid-generic 0003:045E:07F8.0011: input,hidraw5: USB HID v1.11 Device [Microsoft Wired Keyboard 600] on usb-0000:00:14.0-3.3/input1
[  200.533738] r8152 4-1.1:1.0 enx5c857e3d381a: renamed from eth0
[  203.785616] usb 3-3.5: new high-speed USB device number 17 using xhci_hcd
[  203.891329] usb 3-3.5: New USB device found, idVendor=2109, idProduct=8817, bcdDevice= 0.01
[  203.891340] usb 3-3.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  203.891343] usb 3-3.5: Product: USB-C Multiport Adapter
[  203.891346] usb 3-3.5: Manufacturer: VIA Labs, Inc.         
[  203.891349] usb 3-3.5: SerialNumber: 0000000000000001

-> USB Hub disconnected

[  223.737881] usb 3-3: USB disconnect, device number 14
[  223.737893] usb 3-3.2: USB disconnect, device number 15
[  223.781716] usb 4-1: USB disconnect, device number 4
[  223.781732] r8152-cfgselector 4-1.1: USB disconnect, device number 5
[  223.782051] r8152 4-1.1:1.0 enx5c857e3d381a: Stop submitting intr, status -108
[  223.995004] usb 3-3.3: USB disconnect, device number 16
[  224.491184] usb 3-3.5: USB disconnect, device number 17
[  224.847266] show_signal_msg: 7 callbacks suppressed
[  224.847273] iio_info[7586]: segfault at 0 ip 00007f146b0d5fd7 sp 00007ffdc3caea40 error 6 in libusb-1.0.so.0.3.0[7f146b0cf000+f000] likely on CPU 15 (core 7, socket 0)
[  224.847317] Code: c7 47 28 00 00 00 00 48 8b 47 10 48 85 c0 74 b3 48 8b 00 48 83 f8 01 75 aa 48 8b 47 10 48 39 d8 74 36 48 8b 48 28 48 8b 50 20 <48> 89 11 48 89 4a 08 48 c7 40 20 00 00 00 00 48 c7 40 28 00 00 00
[  224.861786] iio_info[7598]: segfault at 0 ip 00007f6d1816dfd7 sp 00007ffc7dd5a380 error 6 in libusb-1.0.so.0.3.0[7f6d18167000+f000] likely on CPU 5 (core 5, socket 0)
[  224.861799] Code: c7 47 28 00 00 00 00 48 8b 47 10 48 85 c0 74 b3 48 8b 00 48 83 f8 01 75 aa 48 8b 47 10 48 39 d8 74 36 48 8b 48 28 48 8b 50 20 <48> 89 11 48 89 4a 08 48 c7 40 20 00 00 00 00 48 c7 40 28 00 00 00
[  224.861830] iio_info[7593]: segfault at 0 ip 00007f5740b94fd7 sp 00007ffeed9918a0 error 6 in libusb-1.0.so.0.3.0[7f5740b8e000+f000] likely on CPU 11 (core 3, socket 0)
[  224.861842] Code: c7 47 28 00 00 00 00 48 8b 47 10 48 85 c0 74 b3 48 8b 00 48 83 f8 01 75 aa 48 8b 47 10 48 39 d8 74 36 48 8b 48 28 48 8b 50 20 <48> 89 11 48 89 4a 08 48 c7 40 20 00 00 00 00 48 c7 40 28 00 00 00
[  224.861954] iio_info[7595]: segfault at 0 ip 00007fd18b512fd7 sp 00007ffd86c476b0 error 6 in libusb-1.0.so.0.3.0[7fd18b50c000+f000] likely on CPU 7 (core 7, socket 0)
[  224.861969] Code: c7 47 28 00 00 00 00 48 8b 47 10 48 85 c0 74 b3 48 8b 00 48 83 f8 01 75 aa 48 8b 47 10 48 39 d8 74 36 48 8b 48 28 48 8b 50 20 <48> 89 11 48 89 4a 08 48 c7 40 20 00 00 00 00 48 c7 40 28 00 00 00

```

tormodvolden, 2024-01-23:
This looks a lot like #1441.

tormodvolden, 2024-01-23:
I am not able to reproduce here. It might depend on timing, how fast the computer and the devices are. Maybe it is limited to USB 3.x / SuperSpeed hubs.

Can you please compile with debug info and see if you get more information about the segfault?
`./configure --enable-tests-build --enable-examples-build CFLAGS="-g"`

tormodvolden, 2024-01-23:
Please also run with `LIBUSB_DEBUG=3` .

You can debug it with gdb also:
 `sudo gdb ./a.out`
At the gdb prompt:
 `run`
Then when it segfaults:
 `backtrace`

BTW, do you compile it in another way than just "make"? Since you run it as a.out instead of ./examples/hotplugtest ...

chiragjethava-cpi, 2024-01-23:
> This looks a lot like #1441. EDIT, well, there it is not crashing.

executable is not showing up seg fault its a seg fault showing up on kernel logs from libusb when USB hub disconnected so there is no logs from gdb as there is no seg fault from running executable
```
[18893.726764] usb 3-3: USB disconnect, device number 14
[18893.726773] usb 3-3.2: USB disconnect, device number 15
[18893.764570] usb 4-1: USB disconnect, device number 44
[18893.764579] r8152-cfgselector 4-1.1: USB disconnect, device number 45
[18893.766257] r8152 4-1.1:1.0 enx5c857e3d381a: Stop submitting intr, status -108
[18893.941841] usb 3-3.3: USB disconnect, device number 16
[18894.385626] usb 3-3.5: USB disconnect, device number 17
[18894.737617] iio_info[78093]: segfault at 0 ip 00007fa73bf17fd7 sp 00007ffc1d763050 error 6 in libusb-1.0.so.0.3.0[7fa73bf11000+f000] likely on CPU 11 (core 3, socket 0)
[18894.737628] Code: c7 47 28 00 00 00 00 48 8b 47 10 48 85 c0 74 b3 48 8b 00 48 83 f8 01 75 aa 48 8b 47 10 48 39 d8 74 36 48 8b 48 28 48 8b 50 20 <48> 89 11 48 89 4a 08 48 c7 40 20 00 00 00 00 48 c7 40 28 00 00 00
[18894.752753] iio_info[78097]: segfault at 0 ip 00007fb972278fd7 sp 00007ffe3ab0db50 error 6 in libusb-1.0.so.0.3.0[7fb972272000+f000] likely on CPU 0 (core 0, socket 0)
[18894.752761] Code: c7 47 28 00 00 00 00 48 8b 47 10 48 85 c0 74 b3 48 8b 00 48 83 f8 01 75 aa 48 8b 47 10 48 39 d8 74 36 48 8b 48 28 48 8b 50 20 <48> 89 11 48 89 4a 08 48 c7 40 20 00 00 00 00 48 c7 40 28 00 00 00
[18894.752790] iio_info[78101]: segfault at 0 ip 00007f156ee42fd7 sp 00007ffec5341680 error 6 in libusb-1.0.so.0.3.0[7f156ee3c000+f000] likely on CPU 15 (core 7, socket 0)
[18894.752800] Code: c7 47 28 00 00 00 00 48 8b 47 10 48 85 c0 74 b3 48 8b 00 48 83 f8 01 75 aa 48 8b 47 10 48 39 d8 74 36 48 8b 48 28 48 8b 50 20 <48> 89 11 48 89 4a 08 48 c7 40 20 00 00 00 00 48 c7 40 28 00 00 00
[18894.752892] iio_info[78088]: segfault at 0 ip 00007f259ef43fd7 sp 00007ffe97477480 error 6 in libusb-1.0.so.0.3.0[7f259ef3d000+f000] likely on CPU 13 (core 5, socket 0)
[18894.752924] Code: c7 47 28 00 00 00 00 48 8b 47 10 48 85 c0 74 b3 48 8b 00 48 83 f8 01 75 aa 48 8b 47 10 48 39 d8 74 36 48 8b 48 28 48 8b 50 20 <48> 89 11 48 89 4a 08 48 c7 40 20 00 00 00 00 48 c7 40 28 00 00 00
```

chiragjethava-cpi, 2024-01-23:
```
$ g++ hotplugdetect.cpp -lusb-1.0

$ sudo ./a.out 
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.006441] [00003446] libusb: error [usbi_get_context] API misuse! Using non-default context as implicit default.
[ 0.006453] [00003446] libusb: debug [libusb_hotplug_register_callback] new hotplug cb 0x5561fa2679c0 with handle 1
[ 0.006459] [00003446] libusb: debug [libusb_hotplug_register_callback] new hotplug cb 0x5561fa267960 with handle 2
[ 0.006463] [00003446] libusb: error [usbi_get_context] API misuse! Using non-default context as implicit default.
[ 0.006469] [00003446] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.006474] [00003446] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.006481] [00003446] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 60000ms

[13.181029] [00003447] libusb: debug [linux_get_device_address] getting address for device: 4-1 detached: 0
[13.181043] [00003447] libusb: debug [linux_get_device_address] scan 4-1
[13.181068] [00003447] libusb: debug [linux_get_device_address] bus=4 dev=26
[13.181071] [00003447] libusb: debug [udev_hotplug_event] udev hotplug event. action: add.
[13.181073] [00003447] libusb: debug [linux_enumerate_device] busnum 4 devaddr 26 session_id 1050
[13.181078] [00003447] libusb: debug [linux_enumerate_device] allocating new device for 4/26 (session 1050)
[13.181098] [00003447] libusb: debug [linux_get_parent_info] dev 0x7ffb88001fa0 (4-1) has parent 0x5561fa269290 (usb4) port 1
[13.181111] [00003446] libusb: debug [usbi_wait_for_events] poll() returned 1
[13.181118] [00003446] libusb: debug [handle_event_trigger] event triggered
[13.181121] [00003446] libusb: debug [handle_event_trigger] hotplug message received
[13.181125] [00003446] libusb: debug [libusb_get_device_descriptor]  
Device attached: 2109:0817
[13.181136] [00003446] libusb: debug [libusb_open] open 4.26
[13.428459] [00003446] libusb: debug [usbi_add_event_source] add fd 7 events 4
[13.428494] [00003446] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[13.428498] [00003446] libusb: debug [handle_events] event sources modified, reallocating event data
[13.428504] [00003446] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[13.434792] [00003447] libusb: debug [linux_get_device_address] getting address for device: 3-3 detached: 0
[13.434812] [00003447] libusb: debug [linux_get_device_address] scan 3-3
[13.434902] [00003447] libusb: debug [linux_get_device_address] bus=3 dev=79
[13.434911] [00003447] libusb: debug [udev_hotplug_event] udev hotplug event. action: add.
[13.434915] [00003447] libusb: debug [linux_enumerate_device] busnum 3 devaddr 79 session_id 847
[13.434941] [00003447] libusb: debug [linux_enumerate_device] allocating new device for 3/79 (session 847)
[13.434996] [00003447] libusb: debug [linux_get_parent_info] dev 0x7ffb88002630 (3-3) has parent 0x5561fa2687e0 (usb3) port 3
[13.435023] [00003446] libusb: debug [usbi_wait_for_events] poll() returned 1
[13.435040] [00003446] libusb: debug [handle_event_trigger] event triggered
[13.435046] [00003446] libusb: debug [handle_event_trigger] hotplug message received
[13.435054] [00003446] libusb: debug [libusb_get_device_descriptor]  
Device attached: 2109:2817
[13.435081] [00003446] libusb: debug [libusb_close]  
[13.435086] [00003446] libusb: debug [usbi_remove_event_source] remove fd 7
[13.435106] [00003446] libusb: debug [libusb_open] open 3.79
[13.435138] [00003446] libusb: debug [usbi_add_event_source] add fd 7 events 4
[13.435148] [00003446] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[13.435154] [00003446] libusb: debug [handle_events] event sources modified, reallocating event data
[13.435163] [00003446] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[13.448004] [00003447] libusb: debug [linux_get_device_address] getting address for device: 4-1 detached: 0
[13.448017] [00003447] libusb: debug [linux_get_device_address] scan 4-1
[13.448051] [00003447] libusb: debug [linux_get_device_address] bus=4 dev=26
[13.448056] [00003447] libusb: debug [udev_hotplug_event] udev hotplug event. action: change.
[13.448060] [00003447] libusb: error [udev_hotplug_event] ignoring udev action change
[13.464048] [00003447] libusb: debug [linux_get_device_address] getting address for device: 3-3 detached: 0
[13.464060] [00003447] libusb: debug [linux_get_device_address] scan 3-3
[13.464087] [00003447] libusb: debug [linux_get_device_address] bus=3 dev=79
[13.464092] [00003447] libusb: debug [udev_hotplug_event] udev hotplug event. action: change.
[13.464093] [00003447] libusb: error [udev_hotplug_event] ignoring udev action change
[13.516345] [00003447] libusb: debug [linux_get_device_address] getting address for device: 4-1 detached: 0
[13.516364] [00003447] libusb: debug [linux_get_device_address] scan 4-1
[13.516429] [00003447] libusb: debug [linux_get_device_address] bus=4 dev=26
[13.516435] [00003447] libusb: debug [udev_hotplug_event] udev hotplug event. action: bind.
[13.529701] [00003447] libusb: debug [linux_get_device_address] getting address for device: 3-3 detached: 0
[13.529720] [00003447] libusb: debug [linux_get_device_address] scan 3-3
[... cut ...]
[  942.808054] Code: c7 47 28 00 00 00 00 48 8b 47 10 48 85 c0 74 b3 48 8b 00 48 83 f8 01 75 aa 48 8b 47 10 48 39 d8 74 36 48 8b 48 28 48 8b 50 20 <48> 89 11 48 89 4a 08 48 c7 40 20 00 00 00 00 48 c7 40 28 00 00 00
[  942.808089] iio_info[13835]: segfault at 0 ip 00007f704d550fd7 sp 00007ffd61f58750 error 6 in libusb-1.0.so.0.3.0[7f704d54a000+f000] likely on CPU 10 (core 2, socket 0)
[  942.808113] Code: c7 47 28 00 00 00 00 48 8b 47 10 48 85 c0 74 b3 48 8b 00 48 83 f8 01 75 aa 48 8b 47 10 48 39 d8 74 36 48 8b 48 28 48 8b 50 20 <48> 89 11 48 89 4a 08 48 c7 40 20 00 00 00 00 48 c7 40 28 00 00 00
chirag@chirag:~$ 

```

mcuee, 2024-01-24:
> I am using sample code and building locally on my host Ubuntu machine:
> 
> ```
> $ g++ hotplugdetect.cpp -lusb-1.0
> 

That is not correct, you are not using latest libusb git but rather your system libusb which may be of old version.

You may want to do the following and then see if you still have the same issue.
```
 git clone https://github.com/libusb/libusb.git
 cd libusb
 ./autogen.sh 
 make
 sudo ./examples/hotplugtest
```

BTW, I cannot reproduce the issue on my Ubutnu 20.04 system.

chiragjethava-cpi, 2024-01-24:
> > I am using sample code and building locally on my host Ubuntu machine:
> > ```
> > $ g++ hotplugdetect.cpp -lusb-1.0
> > ```
> 
> That is not correct, you are not using latest libusb git but rather your system libusb which may be of old version.
> 
> You may want to do the following and then see if you still have the same issue.
> 
> ```
>  git clone https://github.com/libusb/libusb.git
>  cd libusb
> ./bootstrap.sh
> ./configure --enable-tests-build --enable-examples-build CFLAGS="-g"
>  make
>  sudo ./examples/hotplugtest
> ```
> 
> BTW, I cannot reproduce the issue on my Ubutnu 20.04 system.
> 
> Runlog:
> 
> ```
> mcuee@UbuntuSwift3 ~/build/libusb/libusb (master)$ sudo ./examples/hotplugtest 
> Device detached: 05e3:0610
> Device detached: 05e3:0612
> mcuee@UbuntuSwift3 ~/build/libusb/libusb (master)$ sudo ./examples/hotplugtest 
> Device attached: 05e3:0612
> Device attached: 05e3:0610
> mcuee@UbuntuSwift3 ~/build/libusb/libusb (master)$ sudo dmesg | tail -n 16
> [ 1685.944082] usb 3-2: USB disconnect, device number 11
> [ 1686.114793] usb 2-1: USB disconnect, device number 7
> [ 1690.618804] usb 3-2: new high-speed USB device number 12 using xhci_hcd
> [ 1690.647004] usb 2-1: new SuperSpeed USB device number 8 using xhci_hcd
> [ 1690.670381] usb 2-1: New USB device found, idVendor=05e3, idProduct=0612, bcdDevice=92.26
> [ 1690.670390] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [ 1690.670393] usb 2-1: Product: USB3.0 Hub
> [ 1690.670395] usb 2-1: Manufacturer: GenesysLogic
> [ 1690.671468] hub 2-1:1.0: USB hub found
> [ 1690.671745] hub 2-1:1.0: 4 ports detected
> [ 1690.770640] usb 3-2: New USB device found, idVendor=05e3, idProduct=0610, bcdDevice=92.26
> [ 1690.770649] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [ 1690.770652] usb 3-2: Product: USB2.0 Hub
> [ 1690.770655] usb 3-2: Manufacturer: GenesysLogic
> [ 1690.772195] hub 3-2:1.0: USB hub found
> [ 1690.772608] hub 3-2:1.0: 4 ports detected
> ```

Yes I do using native install library thought this is best where I can get some insight related to this issue so posted here:

```
user@user:~$ lsb_release -a
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy
user@user:~$ 
user@user:~$ uname -a
Linux user 6.5.0-14-generic #14~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Nov 20 18:15:30 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
user@user:~$ 

user@user:~$ dpkg -l libusb-1.0*
||/ Name                   Version           Architecture Description
+++-======================-=================-============-===================================================
ii  libusb-1.0-0:amd64     2:1.0.25-1ubuntu2 amd64        userspace USB programming library
ii  libusb-1.0-0-dev:amd64 2:1.0.25-1ubuntu2 amd64        userspace USB programming library development files
ii  libusb-1.0-doc         2:1.0.25-1ubuntu2 all          documentation for userspace USB programming
chirag@chirag:~$ 

user@user:~$ ldd a.out 
	linux-vdso.so.1 (0x00007ffeecef5000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f33c2756000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f33c2400000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f33c272c000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f33c2794000)

```

However I am not able to build locally libusb to test  same with latest libusb:
```
user@user:~/libusb$ ./bootstrap.sh
autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: configure.ac: not using Intltool
autoreconf: configure.ac: not using Gtkdoc
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:23: installing './compile'
configure.ac:26: installing './install-sh'
configure.ac:26: installing './missing'
Makefile.am: installing './INSTALL'
examples/Makefile.am: installing './depcomp'
libusb/Makefile.am:6: error: Libtool library used but 'LIBTOOL' is undefined
libusb/Makefile.am:6:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
libusb/Makefile.am:6:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
libusb/Makefile.am:6:   If 'LT_INIT' is in 'configure.ac', make sure
libusb/Makefile.am:6:   its definition is in aclocal's search path.
parallel-tests: installing './test-driver'
autoreconf: error: automake failed with exit status: 1
```

mcuee, 2024-01-24:
You have to install dependancies like `automake, autoconf, libtool and libudev-dev`.

From the above log, you have installed `automake` and `autoconf`. You need to install `libtool` and `libudeve-dev`.

chiragjethava-cpi, 2024-01-24:
@mcuee  thanks using master branch I didn't see any issue

closed 2024-01-24 by chiragjethava-cpi