libusb/libusb #1445: hotplugtest crash when unplug an external USB hub under macOS

mcuee, 2024-01-24:
From here:
* https://github.com/libusb/libusb/issues/1366#issuecomment-1902491217

Initially no USB devices were attached. Then I pluged in the USB hub (with one USB device attached), then I unpluged the USB hub, which would casue the crash.

```
mcuee@mcuees-Mac-mini libusb % ./examples/listdevs
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000024] [000087e8] libusb: debug [libusb_init_context] created default context
[ 0.000030] [000087e8] libusb: debug [libusb_init_context] libusb v1.0.27.11874-rc1
[ 0.000043] [000087e8] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000114] [000087e9] libusb: debug [darwin_event_thread_main] creating hotplug event source
[ 0.000285] [000087e9] libusb: debug [darwin_event_thread_main] darwin event thread ready to receive events
[ 0.000320] [000087e8] libusb: debug [libusb_get_device_list]  
[ 0.000353] [000087e8] libusb: debug [libusb_exit] destroying default context
[ 0.000366] [000087e9] libusb: debug [darwin_event_thread_main] darwin event thread exiting
[ 0.000473] [000087e8] libusb: debug [usbi_remove_event_source] remove fd 3
mcuee@mcuees-Mac-mini libusb % ./examples/hotplugtest
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000024] [0000881c] libusb: debug [libusb_init_context] created default context
[ 0.000030] [0000881c] libusb: debug [libusb_init_context] libusb v1.0.27.11874-rc1
[ 0.000040] [0000881c] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000115] [0000881d] libusb: debug [darwin_event_thread_main] creating hotplug event source
[ 0.000327] [0000881d] libusb: debug [darwin_event_thread_main] darwin event thread ready to receive events
[ 0.000362] [0000881c] libusb: debug [libusb_hotplug_register_callback] new hotplug cb 0x600001264210 with handle 1
[ 0.000367] [0000881c] libusb: debug [libusb_hotplug_register_callback] new hotplug cb 0x6000012641e0 with handle 2
[ 0.000375] [0000881c] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 0.000377] [0000881c] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.000380] [0000881c] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.000384] [0000881c] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[ 5.267009] [0000881d] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x8fa2b5471
[ 5.267258] [0000881d] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x8fa2b5471
[ 5.278855] [0000881d] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 5.278917] [0000881d] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 5.278971] [0000881d] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 5.278977] [0000881d] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 5.278981] [0000881d] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 5.279018] [0000881d] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 5.279027] [0000881d] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 5.279032] [0000881d] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x1a40
[ 5.279035] [0000881d] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0201
[ 5.279039] [0000881d] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0100
[ 5.279042] [0000881d] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x00
[ 5.279071] [0000881d] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x01
[ 5.279076] [0000881d] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 5.279080] [0000881d] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 5.279132] [0000881d] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 5.279140] [0000881d] libusb: debug [process_new_device] allocating new device in context 0x135f04c00 for with session 0x8fa2b5471
[ 5.279146] [0000881d] libusb: debug [process_new_device] found device with address 31 port = 1 parent = 0x0 at 0x60000327c0ba
[ 5.279240] [0000881c] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 5.279249] [0000881c] libusb: debug [handle_event_trigger] event triggered
[ 5.279253] [0000881c] libusb: debug [handle_event_trigger] hotplug message received
[ 5.279260] [0000881c] libusb: debug [libusb_get_device_descriptor]  
Device attached: 1a40:0201
[ 5.279285] [0000881c] libusb: debug [libusb_open] open 2.31
[ 5.279570] [0000881c] libusb: warning [darwin_open] USBDeviceOpen: another process has device opened for exclusive access
[ 5.279648] [0000881c] libusb: debug [darwin_open] device open for access
[ 5.279658] [0000881c] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 5.279662] [0000881c] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 5.279667] [0000881c] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[ 5.526906] [0000881d] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x8fa8b402c
[ 5.527172] [0000881d] libusb: debug [darwin_get_cached_device] parent sessionID: 0x8fa2b5471
[ 5.527183] [0000881d] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x8fa8b402c/0x2150000 against cached device with sessionID/locationID 0x8fa2b5471/0x2100000
[ 5.527188] [0000881d] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x8fa8b402c
[ 5.537741] [0000881d] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 5.537769] [0000881d] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 5.537774] [0000881d] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 5.537778] [0000881d] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[... 109 lines cut ...]
Device detached: 1a40:0201
[ 2.795870] [00008915] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 2.795876] [00008915] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 2.795882] [00008915] libusb: debug [usbi_wait_for_events] poll() 1 fds with timeout in 60000ms
[ 2.797542] [00008916] libusb: debug [darwin_devices_detached] notifying context 0x146605230 of device disconnect
[ 2.797901] [00008915] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 2.797915] [00008915] libusb: debug [handle_event_trigger] event triggered
[ 2.797921] [00008915] libusb: debug [handle_event_trigger] hotplug message received
[ 2.797929] [00008915] libusb: debug [libusb_get_device_descriptor]  
Device detached: 1a40:0201
[ 2.797941] [00008915] libusb: debug [libusb_exit] destroying default context
zsh: segmentation fault  ./examples/hotplugtest
mcuee@mcuees-Mac-mini libusb % ./examples/listdevs  
1915:1025 (bus 2, device 4) path: 1.5.6
1a40:0201 (bus 2, device 1) path: 1.5
1a40:0201 (bus 2, device 2) path: 1
mcuee@mcuees-Mac-mini libusb % ./examples/testlibusb
Dev (bus 2, device 4): 1915 - 1025 speed: 12M
  Manufacturer:              ZY.Ltd
  Product:                   ZY Control Mic
Dev (bus 2, device 1): 1A40 - 0201 speed: 480M
  Product:                   USB 2.0 Hub [MTT]
Dev (bus 2, device 2): 1A40 - 0201 speed: 480M
  Product:                   USB 2.0 Hub [MTT]
```

mcuee, 2024-01-24:
From @seanm 
> As I've said elsewhere, there are numerous threading bugs that still exist. Thread Sanitizer exposes some of them, so does -Wthread-safety. Once 1.0.27 is out, and more of my PRs are merged, I'll continue https://github.com/libusb/libusb/pull/1419, which will hopefully help.

mcuee, 2024-01-24:
From here
* https://github.com/libusb/libusb/issues/1366#issuecomment-1902509186

If I just unplug the attached USB device on the external HUB, then no issues. The following is with git main without any extra patches.

```
mcuee@mcuees-Mac-mini hotplug_debug % sudo ./examples/hotplugtest
Device detached: 1915:1025 (note: detach)
Device attached: 1915:1025 (note: attach)
```

The issue only appears if I unplug the external hub.

```
mcuee@mcuees-Mac-mini hotplug_debug % sudo ./examples/hotplugtest                                                             
Device detached: 1a40:0201
Device detached: 1a40:0201
AddressSanitizer:DEADLYSIGNAL
=================================================================
==30976==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x0001032c9c60 bp 0x00016d0471d0 sp 0x00016d047130 T0)
==30976==The signal is caused by a WRITE memory access.
==30976==Hint: address points to the zero page.
    #0 0x1032c9c60 in list_del libusbi.h:222
    #1 0x1032c9afc in usbi_hotplug_exit hotplug.c:209
    #2 0x1032bddd4 in libusb_exit core.c:2612
    #3 0x102dbaf24 in main hotplugtest.c:144
    #4 0x18c1f90dc  (<unknown module>)

==30976==Register values:
 x[0] = 0x0000000105f00a40   x[1] = 0x00000001036e00e0   x[2] = 0x000000016d0469a0   x[3] = 0x00000000000001f4  
 x[4] = 0x0000000000000030   x[5] = 0x0000000000000001   x[6] = 0x000000016c84c000   x[7] = 0x0000000000000001  
 x[8] = 0x0000007000020000   x[9] = 0x0000000000000000  x[10] = 0x0000000000000000  x[11] = 0x000000700001ffff  
x[12] = 0x000000016d0469d8  x[13] = 0x5cf5f6c7f63726fc  x[14] = 0x000000016d047240  x[15] = 0x0000000000000000  
x[16] = 0x000000018c574eac  x[17] = 0x00000001037f8728  x[18] = 0x0000000000000000  x[19] = 0x000000016d0474a0  
x[20] = 0x0000000102dba9c0  x[21] = 0x000000016d0475f0  x[22] = 0x00000001031cd910  x[23] = 0x000000016d047670  
x[24] = 0x000000016d0476b0  x[25] = 0x000000018c2785eb  x[26] = 0x0000000000000000  x[27] = 0x0000000000000000  
x[28] = 0x0000000000000000     fp = 0x000000016d0471d0     lr = 0x00000001032c9b00     sp = 0x000000016d047130  
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV libusbi.h:222 in list_del
==30976==ABORTING
zsh: abort      sudo ./examples/hotplugtest
```

mcuee, 2024-01-28:
On the other hand, this is not a regression as it is also present in libusb-1.0.26 release.
* https://github.com/libusb/libusb/issues/1366#issuecomment-1902495347

**So this is not a release blocker for 1.0.27 release.**

sonatique, 2024-01-30:
@mcuee : you may want to try if #1452 helps here.

mcuee, 2024-01-31:
> @mcuee : you may want to try if #1452 helps here.

Unfortunately it does not help.

```
mcuee@mcuees-Mac-mini libusb_pr1452 % ./examples/hotplugtest 
Device detached: 1a40:0201
Device detached: 1a40:0201
zsh: segmentation fault  ./examples/hotplugtest
```

`make check` is okay so no regression.
```
mcuee@mcuees-Mac-mini libusb_pr1452 % make check
Making check in libusb
make[1]: Nothing to be done for `check'.
Making check in examples
make[1]: Nothing to be done for `check'.
Making check in tests
/Library/Developer/CommandLineTools/usr/bin/make  check-TESTS
PASS: stress
PASS: stress_mt
PASS: set_option
PASS: init_context
PASS: macos
============================================================================
Testsuite summary for libusb-1.0 1.0.27-rc2
============================================================================
# TOTAL: 5
# PASS:  5
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
# ERROR: 0
============================================================================
make[1]: Nothing to be done for `check-am'.
```

mcuee, 2024-02-04:
The following test is using @sonatique's Windows hotplug branch.

Mac Mini M1 --> 7-ports powered USB 3.0 Hub --> 13 ports powered USB 2.0 Hub

1) Unplug the 7-ports USB 3.0 Hub --> segfaults

```
mcuee@mcuees-Mac-mini debug_apple_clang % ./examples/hotplugtest                                                                  
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000132] [00025766] libusb: debug [libusb_init_context] created default context
[ 0.000139] [00025766] libusb: debug [libusb_init_context] libusb v1.0.27.11878-rc2
[ 0.000150] [00025766] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000320] [00025768] libusb: debug [darwin_event_thread_main] creating hotplug event source
[ 0.000495] [00025768] libusb: debug [darwin_event_thread_main] darwin event thread ready to receive events
[ 0.000622] [00025766] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x5de703d801
[ 0.000643] [00025766] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x5de703d801
[ 0.002540] [00025766] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002545] [00025766] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002548] [00025766] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0210
[ 0.002550] [00025766] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002551] [00025766] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002553] [00025766] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 0.002555] [00025766] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.002557] [00025766] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002559] [00025766] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x5411
[ 0.002561] [00025766] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002563] [00025766] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002564] [00025766] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002566] [00025766] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002568] [00025766] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002581] [00025766] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002584] [00025766] libusb: debug [process_new_device] allocating new device in context 0x100503c00 for with session 0x5de703d801
[ 0.002588] [00025766] libusb: debug [process_new_device] found device with address 30 port = 2 parent = 0x0 at 0x103800eba
[ 0.002627] [00025766] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x5de7079fff
[ 0.002643] [00025766] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x5de7079fff/0x2600000 against cached device with sessionID/locationID 0x5de703d801/0x2200000
[ 0.002647] [00025766] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x5de7079fff
[ 0.002773] [00025766] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002777] [00025766] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002779] [00025766] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.002780] [00025766] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002782] [00025766] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002784] [00025766] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.002786] [00025766] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.002788] [00025766] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002789] [00025766] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.002791] [00025766] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002793] [00025766] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002795] [00025766] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002796] [00025766] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002798] [00025766] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002807] [00025766] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002810] [00025766] libusb: debug [process_new_device] allocating new device in context 0x100503c00 for with session 0x5de7079fff
[ 0.002813] [00025766] libusb: debug [process_new_device] found device with address 31 port = 6 parent = 0x0 at 0x103800d3a
[ 0.002846] [00025766] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x5de7257664
[ 0.002859] [00025766] libusb: debug [darwin_get_cached_device] parent sessionID: 0x5de7079fff
[ 0.002862] [00025766] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x5de7257664/0x2610000 against cached device with sessionID/locationID 0x5de7079fff/0x2600000
[... cut ...]
x[20] = 0x000000010047e9a8  x[21] = 0x000000016f9833c0  x[22] = 0x0000000100891910  x[23] = 0x000000016f983440  
x[24] = 0x000000016f983480  x[25] = 0x00000001820a062b  x[26] = 0x0000000000000000  x[27] = 0x0000000000000000  
x[28] = 0x0000000000000000     fp = 0x000000016f982ec0     lr = 0x000000010098dc8c     sp = 0x000000016f982e20  
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV libusbi.h:222 in list_del
==26080==ABORTING
zsh: abort      ./examples/hotplugtest
```

mcuee, 2024-02-04:
git main (same as 1.0.27 release) results are the same.

<details>
<summary> Unplug --> Segfault; Plug --> no issues </summary>

```
mcuee@mcuees-Mac-mini arm64_debug % export LIBUSB_DEBUG=4
mcuee@mcuees-Mac-mini arm64_debug % ./examples/hotplugtest                                                                  
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000134] [0002c794] libusb: debug [libusb_init_context] created default context
[ 0.000141] [0002c794] libusb: debug [libusb_init_context] libusb v1.0.27.11882
[ 0.000162] [0002c794] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000320] [0002c796] libusb: debug [darwin_event_thread_main] creating hotplug event source
[ 0.000497] [0002c796] libusb: debug [darwin_event_thread_main] darwin event thread ready to receive events
[ 0.000611] [0002c794] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x609aeabed0
[ 0.000632] [0002c794] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x609aeabed0
[ 0.002327] [0002c794] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002332] [0002c794] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002334] [0002c794] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0210
[ 0.002336] [0002c794] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002338] [0002c794] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002340] [0002c794] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 0.002342] [0002c794] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.002344] [0002c794] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002345] [0002c794] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x5411
[ 0.002347] [0002c794] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002349] [0002c794] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002351] [0002c794] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002353] [0002c794] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002354] [0002c794] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002367] [0002c794] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002370] [0002c794] libusb: debug [process_new_device] allocating new device in context 0x100103c00 for with session 0x609aeabed0
[ 0.002374] [0002c794] libusb: debug [process_new_device] found device with address 6 port = 2 parent = 0x0 at 0x103400f7a
[ 0.002413] [0002c794] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x609b0f574f
[ 0.002430] [0002c794] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x609b0f574f/0x2600000 against cached device with sessionID/locationID 0x609aeabed0/0x2200000
[ 0.002433] [0002c794] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x609b0f574f
[ 0.002557] [0002c794] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002561] [0002c794] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002563] [0002c794] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.002564] [0002c794] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002566] [0002c794] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002568] [0002c794] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.002570] [0002c794] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.002571] [0002c794] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002573] [0002c794] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.002575] [0002c794] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002577] [0002c794] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002579] [0002c794] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002580] [0002c794] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002582] [0002c794] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002592] [0002c794] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002594] [0002c794] libusb: debug [process_new_device] allocating new device in context 0x100103c00 for with session 0x609b0f574f
[ 0.002597] [0002c794] libusb: debug [process_new_device] found device with address 1 port = 6 parent = 0x0 at 0x103400dfa
[ 0.002629] [0002c794] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x609b2ee296
[ 0.002642] [0002c794] libusb: debug [darwin_get_cached_device] parent sessionID: 0x609aeabed0
[ 0.002645] [0002c794] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x609b2ee296/0x2210000 against cached device with sessionID/locationID 0x609b0f574f/0x2600000
[... cut ...]
[ 4.719826] [0002c88c] libusb: debug [darwin_open] device open for access
[ 4.719836] [0002c88c] libusb: debug [libusb_close]  
[ 4.719849] [0002c88c] libusb: debug [libusb_exit] destroying default context
[ 4.719857] [0002c88c] libusb: debug [libusb_unref_device] destroy device 2.25
[ 4.719864] [0002c88c] libusb: debug [libusb_unref_device] destroy device 2.20
[ 4.719897] [0002c88d] libusb: debug [darwin_event_thread_main] darwin event thread exiting
[ 4.721248] [0002c88c] libusb: debug [usbi_remove_event_source] remove fd 3
```

</details>

sonatique, 2024-02-04:
@mcuee could you repeat the experiment with the last instrumented code of e393d8e that I mentioned here: https://github.com/libusb/libusb/pull/1406#issuecomment-1923828942

printf are in the common hotplug code that is used by Darwin as well, so it might help understanding. 

Thanks.

mcuee, 2024-02-04:
> @mcuee could you repeat the experiment with the last instrumented code of [e393d8e](https://github.com/libusb/libusb/commit/e393d8e425d81fd15812c7eff2085dfae41cd816) that I mentioned here: [#1406 (comment)](https://github.com/libusb/libusb/pull/1406#issuecomment-1923828942)
> 
> printf are in the common hotplug code that is used by Darwin as well, so it might help understanding.
> 
> Thanks.

It does not seem to crash with your debug branch.

There are a few warnings when building under Apple macOS.

<details>

```
mcuee@mcuees-Mac-mini debug_apple_clang64 % CFLAGS="-g -fsanitize=address" ../configure --enable-tests-build --enable-examples-build 
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... rm: conftest.dSYM: is a directory
yes
checking for g++... g++
checking whether the compiler supports GNU C++... yes
checking whether g++ accepts -g... yes
checking for g++ option to enable C++11 features... -std=gnu++11
checking for inline... inline
checking for a BSD-compatible install... /opt/homebrew/bin/ginstall -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /opt/homebrew/bin/gmkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports the include directive... yes (GNU style)
checking whether make supports nested variables... yes
checking dependency style of gcc... gcc3
checking dependency style of g++ -std=gnu++11... gcc3
checking build system type... aarch64-apple-darwin23.3.0
checking host system type... aarch64-apple-darwin23.3.0
checking how to print strings... printf
checking for a sed that does not truncate output... /opt/homebrew/bin/gsed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /Library/Developer/CommandLineTools/usr/bin/ld
checking if the linker (/Library/Developer/CommandLineTools/usr/bin/ld) is GNU ld... no
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... rm: conftest.dSYM: is a directory
BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 786432
checking how to convert aarch64-apple-darwin23.3.0 file names to aarch64-apple-darwin23.3.0 format... func_convert_file_noop
checking how to convert aarch64-apple-darwin23.3.0 file names to toolchain format... func_convert_file_noop
checking for /Library/Developer/CommandLineTools/usr/bin/ld option to reload object files... -r
checking for file... file
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... rm: conftest.dSYM: is a directory
no
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... rm: conftest.dSYM: is a directory
ok
checking for sysroot... no
checking for a working dd... /bin/dd
[... 173 lines cut ...]
freed: (0x103d00920)
freed: (0x103d00720)

removed (0x103d00c20) 2.1 = 1 ; parent=(0x0)
will unref (0x103d00c20) 2.1 = 1 ; parent=(0x0)
freed: (0x103d00c20)
```

2) Inserting the hub
```
mcuee@mcuees-Mac-mini debug_apple_clang64 % ./examples/hotplugtest
Device attached: 0bda:5411
Device attached: 0bda:0411
device list begin
(0x105a03320) 2.11 = 1 ; parent=(0x0)
(0x105a034a0) 2.10 = 1 ; parent=(0x0)
device list end
removed (0x105a03320) 2.11 = 1 ; parent=(0x0)
will unref (0x105a03320) 2.11 = 1 ; parent=(0x0)
freed: (0x105a03320)

removed (0x105a034a0) 2.10 = 1 ; parent=(0x0)
will unref (0x105a034a0) 2.10 = 1 ; parent=(0x0)
freed: (0x105a034a0)
```

mcuee, 2024-02-04:
@sonatique 

Maybe you find a fix for this issue. Nice.

Your branch does not have the issue mentioned here.
https://github.com/sonatique/libusb/commit/e393d8e425d81fd15812c7eff2085dfae41cd816

Full debug log under macOS 14.3 (Mac Mini M1 2020)

<details>
<summary> no more crashes under macOS 14.3 when removing the external hub </summary>

```
mcuee@mcuees-Mac-mini debug_apple_clang64 % export LIBUSB_DEBUG=4
mcuee@mcuees-Mac-mini debug_apple_clang64 % ./examples/hotplugtest 
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000164] [00009a22] libusb: debug [libusb_init_context] created default context
[ 0.000170] [00009a22] libusb: debug [libusb_init_context] libusb v1.0.27.11878-rc2
[ 0.000180] [00009a22] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000609] [00009a24] libusb: debug [darwin_event_thread_main] creating hotplug event source
[ 0.000802] [00009a24] libusb: debug [darwin_event_thread_main] darwin event thread ready to receive events
[ 0.001129] [00009a22] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x28cd46b0f
[ 0.001150] [00009a22] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x28cd46b0f
[ 0.002692] [00009a22] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002697] [00009a22] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002700] [00009a22] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0210
[ 0.002702] [00009a22] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002703] [00009a22] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002705] [00009a22] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x02
[ 0.002706] [00009a22] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.002708] [00009a22] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.002709] [00009a22] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x5411
[ 0.002711] [00009a22] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.002713] [00009a22] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.002724] [00009a22] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.002728] [00009a22] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002729] [00009a22] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002746] [00009a22] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002750] [00009a22] libusb: debug [process_new_device] allocating new device in context 0x100403c00 for with session 0x28cd46b0f
[ 0.002755] [00009a22] libusb: debug [process_new_device] found device with address 10 port = 2 parent = 0x0 at 0x103600eba
[ 0.002830] [00009a22] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x28cda3d34
[ 0.002847] [00009a22] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x28cda3d34/0x2600000 against cached device with sessionID/locationID 0x28cd46b0f/0x2200000
[ 0.002850] [00009a22] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x28cda3d34
[ 0.003072] [00009a22] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.003077] [00009a22] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.003078] [00009a22] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0300
[ 0.003080] [00009a22] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.003082] [00009a22] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.003083] [00009a22] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x03
[ 0.003085] [00009a22] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x09
[ 0.003086] [00009a22] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x0bda
[ 0.003088] [00009a22] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0411
[ 0.003090] [00009a22] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0117
[ 0.003091] [00009a22] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.003093] [00009a22] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x02
[ 0.003094] [00009a22] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.003096] [00009a22] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.003104] [00009a22] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.003107] [00009a22] libusb: debug [process_new_device] allocating new device in context 0x100403c00 for with session 0x28cda3d34
[ 0.003110] [00009a22] libusb: debug [process_new_device] found device with address 11 port = 6 parent = 0x0 at 0x103600d3a
[ 0.003175] [00009a22] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x28cf9026d
[ 0.003207] [00009a22] libusb: debug [darwin_get_cached_device] parent sessionID: 0x28cda3d34
[... cut ...]
will unref (0x1034007a0) 2.13 = 1 ; parent=(0x103400c20)
[ 4.134815] [00009a22] libusb: debug [libusb_unref_device] destroy device 2.13
freed: (0x1034007a0)

removed (0x103400c20) 2.10 = 1 ; parent=(0x0)
will unref (0x103400c20) 2.10 = 1 ; parent=(0x0)
[ 4.134843] [00009a22] libusb: debug [libusb_unref_device] destroy device 2.10
freed: (0x103400c20)

[ 4.134922] [00009a24] libusb: debug [darwin_event_thread_main] darwin event thread exiting
[ 4.142608] [00009a22] libusb: debug [usbi_remove_event_source] remove fd 3
```

</details>

mcuee, 2024-02-04:
More tests with four layer of USB hubs (total 15 USB devices).

Mac Mini M1 --> Ugreen USB Type-C hub --> 7 ports powered USB 3.0 hub --> 13 ports powered USB 2.0 hub --> Ugreen 2.0 USB hub with 100Mbps Ethernet port.

```
mcuee@mcuees-Mac-mini debug_apple_clang64 % git diff
diff --git a/examples/hotplugtest.c b/examples/hotplugtest.c
index c3426650..fdbedad0 100644
--- a/examples/hotplugtest.c
+++ b/examples/hotplugtest.c
@@ -138,7 +138,7 @@ int main(int argc, char *argv[])
                return EXIT_FAILURE;
        }
 
-       while (done < 2) {
+       while (done < 15) {
                rc = libusb_handle_events (NULL);
                if (LIBUSB_SUCCESS != rc)
                        printf ("libusb_handle_events() failed: %s\n",

mcuee@mcuees-Mac-mini debug_apple_clang64 % sudo ./examples/hotplugtest
Device attached: 05e3:0612
Device attached: 05e3:0610
Device attached: 0bda:0411
Device attached: 0bda:0411
Device attached: 05e3:0749
Device attached: 03eb:2104
Device attached: 0bda:5411
Device attached: 0bda:5411
Device attached: 1a40:0201
Device attached: 05e3:0608
Device attached: 046d:c52b
Device attached: 1a40:0201
Device attached: 0b95:772b
Device attached: 16c0:05dc
Device attached: 1915:1025
device list begin
(0x103f056a0) 0.15 = 1 ; parent=(0x103f05ca0)
(0x103f058a0) 0.14 = 1 ; parent=(0x103f022a0)
(0x103f05aa0) 0.13 = 1 ; parent=(0x103f022a0)
(0x103f05ca0) 0.12 = 2 ; parent=(0x103f02420)
(0x103f02020) 0.11 = 1 ; parent=(0x103f02420)
(0x103f022a0) 0.10 = 3 ; parent=(0x103f02420)
(0x103f02420) 0.9 = 4 ; parent=(0x103f02720)
(0x103f025a0) 0.8 = 1 ; parent=(0x103f02720)
(0x103f02720) 0.7 = 3 ; parent=(0x103f02fa0)
(0x103f028a0) 0.6 = 1 ; parent=(0x103f02fa0)
(0x103f02aa0) 0.5 = 1 ; parent=(0x103f02ca0)
(0x103f02ca0) 0.4 = 2 ; parent=(0x103f02e20)
(0x103f02e20) 0.3 = 2 ; parent=(0x103f03120)
(0x103f02fa0) 0.2 = 3 ; parent=(0x0)
(0x103f03120) 0.1 = 2 ; parent=(0x0)
device list end
removed (0x103f056a0) 0.15 = 1 ; parent=(0x103f05ca0)
will unref (0x103f056a0) 0.15 = 1 ; parent=(0x103f05ca0)
freed: (0x103f056a0)

removed (0x103f058a0) 0.14 = 1 ; parent=(0x103f022a0)
will unref (0x103f058a0) 0.14 = 1 ; parent=(0x103f022a0)
freed: (0x103f058a0)

removed (0x103f05aa0) 0.13 = 1 ; parent=(0x103f022a0)
will unref (0x103f05aa0) 0.13 = 1 ; parent=(0x103f022a0)
freed: (0x103f05aa0)

removed (0x103f05ca0) 0.12 = 1 ; parent=(0x103f02420)
will unref (0x103f05ca0) 0.12 = 1 ; parent=(0x103f02420)
freed: (0x103f05ca0)

removed (0x103f02020) 0.11 = 1 ; parent=(0x103f02420)
[... 144 lines cut ...]

removed (0x105b004a0) 1.6 = 1 ; parent=(0x105b00ba0)
will unref (0x105b004a0) 1.6 = 1 ; parent=(0x105b00ba0)
freed: (0x105b004a0)

removed (0x105b006a0) 1.5 = 1 ; parent=(0x105b008a0)
will unref (0x105b006a0) 1.5 = 1 ; parent=(0x105b008a0)
freed: (0x105b006a0)

removed (0x105b008a0) 1.4 = 1 ; parent=(0x105b00a20)
will unref (0x105b008a0) 1.4 = 1 ; parent=(0x105b00a20)
freed: (0x105b008a0)

removed (0x105b00a20) 1.3 = 1 ; parent=(0x105b00d20)
will unref (0x105b00a20) 1.3 = 1 ; parent=(0x105b00d20)
freed: (0x105b00a20)

removed (0x105b00ba0) 1.2 = 1 ; parent=(0x0)
will unref (0x105b00ba0) 1.2 = 1 ; parent=(0x0)
freed: (0x105b00ba0)

removed (0x105b00d20) 1.1 = 1 ; parent=(0x0)
will unref (0x105b00d20) 1.1 = 1 ; parent=(0x0)
freed: (0x105b00d20)
```

mcuee, 2024-02-04:
The above log is with AScan. The following is with TScan, `CFLAGS="-g -fsanitize=thread"`.

```
cuee@mcuees-Mac-mini debug_tscan % sudo ./examples/listdevs
Password:
listdevs(6211,0x1db1e1c40) malloc: nano zone abandoned due to inability to reserve vm space.
1915:1025 (bus 1, device 15) path: 1.3.4.5.6
16c0:05dc (bus 1, device 14) path: 1.3.4.4.3
0b95:772b (bus 1, device 13) path: 1.3.4.4.1
046d:c52b (bus 1, device 12) path: 1.3.4.6
1a40:0201 (bus 1, device 11) path: 1.3.4.5
05e3:0608 (bus 1, device 10) path: 1.3.4.4
1a40:0201 (bus 1, device 9) path: 1.3.4
0bda:5411 (bus 1, device 8) path: 1.3.1
0bda:5411 (bus 1, device 7) path: 1.3
03eb:2104 (bus 1, device 6) path: 1.1
05e3:0749 (bus 1, device 5) path: 2.3.1.2
0bda:0411 (bus 1, device 4) path: 2.3.1
0bda:0411 (bus 1, device 3) path: 2.3
05e3:0610 (bus 1, device 2) path: 1
05e3:0612 (bus 1, device 1) path: 2
device list begin
(0x106e00d80) 1.15 = 1 ; parent=(0x106e00f60)
(0x106e00de0) 1.14 = 1 ; parent=(0x106e00fc0)
(0x106e00e40) 1.13 = 1 ; parent=(0x106e00fc0)
(0x106e00f00) 1.12 = 1 ; parent=(0x106e01020)
(0x106e00f60) 1.11 = 2 ; parent=(0x106e01020)
(0x106e00fc0) 1.10 = 3 ; parent=(0x106e01020)
(0x106e01020) 1.9 = 4 ; parent=(0x106e010e0)
(0x106e01080) 1.8 = 1 ; parent=(0x106e010e0)
(0x106e010e0) 1.7 = 3 ; parent=(0x106e01320)
(0x106e01140) 1.6 = 1 ; parent=(0x106e01320)
(0x106e011a0) 1.5 = 1 ; parent=(0x106e01200)
(0x106e01200) 1.4 = 2 ; parent=(0x106e01260)
(0x106e01260) 1.3 = 2 ; parent=(0x106e012c0)
(0x106e01320) 1.2 = 3 ; parent=(0x0)
(0x106e012c0) 1.1 = 2 ; parent=(0x0)
device list end
removed (0x106e00d80) 1.15 = 1 ; parent=(0x106e00f60)
will unref (0x106e00d80) 1.15 = 1 ; parent=(0x106e00f60)
freed: (0x106e00d80)

removed (0x106e00de0) 1.14 = 1 ; parent=(0x106e00fc0)
will unref (0x106e00de0) 1.14 = 1 ; parent=(0x106e00fc0)
freed: (0x106e00de0)

removed (0x106e00e40) 1.13 = 1 ; parent=(0x106e00fc0)
will unref (0x106e00e40) 1.13 = 1 ; parent=(0x106e00fc0)
freed: (0x106e00e40)

removed (0x106e00f00) 1.12 = 1 ; parent=(0x106e01020)
will unref (0x106e00f00) 1.12 = 1 ; parent=(0x106e01020)
freed: (0x106e00f00)

removed (0x106e00f60) 1.11 = 1 ; parent=(0x106e01020)
will unref (0x106e00f60) 1.11 = 1 ; parent=(0x106e01020)
freed: (0x106e00f60)

removed (0x106e00fc0) 1.10 = 1 ; parent=(0x106e01020)
will unref (0x106e00fc0) 1.10 = 1 ; parent=(0x106e01020)
freed: (0x106e00fc0)

removed (0x106e01020) 1.9 = 1 ; parent=(0x106e010e0)
will unref (0x106e01020) 1.9 = 1 ; parent=(0x106e010e0)
freed: (0x106e01020)

removed (0x106e01080) 1.8 = 1 ; parent=(0x106e010e0)
will unref (0x106e01080) 1.8 = 1 ; parent=(0x106e010e0)
freed: (0x106e01080)

[... 161 lines cut ...]

removed (0x106c029a0) 0.6 = 1 ; parent=(0x106c02b80)
will unref (0x106c029a0) 0.6 = 1 ; parent=(0x106c02b80)
freed: (0x106c029a0)

removed (0x106c02a00) 0.5 = 1 ; parent=(0x106c02a60)
will unref (0x106c02a00) 0.5 = 1 ; parent=(0x106c02a60)
freed: (0x106c02a00)

removed (0x106c02a60) 0.4 = 1 ; parent=(0x106c02ac0)
will unref (0x106c02a60) 0.4 = 1 ; parent=(0x106c02ac0)
freed: (0x106c02a60)

removed (0x106c02ac0) 0.3 = 1 ; parent=(0x106c02b20)
will unref (0x106c02ac0) 0.3 = 1 ; parent=(0x106c02b20)
freed: (0x106c02ac0)

removed (0x106c02b80) 0.2 = 1 ; parent=(0x0)
will unref (0x106c02b80) 0.2 = 1 ; parent=(0x0)
freed: (0x106c02b80)

removed (0x106c02b20) 0.1 = 1 ; parent=(0x0)
will unref (0x106c02b20) 0.1 = 1 ; parent=(0x0)
freed: (0x106c02b20)
```

sonatique, 2024-02-04:
@mcuee : that is interesting. I pushed a little fix along the way in my debug branch (added the lock to prevent background hot plug thread to concurrently modify device list during exit, previously talked about in windows hotplug branch). 

This is probably what helps. 

I will look a logs into details later, thanks.

mcuee, 2024-02-04:
@seanm and @tormodvolden 

Maybe you are also interested in taking a look at the branch here, since it seems to fix this issue (maybe a surgical fix).

https://github.com/sonatique/libusb/commit/e393d8e425d81fd15812c7eff2085dfae41cd816

tormodvolden, 2024-02-04:
> @seanm and @tormodvolden
> 
> Maybe you are also interested in taking a look at the branch here, since it seems to fix this issue (maybe a surgical fix).

Isn't this what is in PR #1452?

sonatique, 2024-02-04:
@tormodvolden : let me explain: What @mcuee is referring to is not yet in #1452 . It happens that I made a special debug commit with lots of printf for debugging the remaining seg fault @mcuee is experiencing with my Windows hotplug implementation.
@mcuee used this commit under MacOS and noticed it fixed a bug. It occurred  that I made a commit for doing a "sanity fix" and included it in the branch containing the debug commit, with the aim of validating it later.
I first mentioned the problem related to the fix here: https://github.com/libusb/libusb/pull/1406#issuecomment-1887327705
It seems that this was actually a good idea.
I will move the fix to #1452 with some explanations, since it is related to hotplug_exit

sonatique, 2024-02-05:
@tormodvolden the fix is now included in #1452

mcuee, 2024-02-05:
@sonatique 

Thanks. I will test #1452 this evening or tomorrow.

mcuee, 2024-02-05:
@sonatique 

BTW, the Windows hotplug commit is also good under macOS.
https://github.com/sonatique/libusb/commit/497885df0059f6ede5863b52fecdc09d100f623b

```
mcuee@mcuees-Mac-mini libusb_win_hotplug % ./examples/listdevs          
1915:1025 (bus 2, device 18) path: 2.4.5.6
1a40:0201 (bus 2, device 6) path: 2.4.5
046d:c52b (bus 2, device 12) path: 2.4.6
05e3:0749 (bus 2, device 11) path: 6.1.2
1a40:0201 (bus 2, device 28) path: 2.4
0bda:5411 (bus 2, device 4) path: 2.1
0bda:0411 (bus 2, device 31) path: 6.1
0bda:0411 (bus 2, device 2) path: 6
0bda:5411 (bus 2, device 3) path: 2

mcuee@mcuees-Mac-mini libusb_win_hotplug % ./examples/hotplugtest       
Device detached: 0bda:0411
Device detached: 0bda:5411
Device detached: 0bda:0411
Device detached: 0bda:5411
Device detached: 05e3:0749
Device detached: 1a40:0201
Device detached: 1a40:0201
Device detached: 1915:1025
Device detached: 046d:c52b

mcuee@mcuees-Mac-mini libusb_win_hotplug % ./examples/hotplugtest
Device attached: 0bda:5411
Device attached: 0bda:0411
Device attached: 0bda:0411
Device attached: 05e3:0749
Device attached: 1a40:0201
Device attached: 0bda:5411
Device attached: 046d:c52b
Device attached: 1a40:0201
Device attached: 1915:1025
mcuee@mcuees-Mac-mini libusb_win_hotplug % git diff
diff --git a/examples/hotplugtest.c b/examples/hotplugtest.c
index c22bf747..88856bc1 100644
--- a/examples/hotplugtest.c
+++ b/examples/hotplugtest.c
@@ -138,7 +138,7 @@ int main(int argc, char *argv[])
                return EXIT_FAILURE;
        }
 
-       while (done < 2) {
+       while (done < 9) {
                rc = libusb_handle_events (NULL);
                if (LIBUSB_SUCCESS != rc)
                        printf ("libusb_handle_events() failed: %s\n",
```

mcuee, 2024-02-13:
Latest PR #1452 git is also fine. 
https://github.com/sonatique/libusb/commit/8b20a20a5efd07418d888dea4e8b6d8faf47d5d4

tormodvolden, 2024-03-14:
Just a note: The crash is not due to the unplug event alone, but because hotplugtest at the same time is calling libusb_exit immediately after seeing the unplug event. So I suppose it only appears when leaving the loop in hotplugtest.c.
`       while (done < 2) {
`
Or said differently, if 2 is replaced with a higher number, the plug/unplug cycles go fine until the count is reached, (which happens sooner if a hub with multiple devices is involved). Does that match what you see?

closed 2024-03-14 by tormodvolden

mcuee, 2024-03-15:
> Just a note: The crash is not due to the unplug event alone, but because hotplugtest at the same time is calling libusb_exit immediately after seeing the unplug event. So I suppose it only appears [...]

Yes, you can see in my test I increse the number 2 to a number of devices being plugged out or plugged in (the USB hub and the devices connected to the hub).

tormodvolden, 2024-03-15:
Yes, but if you raise the number to something much higher, so that it doesn't exit, you don't see the crash appearing, right?

commit 6c0ae1ab (2024-03-14): hotplug_exit: Mutex protection of context device list while cleaning it

This prevents concurrent access to the list by the event background
thread, which could still be processing a previous hotplug event and
having to modify the list.

Fixes #1366
Fixes #1445
Closes #1452
References #1406