libusb/libusb #1446: unable to read device descriptor for attached USB device when attached event occurs

ercjethava, 2024-01-24:
-> I am trying to read USB device descriptor information such as Manufacturer and Product string when USB device attached event occurs .
 
Sample code used along with minor changes:
https://github.com/libusb/libusb/blob/master/examples/hotplugtest.c

```
/* -*- Mode: C; indent-tabs-mode:t ; c-basic-offset:8 -*- */
/*
 * libusb example program for hotplug API
 * Copyright © 2012-2013 Nathan Hjelm <hjelmn@mac.com>
 *
 * This library is free software; you can redistribute it and/or
 * modify it under the terms of the GNU Lesser General Public
 * License as published by the Free Software Foundation; either
 * version 2.1 of the License, or (at your option) any later version.
 *
 * This library is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 * Lesser General Public License for more details.
 *
 * You should have received a copy of the GNU Lesser General Public
 * License along with this library; if not, write to the Free Software
 * Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA 02110-1301 USA
 */

#include <stdlib.h>
#include <stdio.h>

#include "libusb.h"

int done = 0;
libusb_device_handle *handle = NULL;

static int LIBUSB_CALL hotplug_callback(libusb_context *ctx, libusb_device *dev, libusb_hotplug_event event, void *user_data)
{
	struct libusb_device_descriptor desc;
	int rc;

	(void)ctx;
	(void)dev;
	(void)event;
	(void)user_data;

	rc = libusb_get_device_descriptor(dev, &desc);
	if (LIBUSB_SUCCESS == rc) {
		printf ("Device attached: %04x:%04x\n", desc.idVendor, desc.idProduct);
	} else {
		printf ("Device attached\n");
		fprintf (stderr, "Error getting device descriptor: %s\n",
			 libusb_strerror((enum libusb_error)rc));
	}

	if (handle) {
		libusb_close (handle);
		handle = NULL;
	}

	rc = libusb_open (dev, &handle);
	if (LIBUSB_SUCCESS != rc) {
		fprintf (stderr, "Error opening device: %s\n",
			 libusb_strerror((enum libusb_error)rc));
	}

	// read manufacturer, product number
	char manu[256], prod[256];
	rc = libusb_get_string_descriptor_ascii(handle, desc.iManufacturer, (unsigned char *)manu, sizeof(manu));
	if (rc < 0) {
		printf("Error getting manufacturer string: %s \n", libusb_error_name(rc));
	} else {
[... 173 lines cut ...]
[18553.749516] usb 4-1: new SuperSpeed USB device number 26 using xhci_hcd
[18553.771713] usb 4-1: New USB device found, idVendor=2109, idProduct=0817, bcdDevice= 5.33
[18553.771723] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[18553.771727] usb 4-1: Product: USB3.0 Hub             
[18553.771730] usb 4-1: Manufacturer: VIA Labs, Inc.         
[18553.773532] hub 4-1:1.0: USB hub found
[18553.773834] hub 4-1:1.0: 4 ports detected
[18554.341163] usb 4-1.1: new SuperSpeed USB device number 27 using xhci_hcd
[18554.362559] usb 4-1.1: New USB device found, idVendor=0bda, idProduct=8153, bcdDevice=30.00
[18554.362563] usb 4-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=6
[18554.362564] usb 4-1.1: Product: USB 10/100/1000 LAN
[18554.362565] usb 4-1.1: Manufacturer: CMI
[18554.362566] usb 4-1.1: SerialNumber: 000001
[18554.445643] r8152-cfgselector 4-1.1: reset SuperSpeed USB device number 27 using xhci_hcd
[18554.495617] r8152 4-1.1:1.0: load rtl8153a-4 v2 02/07/20 successfully
[18554.529865] r8152 4-1.1:1.0 eth0: v1.12.13
[18558.380873] usb 3-3.5: new high-speed USB device number 45 using xhci_hcd
[18558.482483] usb 3-3.5: New USB device found, idVendor=2109, idProduct=8817, bcdDevice= 0.01
[18558.482494] usb 3-3.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[18558.482498] usb 3-3.5: Product: USB-C Multiport Adapter
[18558.482501] usb 3-3.5: Manufacturer: VIA Labs, Inc.         
[18558.482503] usb 3-3.5: SerialNumber: 0000000000000001
[18558.659558] r8152 4-1.1:1.0 enx5c857e3d381a: renamed from eth0

```

mcuee, 2024-01-24:
Maybe you have to wait a bit for the device to get settled with proper driver.

ercjethava, 2024-01-27:
@mcuee 

I have tried adding 10sec delay before accessing attached device handle still same issue seeing.

```
static int LIBUSB_CALL hotplug_callback(libusb_context *ctx, libusb_device *dev, libusb_hotplug_event event, void *user_data)
{
	struct libusb_device_descriptor desc;
	int rc;

	(void)ctx;
	(void)dev;
	(void)event;
	(void)user_data;

	rc = libusb_get_device_descriptor(dev, &desc);
	if (LIBUSB_SUCCESS == rc) {
		printf ("Device attached: %04x:%04x\n", desc.idVendor, desc.idProduct);
	} else {
		printf ("Device attached\n");
		fprintf (stderr, "Error getting device descriptor: %s\n",
			 libusb_strerror((enum libusb_error)rc));
	}

	if (handle) {
		libusb_close (handle);
		handle = NULL;
	}

        printf("wait for 10 seconds\n");
        sleep(10);
        printf("wait for 10 seconds done\n");
	rc = libusb_open (dev, &handle);
	if (LIBUSB_SUCCESS != rc) {
		fprintf (stderr, "Error opening device: %s\n",
			 libusb_strerror((enum libusb_error)rc));
	}

	// read manufacturer, product number
	char manu[256], prod[256];
	rc = libusb_get_string_descriptor_ascii(handle, desc.iManufacturer, (unsigned char *)manu, sizeof(manu));
	if (rc < 0) {
		printf("Error getting manufacturer string: %s \n", libusb_error_name(rc));
	} else {
		printf("Manufacturer: %s\n", manu);
	}

	rc = libusb_get_string_descriptor_ascii(handle, desc.iProduct, (unsigned char *)prod, sizeof(prod));
	if (rc < 0) {
		printf("Error getting product string: %s \n", libusb_error_name(rc));
	} else {
		printf("Product: %s\n", prod);
	}

	return 0;
}
```

```
timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.001345] [000228af] libusb: error [usbi_get_context] API misuse! Using non-default context as implicit default.
[ 0.001351] [000228af] libusb: debug [libusb_hotplug_register_callback] new hotplug cb 0x5573ccf8e9c0 with handle 1
[ 0.001353] [000228af] libusb: debug [libusb_hotplug_register_callback] new hotplug cb 0x5573ccf8e960 with handle 2
[ 0.001355] [000228af] libusb: error [usbi_get_context] API misuse! Using non-default context as implicit default.
[ 0.001358] [000228af] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.001361] [000228af] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.001364] [000228af] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 60000ms

-> device plugged-in
[... 113 lines cut ...]
[10.524802] [000228b0] libusb: debug [udev_hotplug_event] udev hotplug event. action: change.
[10.524804] [000228b0] libusb: error [udev_hotplug_event] ignoring udev action change
[10.552746] [000228b0] libusb: debug [linux_get_device_address] getting address for device: 4-1.1 detached: 0
[10.552776] [000228b0] libusb: debug [linux_get_device_address] scan 4-1.1
[10.552802] [000228b0] libusb: debug [linux_get_device_address] bus=4 dev=19
[10.552805] [000228b0] libusb: debug [udev_hotplug_event] udev hotplug event. action: bind.
[11.037278] [000228b0] libusb: debug [linux_get_device_address] getting address for device: 3-3.5 detached: 0
[11.037289] [000228b0] libusb: debug [linux_get_device_address] scan 3-3.5
[11.037323] [000228b0] libusb: debug [linux_get_device_address] bus=3 dev=36
[11.037328] [000228b0] libusb: debug [udev_hotplug_event] udev hotplug event. action: bind.
wait for 10 seconds done
[15.619205] [000228af] libusb: debug [libusb_open] open 3.33
[15.619282] [000228af] libusb: debug [usbi_add_event_source] add fd 7 events 4
Error getting manufacturer string: LIBUSB_ERROR_BUSY 
Error getting product string: LIBUSB_ERROR_BUSY 
[15.619307] [000228af] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[15.619313] [000228af] libusb: debug [handle_events] event sources modified, reallocating event data
[15.619320] [000228af] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[15.619330] [000228af] libusb: debug [usbi_wait_for_events] poll() returned 1
[15.619334] [000228af] libusb: debug [handle_event_trigger] event triggered
[15.619339] [000228af] libusb: debug [handle_event_trigger] hotplug message received
[15.619347] [000228af] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[15.619352] [000228af] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms

```

mcuee, 2024-01-27:
> @mcuee
> 
> I have tried adding 10sec delay before accessing attached device handle still same issue seeing.

Please provide more details about your OS and device.
For example, you can post the output of `xusb -d xxxx:xxxx" or `lsusb -vvv` about your device. Thanks.

Make sure you use the latest version of libusb release version like libusb-1.0.26, or better use libusb-1.0.27-rc1.
https://github.com/libusb/libusb/releases

ercjethava, 2024-01-27:
@mcuee  I have tried with multiple device seeing same issue.

I have used latest master branch 

```
user@user:~/libusb$ git status
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   examples/hotplugtest.c

no changes added to commit (use "git add" and/or "git commit -a")
user@user:~/libusb$

```

**Test1: attached USB hub to PC**
```
user@user:~/libusb$ sudo ./examples/hotplugtest
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.004630] [000027b6] libusb: error [usbi_get_context] API misuse! Using non-default context as implicit default.
[ 0.004633] [000027b6] libusb: debug [libusb_hotplug_register_callback] new hotplug cb 0x55bec7e419d0 with handle 1
[ 0.004635] [000027b6] libusb: debug [libusb_hotplug_register_callback] new hotplug cb 0x55bec7e41970 with handle 2
[ 0.004636] [000027b6] libusb: error [usbi_get_context] API misuse! Using non-default context as implicit default.
[ 0.004639] [000027b6] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.004640] [000027b6] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.004642] [000027b6] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 60000ms

[ 8.383303] [000027c4] libusb: debug [linux_enumerate_device] busnum 3 devaddr 17 session_id 785
[ 8.383331] [000027c4] libusb: debug [linux_enumerate_device] allocating new device for 3/17 (session 785)
[ 8.383376] [000027c4] libusb: debug [linux_get_parent_info] dev 0x7fb108001fa0 (3-3) has parent 0x55bec7e424d0 (usb3) port 3
[ 8.383411] [000027b6] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 8.383435] [000027b6] libusb: debug [handle_event_trigger] event triggered
[ 8.383442] [000027b6] libusb: debug [handle_event_trigger] hotplug message received
[ 8.383454] [000027b6] libusb: debug [libusb_get_device_descriptor]  
Device attached: 2109:2817
waiting for 10 seconds : start 
[ 8.538102] [000027c4] libusb: debug [linux_enumerate_device] busnum 4 devaddr 8 session_id 1032
[ 8.538132] [000027c4] libusb: debug [linux_enumerate_device] allocating new device for 4/8 (session 1032)
[ 8.538196] [000027c4] libusb: debug [linux_get_parent_info] dev 0x7fb108002630 (4-1) has parent 0x55bec7e43490 (usb4) port 1
[10.234882] [000027c4] libusb: debug [linux_enumerate_device] busnum 3 devaddr 18 session_id 786
[10.234916] [000027c4] libusb: debug [linux_enumerate_device] allocating new device for 3/18 (session 786)
[10.234976] [000027c4] libusb: debug [linux_get_parent_info] dev 0x7fb1080040d0 (3-3.2) has parent 0x7fb108001fa0 (3-3) port 2
[10.235146] [000027c4] libusb: debug [linux_enumerate_device] busnum 4 devaddr 9 session_id 1033
[10.235157] [000027c4] libusb: debug [linux_enumerate_device] allocating new device for 4/9 (session 1033)
[10.235208] [000027c4] libusb: debug [linux_get_parent_info] dev 0x7fb1080048a0 (4-1.1) has parent 0x7fb108002630 (4-1) port 1
[10.394055] [000027c4] libusb: debug [linux_enumerate_device] busnum 3 devaddr 19 session_id 787
[10.394083] [000027c4] libusb: debug [linux_enumerate_device] allocating new device for 3/19 (session 787)
[10.394161] [000027c4] libusb: debug [linux_get_parent_info] dev 0x7fb108004f10 (3-3.3) has parent 0x7fb108001fa0 (3-3) port 3
[12.814608] [000027c4] libusb: debug [linux_enumerate_device] busnum 3 devaddr 20 session_id 788
[12.814631] [000027c4] libusb: debug [linux_enumerate_device] allocating new device for 3/20 (session 788)
[12.814674] [000027c4] libusb: debug [linux_get_parent_info] dev 0x7fb108006310 (3-3.5) has parent 0x7fb108001fa0 (3-3) port 5
waiting for 10 seconds : finished 
[18.383746] [000027b6] libusb: debug [libusb_open] open 3.17
[18.383859] [000027b6] libusb: debug [usbi_add_event_source] add fd 7 events 4
Error getting manufacturer string: LIBUSB_ERROR_BUSY 
Error getting product string: LIBUSB_ERROR_BUSY 
[18.383890] [000027b6] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[18.383896] [000027b6] libusb: debug [handle_events] event sources modified, reallocating event data
[18.383902] [000027b6] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[18.383915] [000027b6] libusb: debug [usbi_wait_for_events] poll() returned 1
[18.383920] [000027b6] libusb: debug [handle_event_trigger] event triggered
[18.383924] [000027b6] libusb: debug [handle_event_trigger] hotplug message received
[18.383933] [000027b6] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[18.383939] [000027b6] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms

[... 425 lines cut ...]
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     138
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               8

```

tormodvolden, 2024-03-14:
The title says "unable to read device descriptor". Does libusb_get_device_descriptor() fail?

mcuee, 2024-04-27:
@ercjethava 

Any update on this issue? I will close this issue if we do not get the reply for a long time (say two months).

mcuee, 2024-06-25:
Converted to discussion for now.