libusb/libusb #1449: umockdev fails when built with --enable-debug-log

sonatique, 2024-01-25:
When I do
```
./configure
make
./tests/umockdev
```
I get a successful run (no ERROR) with the following output:
```
# random seed: R02Sb1078e9ba8805394a1051f93621bf7c1
1..10
# Start of libusb tests
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
# DEBUG: umockdev.vala:110: Created udev test bed /tmp/umockdev.KYRBI2
# DEBUG: umockdev.vala:1294: parsing device description for /devices/usb1
# DEBUG: umockdev.vala:1363: creating device /devices/usb1 (subsystem usb)
# DEBUG: umockdev.vala:1408: create_node_for_device: creating file device /tmp/umockdev.KYRBI2/dev/bus/usb/001/001
# DEBUG: umockdev.vala:765: umockdev_testbed_uevent: lazily initializing uevent_sender
# DEBUG: umockdev.vala:769: umockdev_testbed_uevent: sending uevent add for device /sys/devices/usb1
[ 0.000749] [00354714] libusb: debug [libusb_get_device_list]
[ 0.000781] [00354714] libusb: debug [libusb_get_device_descriptor]
[ 0.000805] [00354714] libusb: debug [libusb_open] open 1.1
[ 0.001385] [00354714] libusb: debug [usbi_add_event_source] add fd 13 events 4
[ 0.001436] [00354714] libusb: debug [libusb_close]
[ 0.001456] [00354714] libusb: debug [usbi_remove_event_source] remove fd 13
[ 0.001498] [00354714] libusb: debug [libusb_get_device_list]
[ 0.001520] [00354714] libusb: debug [libusb_get_device_descriptor]
[ 0.001535] [00354714] libusb: debug [libusb_open] open 1.1
[ 0.001853] [00354714] libusb: debug [usbi_add_event_source] add fd 13 events 4
[ 0.001878] [00354714] libusb: debug [libusb_close]
[ 0.001897] [00354714] libusb: debug [usbi_remove_event_source] remove fd 13
[ 0.001924] [00354714] libusb: debug [libusb_get_device_list]
[ 0.001942] [00354714] libusb: debug [libusb_exit]
[ 0.001955] [00354714] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.002034] [00354714] libusb: debug [usbi_remove_event_source] remove fd 10
[ 0.002059] [00354714] libusb: debug [usbi_remove_event_source] remove fd 9
# DEBUG: umockdev.vala:137: Removing test bed /tmp/umockdev.KYRBI2
# GLib-DEBUG: unsetenv() is not thread-safe and should not be used after threads are created
ok 1 /libusb/open-close
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
# DEBUG: umockdev.vala:110: Created udev test bed /tmp/umockdev.H1ACI2
# DEBUG: umockdev.vala:1294: parsing device description for /devices/usb1
# DEBUG: umockdev.vala:1363: creating device /devices/usb1 (subsystem usb)
# DEBUG: umockdev.vala:1408: create_node_for_device: creating file device /tmp/umockdev.H1ACI2/dev/bus/usb/001/001
# DEBUG: umockdev.vala:765: umockdev_testbed_uevent: lazily initializing uevent_sender
# DEBUG: umockdev.vala:769: umockdev_testbed_uevent: sending uevent add for device /sys/devices/usb1
[ 0.008134] [00354714] libusb: error [usbi_get_context] API misuse! Using non-default context as implicit default.
[ 0.008586] [00354714] libusb: debug [libusb_get_device_list]
[ 0.008610] [00354714] libusb: debug [libusb_exit]
[ 0.008617] [00354714] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.008703] [00354714] libusb: debug [usbi_remove_event_source] remove fd 12
[ 0.008727] [00354714] libusb: debug [usbi_remove_event_source] remove fd 11
# DEBUG: umockdev.vala:137: Removing test bed /tmp/umockdev.H1ACI2
# GLib-DEBUG: unsetenv() is not thread-safe and should not be used after threads are created
ok 2 /libusb/implicit-default
# GLib-DEBUG: setenv()/putenv() are not thread-safe and should not be used after threads are created
# DEBUG: umockdev.vala:110: Created udev test bed /tmp/umockdev.JO3BI2
# DEBUG: umockdev.vala:1294: parsing device description for /devices/usb1
# DEBUG: umockdev.vala:1363: creating device /devices/usb1 (subsystem usb)
# DEBUG: umockdev.vala:1408: create_node_for_device: creating file device /tmp/umockdev.JO3BI2/dev/bus/usb/001/001
# DEBUG: umockdev.vala:765: umockdev_testbed_uevent: lazily initializing uevent_sender
# DEBUG: umockdev.vala:769: umockdev_testbed_uevent: sending uevent add for device /sys/devices/usb1
[ 0.015104] [00354714] libusb: debug [libusb_get_device_list]
[ 0.015130] [00354714] libusb: debug [libusb_get_device_descriptor]
[ 0.015146] [00354714] libusb: debug [libusb_open] open 1.1
[ 0.015614] [00354714] libusb: debug [usbi_add_event_source] add fd 17 events 4
[ 0.015641] [00354714] libusb: debug [libusb_submit_transfer] transfer 0x55be56ee64f0
[ 0.015657] [00354714] libusb: debug [add_to_flying_list] arm timer for timeout in 1ms (first in line)
[ 0.015680] [00354714] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 4
[ 0.016012] [00354714] libusb: debug [libusb_close]
[ 0.016034] [00354714] libusb: error [do_close] Device handle closed while transfer was still being processed, but the device is still connected as far as we know
[... 265 lines cut ...]
# DEBUG: umockdev.vala:769: umockdev_testbed_uevent: sending uevent add for device /sys/devices/usb1
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000023] [00355220] libusb: debug [libusb_init_context] libusb v1.0.27.11874-rc1
[ 0.000052] [00355220] libusb: debug [usbi_add_event_source] add fd 9 events 1
[ 0.000071] [00355220] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000078] [00355220] libusb: debug [usbi_add_event_source] add fd 10 events 1
[ 0.000099] [00355220] libusb: debug [get_kernel_version] reported kernel version is 5.15.0-91-generic
[ 0.000141] [00355220] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000151] [00355220] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000162] [00355220] libusb: debug [op_init] sysfs is available
[ 0.000923] [00355230] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.001269] [00355220] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.001283] [00355220] libusb: debug [linux_get_device_address] scan usb1
[ 0.001327] [00355220] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.001336] [00355220] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.001341] [00355220] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.001391] [00355220] libusb: debug [libusb_init_context] installing new context as implicit default
Bail out! FATAL-ERROR: Searched for "libusb_get_device_list" (4) but no message matched

** (/home/sylvain/libusb/tests/.libs/umockdev:3494432): ERROR **: 11:00:43.866: Searched for "libusb_get_device_list" (4) but no message matched
Trace/breakpoint trap (core dumped)
```

Any idea? Thanks.

mcuee, 2024-01-25:
I can reproduce the issue under Ubuntu 20.04.

```
mcuee@UbuntuSwift3 ../configure --enable-examples-build --enable-tests-build --enable-debug-log
mcuee@UbuntuSwift3 make
mcuee@UbuntuSwift3 ~/build/libusb/libusb/build_debuglog (master)$ ./tests/umockdev 
# random seed: R02S8e62e27d6f6a12ead1b86e294f8430bb
1..10
# Start of libusb tests
# DEBUG: umockdev.vala:123: Created udev test bed /tmp/umockdev.LL46H2
# DEBUG: umockdev.vala:1297: parsing device description for /devices/usb1
# DEBUG: umockdev.vala:1366: creating device /devices/usb1 (subsystem usb)
# DEBUG: umockdev.vala:1411: create_node_for_device: creating file device /tmp/umockdev.LL46H2/dev/bus/usb/001/001
# DEBUG: umockdev.vala:771: umockdev_testbed_uevent: lazily initializing uevent_sender
# DEBUG: umockdev.vala:775: umockdev_testbed_uevent: sending uevent add for device /sys/devices/usb1
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000014] [000066ba] libusb: debug [libusb_init_context] libusb v1.0.27.11874-rc1
[ 0.000033] [000066ba] libusb: debug [usbi_add_event_source] add fd 9 events 1
[ 0.000043] [000066ba] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000047] [000066ba] libusb: debug [usbi_add_event_source] add fd 10 events 1
[ 0.000058] [000066ba] libusb: debug [get_kernel_version] reported kernel version is 5.15.0-91-generic
[ 0.000080] [000066ba] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000084] [000066ba] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000096] [000066ba] libusb: debug [op_init] sysfs is available
[ 0.000302] [000066ca] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.000604] [000066ba] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.000613] [000066ba] libusb: debug [linux_get_device_address] scan usb1
[ 0.000943] [000066ba] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.000949] [000066ba] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.000953] [000066ba] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.001005] [000066ba] libusb: debug [libusb_init_context] installing new context as implicit default
Bail out! FATAL-ERROR: Searched for "libusb_get_device_list" (4) but no message matched

** (/home/mcuee/build/libusb/libusb/build_debuglog/tests/.libs/umockdev:26298): ERROR **: 22:19:39.453: Searched for "libusb_get_device_list" (4) but no message matched
Trace/breakpoint trap (core dumped)
```

sonatique, 2024-01-26:
@mcuee The reason is quite simple actually: umockdev expects to see all debug logs it prints out via a log callback it sets per context (not globally).

However function `libusb_set_log_cb_internal` contains this code:
```
#if !defined(ENABLE_DEBUG_LOGGING)
	if (mode & LIBUSB_LOG_CB_CONTEXT) {
		ctx = usbi_get_context(ctx);
		ctx->log_handler = cb;
	}
```
i.e. setting a per context log callback is ignored when ENABLE_DEBUG_LOGGING is set, i.e. when "--enable-debug-log" is passed to configure.

Therefore in this case umockdev cannot set its per context callback, hence no logs are printed, hence it fails when it first try to assert that an expected log is printed.

To summarize: umockdev ways of checking things required a per context log callback to be set, but "--enable-debug-log" disable this, hence umockdev is not compatible with "--enable-debug-log".

I don't know what is the best way to solve this. For instance we can:
- complement umockdev with an information message + quit at the very beginning of its execution when ENABLE_DEBUG_LOGGING is set, such that test result is "inconclusive" instead of "failed"
- change umockdev for using only global log callback, but this will requires carefully validating several places where code is especially expecting not to see logs belonging to other context than the one it is looking at.
- remove the #if !defined in  `libusb_set_log_cb_internal` but we shall understand why it was made in this way in the first place

Any thought?

tormodvolden, 2024-01-27:
> * complement umockdev with an information message + quit at the very beginning of its execution when ENABLE_DEBUG_LOGGING is set, such that test result is "inconclusive" instead of "failed"

Yes, thanks for the analysis, this solution makes sense. As you are saying, umockdev relies on the per-context log callbacks to do its checks properly, and at the same time we have, as quoted from the documentation:

> If ENABLE_DEBUG_LOGGING is defined then per context callback function will
> never be called.

sonatique, 2024-01-28:
@mcuee @tormodvolden https://github.com/libusb/libusb/pull/1450

sonatique, 2024-01-29:
@mcuee @tormodvolden https://github.com/libusb/libusb/pull/1451 for replacing #1450

commit fdff3cd3 (2024-01-31): configure.ac: Do not build umockdev if --enable-debug-log is used

umockdev relies on the per-context log callbacks to do its checks
properly, and these are not called if ENABLE_DEBUG_LOGGING is defined
(configure --enable-debug-log).

Fixes #1449
closed 2024-01-31 by tormodvolden