libusb/libusb #1453: Misc: To release 1.0.28

tormodvolden, 2024-01-31:
Pending review, may be postponed if not enough reviews:

(Windows)
- #1297 (new API)
- #1181
- #749
- #1227
- #1018 (libusb0.sys)
- #1127

(General)
- #1067
- #890

Important changes merged:

Other changes merged:

mcuee, 2024-02-01:
We also hope to sort out #86 with PR #1406
* #86
* #1406

mcuee, 2024-02-01:
Another long standing issue with libusb is better Android support. Hopefully we will have some further review and updates of the following PR.
* #1164

mcuee, 2024-02-01:
CI related PR may be desired as well.
* #1294 
* #1353 
* #1360

mcuee, 2024-02-02:
Hopefully the following confirmed bugs can be fixed as well.
* 1445

fabiensanglard, 2024-02-05:
Do you think we can add [Add support for SuperSpeed+ Capability Descriptors #1428](https://github.com/libusb/libusb/pull/1428) to 1.0.28?

tormodvolden, 2024-02-05:
Note that this is just a wishlist, and to not lose something off the radar as we approach the release. In principle we'll include what is ready and approved if we can. And waiting for something that we really wanted was not always a good idea. Finally, we have 50 pull requests and we don't need to list them all here.

sonatique, 2024-02-14:
@mcuee 
> Hopefully the following confirmed bugs can be fixed as well.
> 
> * [hotplugtest crash when unplug an external USB hub under macOS #1445](https://github.com/libusb/libusb/issues/1445)
> 
> #1452 does not help to fix #1445 but it may worth being reviewed and merged if deemed correct.
> 
> * [hotplug_exit: removes all parents with a single child remaining in parent tree + lock device list #1452](https://github.com/libusb/libusb/pull/1452)

Actually latest #1452 does fix this issue (more precisely the second commit named "mutex protection of context device list while cleaning it" , as stated here: https://github.com/libusb/libusb/issues/1445#issuecomment-1926939852

mcuee, 2024-02-14:
@sonatique 

Thanks, latest PR #1452 is able to fix #1445 now.
* #1445 
* #1452

mcuee, 2024-02-26:
Hopefully we can get more reviewers and testers to get libusb-1.0.28 release in year 2024.
* https://github.com/libusb/libusb/discussions/1325

martinling, 2024-07-25:
#1512 should be put on the list instead of #1297.

mcuee, 2024-07-25:
> #1512 should be put on the list instead of #1297.

Done.

mcuee, 2024-10-17:
Hopefully more people will be able to help out on testing and reviewing of the current PRs.

mcuee, 2024-11-14:
@tormodvolden and @hjelmn 

Looks like we will not be able to release 1.0.28 within this year if we may not have enough new features or bug fixes merged.

Still just wondering if you think it is still worth to push out one release.

seanm, 2024-11-19:
>...we may not have enough new features or bug fixes merged.

I think we do. I'm pretty sure I've fixed some memory overruns since the last release. That alone is worth a release IMHO.

tormodvolden, 2025-01-11:
Yes, I have been toying with the idea of making a maintenance release (maybe after merging a few of the most straight-forward pending PRs). We have already a good number of small fixes that can be important enough for those who run into them. And a few new API functions were added as well.

seanm, 2025-02-02:
> Yes, I have been toying with the idea of making a maintenance release

Yes, please.  It has been a year and a day now since the last release.

>(maybe after merging a few of the most straight-forward pending PRs). We have already a good number of small fixes that can be important enough for those who run into them. And a few new API functions were added as well.

Why do you not merge these as soon as they have passed review?

For who/what are releases?  For who/what is git master?  In my view, releases are something stable and tested for production use.  Master is for testing and stabilization in preparation for a release.  If you delay merging PRs, there is less opportunity for integration testing.  Individual PRs may be correct on their own, but when you combine them there may be negative interactions. The sooner they are integrated, the better. The more time that passes between integration and release, the better. Integrating then and releasing immediately is riskier.

Nothing has been merged for a 1/4 of a year now. That ought to mean master is pretty stable. Let's just cut a release with current master *now*.

tormodvolden, 2025-02-23:
You are right, we should start preparing a release without trying to squeeze in anything else, it is a perfect moment with a quiet master. I had for instance the Windows raw I/O in mind but need to recheck it and it can also wait for next release. I saw that you also looked at that one recently. Maybe just make an RC first, to encourage some more extensive testing. I don't think we disagree on what releases and master are for. I don't delay merging PRs for other reasons than that I haven't reviewed them yet and/or am unsure if they are ready or should be merged.

patstew, 2025-02-24:
FWIW I and I'm pretty sure a lot of other people would like to see the RAW_IO patches go in. Various versions of it have been good enough for ages, it just needs someone to merge it. Realistically, it's clear the maintainers of this project don't have enough time to make regular releases and review all the PRs. I'm not complaining, it's entirely fair and I'm grateful for the work you do. Nonetheless, if it doesn't go into this release I'm worried it's going to be yet another year before I can use an actual libusb release.

seanm, 2025-02-24:
> You are right, we should start preparing a release without trying to squeeze in anything else, it is a perfect moment with a quiet master.

My preference is still immediate release of what's currently master.

Then immediately after that, merge all those PRs that have been reviewed. Then everything is in master and can be tested together.

tormodvolden, 2025-03-06:
@hjemn, I agree with Sean that now is the perfect time to release what has been baking in master for months. It will be an easy release. We know from experience that even just picking something that looks straight-forward can have unexpected consequences and further delay things. Also, I am totally fine with preparing a new release again in 2-3 months time.

hjelmn, 2025-03-06:
@tormodvolden SGTM. If we already have enough for a new release it is worth cutting it and getting it out there. My main concern is if we want to get that Windows hot-plug PR in anytime soon it would need to be just after we cut a release to avoid any confusion or issues. Get a 1.0.28 out, merge obvious PRs, maybe get 1.0.28 or 1.0.29 (depending on the PRs taken) started, then focus a branch on windows hot-plug to stabilize it.

tormodvolden, 2025-03-06:
> just after we cut a release
Agree very much to this. And yes there are many other PRs that are more straight-forward than the Windows hot-plug (which has known regressions) and can go in for a soonish 1.0.29. Then get it in and hopefully it matures during what can be a longer release cycle again.

tormodvolden, 2025-03-06:
The RC is out and test results are very welcome in #1615.

Neustradamus, 2025-03-08:
Good job to all contributors! :)

tormodvolden, 2025-03-19:
Released.

closed 2025-03-19 by tormodvolden