libusb/libusb #1454: `stress_mt` test improbement

mcuee, 2024-02-02:
From here by @tormodvolden 
https://github.com/libusb/libusb/pull/1447#issuecomment-1918061222

> I think stress_mt needs some serious rework. It needs to differentiate between errors in init, enumeration or control requests.
Also add a third round without the open and control requests.
>
> Ideally it could go through all devices single-threaded and record a baseline, then run multithreaded and see if there is a difference (assuming the same devices are still around). At least for the device count. This wouldn't be difficult to add either. It would be more work to keep track of which devices can be opened and be sent control requests or not.

tormodvolden, 2024-02-02:
I have started on something in #1458.

tormodvolden, 2024-05-07:
On the macOS CI it can also get into an infinite loop it seems like: https://github.com/libusb/libusb/actions/runs/8991096355/job/24697895520 (been running 2 hours now). Not sure if it is our code spinning or if it is stuck in a call.