libusb/libusb #1455: Proper hotplug test utility

mcuee, 2024-02-02:
From here by @tormodvolden 

> I intend to work on a hotplug testing utility after the release, to be used for development (if no one beats me to it).

mcuee, 2024-02-05:
Some improbement ideas.
1) To be able to specify number of devicdes for monitoring.
2) Adopt similar idea to #1425

mcuee, 2026-03-09:
@tormodvolden has some good comments in #1706.

> The hotplugtest kind of expects you would start the program, then plug in devices, and then optionally repeat variations of (plug some out again, in again) then when it believes all devices are unplugged again it will quit.
> The program is not a a stress-test or validation tool, I guess it was originally meant to be a simple example showing how to use the hotplug API.

sonatique, 2026-03-09:
@mcuee : this is my proposal:  https://github.com/libusb/libusb/pull/1776

As I said elsewhere hotplugtest was never meant to be a "test" program (despite its name), and this is actually obvious once you realize it is in the "examples" folder, not in the "tests" folder ;-)

Therefore I didn't add real test, but rathe I tried to make it more user-friendly and exampling more features and things.