libusb/libusb #1459: On an Android device, libusb_bulk_transfer reception is abnormal

gongjianbo, 2024-02-04:
I have an Android Pad (Lenovo Tab M10), call libusb_bulk_transfer reception is abnormal. I tested libusb version 1.0.26 and 1.0.27.
Video can be captured normally in the libuvccamera program, so I think the hardware should be able to be used normally.
Other Android devices don't have this problem.

`int usb_ret = libusb_bulk_transfer(dev_handle, 0x82, pack_buf, pack_len, &actual, 0);`
Log info from libusb 1.0.27.
The Pad successful data:
`
D libusb  : [ 7.238560] [000039d1] libusb: debug [libusb_submit_transfer] transfer 0x6f8d71d770
D libusb  : [ 7.238576] [000039d1] libusb: debug [submit_bulk_transfer] need 2 urbs for new transfer with length 22528
D libusb  : [ 7.238655] [000039d1] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
D libusb  : [ 7.238669] [000039d1] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
D libusb  : [ 7.238682] [000039d1] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 60000ms
D libusb  : [ 7.249291] [000039d1] libusb: debug [usbi_wait_for_events] poll() returned 1
D libusb  : [ 7.249390] [000039d1] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=16384
D libusb  : [ 7.249411] [000039d1] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/2
D libusb  : [ 7.249433] [000039d1] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
D libusb  : [ 7.249446] [000039d1] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
D libusb  : [ 7.249456] [000039d1] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 60000ms
D libusb  : [ 7.253605] [000039d1] libusb: debug [usbi_wait_for_events] poll() returned 1
D libusb  : [ 7.253688] [000039d1] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=6144
D libusb  : [ 7.253704] [000039d1] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 2/2
D libusb  : [ 7.253717] [000039d1] libusb: debug [handle_bulk_completion] all URBs in transfer reaped --> complete!
D libusb  : [ 7.253732] [000039d1] libusb: debug [usbi_handle_transfer_completion] transfer 0x6f8d71d770 has callback 0x6e5f3a2d3c
D libusb  : [ 7.253744] [000039d1] libusb: debug [sync_transfer_cb] actual_length=22528
D libusb  : [ 7.253763] [000039d1] libusb: debug [libusb_free_transfer] transfer 0x6f8d71d770
`

The Pad failed data:
`
D libusb  : [ 6.707303] [000039d1] libusb: debug [libusb_submit_transfer] transfer 0x6f8d71d770
D libusb  : [ 6.707331] [000039d1] libusb: debug [submit_bulk_transfer] need 2 urbs for new transfer with length 22528
D libusb  : [ 6.707437] [000039d1] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
D libusb  : [ 6.707490] [000039d1] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
D libusb  : [ 6.707513] [000039d1] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 60000ms
D libusb  : [ 6.707832] [000039d1] libusb: debug [usbi_wait_for_events] poll() returned 1
D libusb  : [ 6.707919] [000039d1] libusb: debug [reap_for_handle] urb type=3 status=-121 transferred=10
D libusb  : [ 6.708004] [000039d1] libusb: debug [handle_bulk_completion] handling completion status -121 of bulk urb 1/2
D libusb  : [ 6.708042] [000039d1] libusb: debug [handle_bulk_completion] short transfer 10/16384 --> complete!
D libusb  : [ 6.708079] [000039d1] libusb: debug [discard_urbs] URB not found --> assuming ready to be reaped
D libusb  : [ 6.708121] [000039d1] libusb: debug [reap_for_handle] urb type=3 status=-104 transferred=0
D libusb  : [ 6.708150] [000039d1] libusb: debug [handle_bulk_completion] handling completion status -104 of bulk urb 2/2
D libusb  : [ 6.708179] [000039d1] libusb: debug [handle_bulk_completion] abnormal reap: urb status -104
D libusb  : [ 6.708207] [000039d1] libusb: debug [handle_bulk_completion] abnormal reap: last URB handled, reporting
D libusb  : [ 6.708236] [000039d1] libusb: debug [usbi_handle_transfer_completion] transfer 0x6f8d71d770 has callback 0x6e5f3a2d3c
D libusb  : [ 6.708265] [000039d1] libusb: debug [sync_transfer_cb] actual_length=10
D libusb  : [ 6.708299] [000039d1] libusb: debug [libusb_free_transfer] transfer 0x6f8d71d770
`

I tried packets smaller than 16384, but the reception was still abnormal.

Other Android device data:
`
D libusb  : [10.169424] [000066e4] libusb: debug [libusb_submit_transfer] transfer 0x7b01069840
D libusb  : [10.169485] [000066e4] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 22528
D libusb  : [10.169639] [000066e4] libusb: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
D libusb  : [10.169695] [000066e4] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
D libusb  : [10.169737] [000066e4] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 60000ms
D libusb  : [10.183931] [000066e4] libusb: debug [usbi_wait_for_events] poll() returned 1
D libusb  : [10.184089] [000066e4] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=22528
D libusb  : [10.184161] [000066e4] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
D libusb  : [10.184220] [000066e4] libusb: debug [handle_bulk_completion] all URBs in transfer reaped --> complete!
D libusb  : [10.184265] [000066e4] libusb: debug [usbi_handle_transfer_completion] transfer 0x7b01069840 has callback 0x7b1915ad3c
D libusb  : [10.184304] [000066e4] libusb: debug [sync_transfer_cb] actual_length=22528
D libusb  : [10.184357] [000066e4] libusb: debug [libusb_free_transfer] transfer 0x7b01069840
`

mcuee, 2024-02-04:
> I have an Android Pad (Lenovo Tab M10), call libusb_bulk_transfer reception is abnormal. I tested libusb version 1.0.26 and 1.0.27. Video can be captured normally in the libuvccamera program, so I think the hardware should be able to be used normally.
Other Android devices don't have this problem.

Unfortunately it is a wild west for the Android platforms. Since other Android devices do not have the issue, I think this is not a libusb issue but rather related to the specific Lenovo Tab M10.

mcuee, 2024-02-04:
For libuvccamera, I tend to think it can tolerate a few lost frames.

gongjianbo, 2024-02-05:
Maybe there is something problem with this Android Pad. Next I will update the driver of the hardware device and try again.

mcuee, 2024-02-05:
I will close this issue. Please re-open if you find there is an issue related to libusb. Thanks.

closed 2024-02-05 by mcuee