libusb/libusb #1464: Why spin locks for usbi_mutex_static_lock on Windows (instead of CRITICAL_SECTION)?

sonatique, 2024-02-14:
I discovered that usbi_mutex_static_lock is implemented with a "hand-coded" spin lock using InterlockedExchange in threads_windows.h, while `usbi_mutex_lock` is implemented using `EnterCriticalSection`.

In threads_posix.h, both static and regular mutex are implemented with `pthread_mutex_t `

I am wondering why this choice has been made for Windows, specifically for static lock vs regular locks.

Thanks!

sonatique, 2024-02-17:
I looked at the history, and it seems it dates back to the first Windows support implementation by @pbatard of 2010.

@pbatard : do you remember why you chose `LONG` for `usbi_mutex_static_t` and `HANDLE` for `usbi_mutex_t`, and therefore used "manul spin wait" for locking static mutex?

`usbi_mutex_t` has since then been changed to `CRITICAL_SECTION` but LONG and spin has been kept for static.

My concern is that I wish to introduce "try lock" for statically declated mutex, and the safer way (in my opinion) would be to use CRITICAL_SECTION for `usbi_mutex_static_t` , but I have to admit I don't understand the need for both `usbi_mutex_t `and `usbi_mutex_static_t` in the first place and hence I am not sure what would be implied by turning `LONG `into `CRITICAL_SECTION` (or maybe dropping `*_static` altogether?)

Thanks in advance!

pbatard, 2024-02-18:
You'd want to go back to the mailing list at the time and especially [this message](https://sourceforge.net/p/libusb/mailman/message/25189301/) along with the rest of the thread.

I'm afraid it's been way too long for me to recall any specifics, so the best you're ever going to get is what you'll find digging up the mailing list.

sonatique, 2024-02-18:
@pbatard : thanks a lot for your reply. I wasn't aware of that mailing list. I read the message you pointed to me and I think I understood the reason behind the choice, which is basically the one I guessed:

> Win32 mutexes are *never* statically initialized.  They are kernel objects, so a function
> must be called to allocate one.  Since the default context lock must exist
> by default, and is only held briefly, we instead implement it as a spinlock,
> not a mutex.

I am wondering whether we could use DLLMain for initializing a proper mutex object kind of statically, but I am not sure we could have a benefit from it that would pay for the added complexity.

Thanks again.

closed 2024-02-18 by sonatique