libusb/libusb #1465: 1.0.27 fails to build on Solaris/illumos

ptribble, 2024-02-14:
Builing 1.0.27 fails .on Solaris/illumos like so:

```
os/sunos_usb.c: In function 'sunos_get_link':
os/sunos_usb.c:89:6: error: 'larg' undeclared (first use in this function); did you mean 'arg'?
   89 |  if (larg->path) {
      |      ^~~~
      |      arg
os/sunos_usb.c:89:6: note: each undeclared identifier is reported only once for each function it appears in
os/sunos_usb.c:85:15: warning: unused variable 'link_arg' [-Wunused-variable]
   85 |  walk_link_t *link_arg = (walk_link_t *)arg;
      |               ^~~~~~~~

```

On inspection, the reason is clear: some instances of larg have been renamed link_arg; some others haven't.

Simply replacing the remaining instances of larg with link_arg in libusb/os/sunos_usb.c gives a successful build.

seanm, 2024-02-14:
Care to create a pull request?

tormodvolden, 2024-02-14:
Simply running "git grep -w larg" would have avoided this. I am sorry this slipped through my review, I didn't even think about the possibility of it not having been compile-tested. It will teach me merging code "spelling" fixes in an RC freeze.

It also shows that we could need some Solaris/illumos user to sign up to the mailing list (or follow the repo) and at least do a compile check when we announce release candidates...

closed 2024-02-14 by tormodvolden

seanm, 2024-02-14:
>I didn't even think about the possibility of it not having been compile-tested.

Me neither. :(  Looking at libusb's CI... it seems it doesn't include some of the OSes that libusb supports... Why is that?

Could use this to add OpenBSD for example: https://github.com/vmactions/openbsd-vm

mcuee, 2024-02-14:
> > I didn't even think about the possibility of it not having been compile-tested.
> 
> Me neither. :( Looking at libusb's CI... it seems it doesn't include some of the OSes that libusb supports... Why is that?
> 
> Could use this to add OpenBSD for example: https://github.com/vmactions/openbsd-vm

Need to test how fast it goes.

HIDAPI project looked at adding BSD CIs and we found that the github actions are too slow. 
* https://github.com/libusb/hidapi/issues/477
* https://github.com/libusb/hidapi/issues/462

In the end, SourceHut is used.
https://builds.sr.ht/~z3ntu/hidapi

mcuee, 2024-02-14:
I will create a new issue with regard to CI improvements in general.

I think HIDAPI project has done a better job in the CI aspect. Maybe @Youw can help here.

Youw, 2024-02-15:
> HIDAPI project looked at adding BSD CIs and we found that the github actions are too slow.

Based on: https://github.com/vmactions/openbsd-vm/actions/runs/7866389477/job/21460597951
Maybe it worth reconsider Github Actions as a BSD CI. Haven't tried it, but if I read it right - it takes 2min30sec to prepare the VM.
I'd say that's more than acceptable overhead.

ptribble, 2024-02-25:
To follow up, I've signed up for the mailing list so should be able to build-test any new releases.

Looking at vmactions, they have Solaris and OmniOS as well

https://github.com/vmactions/omnios-vm

commit 5c64e525 (2024-02-14): sunos: Fix typos breaking build

Fixup of commit 66e63d6

Fixes #1465