libusb/libusb #1469: Document PGP release signature usage

maltfield, 2024-03-10:
### Expected behaviour

When I go to download the latest version of libusb, I should also see instructions on how to verify the authenticity of the file after download and before install. Or, at least, a link to the document that describes this.

### Actual behaviour

I see no mention about cryptographic authenticity verification on the download page on GitHub

 * https://github.com/libusb/libusb/releases/tag/v1.0.27

I see no mention about cryptographic authenticity verification on the download page on SourceForge:

 * https://sourceforge.net/projects/libusb/files/libusb-1.0/libusb-1.0.27/

I see no mention about cryptographic authenticity verification on the "Downloads" section of the libusb wiki:

 * https://github.com/libusb/libusb/wiki#user-content-Downloads

I see no mention about cryptographic authenticity verification on the "FAQ" section of the libusb wiki:

 * https://github.com/libusb/libusb/wiki/FAQ

I see no mention about cryptographic authenticity verification on the website:

 * https://libusb.info/

### Steps to reproduce
1. Go to https://github.com/libusb/libusb/
2. Click `Releases`
3. Download `libusb-1.0.27.tar.bz2`
4. Download `libusb-1.0.27.tar.bz2.asc`
5. Look around for information on where to get the authentic Release Signing PGP Key for the libusb project
6. ???
7. Get confused and open ticket

### Additional Context

I would recommend adding a new page to your wiki that:

1. Writes-out the full fingerprint of the Release Signing Key
2. Link to a URL where the user can download the Release Signing Key
3. Give the `gpg` command for how a user can fetch the Release Signing Key from the keyservers
4. Links to all the other domains that write-out the full fingerprint of the Release Signing Key (so the user can cross-verify out-of-band)

After the above documentation page is complete, add at least one link to it from the following pages:

 * https://github.com/libusb/libusb/releases/tag/v*
 * https://sourceforge.net/projects/libusb/files/libusb-1.0/libusb-*
 * https://github.com/libusb/libusb/wiki#user-content-Downloads
 * https://github.com/libusb/libusb/wiki/FAQ
 * https://libusb.info/

maltfield, 2024-03-10:
I also recommend adding a `KEYS` file to the root of your repo (located along-side files like `COPYING` and `AUTHORS`), per the [KEYS standard established by Apache](https://infra.apache.org/release-signing)
 * https://infra.apache.org/release-signing

For more best-practices, see also:

1. https://infra.apache.org/release-signing
2. https://docs.opendev.org/opendev/system-config/latest/signing.html
3. https://wiki.debian.org/Subkeys
4. https://riseup.net/en/security/message-security/openpgp/best-practices

mcuee, 2024-03-11:
Reference: 
* https://github.com/vpelletier/python-libusb1/issues/96

And then compare libusb-1.0.27 release and python-libusb1 3.1.0 release, I do not see any difference in terms of the source code archive signing. 
https://github.com/libusb/libusb/releases/tag/v1.0.27
https://github.com/vpelletier/python-libusb1/releases/tag/3.1.0

What is missing here?

mcuee, 2024-03-11:
> I also recommend adding a `KEYS` file to the root of your repo (located along-side files like `COPYING` and `AUTHORS`), per the [KEYS standard established by Apache](https://infra.apache.org/release-signing#keys-policy)

I do not see any repos I am involved doing that (eg: libusb, pyusb, avrdude, openocd, libftdi, hidapi, etc).

maltfield, 2024-03-11:
> I do not see any repos I am involved doing that (eg: libusb, pyusb, avrdude, openocd, libftdi, hidapi, etc).

Check Maven. 100% of the packages on Apache Maven are cryptographically signed with PGP.

maltfield, 2024-03-11:
> What is missing here?

In case it wasn't clear: this ticket is asking for documentation. What's missing is documentation.

mcuee, 2024-03-11:
> > I do not see any repos I am involved doing that (eg: libusb, pyusb, avrdude, openocd, libftdi, hidapi, etc).
> 
> Check Maven. 100% of the packages on Apache Maven's central repository are cryptographically signed with PGP.

At least I do not see anything like that in the official github repo.
https://github.com/apache/maven

mcuee, 2024-03-11:
> > What is missing here?
> 
> Sorry if I was unclear: this ticket is asking for documentation. What's missing is documentation.

Could you provide a reference? Thanks.

maltfield, 2024-03-11:
> At least I do not see anything like that in the official github repo.

A good example is the [Apache Ant KEYS file](https://downloads.apache.org/ant/KEYS)

 * https://downloads.apache.org/ant/KEYS

> They do not even have signed source tarball.

Maven signs their releases. You can find the signatures on their website:

 * https://maven.apache.org/download.cgi#files

Note the text above the table, which provides info to users on how to verify signatures by linking to the documentation

> In order to guard against corrupted downloads/installations, it is highly recommended to [verify the signature](https://www.apache.org/info/verification.html) of the release bundles against the public [KEYS](https://downloads.apache.org/maven/KEYS) used by the Apache Maven developers.

The above text links to the following documentation, which is what this ticket is asking-for to be added to the libusb website and/or wiki:

* https://www.apache.org/info/verification.html#CheckingSignatures

> Could you provide a reference? Thanks.

Does the above answer your question? If not, let me know and I can provide examples of such documentation from other projects, if you'd like.

mcuee, 2024-03-11:
> Does the above answer your question? If not, let me know and I can provide examples of such documentation from other projects, if you'd like.

Thanks for the answer.

I will let @tormodvolden decide how to address the issue you raised. He is in charge of the recent releases like 1.0.25/1.026/1.0.27.

tormodvolden, 2024-03-14:
This is a good point. I have been signing with my key that is listed e.g. at https://launchpad.net/~tormodvolden but it is not mentioned anywhere on our pages.

tormodvolden, 2024-03-14:
You have done a great job at documenting the issue, you're welcome to suggest some wording and templates as well.

maltfield, 2024-03-15:
Here are two more examples of docs from open source projects telling their users how they can cryptographically verify their releases:

 * https://calyxos.org/install/verify/#additional-verification
 * https://docs.featherwallet.org/guides/linux#verifying-the-download-optional

mcuee, 2024-03-17:
I think the above seems to be a good practice.

On the other hand, I am also trying to understand why most projects do not even sign the release.

So I look at libusb github download number statistics.

1) libusb-1.0.26 till today 17-Mar-2024
v1.0.26
  Release Info:
  Published on: 2022-04-10
  Release Author: [tormodvolden](https://github.com/tormodvolden)
  Downloads: 1,616,488
  Download Info:
[Do.not.use.the.tar.gz.--.please.download.the.tar.bz.--.txt](https://github.com/libusb/libusb/releases/download/v1.0.26/Do.not.use.the.tar.gz.--.please.download.the.tar.bz.--.txt) (0 MiB)
Last updated on 2022-04-10 — Downloaded 9,282 times
[libusb-1.0.26-binaries.7z](https://github.com/libusb/libusb/releases/download/v1.0.26/libusb-1.0.26-binaries.7z) (2.87 MiB)
Last updated on 2022-04-10 — Downloaded 121,156 times
[libusb-1.0.26-binaries.7z.asc](https://github.com/libusb/libusb/releases/download/v1.0.26/libusb-1.0.26-binaries.7z.asc) (0 MiB)
Last updated on 2022-04-10 — Downloaded 2,321 times
[libusb-1.0.26.7z](https://github.com/libusb/libusb/releases/download/v1.0.26/libusb-1.0.26.7z) (2.4 MiB)
Last updated on 2023-12-10 — Downloaded 1,737 times
[libusb-1.0.26.7z.asc](https://github.com/libusb/libusb/releases/download/v1.0.26/libusb-1.0.26.7z.asc) (0 MiB)
Last updated on 2023-12-10 — Downloaded 49 times
[libusb-1.0.26.tar.bz2](https://github.com/libusb/libusb/releases/download/v1.0.26/libusb-1.0.26.tar.bz2) (0.59 MiB)
Last updated on 2022-04-10 — Downloaded 1,479,712 times
[libusb-1.0.26.tar.bz2.asc](https://github.com/libusb/libusb/releases/download/v1.0.26/libusb-1.0.26.tar.bz2.asc) (0 MiB)
Last updated on 2022-04-10 — Downloaded 2,231 times

2) libusb 1.0.27 till today 17-Mar-2024

[Latest Release: v1.0.27](https://github.com/libusb/libusb/releases/tag/v1.0.27)
  Release Info:
  Published on: 2024-01-31
  Release Author: [tormodvolden](https://github.com/tormodvolden)
  Downloads: 37,379
  Download Info:
[libusb-1.0.27.7z](https://github.com/libusb/libusb/releases/download/v1.0.27/libusb-1.0.27.7z) (3.91 MiB)
Last updated on 2024-01-31 — Downloaded 8,922 times
[libusb-1.0.27.7z.asc](https://github.com/libusb/libusb/releases/download/v1.0.27/libusb-1.0.27.7z.asc) (0 MiB)
Last updated on 2024-01-31 — Downloaded 128 times
[libusb-1.0.27.tar.bz2](https://github.com/libusb/libusb/releases/download/v1.0.27/libusb-1.0.27.tar.bz2) (0.61 MiB)
Last updated on 2024-01-31 — Downloaded 28,162 times
[libusb-1.0.27.tar.bz2.asc](https://github.com/libusb/libusb/releases/download/v1.0.27/libusb-1.0.27.tar.bz2.asc) (0 MiB)
Last updated on 2024-01-31 — Downloaded 167 times

tormodvolden, 2024-03-17:
It is understandable that the signatures are not downloaded often if we don't tell explain what they are and how to use them. I also think the download numbers are skewed by bots downloading whatever (comes first?), and build scripts etc.

maltfield, 2024-03-17:
> I am also trying to understand why most projects do not even sign the release.

Unfortunately, security is an afterthought for most software developers.

If you don't  sign your releases, then your users cannot know if they downloaded the authentic version or a malicious version of your app. https does not protect users from supply-chain attacks such as Publishing Infrastructure Compromise.

Many open source projects have learned this lesson the hard way. Here is a incomplete list of Publishing Infrastructure Compromise that affected open-source projects in the past years:

 * https://github.com/cncf/tag-security/tree/main/supply-chain-security/compromises

maltfield, 2024-03-22:
@tormodvolden for right now (in the meantime) where can users download your PGP public key? And what is the full fingerprint?

The latest libusb release appears to be signed with a key = `AA0639079EFB61B9`, but that doesn't match any of the three keys listed on your launchpad link:

1. C68187379B23DE9EFC46651E2C80FF56C6830A0E
2. 32EAC3AA2932F7F152850C2A0ECA153961EC0EEC
3. C68187379B23DE9EFC46651E2C80FF56C6830A0E

Also there's no link on your launchpad page to download the actual PGP public key.

tormodvolden, 2024-03-22:
Yes, the third key (you listed it as number 1 also but that must have been a copy-pasto) with fingerprint `C681 8737 9B23 DE9E FC46  651E 2C80 FF56 C683 0A0E` is the correct public key. If you download this one the signature on the releases will be verified. But you are right it is not obvious just looking at the key id used for signing, unless you click on the key on the launchpad page so that you can recognize `AA0639079EFB61B9` being a (signing) subkey of the primary one.

The keys listed on the search result page are download links.

Also, if you search for `AA0639079EFB61B9` on https://keys.openpgp.org the result (with click to download) is `C68187379B23DE9EFC46651E2C80FF56C6830A0E`.

But yes, we can make this clearer and have the fingerprints listed on our web pages.

maltfield, 2024-03-22:
Thanks

> unless you click on the key on the launchpad page

Where exactly do you click? The fingerprints are not clickable for me

tormodvolden, 2024-03-22:
Looks like the fingerprints are not clickable unless you are logged in :(

But this is the URL and it seems publicly available: https://keyserver.ubuntu.com/pks/lookup?fingerprint=on&op=index&search=0xC68187379B23DE9EFC46651E2C80FF56C6830A0E

tormodvolden, 2024-05-26:
I added my key (fingerprint) to the wiki https://github.com/libusb/libusb/wiki#user-content-Downloads

@maltfield If we add this KEYS file (see PR), it should workaround the issue of not getting UIDs from the key servers. If the users first import the KEYS file, they can use one or multiple keyservers for confirming it afterwards.

I am not so fan of the direct link I provided above, since it may expose various UIDs to spam harvesters (which I guess is why the key servers hide them).

maltfield, 2024-05-29:
Great, thanks :)

It looks like your PR for the `KEYS` file has a random html `</p>` that should probably be removed. Otherwise it looks good ;)

> I am not so fan of the direct link I provided above, since it may expose various UIDs to spam harvesters (which I guess is why the key servers hide them).

Actually, I think the reason UIDs are stripped from keys is to bypass the issue of certificate flooding and also to deal with GDPR issues.

 * https://tech.michaelaltfield.net/2019/07/14/mitigating-poisoned-pgp-certificates/

Surprising though it may be, I don't think spam harvesters have learned to scrape key servers....

tormodvolden, 2024-05-29:
Thanks for checking! The random `</p>` was copy-pasted from https://infra.apache.org/release-signing#keys-policy :)

I hope you are right about the spam harvesters, but the linked page had the e-mail addresses spelt out in plain text.

tormodvolden, 2024-07-29:
Now that our KEYS file is in git master, I have added a SIGNING_KEYS.txt file (with detached signature for good measure) to the 1.0.25/1.0.26/1.0.27 release download folders both at GitHub and SourceForge. The file explains how to get the public key (e.g. from git) and check it (against multiple key servers) and how to verify the release files with it.

tormodvolden, 2025-06-08:
Yes, I think this well handled and documented now. If @maltfield or anyone else has suggestions on how to improve it, they are welcome.

closed 2025-06-08 by tormodvolden