libusb/libusb #1471: gpg: selecting card failed: No such device with libusb 1.0.27

Mno-hime, 2024-03-13:
I updated Fedora 39's libusb1 package to 1.0.27-1, and suddenly, my YubiKey (Yubico YubiKey CCID 01 00, version 3.4) can't be detected by GPG 2.4.4:

```
$ gpg2 --card-status
gpg: selecting card failed: No such device
gpg: OpenPGP card not available: No such device
```

The kernel can see the key fine:
```
bře 13 11:58:07 thinkpad kernel: usb 1-2: new full-speed USB device number 28 using xhci_hcd
bře 13 11:58:07 thinkpad kernel: usb 1-2: New USB device found, idVendor=1050, idProduct=0404, bcdDevice= 5.43
bře 13 11:58:07 thinkpad kernel: usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
bře 13 11:58:07 thinkpad kernel: usb 1-2: Product: YubiKey CCID
bře 13 11:58:07 thinkpad kernel: usb 1-2: Manufacturer: Yubico
```

Downgrading to libusb 1.0.26-3 makes GPG to detect the key again.

tormodvolden, 2024-03-13:
Can you please run with env LIBUSB_DEBUG=3 ?

Mno-hime, 2024-03-13:
It seems the variable did not propagate.

With libusb 1.0.27:
```
 export LIBUSB_DEBUG=3 
 gpg2 --card-status
gpg: selecting card failed: No such device
gpg: OpenPGP card not available: No such device
```

tormodvolden, 2024-03-13:
It is quite possible that gpg2 sanitizes the environment. Does it spawn another process for the USB operations? Or load libusb explicitly? I don't see libusb in the ldd output of gpg2 2.4.4 here (Arch).

tormodvolden, 2024-03-13:
scdaemon has also --verbose and --debug-level options. But I guess the libusb debug output is needed as well.

Mno-hime, 2024-03-13:
```
$ LIBUSB_DEBUG=3 /usr/libexec/scdaemon --multi-server --debug-level guru -vvvvv
scdaemon[207119]: enabled debug flags: mpi crypto memory cache memstat hashing ipc card cardio reader app
scdaemon[207119]: listening on socket '/run/user/1000/gnupg/S.scdaemon'
scdaemon[207119]: handler for fd -1 started
scdaemon[207119]: DBG: chan_6 -> OK GNU Privacy Guard's Smartcard server ready
OK GNU Privacy Guard's Smartcard server ready
```

```
$ LIBUSB_DEBUG=3 gpg2 --card-status
gpg: selecting card failed: No such device
gpg: OpenPGP card not available: No such device
```

The problem is that `gpg2 --card-status` keeps starting its own `scdaemon` process:
```
207119 /usr/libexec/scdaemon --multi-server --debug-level guru -vvvvv
207137 scdaemon --multi-server
```

tormodvolden, 2024-03-13:
Please see the scdaemon manual page. You can create a config file with options. Presumably the daemon is started by gpg-agent. Maybe gpg-agent will pass on LIBUSB_DEBUG. gpg-agent also has an option to specify the scdaemon, which you can make a wrapper setting LIBUSB_DEBUG.

Maytbe @LudovicRousseau has an idea on debugging this?

mcuee, 2024-03-13:
@Mno-hime 

Please help to try libusb xusb example first to see if there are behaviors difference between 1.0.26 release and 1.0.27 release. Please post the output of the following command for both releases.
```
xusb -d 1050:0404
```

You can download the release tarball here and build libusb using the following command.
```
./configure --enable-tests-build --enable-examples-build
```

libusb source code release tarball download:
https://github.com/libusb/libusb/releases/tag/v1.0.27
https://github.com/libusb/libusb/releases/tag/v1.0.26

Reference: libusb Wiki troubleshooting guide.
https://github.com/libusb/libusb/wiki/Troubleshooting

LudovicRousseau, 2024-05-09:
Upgrading from Ubuntu 22.04 to 24.04 will not upgrade just libusb.
GnuPG also is upgraded from version 2.2.27 to 2.4.4.

I guess your problem is not with libusb itself.

See https://blog.apdu.fr/posts/2024/04/gnupg-and-pcsc-conflicts-episode-2/

tormodvolden, 2024-05-09:
This should be fairly easy to narrow down by installing libusb 1.0.27 on Ubuntu 22.04 (which has gnupg 2.2.27).

tormodvolden, 2024-05-09:
The gnupg upstream issue: https://dev.gnupg.org/T6871

jussih, 2024-05-09:
My problem was indeed caused by the GPG update as suggested by @LudovicRousseau. I was mislead by arriving in this issue from google search and coincidentally having the same situation with libusb versions as the issue reporter.

Gratitude for providing the leads to the root cause even though the issue had nothing to do with libusb. The next person arriving here from google will now also have the link to the solution.

mcuee, 2024-05-09:
In this case, I will close this issue.

closed 2024-05-09 by mcuee

tormodvolden, 2024-05-09:
In the original post it looked very much like upgrading to libusb 1.027 was the problem, since downgrading to 1.0.25 made it work again, but there could be external factors and coincidences at play. If we don't hear back from @Mno-hime there is not much we can do about it.