libusb/libusb #1473: WINDOWS: libusb doesn't enumerate all USB devices

gregoiregentil, 2024-03-16:
I'm compiling libusb (latest as of March 15, 2024) on mingw64 and the examples 'listdevs.exe' doesn't enumerate all USB devices. The utility USBDeview sees all devices. See the following screenshot:

[image]

Any idea what could be the issue? The USB device vid:0x04e8 pid:0x1234 doesn't have any installed driver. Could it be a problem?

mcuee, 2024-03-16:
Please post libusb debug log after setting the environment variable `LIBUSB_DEBUG=4`. Thanks.

Please do not use screenshot but rather copy paste the full output.

gregoiregentil, 2024-03-16:
Thank you. I pasted the image to show the report of USBDeview.

We are making progress. Here is the log report:

> listdevs.exe
> [timestamp] [threadID] facility level [function call] <message>
> --------------------------------------------------------------------------------
> [ 0.001660] [00002e70] libusb: debug [libusb_init_context] created default context
> [ 0.002235] [00002e70] libusb: debug [libusb_init_context] libusb v1.0.27.11887
> [ 0.002747] [00002e70] libusb: debug [usbi_add_event_source] add HANDLE 0x16c events 0
> [ 0.003312] [00002e70] libusb: debug [usbi_io_init] using timer for timeouts
> [ 0.003949] [00002e70] libusb: debug [usbi_add_event_source] add HANDLE 0x170 events 0
> [ 0.004616] [00002e70] libusb: debug [get_windows_version] Windows 10 64-bit
> [ 0.007339] [00002e70] libusb: debug [htab_create] using 1021 entries hash table
> [ 0.017530] [00002e70] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
> [ 0.018504] [00002e70] libusb: info [winusbx_init] libusbK DLL is not available
> [ 0.022576] [00002e70] libusb: info [windows_init] UsbDk backend is not available
> [ 0.023264] [00002e70] libusb: debug [libusb_get_device_list]
> [ 0.024386] [0000229c] libusb: debug [windows_iocp_thread] I/O completion thread started
> [ 0.049727] [00002e70] libusb: debug [winusb_get_device_list] ENUM pass HUB {F18A0E88-C30C-11D0-8815-00A0C906BED8}
> [ 0.051214] [00002e70] libusb: debug [winusb_get_device_list] ENUM processing USB\ROOT_HUB30\4&35A59F1D&0&0
> [ 0.052901] [00002e70] libusb: debug [winusb_get_device_list] allocating new device for session [46]
> [ 0.054338] [00002e70] libusb: debug [winusb_get_device_list] assigning Root Hub 'USB\ROOT_HUB30\4&35A59F1D&0&0' bus number 1
> [ 0.054937] [00002e70] libusb: debug [winusb_get_device_list] ENUM pass DEV {A5DCBF10-6530-11D2-901F-00C04FB951ED}
> [ 0.056425] [00002e70] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_0CF3&PID_E300\5&2F7DFF3E&0&8
> [ 0.057263] [00002e70] libusb: debug [get_api_type] driver(s): BTHUSB
> [ 0.057865] [00002e70] libusb: debug [get_api_type] upper filter driver(s): BtFilter
> [ 0.058728] [00002e70] libusb: debug [get_api_type] lower filter driver(s): BtFilter
> [ 0.059486] [00002e70] libusb: debug [winusb_get_device_list] allocating new device for session [2F]
> [ 0.060396] [00002e70] libusb: debug [winusb_get_device_list] ENUM pass HCD {3ABF6F2D-71C4-462A-8A92-1E6861E6AF27}
> [ 0.063193] [00002e70] libusb: debug [winusb_get_device_list] ENUM processing PCI\VEN_8086&DEV_9D2F&SUBSYS_380517AA&REV_21\3&11583659&0&A0
> [ 0.064224] [00002e70] libusb: debug [winusb_get_device_list] ENUM pass GEN
> [ 0.065033] [00002e70] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_138A&PID_0094\A00280B91484
> [ 0.065326] [00002e70] libusb: info [winusb_get_device_list] The following device has no driver: 'USB\VID_138A&PID_0094\A00280B91484'
> [ 0.065924] [00002e70] libusb: info [winusb_get_device_list] libusb will not be able to access it
> [ 0.066609] [00002e70] libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_138A&PID_0094\A00280B91484'
> [ 0.067188] [00002e70] libusb: debug [winusb_get_device_list] allocating new device for session [4B]
> [ 0.067996] [00002e70] libusb: debug [init_device] found 1 configurations for device 'USB\VID_138A&PID_0094\A00280B91484' but device is not configured (i.e. current config: 0), ignoring it
> [ 0.068477] [00002e70] libusb: warning [winusb_get_device_list] failed to initialize device 'USB\VID_138A&PID_0094\A00280B91484'[ 0.069086] [00002e70] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_04E8&PID_1234\5&2F7DFF3E&0&6
> [ 0.069713] [00002e70] libusb: info [winusb_get_device_list] The following device has no driver: 'USB\VID_04E8&PID_1234\5&2F7DFF3E&0&6'
> [ 0.070260] [00002e70] libusb: info [winusb_get_device_list] libusb will not be able to access it
> [ 0.070987] [00002e70] libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_04E8&PID_1234\5&2F7DFF3E&0&6'
> [ 0.071546] [00002e70] libusb: debug [winusb_get_device_list] allocating new device for session [4C]
> [ 0.072360] [00002e70] libusb: debug [init_device] found 1 configurations for device 'USB\VID_04E8&PID_1234\5&2F7DFF3E&0&6' but device is not configured (i.e. current config: 0), ignoring it
> [ 0.072842] [00002e70] libusb: warning [winusb_get_device_list] failed to initialize device 'USB\VID_04E8&PID_1234\5&2F7DFF3E&0&6'
> [ 0.084864] [00002e70] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_0CF3&PID_E300\5&2F7DFF3E&0&8
> [ 0.086304] [00002e70] libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_0CF3&PID_E300\5&2F7DFF3E&0&8'
> [ 0.087197] [00002e70] libusb: debug [winusb_get_device_list] found existing device for session [2F]
> [ 0.088612] [00002e70] libusb: debug [init_device] found 1 configurations (current config: 1) for device 'USB\VID_0CF3&PID_E300\5&2F7DFF3E&0&8'
> [ 0.089593] [00002e70] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 177 bytes)
> [ 0.090379] [00002e70] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 8): 'USB\VID_0CF3&PID_E300\5&2F7DFF3E&0&8'
> [ 0.091212] [00002e70] libusb: debug [winusb_get_device_list] ENUM processing USB\ROOT_HUB30\4&35A59F1D&0&0
> [ 0.092126] [00002e70] libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\ROOT_HUB30\4&35A59F1D&0&0'
[... cut ...]
There is no driver for this USB device (which is in upgrade mode). But the following Linux code is working fine and I want to do the same from Windows.

```
libusb_device_handle *devHandle = libusb_open_device_with_vid_pid(context, 0x04e8, 0x1234);
libusb_claim_interface(devHandle, 0);
libusb_bulk_transfer(devHandle, 2, mem, size, &transferred, 0);	
libusb_release_interface(devHandle, 0);
libusb_close(devHandle);

```

How could I patch libusb so that the USB device is reported and libusb_open_device_with_vid_pid works on Windows too? Thank you in advance for any advice.

gregoiregentil, 2024-03-16:
So, I removed lines 116 and 1167:
https://github.com/libusb/libusb/blob/master/libusb/os/windows_winusb.c#L1166

listdevs.exe reports correctly the USB device.

But my code above still doesn't work:

```
libusb_device_handle *devHandle = libusb_open_device_with_vid_pid(context, 0x04e8, 0x1234);
libusb_claim_interface(devHandle, 0);
libusb_bulk_transfer(devHandle, 2, mem, size, &transferred, 0);	
libusb_release_interface(devHandle, 0);
libusb_close(devHandle);
```

> [timestamp] [threadID] facility level [function call] <message>
> --------------------------------------------------------------------------------
> [ 0.000734] [0000073c] libusb: debug [libusb_init_context] libusb v1.0.27.11887
> [ 0.032564] [0000073c] libusb: debug [usbi_add_event_source] add HANDLE 0x214 events 0
> [ 0.033938] [0000073c] libusb: debug [usbi_io_init] using timer for timeouts
> [ 0.036018] [0000073c] libusb: debug [usbi_add_event_source] add HANDLE 0x218 events 0
> [ 0.037080] [0000073c] libusb: debug [get_windows_version] Windows 10 64-bit
> [ 0.038092] [0000073c] libusb: debug [htab_create] using 1021 entries hash table
> [ 0.040484] [0000073c] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
> [ 0.040855] [0000073c] libusb: info [winusbx_init] libusbK DLL is not available
> [ 0.042658] [0000073c] libusb: info [windows_init] UsbDk backend is not available
> [ 0.043057] [0000073c] libusb: debug [libusb_init_context] installing new context as implicit default
> [ 0.043797] [0000073c] libusb: debug [libusb_get_device_list]
> [ 0.043839] [000012b4] libusb: debug [windows_iocp_thread] I/O completion thread started
> [ 0.069862] [0000073c] libusb: debug [winusb_get_device_list] ENUM pass HUB {F18A0E88-C30C-11D0-8815-00A0C906BED8}
> [ 0.071989] [0000073c] libusb: debug [winusb_get_device_list] ENUM processing USB\ROOT_HUB30\4&35A59F1D&0&0
> [ 0.072570] [0000073c] libusb: debug [winusb_get_device_list] allocating new device for session [46]
> [ 0.073888] [0000073c] libusb: debug [winusb_get_device_list] assigning Root Hub 'USB\ROOT_HUB30\4&35A59F1D&0&0' bus number 1
> [ 0.074791] [0000073c] libusb: debug [winusb_get_device_list] ENUM pass DEV {A5DCBF10-6530-11D2-901F-00C04FB951ED}
> [ 0.076372] [0000073c] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_0CF3&PID_E300\5&2F7DFF3E&0&8
> [ 0.077425] [0000073c] libusb: debug [get_api_type] driver(s): BTHUSB
> [ 0.078208] [0000073c] libusb: debug [get_api_type] upper filter driver(s): BtFilter
> [ 0.079052] [0000073c] libusb: debug [get_api_type] lower filter driver(s): BtFilter
> [ 0.079833] [0000073c] libusb: debug [winusb_get_device_list] allocating new device for session [2F]
> [ 0.080929] [0000073c] libusb: debug [winusb_get_device_list] ENUM pass HCD {3ABF6F2D-71C4-462A-8A92-1E6861E6AF27}
> [ 0.081717] [0000073c] libusb: debug [winusb_get_device_list] ENUM processing PCI\VEN_8086&DEV_9D2F&SUBSYS_380517AA&REV_21\3&11583659&0&A0
> [ 0.083203] [0000073c] libusb: debug [winusb_get_device_list] ENUM pass GEN
> [ 0.084114] [0000073c] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_138A&PID_0094\A00280B91484
> [ 0.084451] [0000073c] libusb: info [winusb_get_device_list] The following device has no driver: 'USB\VID_138A&PID_0094\A00280B91484'
> [ 0.085157] [0000073c] libusb: info [winusb_get_device_list] libusb will not be able to access it
> [ 0.091172] [0000073c] libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_138A&PID_0094\A00280B91484'
> [ 0.097135] [0000073c] libusb: debug [winusb_get_device_list] allocating new device for session [4B]
> [ 0.098414] [0000073c] libusb: debug [init_device] Gregoire found 1 configurations for device 'USB\VID_138A&PID_0094\A00280B91484' but device is not configured (i.e. current config: 0), ignoring it
> [ 0.099182] [0000073c] libusb: debug [init_device] found 1 configurations (current config: 0) for device 'USB\VID_138A&PID_0094\A00280B91484'
> [ 0.100007] [0000073c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 53 bytes)
> [ 0.100756] [0000073c] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 5): 'USB\VID_138A&PID_0094\A00280B91484'
> [ 0.108400] [0000073c] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_04E8&PID_1234\5&2F7DFF3E&0&6
> [ 0.109933] [0000073c] libusb: info [winusb_get_device_list] The following device has no driver: 'USB\VID_04E8&PID_1234\5&2F7DFF3E&0&6'
> [ 0.110970] [0000073c] libusb: info [winusb_get_device_list] libusb will not be able to access it
> [ 0.112150] [0000073c] libusb: info [get_guid] no DeviceInterfaceGUID registered for 'USB\VID_04E8&PID_1234\5&2F7DFF3E&0&6'
> [ 0.113020] [0000073c] libusb: debug [winusb_get_device_list] allocating new device for session [4C]
> [ 0.114090] [0000073c] libusb: debug [init_device] Gregoire found 1 configurations for device 'USB\VID_04E8&PID_1234\5&2F7DFF3E&0&6' but device is not configured (i.e. current config: 0), ignoring it
> [ 0.114879] [0000073c] libusb: debug [init_device] found 1 configurations (current config: 0) for device 'USB\VID_04E8&PID_1234\5&2F7DFF3E&0&6'
> [ 0.115767] [0000073c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 32 bytes)
> [ 0.116663] [0000073c] libusb: debug [init_device] (bus: 1, addr: 4, depth: 1, port: 6): 'USB\VID_04E8&PID_1234\5&2F7DFF3E&0&6'
> [ 0.117572] [0000073c] libusb: debug [winusb_get_device_list] ENUM processing USB\VID_0CF3&PID_E300\5&2F7DFF3E&0&8
[... cut ...]
> [ 0.365788] [0000073c] libusb: debug [libusb_unref_device] destroy device 1.2
> [ 0.366851] [0000073c] libusb: debug [libusb_unref_device] destroy device 1.0
> Cannot find USB device
> [ 0.369039] [0000073c] libusb: debug [libusb_exit]
> [ 0.379486] [000012b4] libusb: debug [windows_iocp_thread] I/O completion thread exiting
> [ 0.381969] [0000073c] libusb: debug [usbi_remove_event_source] remove HANDLE 0x218
> [ 0.382291] [0000073c] libusb: debug [usbi_remove_event_source] remove HANDLE 0x214
> 

What else would I have to patch to make it work?

tormodvolden, 2024-03-16:
As the warning message says "The following device has no driver: [...] libusb will not be able to access it", libusb cannot access a device on Windows if there is no driver registered for it. The operating system only allows applications to access interfaces set up by a driver. On Linux an application can access devices directly (and with libusb preferably when there is no driver interfering...).

You can use zadig.exe to associate a libusb-compatible driver (e.g. WinUSB) with the device, then it should work.

gregoiregentil, 2024-03-16:
Everything you are saying is correct. After installing WinUSB via zadig.exe or the Computer Management->Device Manager console, the libusb application is working.

I'm closing the bug. Thank you.

Nevertheless, I still have a question: how could I create a script or a command-line app that registers the WinUSB driver for my device without using a UI app (such as zadig.exe or device manager)? I want this to be automatic for my user.

closed 2024-03-16 by gregoiregentil

mcuee, 2024-03-17:
@gregoiregentil 

You can adapt libwdi example `wdi-simple` for your use case. It is basically the command line version of Zadig.
https://github.com/pbatard/libwdi/tree/master/examples