libusb/libusb #1476: Error: Failed to set usb camera brightness!

edu118, 2024-03-26:
I used the libusb library to set the usb camera brightness，failed.
code as follow：

#include <opencv2/opencv.hpp>
#include <libusb-1.0/libusb.h>

using namespace cv;
using namespace std;

#define VENDOR_ID 0x1bcf
#define PRODUCT_ID 0x0b09

int main() {
    // 初始化相机
    cv::VideoCapture cap(0);
    if (!cap.isOpened()) {
        std::cerr << "Error: Failed to open camera!" << std::endl;
        return -1;
    }

    // 创建窗口
    cv::namedWindow("Camera", cv::WINDOW_AUTOSIZE);

    // 控制亮度
    libusb_device_handle *dev_handle;
    libusb_init(NULL);
    dev_handle = libusb_open_device_with_vid_pid(NULL, VENDOR_ID, PRODUCT_ID);
    if (dev_handle == NULL) {
        std::cerr << "Error: Failed to open USB device!" << std::endl;
        return -1;
    }
    // 设置亮度值（这里的值需要根据相机和协议进行调整）
    //uint8_t brightness = 128; // 0-255
    uint8_t brightness = 50;
    int result = libusb_control_transfer(dev_handle, 0x21, 0x0A, 0x00, 0x00, &brightness, 1, 0);
    if (result < 0) {
        std::cerr << "Error: Failed to set brightness!" << std::endl;
        return -1;
    }

    // 捕获和显示图像
    cv::Mat frame;
    while (true) {
        cap >> frame;
        if (frame.empty()) {
            std::cerr << "Error: Failed to capture frame!" << std::endl;
            break;
        }
        cv::imshow("Camera", frame);

        // 按下ESC键退出循环
        if (cv::waitKey(1) == 27) {
            break;
        }
    }

    // 关闭相机和USB设备
    cap.release();
    libusb_close(dev_handle);
    libusb_exit(NULL);

    // 关闭窗口
    cv::destroyAllWindows();

    return 0;
}

Error: Failed to set brightness!

tormodvolden, 2024-04-02:
Please provide a debug log. Does it work if you comment out the openCV code? If the camera is accessed via a kernel driver, it may conflict with the direct access from your program.

tormodvolden, 2024-04-02:
This is a UVC camera: https://linux-hardware.org/?id=usb:1bcf-0b09
So you can try a tool like `qv4l2` to control it.

closed 2024-04-27 by mcuee