libusb/libusb #1478: Possible type error in darwin_usb.c?

cmsj, 2024-04-03:
Howdy!

I was reading dawrin_usb.c to learn about how libusb works on macOS and I noticed that on [line 39](https://github.com/libusb/libusb/blob/master/libusb/os/darwin_usb.c#L639) the `session` variable is declared as a `UInt64`, but on the following line, `get_ioregistry_value_number()` is called and told the type should be `kCFNumberSInt64Type`.

So it seems like it's fetching a signed int64 and then storing it in an unsigned int64?

I wasn't able to find any documentation for the `sessionID` (or `locationID`) IORegistry properties, which makes me wonder if they are private API? Either way, the disparate typing confused me and I figured I'd report it in case it's a bug.

seanm, 2024-04-04:
I fixed that a few months ago... are you looking in master or a release?  It was perhaps merged after the last release.

seanm, 2024-04-04:
Oh wait, I read too fast.  The signed-unsigned discrepancy isn't a problem.

They aren't private API, Apple's docs just suck, especially for anything that's Mac-only.

closed 2024-04-09 by cmsj