libusb/libusb #1485: Architecture mismatch error

tkheard, 2024-04-27:
Using macport, I attempted to install qdmr on my M1 Macbook running Sonoma 14.4.1, resulting in:

Error: Cannot install qdmr for the arch 'arm64' because
Error: its dependency libusb is only installed for the arch 'i386'
Error: and the configured universal_archs 'x86_64 i386' are not sufficient.
Error: Unable to execute port qdmr: architecture mismatch

Youw, 2024-04-27:
This repo doesn't contain macport recipe for libusb, we don't control it.
I believe libusb recipe is located at: https://github.com/macports/macports-ports/tree/master/devel/libusb

closed 2024-04-27 by Youw

tkheard, 2024-04-27:
Thanks. Sorry to have bothered you.

Thomas Heard
***@***.***

seanm, 2024-04-27:
> Error: its dependency libusb is only installed for the arch 'i386'

Somehow your libusb was built only for old 32-bit Intel and needs to be rebuilt for arm64.