libusb/libusb #1487: Asynchronous isochronous data transfer problem with libusb-1.0.27 on Windows

kuzzh, 2024-04-28:
Hi libusb team,

I'm using libusb-1.0.27 to read PCM data from a microphone device that supports the UAC protocol. I've tested it on both Windows 10 and Ubuntu (18.04.6), and the results show that they can both retrieve PCM data properly. However, on Windows, when I playback this PCM data, there is significant noise, and the recorded sound becomes very faint, while everything works fine on Ubuntu. I've tried different drivers on Windows, including WinUSB (v6.1.7600.16385), libusb-win32 (v1.2.7.3), libusbK (v3.1.0.0), but the results are the same. The code used on both platforms is exactly the same.

- My test code

    ```cpp
    const uint16_t VID = 0x1403;
    const uint16_t PID = 0x3592;
    const int IFACE_NUM = 1;
    const int ALTERNATE_SETTING = 1;
    const int EP_ISO_IN = 0x82;
    const int PACKET_SIZE = 294;
    const int NUM_TRANSFERS = 10;
    const int NUM_PACKETS = 10;

    static FILE* fout = NULL;
    static int bFirst = 1;

    static void WINAPI transfer_callback(struct libusb_transfer* xfr)
    {
        if (bFirst)
        {
            bFirst = 0;
            fout = fopen("./output_data.pcm", "w+");
            if (fout == NULL)
            {
                printf("erro to open file[%d %s] \n", __LINE__, __FUNCTION__);
            }
        }

        int  i = 0;
        if (fout)
        {
            for (i = 0; i < xfr->num_iso_packets; i++)
            {
                struct libusb_iso_packet_descriptor* pack = &xfr->iso_packet_desc[i];
                if (pack->status != LIBUSB_TRANSFER_COMPLETED)
                {
                    printf("erro transfer status %d %s [%d %s]\n", pack->status, libusb_error_name(pack->status), __LINE__, __FUNCTION__);
                    break;
                }

                const uint8_t* data = libusb_get_iso_packet_buffer_simple(xfr, i);
                printf("get out data %d [%d %s]\n", pack->actual_length, __LINE__, __FUNCTION__);

                fwrite(data, 1, pack->actual_length, fout);
            }

        }

        if (libusb_submit_transfer(xfr) < 0) {
            printf("error re-submitting !!!!!!!exit ----------[%d %s]\n", __LINE__, __FUNCTION__);
            exit(1);
        }
        else
        {
            printf("re-submint ok !\n");
        }
    }

    int main(int argc, char* argv[]) {

        int ret = libusb_init(NULL);

        libusb_set_debug(NULL, LIBUSB_LOG_LEVEL_DEBUG);

        if (ret < 0) {
            printf("Initialize libusb failed: %s\n", libusb_error_name(ret));
            return -1;
[... 424 lines cut ...]
                ------ String Descriptor 2 ------
    bLength                  : 0x1E (30 bytes)
    bDescriptorType          : 0x03 (String Descriptor)
    Language 0x0409          : "Comica_VM30 TX"
    Data (HexDump)           : 1E 03 43 00 6F 00 6D 00 69 00 63 00 61 00 5F 00   ..C.o.m.i.c.a._.
                            56 00 4D 00 33 00 30 00 20 00 54 00 58 00         V.M.3.0. .T.X.
                ------ String Descriptor 3 ------
    bLength                  : 0x18 (24 bytes)
    bDescriptorType          : 0x03 (String Descriptor)
    Language 0x0409          : "MA315614543"
    Data (HexDump)           : 18 03 4D 00 41 00 33 00 31 00 35 00 36 00 31 00   ..M.A.3.1.5.6.1.
                            34 00 35 00 34 00 33 00                           4.5.4.3.
                ------ String Descriptor 5 ------
    bLength                  : 0x1E (30 bytes)
    bDescriptorType          : 0x03 (String Descriptor)
    Language 0x0409          : "Comica_VM30 TX"
    Data (HexDump)           : 1E 03 43 00 6F 00 6D 00 69 00 63 00 61 00 5F 00   ..C.o.m.i.c.a._.
                            56 00 4D 00 33 00 30 00 20 00 54 00 58 00         V.M.3.0. .T.X.
                ------ String Descriptor 6 ------
    bLength                  : 0x1E (30 bytes)
    bDescriptorType          : 0x03 (String Descriptor)
    Language 0x0409          : "Comica_VM30 TX"
    Data (HexDump)           : 1E 03 43 00 6F 00 6D 00 69 00 63 00 61 00 5F 00   ..C.o.m.i.c.a._.
                            56 00 4D 00 33 00 30 00 20 00 54 00 58 00         V.M.3.0. .T.X.
    ```

mcuee, 2024-04-28:
There are multiple issues of Windows ISOC transfer with libusb. 
https://github.com/libusb/libusb/issues?q=is%3Aopen+is%3Aissue+label%3Aisoc

You can try PR #749 to see if tha helps or not.

If not, you may consider libusbK API instead (Windows only).
https://github.com/mcuee/libusbk

kuzzh, 2024-04-29:
@mcuee Thanks for your reply. I just tested it, this PR didn't fix the issue, it couldn't even receive data properly. I`ll try libusbk later.