libusb/libusb #1491: LIBUSB_TRANSFER_TIMED_OUT for stream transfers.

PKM01, 2024-05-03:
Hi libusb team,

I'm currently utilizing libusb version 1.0.27 for stream transfer with my USB device (SanDisk SSD) on Ubuntu 22.04.4 LTS . Specifically, I'm sending an inquiry command to the UAS endpoints. While I can confirm that the command is successfully transmitted, unfortunately, I'm encountering an issue where I'm unable to receive the data in return. The error received is LIBUSB_TRANSFER_TIMED_OUT. I would greatly appreciate any insights or assistance regarding why the inquiry data isn't being received. Below I have provided the code, transfer logs and device information.

CODE:

#include <stdarg.h>
#include <stdbool.h>
#include <stdio.h>
#include <stdint.h>
#include <stdlib.h>
#include <string.h>
#include <time.h>

#include <libusb-1.0/libusb.h>

static void perr(char const *format, ...)
{
    va_list args;

    va_start (args, format);
    vfprintf(stderr, format, args);
    va_end(args);
}

#define ERR_EXIT(errcode) do { perr("   %s\n", libusb_strerror((enum libusb_error)errcode)); return -1; } while (0)
#define CALL_CHECK(fcall) do { int _r=fcall; if (_r < 0) ERR_EXIT(_r); } while (0)
#define CALL_CHECK_CLOSE(fcall, hdl) do { int _r=fcall; if (_r < 0) { libusb_close(hdl); ERR_EXIT(_r); } } while (0)
#define B(x) (((x)!=0)?1:0)
#define be_to_int32(buf) (((buf)[0]<<24)|((buf)[1]<<16)|((buf)[2]<<8)|(buf)[3])

#define RETRY_MAX                     5
#define INQUIRY_LENGTH                0x24

#define INTERFACE_NUMBER 0 
#define ALTERNATE_SETTING 1 

#define UAS_EP_COMMAND_OUT  0x04 
#define UAS_EP_STATUS_IN    0x83 
#define UAS_EP_DATA_IN      0x81 
#define UAS_EP_DATA_OUT     0x02 

// SCSI Inquiry command
#define SCSI_INQUIRY_OPCODE 0x12
#define SCSI_INQUIRY_ALLOCATION_LENGTH 0x24 

struct command_iu {
    uint8_t iu_id;
    uint8_t rsvd1;
    uint16_t tag;
    uint8_t prio_attr;
    uint8_t rsvd5;
    uint8_t len;
    uint8_t rsvd7;
    uint8_t lun[8]; 
    uint8_t cdb[16];
} __attribute__((__packed__));

static enum test_type {
    USE_GENERIC,
    USE_SCSI,
} test_mode;

static uint16_t VID, PID;

uint8_t inquiry_buffer[64];

static void LIBUSB_CALL inquiry_data_transfer_cb(struct libusb_transfer *transfer) {

    if (transfer->status == LIBUSB_TRANSFER_COMPLETED) {
[... 377 lines cut ...]
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
        MaxStreams             32
        Status pipe (0x02)
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
        Command pipe (0x01)

tormodvolden, 2024-05-07:
> I'm unable to receive the data in return

Are you sure the device is returning anything and not just bailing out?

PKM01, 2024-05-08:
I observed the packet transfer on Teledyne LeCroy. The command is transmitted successfully, followed by the device sending an acknowledgment. To receive data from the device, the host must then send an acknowledgment back to the device. However, in this instance, the host isn't sending the acknowledgment to the device. Additionally, I utilized the synchronous API libusb_bulk_transfer to send the same command, and it functions properly with the device returning the correct inquiry data.

GermanAizek, 2024-07-30:
same problem in IMSProg (https://github.com/bigbigmdm/IMSProg), not detect chip.

```
cb_out: error: LIBUSB_TRANSFER_TIMED_OUT
config_stream: Failed to write 3 bytes
Could not configure stream interface.
Found programmer device: WinChipHead (WCH) - CH341A
Device revision is 3.0.4

cb_out: error: LIBUSB_TRANSFER_TIMED_OUT
config_stream: Failed to write 3 bytes
Could not configure stream interface.
Found programmer device: WinChipHead (WCH) - CH341A
Device revision is 3.0.4

cb_out: error: LIBUSB_TRANSFER_TIMED_OUT
config_stream: Failed to write 3 bytes
Could not configure stream interface.
Found programmer device: WinChipHead (WCH) - CH341A
Device revision is 3.0.4

cb_out: error: LIBUSB_TRANSFER_TIMED_OUT
config_stream: Failed to write 3 bytes
Could not configure stream interface.
Found programmer device: WinChipHead (WCH) - CH341A
Device revision is 3.0.4
```

GermanAizek, 2024-07-30:
this problem is common and may be related to libusb version.
https://tracker.pureos.net/T692

mcuee, 2024-07-30:
Most likely this has nothing to do with libusb, but rather with the device or application.

PKM01, 2024-07-31:
Yes, the issue is nothing to do with the libusb. Actually the tag value is getting converted to little endian due to which there was a tag mismatch with the stream id on the data IN pipe. I have explicitly converted the little endian to big endian and it is working.

closed 2024-07-31 by mcuee