libusb/libusb #1492: libusb_get_bos_descriptor() fails to check for bcdUSB > 0x0200 before querying device for BOS descriptor

njk42, 2024-05-03:
It is my (hopefully correct) understanding that only USB 2.0 LPM and later standards (bcdUSB > 0x0200) require devices to support BOS descriptor query.  However, the libusb function `libusb_get_bos_descriptor()` does not contain logic to first check the bcdUSB value is high enough, and therefore still queries devices with bcdUSB <= 0x0200 for a BOS descriptor.

For older USB 1.0 or USB 2.0 (_not_ LPM) devices, the descriptor type = 0x0f could be undefined and requesting it could cause abnormal device behavior and the library could return LIBUSB_ERROR_TIMEOUT or LIBUSB_ERROR_PIPE when it should be returning LIBUSB_ERROR_NOT_FOUND according to the doxygen comment.

Some background:

https://techcommunity.microsoft.com/t5/microsoft-usb-blog/usb-2-1-2-0-1-1-device-enumeration-changes-in-windows-8/ba-p/270775

https://microchip.my.site.com/s/article/Does-a-USB2-1-Specification-Exist

mcuee, 2024-05-21:
> We should maybe make our use of libusb_get_bos_descriptor() in xusb and testlibusb conditional on bcdUSB.

Good idea.

tormodvolden, 2024-05-26:
It was already tested for in testlibusb. I added the same check in xusb.

The question remains whether we should map pipe error to LIBUSB_ERROR_NOT_FOUND if bcdUSB < 0x0201.