libusb/libusb #1494: stress_mt failures on Linux

tormodvolden, 2024-05-07:
We have previously seen errors on macOS on virtual machines, but here we have something I can reproduce. Testing on a Lenovo L13 with Intel i5-1135G7 (8 cores), Ubuntu 24.04 and its kernel 6.8.1, I get with latest git and with 1.0.27:
```
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
No access to device 1d6b:0003, skipping transfer tests.
No access to device 06cb:00bd, skipping transfer tests.
No access to device 13d3:56bc, skipping transfer tests.
No access to device 192f:0416, skipping transfer tests.
No access to device 04d9:1829, skipping transfer tests.
No access to device 05e3:0608, skipping transfer tests.
No access to device 05e3:0608, skipping transfer tests.
No access to device 046d:0a56, skipping transfer tests.
No access to device 8087:0026, skipping transfer tests.
No access to device 1d6b:0002, skipping transfer tests.
No access to device 1d6b:0003, skipping transfer tests.
No access to device 1d6b:0002, skipping transfer tests.
Thread 2 failed (iteration 1): LIBUSB_ERROR_IO
Thread 3 failed (iteration 1): LIBUSB_ERROR_IO
Thread 5 failed (iteration 1): LIBUSB_ERROR_IO
All threads discovered 13 devices (12 not opened)
All done, 3 errors
```
Or
```
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
No access to device 1d6b:0003, skipping transfer tests.
No access to device 06cb:00bd, skipping transfer tests.
No access to device 13d3:56bc, skipping transfer tests.
No access to device 192f:0416, skipping transfer tests.
No access to device 04d9:1829, skipping transfer tests.
No access to device 05e3:0608, skipping transfer tests.
No access to device 05e3:0608, skipping transfer tests.
No access to device 046d:0a56, skipping transfer tests.
No access to device 8087:0026, skipping transfer tests.
No access to device 1d6b:0002, skipping transfer tests.
No access to device 1d6b:0003, skipping transfer tests.
No access to device 1d6b:0002, skipping transfer tests.
Thread 0 failed (iteration 1): LIBUSB_ERROR_IO
Device count mismatch: Thread 1 discovered 13 devices instead of 0
Thread 2 failed (iteration 1): LIBUSB_ERROR_IO
Thread 7 failed (iteration 1): LIBUSB_ERROR_IO
All done, 4 errors
```

tormodvolden, 2024-05-07:
On the macOS CI it can also get into an infinite loop it seems like: https://github.com/libusb/libusb/actions/runs/8991096355/job/24697895520 (been running 2 hours now).

mcuee, 2024-05-21:
For Windows, we have two more ignored errors, I tend to think `LIBUSB_ERROR_NOT_FOUND` is not a normal error as well. But I need to review my past test results.
```
#if defined(PLATFORM_WINDOWS)
				    || open_err == LIBUSB_ERROR_NOT_SUPPORTED
				    || open_err == LIBUSB_ERROR_NOT_FOUND
#endif
```

mcuee, 2024-05-29:
> On the macOS CI it can also get into an infinite loop it seems like: https://github.com/libusb/libusb/actions/runs/8991096355/job/24697895520 (been running 2 hours now).

Same for this one (running for more than 5 hours)
https://github.com/libusb/libusb/actions/runs/9276043843

tormodvolden, 2024-05-29:
I can reproduce it also with debug logging:
```
[ 2.949947] [0001427f] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 2.949967] [0001427f] libusb: debug [reap_for_handle] urb type=2 status=-71 transferred=0
[ 2.949952] [0001427d] libusb: error [get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/003/003, errno=13
[ 2.949972] [0001427f] libusb: debug [handle_control_completion] handling completion status -71
[ 2.949975] [0001427d] libusb: error [get_usbfs_fd] libusb requires write access to USB device nodes
[ 2.949996] [0001427d] libusb: debug [libusb_open] open 3.3 returns -3
No access to device 13d3:56bc, skipping transfer tests.
[ 2.950006] [0001427d] libusb: debug [libusb_get_device_descriptor]  
[ 2.950010] [0001427d] libusb: debug [libusb_open] open 3.11
[ 2.949978] [0001427f] libusb: debug [handle_control_completion] low-level bus error -71
```
Error -71 is -EPROTO (see also https://www.kernel.org/doc/html/latest/driver-api/usb/error-codes.html)

It turned out this was always related to a "Alcor Micro Corp. AU9540 Smartcard Reader". I think it gets stressed by all the descriptor requests. When I disable the descriptor requests for this device, I don't get any errors any longer.

mcuee, 2025-06-06:
One more instance of hanging on macOS.
https://github.com/libusb/libusb/actions/runs/15444830636/job/43612934989
```
Making check in tests
/Applications/Xcode_15.4.app/Contents/Developer/usr/bin/make  check-TESTS
PASS: stress
(hang here)
```

mcuee, 2025-06-06:
On my libusb repo, I can see one failed macOS github action today.
https://github.com/mcuee/libusb/actions/runs/15489793357/job/43612457676

```
Making check in tests
/Applications/Xcode_15.4.app/Contents/Developer/usr/bin/make  check-TESTS
PASS: stress
FAIL: stress_mt
PASS: set_option
PASS: init_context
PASS: macos
============================================================================
Testsuite summary for libusb-1.0 1.0.29
============================================================================
# TOTAL: 5
# PASS:  4
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0
============================================================================
See tests/test-suite.log for debugging.
Some test(s) failed.  Please report this to libusb-devel@lists.sourceforge.net,
together with the test-suite.log file (gzipped) and your system
make[3]: *** [test-suite.log] Error 1
information.  Thanks.
============================================================================
make[2]: *** [check-TESTS] Error 2
make[1]: *** [check-am] Error 2
make: *** [check-recursive] Error 1
=============================================
   libusb-1.0 1.0.29: tests/test-suite.log
=============================================

# TOTAL: 5
# PASS:  4
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0

System information (uname -a): Darwin 23.6.0 Darwin Kernel Version 23.6.0: Thu Apr 24 20:27:23 PDT [202](https://github.com/mcuee/libusb/actions/runs/15489793357/job/43612457676#step:5:203)5; root:xnu-10063.141.1.705.2~1/RELEASE_ARM64_VMAPPLE

.. contents:: :depth: 2

FAIL: stress_mt
===============

Thread 2 failed (iteration 1): LIBUSB_ERROR_OTHER
Running multithreaded init/exit test...
Starting 8 threads
Running multithreaded init/exit test with enumeration...
Starting 8 threads
All threads discovered 2 devices (0 not opened)
All done, 1 errors
FAIL stress_mt (exit status: 1)

Error: Process completed with exit code 1.
```

seanm, 2025-06-06:
With Thread Sanitizer, stress_mt always fails, reliably.

tormodvolden, 2025-06-08:
Can you get it to fail locally? Can you see where it fails?

The hangs we see on the macOS CI is maybe a dead-lock in the enumeration?

seanm, 2025-06-09:
> Can you get it to fail locally? Can you see where it fails?

Yes & yes, I posted a backtrace in some other Issue or PR months ago.  I tried to debug it, but it was not immediately obvious (often the case with threading issues).

1.5 years ago (!) I did work to [improve thread safety](https://github.com/libusb/libusb/pull/1461#issuecomment-2740763597), but, frankly, if things don't get merged, I'm not motivated to spend any more time on it.