libusb/libusb #1495: Wrong permissions on device access to USB-connected UPS

sarroutbi, 2024-05-10:
Hello. After upgrading to Ubuntu 24.04 (libusb-1.0.27) I am facing issues with libusb. I am getting next errors:

```bash
...
   0.021815	[D2] libusb1: No appropriate HID device found
   0.021828	libusb1: Could not open any HID devices: insufficient permissions on everything
   0.021845	No matching HID UPS found
...
```

The unit was working perfectly on Ubuntu 22.04. Access just broke after update. Is there anyway to debug this issue? I have tried reinstalling libusb, but had no success.

Thanks in advance

tormodvolden, 2024-05-10:
These messages are not from libusb 1.0.27 so you will have to provide more information. Please see the FAQ in the wiki.

tormodvolden, 2024-05-10:
https://github.com/libusb/libusb/wiki/Troubleshooting