libusb/libusb #1503: FAIL: USB get_lock not found

peterwillcn, 2024-05-27:
```
 [2024-05-27 08:05:08] ICA 0 usb transfer err:(-9) LIBUSB_ERROR_PIPE
 [2024-05-27 08:05:08] FAIL: USB get_lock not found (1:2)
 [2024-05-27 08:05:10] ICA 0 SendWork usb write err:(-7) LIBUSB_ERROR_TIMEOUT
 [2024-05-27 08:05:10] ICA 0 attempted reset got err:(0) LIBUSB_SUCCESS
 [2024-05-27 08:05:16] ICA 0 usb transfer err:(-9) LIBUSB_ERROR_PIPE
 [2024-05-27 08:05:16] FAIL: USB get_lock not found (1:2)
 [2024-05-27 08:05:17] ICA 0 SendWork usb write err:(-7) LIBUSB_ERROR_TIMEOUT
 [2024-05-27 08:05:18] ICA 0 attempted reset got err:(0) LIBUSB_SUCCESS
 [2024-05-27 08:05:23] ICA 0 usb transfer err:(-9) LIBUSB_ERROR_PIPE
 [2024-05-27 08:05:23] FAIL: USB get_lock not found (1:2)
 [2024-05-27 08:05:25] ICA 0 SendWork usb write err:(-7) LIBUSB_ERROR_TIMEOUT
 [2024-05-27 08:05:25] ICA 0 attempted reset got err:(0) LIBUSB_SUCCESS
 [2024-05-27 08:05:30] ICA 0 usb transfer err:(-9) LIBUSB_ERROR_PIPE
 [2024-05-27 08:05:30] FAIL: USB get_lock not found (1:2)
 [2024-05-27 08:05:32] ICA 0 SendWork usb write err:(-7) LIBUSB_ERROR_TIMEOUT
 [2024-05-27 08:05:32] ICA 0 attempted reset got err:(0) LIBUSB_SUCCESS
 [2024-05-27 08:05:37] ICA 0 usb transfer err:(-9) LIBUSB_ERROR_PIPE
 [2024-05-27 08:05:37] FAIL: USB get_lock not found (1:2)
 [2024-05-27 08:05:39] ICA 0 SendWork usb write err:(-7) LIBUSB_ERROR_TIMEOUT
 [2024-05-27 08:05:39] ICA 0 attempted reset got err:(0) LIBUSB_SUCCESS
 [2024-05-27 08:05:44] ICA 0 usb transfer err:(-9) LIBUSB_ERROR_PIPE
 [2024-05-27 08:05:44] FAIL: USB get_lock not found (1:2)
 [2024-05-27 08:05:46] ICA 0 SendWork usb write err:(-7) LIBUSB_ERROR_TIMEOUT
 [2024-05-27 08:05:46] ICA 0 attempted reset got err:(0) LIBUSB_SUCCESS
```

mcuee, 2024-05-27:
@peterwillcn 

Please follow the troubleshooting guide and provide more information about the device, OS and libusb version.

tormodvolden, 2024-05-27:
And you are aware that none of these messages come from libusb?

closed 2024-05-29 by peterwillcn