libusb/libusb #1504: lose packet when read data on Android platform

yymjr, 2024-06-01:
I have two uvc camera devices. The preview of one device is invalid and the other is flickering when I use libusb and libuvc code of master brach on Android plathorm.

The sample code can run as Ubuntu and MacOs. I found that the main reason is that preview data is received is incomplete. The received data length is 590172 When running on Ubuntu, but it's changes at any times on Android.

run log(Android)
[log.txt](https://github.com/user-attachments/files/15520642/log.txt)

mcuee, 2024-06-02:
Just wondering why do you need to use libusb? Which version of libusb and libuvc are you using?

The issue may not be related to libusb though. isoc transfer seems to be a generic issue for Android.

Similar issues.
https://github.com/libusb/libusb/issues?q=is%3Aopen+is%3Aissue+label%3Aandroid+label%3Aisoc

yymjr, 2024-06-03:
because I need to use uvc camera. use the latest version of libusb(v1.0.27) and libuvc(v0.0.7). you can't buy this Android device(Android 12) on the market.
If I avoid using isoc transfer,will it work normally? 

It seem to be a generic issue of Android. How can I solve this problem? Can you give me some advice?

mcuee, 2024-06-03:
My suggestion is not to use libusb/libuvc under Android. Sorry but I do not know what will be a good solution since I do not know much about Android.

Youw, 2024-06-03:
> because I need to use uvc camera

> My suggestion is not to use libusb/libuvc under Android

[Camera2 API (NDK Camera)](https://developer.android.com/ndk/reference/group/camera) should be your best choice on Android. There are [samples](https://github.com/android/ndk-samples/tree/main/camera) available.

I've also successfully used v4l2 Linux API to use UVC cameras.

But Camera2/NDKCamera is the only way to avoid SEPolicy issues accessing USB/v4l2 devices directly.

yymjr, 2024-06-03:
> [Camera2 API (NDK Camera)](https://developer.android.com/ndk/reference/group/camera) should be your best choice on Android. There are [...]

Thanks for you replay. Don't worry about permissions, I can modify system source. But it seems that v4l2 is not used in the example you provided.

yymjr, 2024-06-03:
> My suggestion is not to use libusb/libuvc under Android. Sorry but I do not know what will be a good solution since I do not know much about Android.
> 
> I will suggest you to ask in libusb-devel mailing list.

OK,Thanks for your help.

yymjr, 2024-06-04:
I found external camera provide [information](https://source.android.com/docs/core/camera/external-usb-cameras), so close this issues.

closed 2024-06-04 by yymjr

Youw, 2024-06-04:
> I found external camera provide [information](https://source.android.com/docs/core/camera/external-usb-cameras), so close this issues.

Yeap, if you're a vendor of an Android device and need to get/provide access to an external USB/UVC camera - that's the way to go.