libusb/libusb #1517: libusb_set_configuration error:LIBUSB_ERROR_BUSY

Mrf2233, 2024-06-18:
On linux I first got the device descriptor (there are 5 optional configurations), after opening the device I set `libusb_set_auto_detach_kernel_driver` to 1, then I used `libusb_get_configuration` to return a value of 4, then I used `libusb_set_ configuration` returned` LIBUSB_ERROR_BUSY` when I used` libusb_set_configuration` to set the value to 5.

I went to check who was using the usb device

`COMMAND     PID    USER   FD   TYPE  DEVICE SIZE/OFF NODE NAME
gvfsd-gph 16633 forlinx   14u   CHR 189,770      0t0 1764 /dev/bus/usb/007/003`

This is what seems to be the default feature for viewing phone pictures.

I'm looking at the logs. Everything's fine.
I don't know how to switch configurations anymore :(.

`[ 0.000048] [0000427e] libusb: debug [libusb_init_context] libusb v1.0.27.11882
"[ 0.000048] [0000427e] libusb: debug [libusb_init_context] libusb v1.0.27.11882"
[ 0.001020] [0000427e] libusb: debug [usbi_add_event_source] add fd 4 events 1
"[ 0.001020] [0000427e] libusb: debug [usbi_add_event_source] add fd 4 events 1"
[ 0.001053] [0000427e] libusb: debug [usbi_io_init] using timer for timeouts
"[ 0.001053] [0000427e] libusb: debug [usbi_io_init] using timer for timeouts"
[ 0.001070] [0000427e] libusb: debug [usbi_add_event_source] add fd 5 events 1
"[ 0.001070] [0000427e] libusb: debug [usbi_add_event_source] add fd 5 events 1"
[ 0.001096] [0000427e] libusb: debug [get_kernel_version] reported kernel version is 5.10.160
"[ 0.001096] [0000427e] libusb: debug [get_kernel_version] reported kernel version is 5.10.160"
[ 0.001160] [0000427e] libusb: debug [op_init] found usbfs at /dev/bus/usb
"[ 0.001160] [0000427e] libusb: debug [op_init] found usbfs at /dev/bus/usb"
[ 0.001181] [0000427e] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
"[ 0.001181] [0000427e] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes"
[ 0.001206] [0000427e] libusb: debug [op_init] sysfs is available
"[ 0.001206] [0000427e] libusb: debug [op_init] sysfs is available"
[ 0.010419] [0000427e] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
"[ 0.010419] [0000427e] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0"
[ 0.010457] [0000427e] libusb: debug [linux_get_device_address] scan usb1
"[ 0.010457] [0000427e] libusb: debug [linux_get_device_address] scan usb1"
[ 0.010552] [0000427e] libusb: debug [linux_get_device_address] bus=1 dev=1
"[ 0.010552] [0000427e] libusb: debug [linux_get_device_address] bus=1 dev=1"
[ 0.010574] [0000427e] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
"[ 0.010574] [0000427e] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257"
[ 0.010591] [0000427e] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
"[ 0.010591] [0000427e] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)"
[ 0.010862] [0000427e] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
"[ 0.010862] [0000427e] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0"
[ 0.010887] [0000427e] libusb: debug [linux_get_device_address] scan usb3
"[ 0.010887] [0000427e] libusb: debug [linux_get_device_address] scan usb3"
[ 0.010964] [0000427e] libusb: debug [linux_get_device_address] bus=3 dev=1
"[ 0.010964] [0000427e] libusb: debug [linux_get_device_address] bus=3 dev=1"
[ 0.010983] [0000427e] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
"[ 0.010983] [0000427e] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769"
[ 0.011000] [0000427e] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
"[ 0.011000] [0000427e] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)"
[ 0.011242] [0000427e] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
"[ 0.011242] [0000427e] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0"
[ 0.011265] [0000427e] libusb: debug [linux_get_device_address] scan usb2
"[ 0.011265] [0000427e] libusb: debug [linux_get_device_address] scan usb2"
[ 0.011339] [0000427e] libusb: debug [linux_get_device_address] bus=2 dev=1
"[ 0.011339] [0000427e] libusb: debug [linux_get_device_address] bus=2 dev=1"
[ 0.011358] [0000427e] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
"[ 0.011358] [0000427e] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513"
[ 0.011375] [0000427e] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
"[ 0.011375] [0000427e] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)"
[ 0.011612] [0000427e] libusb: debug [linux_get_device_address] getting address for device: usb4 detached: 0
"[ 0.011612] [0000427e] libusb: debug [linux_get_device_address] getting address for device: usb4 detached: 0"
[ 0.011635] [0000427e] libusb: debug [linux_get_device_address] scan usb4
"[ 0.011635] [0000427e] libusb: debug [linux_get_device_address] scan usb4"
[ 0.011710] [0000427e] libusb: debug [linux_get_device_address] bus=4 dev=1
"[ 0.011710] [0000427e] libusb: debug [linux_get_device_address] bus=4 dev=1"
[ 0.011730] [0000427e] libusb: debug [linux_enumerate_device] busnum 4 devaddr 1 session_id 1025
"[ 0.011730] [0000427e] libusb: debug [linux_enumerate_device] busnum 4 devaddr 1 session_id 1025"
[ 0.011747] [0000427e] libusb: debug [linux_enumerate_device] allocating new device for 4/1 (session 1025)
"[ 0.011747] [0000427e] libusb: debug [linux_enumerate_device] allocating new device for 4/1 (session 1025)"
[ 0.012015] [0000427e] libusb: debug [linux_get_device_address] getting address for device: usb7 detached: 0
"[ 0.012015] [0000427e] libusb: debug [linux_get_device_address] getting address for device: usb7 detached: 0"
[... 27 lines cut ...]
"[ 0.012997] [0000427e] libusb: debug [linux_enumerate_device] busnum 8 devaddr 1 session_id 2049"
[ 0.013014] [0000427e] libusb: debug [linux_enumerate_device] allocating new device for 8/1 (session 2049)
"[ 0.013014] [0000427e] libusb: debug [linux_enumerate_device] allocating new device for 8/1 (session 2049)"
[ 0.013283] [0000427e] libusb: debug [linux_get_device_address] getting address for device: usb5 detached: 0
"[ 0.013283] [0000427e] libusb: debug [linux_get_device_address] getting address for device: usb5 detached: 0"
[ 0.013307] [0000427e] libusb: debug [linux_get_device_address] scan usb5
"[ 0.013307] [0000427e] libusb: debug [linux_get_device_address] scan usb5"
[ 0.013386] [0000427e] libusb: debug [linux_get_device_address] bus=5 dev=1
"[ 0.013386] [0000427e] libusb: debug [linux_get_device_address] bus=5 dev=1"
[ 0.013407] [0000427e] libusb: debug [linux_enumerate_device] busnum 5 devaddr 1 session_id 1281
"[ 0.013407] [0000427e] libusb: debug [linux_enumerate_device] busnum 5 devaddr 1 session_id 1281"
[ 0.013424] [0000427e] libusb: debug [linux_enumerate_device] allocating new device for 5/1 (session 1281)
"[ 0.013424] [0000427e] libusb: debug [linux_enumerate_device] allocating new device for 5/1 (session 1281)"
[ 0.013691] [0000427e] libusb: debug [linux_get_device_address] getting address for device: usb6 detached: 0
"[ 0.013691] [0000427e] libusb: debug [linux_get_device_address] getting address for device: usb6 detached: 0"
[ 0.013715] [0000427e] libusb: debug [linux_get_device_address] scan usb6
"[ 0.013715] [0000427e] libusb: debug [linux_get_device_address] scan usb6"
[ 0.013794] [0000427e] libusb: debug [linux_get_device_address] bus=6 dev=1
"[ 0.013794] [0000427e] libusb: debug [linux_get_device_address] bus=6 dev=1"
[ 0.013816] [0000427e] libusb: debug [linux_enumerate_device] busnum 6 devaddr 1 session_id 1537
"[ 0.013816] [0000427e] libusb: debug [linux_enumerate_device] busnum 6 devaddr 1 session_id 1537"
[ 0.013833] [0000427e] libusb: debug [linux_enumerate_device] allocating new device for 6/1 (session 1537)
"[ 0.013833] [0000427e] libusb: debug [linux_enumerate_device] allocating new device for 6/1 (session 1537)"
[ 0.014134] [0000427e] libusb: debug [libusb_init_context] installing new context as implicit default
"[ 0.014134] [0000427e] libusb: debug [libusb_init_context] installing new context as implicit default"`

tormodvolden, 2024-06-18:
Your log has every line repeated in quotes, it makes it harder to read than necessary.

If gvfsd-gphoto2 is using the USB device, and has the device node open, maybe that's why you can't set a different configuration? I think it would make sense that another program cannot set a different configuration and change everything under your feet if you already have the device open.

Mrf2233, 2024-06-20:
> Your log has every line repeated in quotes, it makes it harder to read than necessary.
>
> If gvfsd-gphoto2 is using the USB device, and has the device node open, maybe that's why you can't set a [...]

I actually killed the process manually and it's still in a BUSY state.