libusb/libusb #1522: [winusbx_submit_control_transfer] ControlTransfer failed: [87] The parameter is incorrect.

raybbian, 2024-06-24:
Hello, I am trying to use libusb and libusb-win32 (1.4.0.0) to create a network tether between a windows and iDevice. In order to enable this, the iDevice USB configuration needs to be set to configuration 5 (1-indexed). However, I was not able to properly set any configuration on this device other than configuration 0 and configuration 1, the default.

These are the related log entries when trying to set the configuration.
```
[ 0.141778] [0000236c] libusb: debug [libusb_set_configuration] configuration 4
[ 0.141805] [0000236c] libusb: debug [libusb_submit_transfer] transfer 000001DCF14DB5A0
[ 0.141831] [0000236c] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 0.141863] [0000236c] libusb: debug [composite_submit_control_transfer] using interface 0
[ 0.141889] [0000236c] libusb: debug [libusb_claim_interface] interface 0
[ 0.141929] [0000236c] libusb: debug [winusbx_claim_interface] claimed interface 0
[ 0.141954] [0000236c] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 02 to interface 0
[ 0.141982] [0000236c] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 81 to interface 0
[ 0.142007] [0000236c] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 83 to interface 0
[ 0.142033] [0000236c] libusb: debug [auto_claim] auto-claimed interface 0 for control request
[ 0.142059] [0000236c] libusb: debug [winusbx_submit_control_transfer] will use interface 0
[ 0.142037] [000044a8] libusb: debug [windows_iocp_thread] ignoring overlapped 00000001000FCCA0 for handle 000001DCF2F76CF0
[ 0.142122] [000044a8] libusb: debug [windows_iocp_thread] ignoring overlapped 00000001000FCCA0 for handle 000001DCF2F76CF0
[ 0.142155] [000044a8] libusb: debug [windows_iocp_thread] ignoring overlapped 00000001000FCCA0 for handle 000001DCF2F76CF0
[ 0.142187] [000044a8] libusb: debug [windows_iocp_thread] ignoring overlapped 00000001000FCCA0 for handle 000001DCF2F76CF0
[ 0.142316] [0000236c] libusb: warning [winusbx_submit_control_transfer] ControlTransfer failed: [87] The parameter is incorrect.
[ 0.142352] [0000236c] libusb: debug [libusb_release_interface] interface 0
[ 0.142380] [0000236c] libusb: debug [auto_release] auto-released interface 0
[ 0.142404] [0000236c] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.142430] [0000236c] libusb: debug [libusb_free_transfer] transfer 000001DCF14DB5A0
```

Unfortunately, neither WinUSB nor libusbK seem to have the functionality for me to set an alternate configuration, so I am unable to test if it is an issue with the driver itself.

raybbian, 2024-06-25:
Additional Information:
libusb0.dll is version 1.4.0.0
libusbk.dll is version 3.1.0.0
libusb0.sys is vesrion 1.4.0.0

mcuee, 2024-06-25:
You may want to use libusb-win32 API itself to see if that works or not. If it works, then the issue is with libusb-1.0 or libusbK.dll.

Then you try libusbK API, if it works, then the issue is with libusb-1.0.dll (libusb project).

raybbian, 2024-06-26:
I was able to successfully use the libusb-win32 API to set the device's configuration. It may have been a driver mixup issue - after installing the libusb-win32 driver for the entire composite device I was able set the configuration.

I also noticed that if I only installed the libusb-win32 driver for one of the device "interfaces" (configuration? as they show up in device manager), I was able to access the device descriptor and send control messages to the device. I was even able to set the device to a different configuration. However, all of the device configurations were duplicates of each other when accessed, only corresponding to the device I installed the driver for. Is this intended behavior?

Regarding my initial task, sending acontrol message to the iPad only having installed the driver on one of its configurations has allowed me to expose two pairs of NCM Data and NCM Control devices to show up in device manager (although Windows for some reason does not recognize the NCM Data interfaces properly, so no connection). I'm going to close as the issue has been resolved, but would you happen to know if it was possible to use libusb itself to communicate with these CDC NCM devices directly? Installing the driver on only one "interface/configuration" seems quite hacky.

closed 2024-06-26 by raybbian