libusb/libusb #1528: macOS: usb device interface is undefined.

emAsif, 2024-07-02:
After installing usb library in project, when I try to access the usb device it shows all the information except for device interface which is undefined. I think the device is bound to kernal and since its an HID device I cannot deattach it using root privilege but still the device interface being undefined is strange. I'm new to mac kernal interaction and did some online research and found that this is well known issue with mac usb connection. I found a resource which talks about IOKit extension [https://developer.apple.com/library/archive/documentation/DeviceDrivers/Conceptual/USBBook/USBDeviceInterfaces/USBDevInterfaces.html#//apple_ref/doc/uid/TP40002645-TPXREF101](https://github.com/node-usb/node-usb/issues/url) and there are some other options like sudo kextunload which is not long term solution and also didn't work for my mac.

in logs i found this error
`Microsoft Teams @(null): AppleUSBHostUserClient::openGated: failed to open GPS Plus@00120000: provider is already opened for exclusive access by pid 14556, Google Chrome.`
and sometime error is 
`Google Chrome@(null): AppleUSBHostUserClient::openGated: failed to open GPS Plus@00120000: provider is already opened for exclusive access by pid 5276, node`
but i think this is not realted to inteface being undefined issue.

Additional Information
device type: usb-2.0-tracking-device
Operating System: macOS Big Sonoma
Runtime (node -v: 18.2.2 , electron -v: 31.1.0,):
node-usb: (2.13.0):

mcuee, 2024-07-02:
Please take note that node-usb is not under libusb. Please raise the issue there.

mcuee, 2024-07-02:
Reference: issue already raised under node-usb.
* https://github.com/node-usb/node-usb/issues/786

mcuee, 2024-07-02:
For USB HID project, the recommendation is to use HIDAPI instead.

Reference:
https://github.com/libusb/libusb/wiki/FAQ#user-content-Does_libusb_support_USB_HID_devices

mcuee, 2024-07-03:
Sorry I can not help you much here. But I will convert this to a discussion and hopefully others can help you better.

You can also try to post the issue to hidapi project -- but please do not use a wrapper, use pure C when you want to report an issue.