libusb/libusb #1535: Multiple libusb_*_timeout() API function signatures use non-const 'struct timeval *'

dlmiles, 2024-07-13:
Either these should be `const struct timeval *` (indicating to the caller libusb does not modify the data).

OR

The documentation should describe the nature of the modification or clarify that in the process of invoking the API call it needs to use the storage for working data.

seanm, 2024-07-20:
@dlmiles did you try making them const?  Does it still compile?  i.e. is libusb actually modifying those `struct timeval`s?

These could be made conditionally const with `LIBUSB_API_VERSION` checks.

Something like:

```
#if LIBUSB_API_VERSION >= xxx
  #define LIBUSB_FUTURE_CONST const
#else
  #define LIBUSB_FUTURE_CONST
#endif
```

then changing the APIs to something like:

```
int LIBUSB_CALL libusb_wait_for_event(libusb_context *ctx, LIBUSB_FUTURE_CONST struct timeval *tv);
```

dlmiles, 2024-07-20:
> @dlmiles did you try making them const? Does it still compile? i.e. is libusb actually modifying those `struct timeval`s?

I was going to take a more details look under to bonnet to better understand if this is possible (for all platforms and scenarios) or if it depends.  It all depends of the values of the respective APIs are copied or used as-is.  Some linux selectable APIs will provide back the remaining time from the original blocking timeout, while other APIs don't.

Also a look at Windows. to implement the _WIN32 parts for https://github.com/libusb/libusb/discussions/1537