libusb/libusb #1536: Transfer overflow on linux

Mrf2233, 2024-07-16:
I connect an iPhone5 to a linux device.
 After finding the device and listing its configuration, I use `libusb_set_configuration` to select the desired configuration then use `libusb_claim_interface` and `libusb_set_interface_alt_setting` to set the interface and altsetting.
After I found the endpoint using  `libusb_control_transfer` to try to send a control message to the endpoint, it also worked.
The error occurs when receiving a packet of length 4 using `libusb_bulk_transfer`.(Could it be that no data is being received?)
 I'm sorry to bother you as I don't really understand the Log output.

This is the interface information I chose, endpoint "0x86" is what I need to use.
```c++
Interface index: 2
	Number of alternate settings:  1
		Altsetting index: 0
		Interface Class:  "0xff"
		InterfaceSub Class:  "0x2a"
		Interface Number:  2
		Alternate settings:  0
		Number of endpoints:  2
			Endpoint index: 0
			   address:  "0x86"
			   bDescriptorType 5
			   type: 2
			   wMaxPacketSize: 512
			Endpoint index: 1
			   address:  "0x05"
			   bDescriptorType 5
			   type: 2
			   wMaxPacketSize: 512
``` 

This is the code to use `libusb_control_transfer`：
```c++
bool CleanFeature() {
  if (m_readEndpoint > 0) {
    int err = libusb_control_transfer(m_handle, 0x02, 0x01, 0, m_readEndpoint,
                                      nullptr, 0, 1000);//0x86
    if (err != LIBUSB_SUCCESS) {
      qCritical("Failed sending control transfer  for clear feature:%s",
                libusb_error_name(err));
      return false;
    }
  }
  if (m_writeEndpoint > 0) {
    int err = libusb_control_transfer(m_handle, 0x02, 0x01, 0, m_writeEndpoint,
                                      nullptr, 0, 1000);//0x05
    if (err != LIBUSB_SUCCESS) {
      qCritical("Failed sending control transfer  for clear feature:%s",
                libusb_error_name(err));
      return false;
    }
  }
  return true;
}
``` 

This is the log generated after using `libusb_control_transfer`:
```c++
[ 0.010014] [0000381f] libusb: debug [libusb_submit_transfer] transfer 0x5556c29598
[ 0.010017] [0000381f] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 0.010035] [0000381f] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.010037] [0000381f] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.010040] [0000381f] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[ 0.010127] [0000381f] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.010131] [0000381f] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=0
[ 0.010133] [0000381f] libusb: debug [handle_control_completion] handling completion status 0
[ 0.010134] [0000381f] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.010137] [0000381f] libusb: debug [usbi_handle_transfer_completion] transfer 0x5556c29598 has callback 0x7ff559d690
[ 0.010139] [0000381f] libusb: debug [sync_transfer_cb] actual_length=0
[ 0.010145] [0000381f] libusb: debug [libusb_free_transfer] transfer 0x5556c29598
[ 0.010147] [0000381f] libusb: debug [libusb_submit_transfer] transfer 0x5556c29598
[ 0.010148] [0000381f] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 0.010154] [0000381f] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.010155] [0000381f] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[ 0.010240] [0000381f] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.010242] [0000381f] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=0
[ 0.010244] [0000381f] libusb: debug [handle_control_completion] handling completion status 0
[ 0.010245] [0000381f] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.010247] [0000381f] libusb: debug [usbi_handle_transfer_completion] transfer 0x5556c29598 has callback 0x7ff559d690
[ 0.010248] [0000381f] libusb: debug [sync_transfer_cb] actual_length=0
[ 0.010250] [0000381f] libusb: debug [libusb_free_transfer] transfer 0x5556c29598
``` 

This is the log generated after using `libusb_bulk_transfer`:
```c++
[ 0.656466] [000037a9] libusb: debug [libusb_submit_transfer] transfer 0x7f88004a78
[ 0.656474] [000037a9] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 0.656479] [000037a9] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 4
[ 0.656492] [000037a9] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.656494] [000037a9] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[ 0.656571] [000037a9] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 0.656575] [000037a9] libusb: debug [reap_for_handle] urb type=3 status=-75 transferred=0
[ 0.656576] [000037a9] libusb: debug [handle_bulk_completion] handling completion status -75 of bulk urb 1/1
[ 0.656578] [000037a9] libusb: debug [handle_bulk_completion] overflow, actual_length=0
[ 0.656580] [000037a9] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.656582] [000037a9] libusb: debug [usbi_handle_transfer_completion] transfer 0x7f88004a78 has callback 0x7ff559d690
[ 0.656583] [000037a9] libusb: debug [sync_transfer_cb] actual_length=0
[ 0.656586] [000037a9] libusb: debug [libusb_free_transfer] transfer 0x7f88004a78
``` 

If more codes or info is needed let me know and I'll add it right away!

tormodvolden, 2024-07-22:
> I connect an iPhone5 to a linux device.

I suppose, in USB terms, you have a iPhone5 device and a Linux host! Are you sure the device allows bulk transfers with size 4? Maybe it always returns a larger packet so that the buffer allocated with your bulk transfer request is too small? You haven't shown your bulk transfer code. Is there anything in the dmesg output? Which USB drivers are involved? Does it make a difference if you use a USB 2.0 hub in between (if this is USB 3 (?) devices/hosts)?

Mrf2233, 2024-08-14:
My problem, ios devices need to be paired before they can transfer data.

closed 2024-08-14 by Mrf2233