libusb/libusb #1546: Bulk Async received error data when start

chenzhaoting, 2024-08-03:
[image]
[image]
when start transfer somedata,there is a error data in callback,but after that all receive was  corrrect.

mcuee, 2024-08-03:
Please read the guide below and provide more information about the OS, libusb version used and info about your device.
https://github.com/libusb/libusb/wiki/Troubleshooting

chenzhaoting, 2024-08-03:
> Please read the guide below and provide more information about the OS, libusb version used and info about your device. https://github.com/libusb/libusb/wiki/Troubleshooting
> 
> I also think Bus Hound is not a good tool. You may need a real HW USB analyzer to handle low level issues.

> Please read the guide below and provide more information about the OS, libusb version used and info about your device. https://github.com/libusb/libusb/wiki/Troubleshooting
> 
> I also think Bus Hound is not a good tool. You may need a real HW USB analyzer to handle low level issues.

use The Latest release Vs2022 libusb-1.0.27.7z MS64, and the event is processed  by a  thread.
#include "winusbdriver.h"
#include "QDebug"
#include "usbthread.h"

#define     DBG(X)  qDebug()<<(X)<<endl

void WinUSBDriver::close_usb1()
{
    release_usb();
}

WinUSBDriver::WinUSBDriver()
{
 
}

#define USB_VID         0x07E4          //USB的产商ID
#define USB_PID         0x07E4          //USB的产品ID

#define EP0ADDR         0x01            //Write端口0地址，通道0
#define EP1ADDR         0x81            //Read 端口1地址，通道1
#define EP2ADDR         0x02            //Write端口2地址，通道2
#define EP3ADDR         0x86            //Read 端口3地址，通道3

#define USB_TIMEOUT     200            //传输数据的时间延迟

libusb_context* m_libusb_context;

typedef union convert
{
    unsigned char buf[4];
    int32_t data;
}union_cov;

uint8_t *ptr;

void LIBUSB_CALL WinUSBDriver::completeCallback(libusb_transfer *xfer)
{
    WinUSBDriver *driver = (WinUSBDriver *)xfer->user_data;
    
    switch(xfer->status)
    {
        case LIBUSB_TRANSFER_COMPLETED:
        //DBG("recv %d",xfer->actual_length);
         //qDebug()<<"recv_len:"<<xfer->actual_length<<endl;

            if(xfer->actual_length > 0){
                
                driver->recv_len = xfer->actual_length;
                memcpy(driver->recv_buf_tmp,xfer->buffer,xfer->actual_length);

                for (int64_t i = 0; i < xfer->actual_length; i += 4)
                {
                    union_cov test1;
                    test1.buf[0] = xfer->buffer[i];
                    test1.buf[1] = xfer->buffer[i + 1];
                    test1.buf[2] = xfer->buffer[i + 2];
                    test1.buf[3] = xfer->buffer[i + 3];

                    if (test1.data == 0)
                    {
[... 260 lines cut ...]
    while(true)
    {
        int ret = 0;
       
        if (is_stop == false)
        {
          //  qDebug() << "libusb thread "<< is_stop;
            ret = libusb_handle_events(NULL);
           // qDebug() << "libusb thread out ";
        }
       
  
        
        msleep(10);
       // QMutexLocker locker(&m_mutex);
        if (is_stop || ret != 0)
        {
           // qDebug() << "begin usb thread exit";
            break;
        }
    }
    qDebug()<<"usb thread exit";
    emit close_usb();

}

chenzhaoting, 2024-08-03:
[image]

tormodvolden, 2024-08-20:
Please see https://libusb.sourceforge.io/api-1.0/group__libusb__asyncio.html first, then
https://libusb.sourceforge.io/api-1.0/group__libusb__poll.html#ga4989086e3f0327f3886a4c474ec7c327
(I think we need to update the first text to not use libusb_handle_events.)

Your threadFunction() should use libusb_handle_events_completed().

Since libusb_handle_events_completed() uses blocking mode there is no reason to sleep in the loop. You should process events as fast as possible and more often than every 10 seconds.

To keep things simple, please change the `#if` in  your example to use threadFunction() directly instead of your handling thread.

closed 2024-09-01 by mcuee