libusb/libusb #1547: The receive timestamp interval is uneven on windows

HM281099, 2024-08-07:
1. When calling 16 asynchronous bulk transfer interfaces to receive data
2.  Invoke libusb_submit_transfer in the libusb_fill_bulk_transfer callback
3.  only the received timestamp is printed in the callback 
4.  When the device sends data to the PC at 90FPS, the time stamp interval for capturing packets through wireshark is normal
5.  The frame interval at the PC end is not maintained within 11ms. In the figure below, the vertical axis is the frame interval and the horizontal axis is the frame number
[image]

GetQueuedCompletionStatus interface through the analysis of printing efficiency is not high, cause data could not be immediately reported to the application layer， I don't know if you have the same problem, thank you very much #

mcuee, 2024-08-07:
Just wondering what you are trying to achieve here. Thanks.

HM281099, 2024-08-07:
thanks for you reply

closed 2024-08-07 by HM281099