libusb/libusb #1548: nut_libusb_open get `some values` failed, retrying...

masterwishx, 2024-08-07:
Not sure if related to libusb , but maybe any help from here would help to recognize some values for libusb NUT for Eaton 9E UPS
https://github.com/networkupstools/nut/pull/2571
https://github.com/networkupstools/nut/issues/1925

mcuee, 2024-08-07:
Please help to provide more info -- OS, libusb version used, more info about the device, etc.
https://github.com/libusb/libusb/wiki/Troubleshooting

masterwishx, 2024-08-07:
> Please help to provide more info -- OS, libusb version used, more info about the device, etc. https://github.com/libusb/libusb/wiki/Troubleshooting

Oh I'm sorry, this is:
Unraid x64 v6.12.11 (Slackware 15)
libusb-1.0.26(api: 0x1000108)
Driver: usbhid-ups
NUT plugin for Eaton 9E2000i ups

masterwishx, 2024-08-07:
**Again, im not sure that is issue with libusb , but if you can confirm or give a hint ?** 

So using NUT plugin for Unraid by @desertwitch   `nut-2.8.2-x86_64-1stable.ssl11`

also in https://github.com/networkupstools/nut/pull/2562
the `networkupstools` developer @jimklimov  tryed to help me recognize usb device Eaton 9E 2000i UPS , that now is: `unknown 2000` for device.model and ups.model. But lsusb shows the info : 

```
Bus 003 Device 002: ID 0463:ffff MGE UPS Systems UPS
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0463 MGE UPS Systems
  idProduct          0xffff UPS
  bcdDevice            2.02
  iManufacturer           1 EATON
  iProduct                2 Eaton 9E
  iSerial                 4 GExxxxxx  #hidden by me
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               20mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode           33 US
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength    1014
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              20
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

but :  

nut_libusb_open get iManufacturer failed, retrying...
nut_libusb_open get iProduct failed, retrying...
nut_libusb_open get iSerialNumber failed, retrying...

https://github.com/user-attachments/files/16457007/nut.log `usbhid-ups` log

https://github.com/networkupstools/nut/pull/2571#issuecomment-2263182806 more info from developer

tormodvolden, 2024-08-20:
Please see https://github.com/libusb/libusb/wiki/FAQ#user-content-Does_libusb_support_USB_HID_devices

jimklimov, 2024-08-20:
Thanks for the link. Currently NUT does not use HIDAPI, and I don't think anyone offered to contribute code to that extent, although that would be welcome :)

NUT is a libusb user since 2003 (maybe before) though, with shims to use either libusb-0.1 or 1.0 in mainline for a couple of years now, and not only HID devices in the loop.

If the API concepts are similar, same shims could be expanded to cover HIDAPI, I suppose, but I am probably not the one to figure out and do it well :( Despite the years of maintainership of largely other people's code, I am sadly vague about even the terminology, so please bear with me.

Back to the original topic: from what I've gathered from the logs and discussions about this issue in NUT tracker, detailed e.g. here: https://github.com/networkupstools/nut/pull/2571#issuecomment-2263182806 our driver gets the report descriptor of about 2KB in size, and walks it to interpret particular reports. While there are no issues with most of the data points, queries for these 3 identification strings return an "Input/Output Error" code.
* The "unknown 2000" placeholder of a name is constructed by the driver from a NULL model and known wattage of the UPS.
* The device has MGE/Eaton identification (vendorid 0x0463) which could have been subject to "quirks" historically. I think we did rule this out for the case in question: the kernel parameter `usbhid.quirks=0x0463:0xffff:0x08` recommended elsewhere did not help.

The puzzle here is that the same approach works for most of the other devices quite well (not 100%, I occasionally do see similar reports - usually in conjunction with inadequate `udev` setup or virtualization pass-through getting in the way). Tools like `lsusb`, `udevadm info` and allegedly `apcupsd` do not have any problem reporting those strings; at least the latter is AFAIK a pure libusb consumer too.

So I guess the question for this tracker is whether we have stumbled on some well-known gardening equipment here, "do X or don't do Y, and the ailment is gone" sort of stuff?

jimklimov, 2024-08-20:
FWIW, HIDAPI was mentioned in the NUT issue tracker... once I think - right in the thread about getting libusb-1.0 into the mainline codebase: https://github.com/networkupstools/nut/issues/300#issuecomment-254202247

tormodvolden, 2024-08-20:
> queries for these 3 identification strings return an "Input/Output Error" code.

Are your queries control transfers for requesting the device strings? I suppose you get the same error when running examples/xusb on the device?

For getting these strings from the OS instead, there is work going on in PR #1532. I think lsusb and udevadm get these string values from the udev database without sending USB requests.

jimklimov, 2024-08-20:
Thanks for the quick reply! I'll try to track down the answer to how the info is queried, though likely tomorrow (now commuting home).

I would defer to @masterwishx about building and running the example program against the actual device.

masterwishx, 2024-08-20:
> Thanks for the quick reply! I'll try to track down the answer to how the info is queried, though likely tomorrow (now commuting home).
> 
> I would defer to @masterwishx about building and running the example program against the actual device.

I'm here to test when needed just need @desertwitch to compile or provide me config for compile in case.

BTW now using the preview build for nut plugin from master where you added pr for Eaton 9E.

tormodvolden, 2024-08-20:
To build xusb and other examples:
```
./configure --enable-tests-build --enable-examples-build     # Or if from a git checkout: ./autogen.sh
make
./examples/xusb -d 0x0463:0xffff    # or whatever the VID:PID is   # -d enables debug output
```

masterwishx, 2024-08-20:
> To build xusb and other examples:
> 
> ```
> ./configure --enable-tests-build --enable-examples-build     # Or if from a git checkout: ./autogen.sh
> make
> ./examples/xusb -d 0x0463:0xffff    # or whatever the VID:PID is   # -d enables debug output
> ```

im using Unraid (Slackware 15) and NUT plugin for Unraid maintained by @desertwitch , so i have build.slack from his package but also need config file that he is using , but he was kindly compiled for me before for pr's  by @jimklimov

masterwishx, 2024-08-20:
> To build xusb and other examples:
> 
> ```
> ./configure --enable-tests-build --enable-examples-build     # Or if from a git checkout: ./autogen.sh
> make
> ./examples/xusb -d 0x0463:0xffff    # or whatever the VID:PID is   # -d enables debug output
> ```

Oh do you mean for xusb, but how it will work with nut?

tormodvolden, 2024-08-20:
You can run it independently of nut, to see if you can read out the device descriptor strings, or if you get some error.

masterwishx, 2024-08-21:
> You can run it independently of nut, to see if you can read out the device descriptor strings, or if you get some error.

Thanks i will try

masterwishx, 2024-08-21:
> To build xusb and other examples:
> 
> ```
> ./configure --enable-tests-build --enable-examples-build     # Or if from a git checkout: ./autogen.sh
> make
> ./examples/xusb -d 0x0463:0xffff    # or whatever the VID:PID is   # -d enables debug output
> ```

@jimklimov do you think it can help?
Also I should stop nut plugin befor?

tormodvolden, 2024-08-21:
You should run this independently of nut, so yes, stop nut while doing these tests.

tormodvolden, 2024-08-21:
@jimklimov 
- I looked at above linked nut.log, but there is no libusb debug output in it. You can for instance at an appropriate debug level also enable libusb debug output, that would help in debugging these cases. For now can you please explain masterwish how to set env var LIBUSB_DEBUG=4 for the process? I don't know how the process is started.
- the log says libusb 1.0.26 is used. Slackware 15 is at 1.0.24 according to repology. Are you bundling another libusb? It would be good if this can be tested with 1.0.27 as well.
- are libusb hotplug callbacks in use?
- "same approach works for most of the other devices quite well" - does this correlate with devices using HID and not?

masterwishx, 2024-08-21:
> @jimklimov
>
> * I looked at above linked nut.log, but there is no libusb debug output in it. You can for instance at an appropriate debug level also enable libusb debug output, that would help in [...]

Unraid v6.12.11 that I'm using has libusb v1.0.26, I think I tryed to use v1.0.27 but it's not working with current unraid version.

But like I said befor Unraid 7 beta2 has libusb-1.0.27.
But I need update Unraid for check it.

Maybe @desertwitch also know how to debug libusb in Unraid?

tormodvolden, 2024-08-21:
@masterwishx Please try 1.0.27 then. You can run "sudo make install" after the "make" above if you have the 1.0.27 or git master sources, or get it from your distribution - sometimes there are possibilities of upgrading single packages.

But please test xusb as instructed above first. Then afterwards you can back to debugging nut again.

desertwitch, 2024-08-21:
Sorry I don't really have anything useful to add here nor the time to help out with further diagnosis, considering this is for the most part an optical and not a functional, road-blocking issue for you (as we've established in the various other issues and repositories this has popped up in over the past weeks). I really don't mean this in a rude way but please, for sake of general etiquette, don't tag and ping me continuously everywhere - I would chime in if I had something useful to add or time to help out. For what it's worth, the compilation process on Slackware (which is the underlying OS of Unraid) is the same as with most other distributions - no special configurations or anything are really needed (those of my packaging process are unneeded for testing).

masterwishx, 2024-08-21:
> Sorry I don't really have anything useful to add here nor the time to help out with further diagnosis, considering this is for the most part an optical and not a functional, road-blocking issue for [...]

OK, I'm really sorry for this, but it's not only optic problem but some values not avaliable becose of it. Optic problem not important for me. Sorry again I got it. Thanks for your help

masterwishx, 2024-08-21:
> @masterwishx Please try 1.0.27 then. You can run "sudo make install" after the "make" above if you have the 1.0.27 or git master sources, or get it from your distribution - sometimes there are [...]

Thanks will try

jimklimov, 2024-08-22:
> Are your queries control transfers for requesting the device strings? I suppose you get the same error when running examples/xusb on the device?
>
> For getting these strings from the OS instead, there [...]

> Thanks for the quick reply! I'll try to track down the answer to how the info is queried, though likely tomorrow (now commuting home).

So, by the looks of it, in the initial device identification (where we loop to check if vendorid, productid and other criteria tell us that this driver likes this device), we:
* first get the numeric values using the library's `libusb_get_device_descriptor()` method https://github.com/networkupstools/nut/blob/master/drivers/libusb1.c#L297 (or `usb_open` with libusb-0.1 builds, at https://github.com/networkupstools/nut/blob/master/drivers/libusb0.c#L335);
* then if the basics are okay, go with a fine comb over other criteria, including `libusb_get_string_descriptor_ascii()` for `iManufacturer`, `iProduct` and `iSerialNumber` at https://github.com/networkupstools/nut/blob/master/drivers/libusb1.c#L395 and below (or `usb_get_string_simple()` in libusb-0.1 builds, at https://github.com/networkupstools/nut/blob/master/drivers/libusb0.c#L401)
  * elsewhere in actual driver code, for libusb-1.0 capable builds the `usb_get_string_simple` name is still used - but builds as a shim for `libusb_get_string_descriptor_ascii()` at https://github.com/networkupstools/nut/blob/master/drivers/usb-common.h#L348-L370
* finally, if the device is fully accepted by the criteria an a `callback` method as defined by individual drivers, a walk of the HID descriptor begins at https://github.com/networkupstools/nut/blob/master/drivers/libusb1.c#L686 (by discovering which of two ways to get one is fruitful), and a `callback` method as defined by individual drivers is called at https://github.com/networkupstools/nut/blob/master/drivers/libusb1.c#L842 to finally check if the driver likes what it sees.
  * in the `nut.log` file attached by OP above, the line `Aug  1 16:04:51 DaRKNaS rc.nut:    3.819549#011[D2:14407] Report Descriptor size = 2053` and the large data-dump below comes from the driver's callback; the text above is from `libusb1.c` (or `libusb0.c` in older builds)

In case of `usbhid-ups` driver, the callback method is at https://github.com/networkupstools/nut/blob/master/drivers/usbhid-ups.c#L1629

The big data dump ought to be `HIDDumpTree()` called at https://github.com/networkupstools/nut/blob/master/drivers/usbhid-ups.c#L1704 implemented at https://github.com/networkupstools/nut/blob/master/drivers/libhid.c#L406 and its visible work ends at log line `Aug  1 16:04:53 DaRKNaS rc.nut:    5.523237#011[D2:14407] get_model_name(unknown, 2000)`

I think that's all the digging I can wrap my head around at the moment...

masterwishx, 2024-08-24:
> To build xusb and other examples:
> 
> ```
> ./configure --enable-tests-build --enable-examples-build     # Or if from a git checkout: ./autogen.sh
> make
> ./examples/xusb -d 0x0463:0xffff    # or whatever the VID:PID is   # -d enables debug output
> ```

have error when trying `./configure --enable-tests-build --enable-examples-build     ` :  

[image]

masterwishx, 2024-08-24:
[image]

jimklimov, 2024-08-24:
Did you install the build toolkits? I suppose those listed in NUT `docs/config-prereqs.txt` would cover a lot (and more, for NUT's other dependencies) if not all libusb needs.

masterwishx, 2024-08-24:
> Did you install the build toolkits? I suppose those listed in NUT `docs/config-prereqs.txt` would cover a lot (and more, for NUT's other dependencies) if not all libusb needs.

No, I had only experience to compile in Slackware VM or same but in docker AutoSlackPack.
Thanks i will check it, but I think we talked about libusb compiling for check xusb how it throw error...

jimklimov, 2024-08-24:
Well, the screenshots indicate you lack parts of compiler toolkit (e.g. `as` - assembly parser used along the way of making binaries), hence the early-stage  `configure` errors.

masterwishx, 2024-08-24:
it seems `binutils` , but now also error i will try all other from `docs/config-prereqs.txt`

masterwishx, 2024-08-24:
i moved compiling from plain Unraid (because of error on  `make` `(make: error while loading shared libraries: libgc.so.1: cannot open shared object file: No such file or directory)` after installed a lot of packages ....

To docker image https://github.com/SpaceinvaderOne/AutoSlackPack and have next error :  
 
[image]

[image]

Log: 
[config.log](https://github.com/user-attachments/files/16736904/config.log)

masterwishx, 2024-08-24:
So i compiled in docker then runned from terminal `./examples/xusb -d 0x0463:0xffff ` :   

```
Using libusb v1.0.26.11724

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000005] [0000482d] libusb: debug [libusb_init] created default context
[ 0.000011] [0000482d] libusb: debug [libusb_init] libusb v1.0.26.11724
[ 0.000033] [0000482d] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000040] [0000482d] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000046] [0000482d] libusb: debug [usbi_add_event_source] add fd 4 events 1
[ 0.000054] [0000482d] libusb: debug [get_kernel_version] reported kernel version is 6.1.106-Unraid
[ 0.000074] [0000482d] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000077] [0000482d] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000081] [0000482d] libusb: debug [op_init] sysfs is available
[ 0.000229] [00004849] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.001091] [0000482d] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.001096] [0000482d] libusb: debug [linux_get_device_address] scan usb3
[ 0.001129] [0000482d] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.001131] [0000482d] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.001134] [0000482d] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.001193] [0000482d] libusb: debug [linux_get_device_address] getting address for device: 3-3 detached: 0
[ 0.001204] [0000482d] libusb: debug [linux_get_device_address] scan 3-3
[ 0.001227] [0000482d] libusb: debug [linux_get_device_address] bus=3 dev=2
[ 0.001246] [0000482d] libusb: debug [linux_enumerate_device] busnum 3 devaddr 2 session_id 770
[ 0.001252] [0000482d] libusb: debug [linux_enumerate_device] allocating new device for 3/2 (session 770)
[ 0.001281] [0000482d] libusb: debug [linux_get_parent_info] dev 0x41aec0 (3-3) has parent 0x41ca00 (usb3) port 3
[ 0.001322] [0000482d] libusb: debug [linux_get_device_address] getting address for device: 3-4 detached: 0
[ 0.001326] [0000482d] libusb: debug [linux_get_device_address] scan 3-4
[ 0.001350] [0000482d] libusb: debug [linux_get_device_address] bus=3 dev=3
[ 0.001352] [0000482d] libusb: debug [linux_enumerate_device] busnum 3 devaddr 3 session_id 771
[ 0.001358] [0000482d] libusb: debug [linux_enumerate_device] allocating new device for 3/3 (session 771)
[ 0.001378] [0000482d] libusb: debug [linux_get_parent_info] dev 0x41b0e0 (3-4) has parent 0x41ca00 (usb3) port 4
[ 0.001424] [0000482d] libusb: debug [linux_get_device_address] getting address for device: 3-6 detached: 0
[ 0.001431] [0000482d] libusb: debug [linux_get_device_address] scan 3-6
[ 0.001452] [0000482d] libusb: debug [linux_get_device_address] bus=3 dev=4
[ 0.001526] [0000482d] libusb: debug [linux_enumerate_device] busnum 3 devaddr 4 session_id 772
[ 0.001530] [0000482d] libusb: debug [linux_enumerate_device] allocating new device for 3/4 (session 772)
[ 0.001572] [0000482d] libusb: debug [linux_get_parent_info] dev 0x41b190 (3-6) has parent 0x41ca00 (usb3) port 6
[ 0.001597] [0000482d] libusb: debug [linux_get_device_address] getting address for device: usb4 detached: 0
[ 0.001600] [0000482d] libusb: debug [linux_get_device_address] scan usb4
[ 0.001627] [0000482d] libusb: debug [linux_get_device_address] bus=4 dev=1
[ 0.001630] [0000482d] libusb: debug [linux_enumerate_device] busnum 4 devaddr 1 session_id 1025
[ 0.001644] [0000482d] libusb: debug [linux_enumerate_device] allocating new device for 4/1 (session 1025)
[ 0.001699] [0000482d] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.001702] [0000482d] libusb: debug [linux_get_device_address] scan usb1
[ 0.001724] [0000482d] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.001726] [0000482d] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.001742] [0000482d] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.001791] [0000482d] libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
[ 0.001794] [0000482d] libusb: debug [linux_get_device_address] scan 1-1
[ 0.001832] [0000482d] libusb: debug [linux_get_device_address] bus=1 dev=2
[ 0.001835] [0000482d] libusb: debug [linux_enumerate_device] busnum 1 devaddr 2 session_id 258
[ 0.001840] [0000482d] libusb: debug [linux_enumerate_device] allocating new device for 1/2 (session 258)
[ 0.001860] [0000482d] libusb: debug [linux_get_parent_info] dev 0x40aa70 (1-1) has parent 0x40a890 (usb1) port 1
[ 0.001910] [0000482d] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.001913] [0000482d] libusb: debug [linux_get_device_address] scan usb2
[ 0.001936] [0000482d] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.001939] [0000482d] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.001944] [0000482d] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.001997] [0000482d] libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
[ 0.002001] [0000482d] libusb: debug [linux_get_device_address] scan 2-1
[ 0.002025] [0000482d] libusb: debug [linux_get_device_address] bus=2 dev=2
[... cut ...]
[ 2.736834] [0000482d] libusb: debug [libusb_unref_device] destroy device 4.1
[ 2.736838] [0000482d] libusb: debug [libusb_unref_device] destroy device 3.4
[ 2.736840] [0000482d] libusb: debug [libusb_unref_device] destroy device 3.3
[ 2.736842] [0000482d] libusb: debug [libusb_unref_device] destroy device 3.2
[ 2.736844] [0000482d] libusb: debug [libusb_unref_device] destroy device 3.1
[ 2.736847] [0000482d] libusb: debug [usbi_remove_event_source] remove fd 4
[ 2.736852] [0000482d] libusb: debug [usbi_remove_event_source] remove fd 3
```

tormodvolden, 2024-08-24:
Thanks for your efforts! It looks like xusb also has issues with the string descriptors. So we can focus on libusb and your device and forget about nut for now.
> Using libusb v1.0.26.11724

Can you please also try 1.0.27? Or git master?

Please apply this patch to examples/xusb.c (if editing manually just add the "+" lines without the plus:
```
diff --git a/examples/xusb.c b/examples/xusb.c
index 239450ce..05b6577b 100644
--- a/examples/xusb.c
+++ b/examples/xusb.c
@@ -960,6 +960,11 @@ static int test_device(uint16_t vid, uint16_t pid)
                if (string_index[i] == 0) {
                        continue;
                }
+               r = libusb_get_string_descriptor(handle, 0, 0, (unsigned char*)string, 4);
+               printf("langid for string %d returned %d\n", string_index[i], r);
+               if (r > 0)
+                       display_buffer_hex((unsigned char*) string, r);
+
                if (libusb_get_string_descriptor_ascii(handle, string_index[i], (unsigned char*)string, sizeof(string)) > 0) {
                        printf("   String (0x%02X): \"%s\"\n", string_index[i], string);
                }
```

masterwishx, 2024-08-24:
> Thanks for your efforts! It looks like xusb also has issues with the string descriptors. So we can focus on libusb and your device and forget about nut for now.
>
>
> Can you please also try 1.0.27? Or [...]

is it ok that i had error on compile ? posted log here :  https://github.com/libusb/libusb/issues/1548#issuecomment-2308410384
 `conftest.cpp:23:10: fatal error: ac_nonexistent.h: No such file or directory
   23 | #include <ac_nonexistent.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.`

tormodvolden, 2024-08-24:
"configure" runs a lot of compilation tests to "try out" the compiler so some of these will fail. As long as "configure" finishes with success, it is fine.

masterwishx, 2024-08-24:
its not finished with success but stopped in 1.0.26  also now 1.0.27 with :   
```
config.status: creating config.h
config.status: executing depfiles commands
config.status: executing libtool commands   <-   last line 
```

masterwishx, 2024-08-24:
but it worked i think but now i have :  

```./examples/xusb: line 202: cd: /app/libusb-1.0.27/examples: No such file or directory
/usr/bin/ld: cannot find crt1.o: No such file or directory
/usr/bin/ld: cannot find crti.o: No such file or directory
/usr/bin/ld: cannot find xusb.o: No such file or directory
/usr/bin/ld: cannot find ../libusb/.libs/libusb-1.0.so: No such file or directory
collect2: error: ld returned 1 exit status```

tormodvolden, 2024-08-24:
That is some libtool issue. Try "make clean" before "make" again, or start over from scratch.

masterwishx, 2024-08-24:
> make clean

> /app/libusb-1.0.27/examples: No such file or directory

it seems becose it stored container path : ` /app/libusb-1.0.27/examples ` and im running from unraid terminal but it worked befor with 1.0.26

masterwishx, 2024-08-24:
tryed some times but get the same , also have:    

[image]

```
./examples/xusb: line 202: cd: /app/libusb/examples: No such file or directory
/usr/bin/ld: cannot find crt1.o: No such file or directory
/usr/bin/ld: cannot find crti.o: No such file or directory
/usr/bin/ld: cannot find xusb.o: No such file or directory
/usr/bin/ld: cannot find ../libusb/.libs/libusb-1.0.so: No such file or directory
collect2: error: ld returned 1 exit status
```

howere v 1.0.26 worked , here also have in compile last line , but no success:   `config.status: executing libtool commands`

tormodvolden, 2024-08-24:
Do "make clean" again and this time do "make CFLAGS=-static". Then the examples/xusb binary will have libusb embedded and you can copy it over and run anywhere without libtool coming in the way.

masterwishx, 2024-08-24:
> Do "make clean" again and this time do "make CFLAGS=-static". Then the examples/xusb binary will have libusb embedded and you can copy it over and run anywhere without libtool coming in the way.

Thanks, I will try it

masterwishx, 2024-08-25:
> Do "make clean" again and this time do "make CFLAGS=-static". Then the examples/xusb binary will have libusb embedded and you can copy it over and run anywhere without libtool coming in the way.

Thanks for advice ,have this after runned :  
```
Using libusb v1.0.27.11882

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000014] [00000ce1] libusb: debug [libusb_init_context] created default context
[ 0.000029] [00000ce1] libusb: debug [libusb_init_context] libusb v1.0.27.11882
[ 0.000054] [00000ce1] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000077] [00000ce1] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000080] [00000ce1] libusb: debug [usbi_add_event_source] add fd 4 events 1
[ 0.000106] [00000ce1] libusb: debug [get_kernel_version] reported kernel version is 6.1.106-Unraid
[ 0.000127] [00000ce1] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000130] [00000ce1] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000143] [00000ce1] libusb: debug [op_init] sysfs is available
[ 0.000338] [00000ce2] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.001273] [00000ce1] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.001280] [00000ce1] libusb: debug [linux_get_device_address] scan usb3
[ 0.001311] [00000ce1] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.001329] [00000ce1] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.001340] [00000ce1] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.001401] [00000ce1] libusb: debug [linux_get_device_address] getting address for device: 3-3 detached: 0
[ 0.001406] [00000ce1] libusb: debug [linux_get_device_address] scan 3-3
[ 0.001428] [00000ce1] libusb: debug [linux_get_device_address] bus=3 dev=2
[ 0.001431] [00000ce1] libusb: debug [linux_enumerate_device] busnum 3 devaddr 2 session_id 770
[ 0.001435] [00000ce1] libusb: debug [linux_enumerate_device] allocating new device for 3/2 (session 770)
[ 0.001462] [00000ce1] libusb: debug [linux_get_parent_info] dev 0x438e00 (3-3) has parent 0x43a940 (usb3) port 3
[ 0.001489] [00000ce1] libusb: debug [linux_get_device_address] getting address for device: 3-4 detached: 0
[ 0.001492] [00000ce1] libusb: debug [linux_get_device_address] scan 3-4
[ 0.001514] [00000ce1] libusb: debug [linux_get_device_address] bus=3 dev=3
[ 0.001517] [00000ce1] libusb: debug [linux_enumerate_device] busnum 3 devaddr 3 session_id 771
[ 0.001519] [00000ce1] libusb: debug [linux_enumerate_device] allocating new device for 3/3 (session 771)
[ 0.001542] [00000ce1] libusb: debug [linux_get_parent_info] dev 0x439020 (3-4) has parent 0x43a940 (usb3) port 4
[ 0.001566] [00000ce1] libusb: debug [linux_get_device_address] getting address for device: 3-6 detached: 0
[ 0.001570] [00000ce1] libusb: debug [linux_get_device_address] scan 3-6
[ 0.001592] [00000ce1] libusb: debug [linux_get_device_address] bus=3 dev=4
[ 0.001596] [00000ce1] libusb: debug [linux_enumerate_device] busnum 3 devaddr 4 session_id 772
[ 0.001605] [00000ce1] libusb: debug [linux_enumerate_device] allocating new device for 3/4 (session 772)
[ 0.001634] [00000ce1] libusb: debug [linux_get_parent_info] dev 0x4390d0 (3-6) has parent 0x43a940 (usb3) port 6
[ 0.001660] [00000ce1] libusb: debug [linux_get_device_address] getting address for device: usb4 detached: 0
[ 0.001663] [00000ce1] libusb: debug [linux_get_device_address] scan usb4
[ 0.001686] [00000ce1] libusb: debug [linux_get_device_address] bus=4 dev=1
[ 0.001689] [00000ce1] libusb: debug [linux_enumerate_device] busnum 4 devaddr 1 session_id 1025
[ 0.001692] [00000ce1] libusb: debug [linux_enumerate_device] allocating new device for 4/1 (session 1025)
[ 0.001734] [00000ce1] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.001738] [00000ce1] libusb: debug [linux_get_device_address] scan usb1
[ 0.001759] [00000ce1] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.001762] [00000ce1] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.001765] [00000ce1] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.001816] [00000ce1] libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
[ 0.001819] [00000ce1] libusb: debug [linux_get_device_address] scan 1-1
[ 0.001848] [00000ce1] libusb: debug [linux_get_device_address] bus=1 dev=2
[ 0.001858] [00000ce1] libusb: debug [linux_enumerate_device] busnum 1 devaddr 2 session_id 258
[ 0.001866] [00000ce1] libusb: debug [linux_enumerate_device] allocating new device for 1/2 (session 258)
[ 0.001930] [00000ce1] libusb: debug [linux_get_parent_info] dev 0x4289b0 (1-1) has parent 0x4287d0 (usb1) port 1
[ 0.001956] [00000ce1] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.001959] [00000ce1] libusb: debug [linux_get_device_address] scan usb2
[ 0.001984] [00000ce1] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.001987] [00000ce1] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.001995] [00000ce1] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.002035] [00000ce1] libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
[... cut ...]
[ 2.800303] [00000ce1] libusb: debug [libusb_unref_device] destroy device 3.4
[ 2.800305] [00000ce1] libusb: debug [libusb_unref_device] destroy device 3.3
[ 2.800307] [00000ce1] libusb: debug [libusb_unref_device] destroy device 3.2
[ 2.800310] [00000ce1] libusb: debug [libusb_unref_device] destroy device 3.1
[ 2.800322] [00000ce2] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
[ 2.800392] [00000ce1] libusb: debug [usbi_remove_event_source] remove fd 4
[ 2.800398] [00000ce1] libusb: debug [usbi_remove_event_source] remove fd 3
```

masterwishx, 2024-08-25:
[config libusb-1.0.27.log](https://github.com/user-attachments/files/16740742/config.libusb-1.0.27.log)
[config libusb-1.0.26.log](https://github.com/user-attachments/files/16740743/config.libusb-1.0.26.log)

@tormodvolden please check compile logs , becose as i said befor i dont have success on compile , its just stop when:  
```
config.status: creating config.h
config.status: executing depfiles commands
config.status: executing libtool commands     **<-   last line**
```

masterwishx, 2024-08-25:
> [config libusb-1.0.27.log](https://github.com/user-attachments/files/16740742/config.libusb-1.0.27.log) [config [...]

Or as you mentioned it seems also OK?

tormodvolden, 2024-08-25:
Thanks, now I can see what is wrong:
```
langid for string 1 returned 4

  00000000  04 00 03 00            
```
This is a bug in the device firmware. It should have returned a valid descriptor listing which languages are supported for its string descriptors. On a typical device that includes the code 0x0409 for en_US language, which would give the byte sequence 04 (length of descriptor) 03 (type of descriptor) 09 04 (0x0409 in little-endian).
In this case the returned descriptor type is 0 which is not a valid value, and libusb rejects it. I am pretty sure the language code is not meant to be 0x0003 either. I think the firmware authors goofed and returned the 16-bit numbers 04 and 03 instead and maybe the actual language code (e.g. 0x0409) follows these first 4 bytes.

jimklimov, 2024-08-25:
CC @arnaudquette-eaton

masterwishx, 2024-08-26:
> maybe the actual language code (e.g. 0x0409) follows these first 4 bytes.

Should I check also next 4 bytes if it will help?

```r = libusb_get_string_descriptor(handle, 0, 0, (unsigned char*)string, 8); ```

tormodvolden, 2024-08-26:
Yes, thanks, it would be interesting!

masterwishx, 2024-08-26:
Strange, have the same ? 

```
Using libusb v1.0.27.11882

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000025] [000004d3] libusb: debug [libusb_init_context] created default context
[ 0.000046] [000004d3] libusb: debug [libusb_init_context] libusb v1.0.27.11882
[ 0.000078] [000004d3] libusb: debug [usbi_add_event_source] add fd 3 events 1
[ 0.000106] [000004d3] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000113] [000004d3] libusb: debug [usbi_add_event_source] add fd 4 events 1
[ 0.000153] [000004d3] libusb: debug [get_kernel_version] reported kernel version is 6.1.106-Unraid
[ 0.000185] [000004d3] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000200] [000004d3] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000211] [000004d3] libusb: debug [op_init] sysfs is available
[ 0.001326] [000004d3] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.001348] [000004d3] libusb: debug [linux_get_device_address] scan usb3
[ 0.001381] [000004d3] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.001404] [000004d3] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.001431] [000004d3] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.001533] [000004d3] libusb: debug [linux_get_device_address] getting address for device: 3-3 detached: 0
[ 0.001547] [000004d3] libusb: debug [linux_get_device_address] scan 3-3
[ 0.001576] [000004d3] libusb: debug [linux_get_device_address] bus=3 dev=2
[ 0.001585] [000004d3] libusb: debug [linux_enumerate_device] busnum 3 devaddr 2 session_id 770
[ 0.001592] [000004d3] libusb: debug [linux_enumerate_device] allocating new device for 3/2 (session 770)
[ 0.001629] [000004d3] libusb: debug [linux_get_parent_info] dev 0x43a6e0 (3-3) has parent 0x439820 (usb3) port 3
[ 0.001669] [000004d3] libusb: debug [linux_get_device_address] getting address for device: 3-4 detached: 0
[ 0.001678] [000004d3] libusb: debug [linux_get_device_address] scan 3-4
[ 0.001707] [000004d3] libusb: debug [linux_get_device_address] bus=3 dev=3
[ 0.001719] [000004d3] libusb: debug [linux_enumerate_device] busnum 3 devaddr 3 session_id 771
[ 0.001726] [000004d3] libusb: debug [linux_enumerate_device] allocating new device for 3/3 (session 771)
[ 0.001752] [000004d3] libusb: debug [linux_get_parent_info] dev 0x43a900 (3-4) has parent 0x439820 (usb3) port 4
[ 0.001800] [000004d3] libusb: debug [linux_get_device_address] getting address for device: 3-6 detached: 0
[ 0.001811] [000004d3] libusb: debug [linux_get_device_address] scan 3-6
[ 0.001838] [000004d3] libusb: debug [linux_get_device_address] bus=3 dev=4
[ 0.001847] [000004d3] libusb: debug [linux_enumerate_device] busnum 3 devaddr 4 session_id 772
[ 0.001853] [000004d3] libusb: debug [linux_enumerate_device] allocating new device for 3/4 (session 772)
[ 0.001896] [000004d3] libusb: debug [linux_get_parent_info] dev 0x43ab20 (3-6) has parent 0x439820 (usb3) port 6
[ 0.001948] [000004d3] libusb: debug [linux_get_device_address] getting address for device: usb4 detached: 0
[ 0.001959] [000004d3] libusb: debug [linux_get_device_address] scan usb4
[ 0.001988] [000004d3] libusb: debug [linux_get_device_address] bus=4 dev=1
[ 0.001996] [000004d3] libusb: debug [linux_enumerate_device] busnum 4 devaddr 1 session_id 1025
[ 0.002009] [000004d3] libusb: debug [linux_enumerate_device] allocating new device for 4/1 (session 1025)
[ 0.002073] [000004d3] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.002085] [000004d3] libusb: debug [linux_get_device_address] scan usb1
[ 0.002114] [000004d3] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.002123] [000004d3] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.002129] [000004d3] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.002183] [000004d3] libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
[ 0.002194] [000004d3] libusb: debug [linux_get_device_address] scan 1-1
[ 0.002220] [000004d3] libusb: debug [linux_get_device_address] bus=1 dev=2
[ 0.002228] [000004d3] libusb: debug [linux_enumerate_device] busnum 1 devaddr 2 session_id 258
[ 0.002235] [000004d3] libusb: debug [linux_enumerate_device] allocating new device for 1/2 (session 258)
[ 0.002278] [000004d3] libusb: debug [linux_get_parent_info] dev 0x428790 (1-1) has parent 0x43a590 (usb1) port 1
[ 0.002324] [000004d3] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.002338] [000004d3] libusb: debug [linux_get_device_address] scan usb2
[ 0.002364] [000004d3] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.002373] [000004d3] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.002394] [000004d3] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.002451] [000004d3] libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
[ 0.002461] [000004d3] libusb: debug [linux_get_device_address] scan 2-1
[ 0.002473] [000004d4] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.002507] [000004d3] libusb: debug [linux_get_device_address] bus=2 dev=2
[... cut ...]
[ 2.803932] [000004d3] libusb: debug [libusb_unref_device] destroy device 3.4
[ 2.803934] [000004d3] libusb: debug [libusb_unref_device] destroy device 3.3
[ 2.803936] [000004d3] libusb: debug [libusb_unref_device] destroy device 3.2
[ 2.803938] [000004d3] libusb: debug [libusb_unref_device] destroy device 3.1
[ 2.803948] [000004d4] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
[ 2.804023] [000004d3] libusb: debug [usbi_remove_event_source] remove fd 4
[ 2.804029] [000004d3] libusb: debug [usbi_remove_event_source] remove fd 3
```

masterwishx, 2024-08-26:
[image]

jimklimov, 2024-08-26:
Seems the reply is 4 bytes long, unbothered by being given an 8-byte buffer (just not using all of it).

I wonder how those other programs got the strings though - ignored bogus output and assumed English? Had some other query? There's some other option to return 8/16-bit words instead of just 8-bit we expect here?..

tormodvolden, 2024-08-26:
The Linux kernel makes great efforts to workaround buggy devices. In this case, with a broken language identifier, it will assume en_US and request the en_US version of the string descriptor.

If this is important to nut, I would recommend implementing a custom version of libusb_get_string_descriptor_ascii() which falls back to en_US (or simply assumes en_US in the case of Eaton devices).

masterwishx, 2024-08-26:
> The Linux kernel makes great efforts to workaround buggy devices. In this case, with a broken language identifier, it will assume en_US and request the en_US version of the string descriptor.
>
> If [...]

@tormodvolden Thanks a lot for your help I think I can close the issue now...

@jimklimov I hope finally we found where the bug was...
If you need test on fixing, I'm still here..
Please confirm I can close issue here.

jimklimov, 2024-08-26:
I guess we can at least explore the path. I'm traveling for the bext half a month so not sure when I'd get some quality computer time though.

For the sake of future versions, I hope @arnaudquette-eaton can pass the bug to firmware team. :)

tormodvolden, 2024-08-26:
On the libusb side we should maybe add a warning to make this easier to spot the next time.

tormodvolden, 2024-08-28:
@arnaudquette-eaton Do you perform any USB compliance testing on the devices?

tormodvolden, 2024-09-03:
@aquette You don't have to disclose any individual company's practices :) but I am curious about what manufacturers typically would do. I thought they had to run through some standard compliance tests before they can print the USB logo on their devices. There is for instance a test program at https://www.usb.org/compliancetools#anchor_usb3cv that according to the documentation will check the langid descriptor and string descriptors among other things.

masterwishx, 2024-09-04:
> @aquette You don't have to disclose any individual company's practices :) but I am curious about what manufacturers typically would do. I thought they had to run through some standard compliance [...]

I don't think I have any usb logo on my ups 9E2000i only usb port. :) 

If for real, as you can see from NUT issue it was not solved for 1.5 years until you helped, however eaton devs was there :(

arnaudquette-eaton, 2024-09-04:
@masterwishx as an OSS leader and dev, I'm retired for 6 years+
As an Eaton guy, I'd say even more... it's only persistence of vision ;)
And I was the only dev, and mainly not with Eaton cap on!
more seriously, I'm too busy at work, and far away from SW communication for almost a decade.
Still, I do my best to continue supporting my baby NUT... And still have that specific issue in mind

@tormodvolden would need to recheck. Keep in mind that I committed all these USB related things (kernel, libusb, NUT USB driver) 20 years ago... And as for all my OSS benevolent work (despite a potential interest for my employers), once done, I moved to something else...

masterwishx, 2024-09-27:
@tormodvolden sorry to bother you again as this issue was solved and maybe need to close it ? 

But have other issue if it is :  
seems i have a lot of `nut_libusb_get_interrupt: Connection timed out` in debug log and 
maybe also affect on https://github.com/networkupstools/nut/issues/2643
or  its just closing connection every time for cycle of reading data?

Mostly looks like this :  
```
send_to_all: SETINFO driver.state "quiet"
Sep 27 09:37:30 DaRKNaS usbhid-ups[24218]: [D6:24218] send_to_all: write 29 bytes to socket 16 succeeded (ret=29): SETINFO driver.state "quiet"
Sep 27 09:37:31 DaRKNaS usbhid-ups[24218]: [D5:24218] send_to_all: SETINFO driver.state "updateinfo"
Sep 27 09:37:31 DaRKNaS usbhid-ups[24218]: [D6:24218] send_to_all: write 34 bytes to socket 16 succeeded (ret=34): SETINFO driver.state "updateinfo"
Sep 27 09:37:31 DaRKNaS usbhid-ups[24218]: [D1:24218] upsdrv_updateinfo...
Sep 27 09:37:32 DaRKNaS usbhid-ups[24218]: [D2:24218] nut_libusb_get_interrupt: Connection timed out
Sep 27 09:37:32 DaRKNaS usbhid-ups[24218]: [D1:24218] Got 0 HID objects...
Sep 27 09:37:32 DaRKNaS usbhid-ups[24218]: [D1:24218] Quick update...
Sep 27 09:37:32 DaRKNaS usbhid-ups[24218]: [D4:24218] 
```

[image]

[4 more replies]