libusb/libusb #1561: libusb compatibility with macOS

sanjana-j-wdc, 2024-09-02:
Are there any versions of macOS on which libusb does not work? I want to send admin SCSI commands to USB mass storage device.

tormodvolden, 2024-09-02:
I am not sure what the oldest, officially supported macOS version is, but libusb/os/darwin_usb.c and libusb/os/darwin_usb.h does version detection that falls back to macOS 10.0, with "interface version" 220 and "device version" 197. In practice, your best bet is looking at what macports and homebrew ships, because they do additional efforts to support older OS versions than we do upstream. Of course, on older platforms some features may not be supported.

tormodvolden, 2024-09-02:
examples/xusb.c sends SCSI commands to USB mass storage devices. In general we don't handheld people venturing into this subject, and often it is obvious from the start they don't realize the complexity involved.

In principle you can do anything USB with libusb, bar performance, but it is userspace and not meant to compete or interfere with kernel drivers.

mcuee, 2024-09-02:
The issue with macOS and libusb is that no easy to detach the kernel driver. So this is a no go.
* https://github.com/libusb/libusb/issues/1014
* https://github.com/libusb/libusb/wiki/FAQ#how-can-i-run-libusb-applications-under-mac-os-x-if-there-is-already-a-kernel-extension-installed-for-the-device-and-claim-exclusive-access