libusb/libusb #1564: Behringer UMC204HD "reboot" cycle

RJVB, 2024-09-11:
I have a Behringer UMC204HD USB DAC connected via a USB3 port on a Thunderbolt dock which gets into what looks like an endless reboot loop as long as I run an application that uses libusb. The example that interests me is the `usbmuxd` implementation from `libimobiledevice` but the same thing happens when I activate WebUSB in Chromium.

I presume that the underlying error is in the device firmware. If so it will probably not get fixed because Behringer cannot reproduce the issue.

Ignoring the issue is not an option as the power circuits in the DAC keep getting restarted leading to pops in attached speakers or phones (and lots of System.log pollution). For now my workaround is to disconnect the device when I need usbmuxd (to access data on an iPhone no longer supported on my Mac OS version) but that isn't satisfactory.

I see 2 better workarounds
- blacklist the device at the libusb level. This seems like a reasonable feature on platforms where libusb isn't more or less the system USB library and access to most devices will probably go through the system equivalent(?)
- blacklist or ignore it in usbmuxd

The latter will presumably be a lot easier for me to implement myself of libusb already has a way for client software to indicate that it's only interested in devices of a specific vendor, for instance? (This assumes devices don't get an init/reset command just to read their IDs...)

I presume libusb doesn't send initialisation commands to all attached devices continuously, so what sequence of events at what place should we be looking for? Is there a code path that keeps getting executed until a device responds in the desired way and that I could shortcut as an initial approach?

Pointers would be appreciated here given that I am familiar with neither libusb nor usbmuxd.

tormodvolden, 2024-09-11:
libusb doesn't do any communication with the device unless the application calls for it. IDs are read out by the OS and libusb only looks through the OS information during its enumeration (plus some hub requests on Windows). And there is no blacklisting at "libusb" level. You can block things at udev level in Linux, like udev rules, or in usbmuxd.

Also, a DAC like this is usually taken care of by an OS driver, and libusb won't have access to it.

RJVB, 2024-09-11:
On Wednesday September 11 2024 14:41:34 Tormod Volden wrote:

We're talking about the Mac OS here.

>libusb doesn't do any communication with the device unless the application calls for it. IDs are read out by the OS and libusb only looks through the OS information during its enumeration 
>(plus some hub requests on Windows).

And on Mac. After spending an hour debugging the listdevs example, zooming in on what happens I can say that the call that triggers a single instance of the issue is `darwin_request_descriptor()`.

>And there is no blacklisting at "libusb" level. You can block things at udev level in Linux, like udev rules

Again, Mac. No udev. That's why I said that blacklisting "seems like a reasonable feature on platforms where libusb isn't more or less the system USB library", because there is no blacklisting at the system level (= udev).
Which would go a step beyond what I need anyway because I do want to be able to use my DAC. If not I'd just leave it disconnected, don't you think?

>or in usbmuxd.

That would clearly be too late, given that the issue is already triggered in the libusb init routine.

>Also, a DAC like this is usually taken care of by an OS driver, and libusb won't have access to it.

Maybe on Linux, but on Mac I see all USB devices in the output from listdevs, even a few I cannot even find in the System Information application's USB panel.

I can get the Vendor,Product pair (0x1397,0x0508) without triggering the issue so I'm now seeing if I can hack in a device-specific ignore in my copy of libusb... it seems that an early error return from `darwin_cache_device_descriptor()` and (for good measure) from `darwin_request_descriptor()` solves my issue. I'm using kIOReturnNotPermitted and the libusb translation of that, which seems like the most appropriate way to signal the device as off-limits.

tormodvolden, 2024-09-11:
Yes, some of what I said is not true for the macOS port. darwin_request_descriptor() will try to communicate with the device.

RJVB, 2024-09-11:
>Yes, some of what I said is not true for the macOS port. darwin_request_descriptor() will try to communicate with the device.

And my guess would be that usbmuxd in its default mode (hotplug support ON) causes a regular call to that function.

tormodvolden, 2024-09-12:
Yes, all calls to libusb_init\* will cause these requests (unless the LIBUSB_OPTION_NO_DEVICE_DISCOVERY flag is set).

It is done to get a copy of the device descriptor. The libusb API [1] guarantees that all device descriptors for discovered devices are available without hardware access... after initialization.

In the case of macOS it seems that 95% of the device descriptor values are easily available through the IOUSBDeviceInterface GET\* functions (from the OS, without causing USB I/O). I don't know about the remaining 5% (bMaxPacketSize0, bcdUSB). IMO it would be a good improvement if all values could be read out to synthesize a device descriptor instead of sending a control transfer to each and every device.

[1] https://libusb.sourceforge.io/api-1.0/group__libusb__desc.html#ga5e9ab08d490a7704cf3a9b0439f16f00
[2] https://opensource.apple.com/source/IOUSBFamily/IOUSBFamily-226.4.3/Documentation/IOUSBLib/Functions/DeviceInterface.html

tormodvolden, 2024-09-12:
I suppose GetConfigurationDescriptorPtr only returns the configuration descriptor itself, and not the whole configuration with all descriptors?

RJVB, 2024-09-12:
On Thursday September 12 2024 02:38:40 Tormod Volden wrote:
>I suppose GetConfigurationDescriptorPtr only returns the configuration descriptor itself, and not the whole configuration with all descriptors?

The inline documentation suggests otherwise:
```
/*! @function GetConfigurationDescriptorPtr
    @abstract Return a pointer to a configuration descriptor for a given index.
    @discussion The device does not have to be open.
    @param self Pointer to an IOUSBDeviceInterface for one IOService.
    @param configIndex The index of the desired config descriptor.
    @param desc Pointer to pointer to an IOUSBConfigurationDescriptor.
    @result Returns kIOReturnSuccess if successful, kIOReturnNoDevice if there is no connection to an IOService.
*/
```

You're already using the function (in `get_configuration_index()`; I'd have to resume debugging to see if that function gets called before the incriminated call.

Re: opening the device or not: 1st thing I tried was not opening it, which didn't seem to make any difference.

tormodvolden, 2024-09-12:
Yes, GetConfigurationDescriptorPtr should not create extra I/O. My question was if it returns (from the OS cache) the complete configuration, which is a concatenation of all sub-descriptors. But that was a brainfart anyway, because the device descriptor isn't a sub-descriptor, rather a super-descriptor. So please just disregard that question.

tormodvolden, 2024-09-12:
I wonder if the last missing piece bcdUSB can be had from the IO registry or somewhere else?

RJVB, 2024-09-12:
>I wonder if the last missing piece bcdUSB can be had from the IO registry or somewhere else?

Don't ask me...

I was wondering if it would be an idea to have a halfway option between the default and LIBUSB_OPTION_NO_DEVICE_DISCOVERY  that just guarantees to discover the information accessible on the given platform without actually accessing the devices. And then saw that according to libusb.h, LIBUSB_OPTION_NO_DEVICE_DISCOVERY  is valid only on Linux?
The idea for that halfway option was to give client software a little more margin in indicating what's important to them and take care of platform specifics themselves. It could also provide the smallest common subset of properties that's available "for free" on all supported platforms (unless that's an empty set) ; you'd expect that that would be the properties most likely to be of interest for client software?

I still think though that support for a simple, user-defined blacklist would be the best solution. Or, comparable in this case, a filter option that discover/initialise only devices or a given vendor and/or product type (in my/usbmuxd case that would be only Apple devices, by definition).

tormodvolden, 2024-09-13:
I have a sketch of the idea at https://github.com/tormodvolden/libusb/tree/darwin_ioless_enum . It is not a PR yet, because it hasn't even been compile-tested, and could well have some typos and misconstructions. If you could beat it into compiling and testing it, it would be great. For now it hard-codes bcdUSB to 2.00 so it is obviously only for testing.

RJVB, 2024-09-13:
Will try, thanks!

tormodvolden, 2024-09-13:
I see the CI ran it now even if it wasn't a PR. And failed of course.

tormodvolden, 2024-09-13:
It would be great if you can run `xusb -d`, which will provide debug output from the initialization/enumeration.

RJVB, 2024-09-13:
On Friday September 13 2024 12:39:58 Tormod Volden wrote:
>  It would be great if you can run `xusb -d`, which will provide debug output from the initialization/enumeration.

will do. I haven't gotten around to this yet, maybe later tonight.

RJVB, 2024-09-13:
Well this looks promising!

# libusb 1.0.27 with my device blacklist hack:
```
> list devs
0080:a001 (bus 129, device 3) path: 4
0d8c:013a (bus 128, device 2) path: 3
046d:c339 (bus 253, device 4) path: 1.2
05ac:821a (bus 250, device 7) path: 1.1.3
0a5c:4500 (bus 250, device 4) path: 1.1
0424:2513 (bus 250, device 2) path: 1
0424:2513 (bus 253, device 2) path: 1
05ac:8007 (bus 128, device 129)
05ac:8007 (bus 129, device 128)
05ac:8006 (bus 253, device 1)
05ac:8006 (bus 250, device 1)
```

#darwin_ioless_enum branch:
```
> listdevs
0080:a001 (bus 129, device 3) path: 4
0d8c:013a (bus 128, device 2) path: 3
1397:0508 (bus 128, device 1) path: 2
046d:c339 (bus 253, device 4) path: 1.2
05ac:821a (bus 250, device 7) path: 1.1.3
05ac:0252 (bus 250, device 5) path: 1.2
05ac:8509 (bus 250, device 3) path: 2
05ac:8242 (bus 253, device 3) path: 1.1
0a5c:4500 (bus 250, device 4) path: 1.1
0424:2513 (bus 250, device 2) path: 1
0424:2513 (bus 253, device 2) path: 1
05ac:8007 (bus 128, device 129)
05ac:8007 (bus 129, device 128)
05ac:8006 (bus 253, device 1)
05ac:8006 (bus 250, device 1)
```

And no DAC reboot or any other kind of reaction!

`xusb -d` output from the #darwin_ioless_enum branch:
[xusb-darwin_ioless_enum.txt](https://github.com/user-attachments/files/16999953/xusb-darwin_ioless_enum.txt)

and for comparison, `xusb -d` output from v1.0.27 with my hack:
[xusb.txt](https://github.com/user-attachments/files/16999957/xusb.txt)

(note the couple of failure messages in there which don't seem to be in the ioless version).

I haven't yet gotten around to testing if `uxbmuxd` works correctly with this variant, will try to do that this weekend.

tormodvolden, 2024-09-14:
Great, thanks a lot for the testing! If only we could find a way to get the bcdUSB values correct...

A part from the DAC, there are 3 devices showing up that were skipped in the original code due to failing requests:
```
05ac:8242: pipe is stalled (e000404f). skipping device
05ac:8509: device not responding (e00002ed). skipping device
05ac:0252: pipe is stalled (e000404f). skipping device
```
I suppose these are built-in hubs or similar.

There is also a slight performance improvement. Can you please run tests/stress and compare the running times?

RJVB, 2024-09-14:
On Saturday September 14 2024 01:09:53 Tormod Volden wrote:

> If only we could find a way to get the bcdUSB values correct...

I really have no idea, sorry.

I haven't looked at the API: is this value accessible directly via a structure or only via a getter method? In the latter case you could always decide to do the required calls if the value hasn't been cached yet. Barring that you probably have no other choice then to add an initialisation flag to enable the ioless variant, if you don't want to change the API beyond that optional extension...

FWIW, usbmuxd doesn't consult `bcdUSB` at all.

>I suppose these are built-in hubs or similar.

Yes, they are.

>There is also a slight performance improvement. Can you please run tests/stress and compare the running times?

Not going out to a device to query it is bound to speed things up. Esp. on Mac I'm tempted to say. Things have improved a lot but I can remember a time where all USB activity seemed to happen on a single kernel thread that could become a real bottleneck.

* ioless (built with -O0 -g ! and the DAC disconnected for comparison's sake):
```
> stress
Starting test run: init_and_exit...
Success (0)
Starting test run: get_device_list...
Success (0)
Starting test run: many_device_lists...
Success (0)
Starting test run: default_context_change...
Success (0)
---
Ran 4 tests
Passed 4 tests
Failed 0 tests
Error in 0 tests
Skipped 0 tests
27.221 user_cpu 71.784 kernel_cpu 1:39.51 total_time 99.4%CPU {2793472M 0F 977185R 0I 5O 3k 5w 187889c}
```

Current 1.0.27 with my hack:
```
> stress
Starting test run: init_and_exit...
Success (0)
Starting test run: get_device_list...
Success (0)
Starting test run: many_device_lists...
Success (0)
Starting test run: default_context_change...
Success (0)
---
Ran 4 tests
Passed 4 tests
Failed 0 tests
Error in 0 tests
Skipped 0 tests
31.123 user_cpu 92.355 kernel_cpu 2:20.63 total_time 87.7%CPU {2523136M 18F 835683R 1I 1O 3k 8w 324023c}
```

tormodvolden, 2024-09-14:
I found a dump of SDK headers on GitHub, and it seems that from SDK 10.11 there is a "bcdUSB" property that can be used (which was not present in the older headers I had been looking at). So please test the latest version of my branch. The debug log should tell if this works (if the property is not found, it falls back to the descriptor fetching from the device).

RJVB, 2024-09-14:
On Saturday September 14 2024 15:07:43 Tormod Volden wrote:
>I found a dump of SDK headers on GitHub, and it seems that from SDK 10.11 there is a "bcdUSB" property that can be used (which was not present in the older headers I had been looking at).

I'm running 10.9.5 ... though I do have a `GetbcdUsb()` function defined in /System/Library/Frameworks/Kernel.framework/Versions/A/Headers/IOKit/usb/IOUSBDevice.h

>So please test the latest version of my branch. The debug log should tell if this works (if the property is not found, it falls back to the descriptor fetching from the device).

So when exactly does the "descriptor fetching from the device" take place on pre-10.11 systems? During the previously ioless init step, or JIT-style when the property is actually required?

(BTW, did you force-push to your repo? I get a merge conflict while pulling, yet didn't change anything to the clone.)

tormodvolden, 2024-09-14:
I think GetbcdUsb() is only available to kernel drivers.

The descriptor fetching takes place in the init step (if looking up the property fails).

I didn't force-push now, but once 14 hours ago. In any case, try "git fetch" instead of "git pull", then git reset --hard FETCH_HEAD

tormodvolden, 2024-09-14:
If you run it on 10.9.5 I'd expect it to not find the "bcdUSB" property and therefore fall back to the descriptor fetching that upsets your DAC.

tormodvolden, 2024-09-14:
You can also try the `ioreg` command to see if it displays any bcdUSB info.

RJVB, 2024-09-14:
>If you run it on 10.9.5 I'd expect it to not find the "bcdUSB" property and therefore fall back to the descriptor fetching that upsets your DAC.

Yup, exactly what happens :(

Re: kernel drivers: would it be disproportional to write one that exposes all the information to userspace with a simpler API than IOKit's?

From what I understand, bcdUSB indicates if a device is USB 1, 2 or 3. That's a question the bus should be able to answer without having to consult the device!

tormodvolden, 2024-09-14:
> From what I understand, bcdUSB indicates if a device is USB 1, 2 or 3. That's a question the bus should be able to answer without having to consult the device!

You would think so. (Note there is 1.1, 2.0, 2.1, 3.0 at least).

RJVB, 2024-09-14:
On Saturday September 14 2024 16:25:45 Tormod Volden wrote:
>You can also try the `ioreg` command to see if it displays any bcdUSB info.

Not under that name:

```
   +-o UMC204HD ***@***.***  <class IOUSBDevice, id 0x10009b474, registered, matched, active, busy 0 (594 ms), retain 16>
        {
          "sessionID" = 642943485319711
          "iManufacturer" = 1
          "bNumConfigurations" = 1
          "idProduct" = 1288
          "bcdDevice" = 309
          "Bus Power Available" = 250
          "bMaxPacketSize0" = 64
          "USB Product Name" = "UMC204HD 192k"
          "iProduct" = 2
          "iSerialNumber" = 0
          "USB Address" = 5
          "bDeviceClass" = 239
          "bDeviceSubClass" = 2
          "locationID" = 18446744071564165120
          "IOUserClientClass" = "IOUSBDeviceUserClientV2"
          "PortNum" = 2
          "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
          "bDeviceProtocol" = 1
          "USB Vendor Name" = "BEHRINGER"
          "Device Speed" = 2
          "idVendor" = 5015
          "PortNumberOffset" = 0
          "Requested Power" = 250
          "IOGeneralInterest" = "IOCommand is not serializable"
          "Low Power Displayed" = No
        }
```

seanm, 2024-09-15:
> Re: kernel drivers: would it be disproportional to write one that exposes all the information to userspace with a simpler API than IOKit's?

Apple has been reducing access to kernel space in every release, writing such a thing would not likely have a long future. :(

RJVB, 2024-09-15:
>Apple has been reducing access to kernel space in every release, writing such a thing would not likely have a long future. :(

Even for kernel extensions, once you figure out how to get them to be loaded? (I'm tempted to say any open/free desktop computing effort on Mac probably won't have a long future anymore but that's another can of worms.)

Either way, in this case I see it more as a specific solution for older OS releases that might help keep the legacy code-paths in libusb (committed or as local/contrib patches) simpler.

seanm, 2024-09-15:
> Even for kernel extensions, once you figure out how to get them to be loaded?

Yes. You can see the trajectory here:

https://support.apple.com/en-ca/guide/deployment/depa5fb8376f/web

RJVB, 2024-09-15:
>Yes. You can see the trajectory here:

But again, there's no need for any of this on later OS versions where all device description data is available from the OS without having to go ask the device(s) themselves. That means 10.9 and before. And on 10.9 it's still perfectly possible to mess with kexts so the question was/is really more about the effort required to write one. And if the missing information is indeed available "ioless" (ideally we'd have an answer to that before starting to develop a kext).

I've started asking around elsewhere about this.

RJVB, 2024-09-15:
Anyway, if I look at the implementation of that IOUSBDevice class it looks like it'll have the same effect:
https://github.com/opensource-apple/IOUSBFamily/blob/7fec39743a5f05254c7e5366d9b0223ecc5dce71/IOUSBFamily/Classes/IOUSBDevice.cpp#L1890